arendst/Tasmota #15721: WebQuery: Body contains extra blank preceding the HTTP body when using `text/plain`

pfisterer, 2022-05-31:
### PROBLEM DESCRIPTION

I'm using [WebQuery](https://tasmota.github.io/docs/Commands/#wi-fi) in a Rule in order to send the state change to a web server. 

However, the actual **HTTP request includes a blank preceding the actual body**. This prevents the request to be parsed correctly on the server side.

The rule is as follows: `Rule1 ON POWER1#state=1 DO WebQuery http://my.server.name:3001/rest/items/eg_aussen_bewaesserung_ventil_zentral POST [Authorization: Basic XXXredactedXXX=|Accept: application/json|Content-Type: text/plain] ON ENDON`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 NodeMCU Development Board
- [x] Tasmota binary firmware version number used: Tasmota 11.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Web Installer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:36:00.920 CMD: Backlog Template; Module; GPIO 255
12:36:00.923 SRC: WebConsole from 192.168.178.20
12:36:00.926 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
12:36:00.942 SRC: Backlog
12:36:00.945 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
12:36:00.949 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:36:01.165 SRC: Backlog
12:36:01.168 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
12:36:01.171 RSL: RESULT = {"Module":{"0":"Generic"}}
12:36:01.415 SRC: Backlog
12:36:01.417 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
12:36:01.425 RSL: RESULT = {"GPIO0":{"257":"Relay_i2"},"GPIO1":{"0":"None"},"GPIO2":{"263":"Relay_i8"},"GPIO3":{"0":"None"},"GPIO4":{"260":"Relay_i5"},"GPIO5":{"262":"Relay_i7"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"258":"Relay_i3"},"GPIO13":{"259":"Relay_i4"},"GPIO14":{"261":"Relay_i6"},"GPIO15":{"0":"None"},"GPIO16":{"256":"Relay_i1"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
12:36:36.278 CMD: Backlog Rule1; Rule2; Rule3
12:36:36.280 SRC: WebConsole from 192.168.178.20
12:36:36.283 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
12:36:36.297 SRC: Backlog
12:36:36.300 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
12:36:36.305 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":209,"Free":302,"Rules":"ON POWER1#state=1 DO WebQuery http://my.server.name:3001/rest/items/eg_aussen_bewaesserung_ventil_zentral POST [Authorization: Basic XXXredactedXXX|Accept: application/json|Content-Type: text/plain] ON ENDON"}}
12:36:36.535 SRC: Backlog
12:36:36.537 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
12:36:36.542 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
12:36:36.784 SRC: Backlog
12:36:36.787 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
12:36:36.792 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:37:39.624 CMD: Status 0
12:37:39.627 SRC: WebConsole from 192.168.178.20
12:37:39.629 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
12:37:39.635 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","","","",""],"Topic":"tasmota_041CD7","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:37:39.670 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:53:11","StartupUTC":"2022-05-31T10:44:28","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-05-31T11:33:41","SaveCount":30,"SaveAddress":"F6000"}}
12:37:39.705 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022-04-13T06:40:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"387/699"}}
12:37:39.722 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
12:37:39.752 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,56,62","Sensors":"1,2,3,4,5,6"}}
12:37:39.786 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-041CD7-7383","IPAddress":"192.168.178.43","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"38:2B:78:04:1C:D7","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:37:39.812 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_041CD7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:37:39.832 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-05-31T11:37:39","Local":"2022-05-31T12:37:39","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:43"}}
12:37:39.850 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-05-31T12:37:39"}}
12:37:39.860 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-05-31T12:37:39","Uptime":"0T00:53:11","UptimeSec":3191,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"blubb4321","BSSId":"7C:FF:4D:44:B9:8A","Channel":6,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:04"}}}
12:37:43.378 RSL: STATE = {"Time":"2022-05-31T12:37:43","Uptime":"0T00:53:15","UptimeSec":3195,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"XXXX","BSSId":"7C:FF:4D:44:B9:8A","Channel":6,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:04"}}
[... 34 lines cut ...]
User-Agent: curl/7.83.0
Accept: application/json
Content-Type: text/plain
Content-Length: 2

ON
```

This is the output received by the server from Tasmota's WebQuery (please note the blank before `ON`):

```console
POST /rest/items/eg_aussen_bewaesserung_ventil_zentral HTTP/1.1
Host: my.server.name:3001
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Authorization: Basic XXXredactedXXX
Accept: application/json
Content-Type: text/plain
Content-Length: 3

 ON
```

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-05-31:
The space is actually present in the command. I suppose the documentation is misleading. Can you try

`Rule1 ON POWER1#state=1 DO WebQuery http://my.server.name:3001/rest/items/eg_aussen_bewaesserung_ventil_zentral POST [Authorization: Basic XXXredactedXXX=|Accept: application/json|Content-Type: text/plain]ON ENDON`

I.e. without any space betwenn `]` and `ON`

pfisterer, 2022-05-31:
@s-hadinger Great, thanks a million. Now it works. I thought this was one of the things I've tried. Maybe this should be included in the documentation.

closed 2022-05-31 by pfisterer