arendst/Tasmota #15760: Hiking Single Phase 65A DIN Rail Energy Meter (DDS238-2): MQTT wrong VAr (0VAr), PowerFactor (1,0)

g6094199, 2022-06-05:
### PROBLEM DESCRIPTION
MQTT sending wrong power factor cosPhi and reactive power. cosPhi is alway 1,0 and thus VAr must be 0W. But this cant be true since Power and Apparent Power are NOT the same. Which tells that cosPhi CANT be 1,0, which must be true in reallife. So something must be miscalculated in tasmota.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Hiking Single Phase 65A DIN Rail Energy Meter (DDS238-2
- [x] Tasmota binary firmware version number used: 11.1.0.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Install and setup like in [template](https://templates.blakadder.com/hiking_DDS238-2.html)

### EXPECTED BEHAVIOUR
VAr is almost never 0VAr und thus cosPhi can be 1,0,  physically almost impossible in reallife.

### SCREENSHOTS
[image]

arendst, 2022-06-05:
Ever heard of rounding numbers? Your difference is 0.7VA. So power factor is 226,7W / 227,4VA = 0.997. With a fixed resolution set to 2 this is rounded up to 1.

sfromis, 2022-06-05:
If you want more decimals on the power factor, you can modify that number in the source code
https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_03_energy.ino#L1225
`EnergyFormat(value3_chr, power_factor, 2));`

Jason2866, 2022-06-05:
Closing since answered and does work as designed

closed 2022-06-05 by Jason2866