arendst/Tasmota #15761: Shutters and Blinds Interlock Hazard

xbln, 2022-06-06:
### PROBLEM DESCRIPTION
With Shutters and Blinds it may happen that the Interlock State between two relays is lost after reboot. Initial operation will then lead to a short circuit and destroys the equipment, because ShutterMode 0 will autodetect the wrong mode according to the lost Interlock State. It then will switch both relais to ON. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used: Sonoff T1 2 Switches
- [x] Tasmota binary firmware version number used: 11.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled

### TO REPRODUCE
Can easily be reproduced.

### EXPECTED BEHAVIOUR
For safety reasons ShutterMode 0 (autodetect) should be disposed. ShutterMode1 should become default and with ShutterMode 1 Interlock 1,2 should engaged by default.

A big exclamation sign in documentation should warn about possible hazards when Interlock is unintentionally disengaged (e. g. when status is lost after reboot). Users should always make sure that Interlock is permanently engaged when applying mains to the shutter.

barbudor, 2022-06-06:
> Can easily be reproduced.

I can't see how the specific "Interlock" setting can be lost without resetting the whole device configuration which would also loose the ShutterMode

Can you please elaborate ?
Thanks

xbln, 2022-06-06:
Sorry, Barbudor, for the unclear reproduction state. In fact, I tried to
reproduce the situation but was unable. I can only say that I once
lost parts of my first shutter configuration including Interlock after
reboot.

I appreciate the work regarding shutters and blinds very much. Tremendous
work!

Regards

Martin

ascillato, 2022-06-06:
Hi,

You said

>TO REPRODUCE
>Can easily be reproduced.

And

> I tried to reproduce the situation but was unable.

So, is it possible that you have sent the interlock command by mistake (or as a retained message) and that was the origin or your problem?

If any of the Tasmota recovery methods was triggered, it should have set everything to default.

Please, can you recall why the device was rebooted? Energy failure, configuration?

xbln, 2022-06-06:
No. I first made the configuration for ShutterRelais1 and Interlock and
then rebooted (as suggested in the documentation:  *After setting the
options for shutter mode, the device must be rebooted.* Otherwise, the
sliders won't be available in the web UI, and the ShutterOpenDuration<x>and
ShutterCloseDuration<x> commands will report "Shutter unknown".)

When the device came back, Shutter-commands were available. Unfortunately I
didn´t recheck ShutterMode1 and Interlock. So when I pushed buttons, power
was on both outputs and the shutter-motor died.

As I wrote before, I couldn´t reproduce the situation. I also switched
SetOption80 on and off, rebooted, made new configurations but didn´t get
the fault again.

I still would recommend to put a warning in the docs. It is always
advisable to double-check ShutterMode and Interlock before applying mains.
Is that autodiscovery for the ShutterMode really useful? It might be safer
to make a simple default configuration with Interlock engaged by default.

Best

Martin

stefanbode, 2022-06-16:
Making shuttermode 1 a default is an idea but will break some stuff. I would suggest to change the documentation how to wire the relays you showed. These relays have the benefit that they have two outputs and if you do it right even with shuttermode 0 you cannot create a short circuit. My devices have never a lost of the interlock. There was a complete lost of configuration sometimes but this does not create electrical issues. The right way to wire is already documented. I would just remove the other one. In this case if both relays go on just the second does not get power.

stefanbode, 2022-06-16:
I see rapi3 already posted the image here

stefanbode, 2022-06-16:
I changed the docs and removed the dangerous wireing image. Now only the good one is left.

https://user-images.githubusercontent.com/34340210/65997877-3517e180-e468-11e9-9b8c-2f0787f977f6.png

amedeodivito, 2022-06-17:
the same configuration for the rolling shutters, with version 11.1.0 it works, with version 12.0.1 when the keys are pressed, wemos d1 mini resets itself

baennjer, 2022-06-18:
@stefanbode concerning wiring. I think something was missmatched because your wiring diagram is the one for shuttermode 2 (Relay1: ON/OFF, Relay2: UP/DOWN) wich is safe in any way. 
In Shuttermode 1 (Relay1: UP/OFF, Relay2: DOWN/OFF) it can happen that u destroy the motor when both relays are active if the wiring is not optimized. See image for optimal wiring in shuttermode 1.
[image]

stefanbode, 2022-06-18:
Yes you're right. Can I use your image for the documentation?

stefanbode, 2022-06-18:
@amedeodivito , @arendst : if there is a change with 12 this is not the code of the shutter driver. Any idea what could be the issue? Still using 11 here in production. If there is a reproduce able reset with buttons pressed please upload some log files for further investigation.

baennjer, 2022-06-18:
@stefanbode yes but use this one as I dont have rights on the pic of the relay board
[image]

rapi3, 2022-06-18:
There you have a new picture I took free to use as you want:
[image]

baennjer, 2022-06-18:
@rapi3  thank you very much for this offer, but I already did a quick powerpoint drawing.

amedeodivito, 2022-06-18:
....... If there is a reproduce able reset with buttons pressed please upload some log files for further investigation.

the error occurs when one of the two keys (up / down) is pressed
[Serial_Log.txt](https://github.com/arendst/Tasmota/files/8933735/Serial_Log.txt)

xbln, 2022-06-19:
I usually trust software devices even when mains is concerned. There are millions of such devices around which work as expected, no hardware protection is needed. It is sometimes difficult to add an extra relais to the installation - here a Sonoff T1-2 wall switch.
In my case there has been a reset while configuring which was not really noticed by me. The problem was that I din´t double check  everything - especially the interlock state - before applying mains. A big warning in the docs would have been helpful in my case.