arendst/Tasmota #15763: Azure IoT Hub permanently disconnecting

martinren87, 2022-06-06:
### PROBLEM DESCRIPTION

I am connecting to Azure IoT Hub and once it connects, it inmediatly disconnects and reconnects. This is an infinite loop. I don´t know if this is intended to work like this or we are having some trouble at any point. 

Using latest firmware in an ESP32.

I copy here the output of the serial monitor. Note that I just copy 2 loops, but it keeps doing the same forever:

```
14:27:26.855 MQT: Attempting connection...
14:27:26.860 MQT: Authenticating with an Azure IoT Hub Token
14:27:26.865 MQT: sha256 dataToSign is 'XXXXXXXXXX.azure-devices.net%2Fdevices%2FIntegrar_Energia_Piloto
1654612046'
14:27:26.877 MQT: sha256 signature is 'XXXXXXXXXX'
14:27:26.879 MQT: Azure IoT Hub SAS Token is 'SharedAccessSignature sr=XXXXXXXXXX.azure-devices.net%2Fdevices%2FIntegrar_Energia_Piloto&sig=XXXXXXXXXX'
14:27:29.753 MQT: TLS connected in 2893 ms, stack low mark 1108
14:27:29.758 MQT: MFLN not supported by TLS server
14:27:29.762 MQT: Server fingerprint: F1 6C DE FE 8F B2 81 81 70 7B 02 F5 A2 8B E0 E3 AF 5B XX XX
14:27:29.763 MQT: Connected
14:27:29.778 MQT: Invalid JSON for topic 'Integrar_Energia_Piloto/LWT', not sending to Azure IoT Hub
14:27:29.779 MQT: Integrar_Energia_Piloto/LWT = Online (retained)
14:27:29.794 MQT: Invalid JSON for topic 'Integrar_Energia_Piloto/cmnd/POWER', not sending to Azure IoT Hub
14:27:29.795 MQT: Integrar_Energia_Piloto/cmnd/POWER = 
14:27:29.810 MQT: Subscribe to Integrar_Energia_Piloto/cmnd/#
14:27:29.817 CFG: CR 515/699, Busy 0
14:27:29.818 CFG: CR 515/699, Busy 0
14:27:29.869 MQT: tasmota/discovery/84CCA82C18A0/config = {"ip":"192.168.1.144","dn":"Tasmota","fn":["Tasmota","Tasmota2","Tasmota3","Tasmota4",null,null,null,null],"hn":"Integrar-Energia-Piloto-6304","mac":"84CCA82C18A0","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"11.1.0.3","t":"Integrar_Energia_Piloto","ft":"Integrar_Energia_Piloto","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":1,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
14:27:29.956 MQT: tasmota/discovery/84CCA82C18A0/sensors = {"sn":{"Time":"2022-06-06T14:27:29","ESP32":{"Temperature":53.3},"TempUnit":"C"},"ver":1} (retained)
14:27:30.801 MQT: Attempting connection...
14:27:30.807 MQT: Authenticating with an Azure IoT Hub Token
14:27:30.812 MQT: sha256 dataToSign is 'XXXXXXXXXX.azure-devices.net%2Fdevices%2FIntegrar_Energia_Piloto
1654612050'
14:27:30.824 MQT: sha256 signature is 'XXXXXXXXXX'
14:27:30.826 MQT: Azure IoT Hub SAS Token is 'SharedAccessSignature sr=XXXXXXXXXX.azure-devices.net%2Fdevices%2FIntegrar_Energia_Piloto&sig=XXXXXXXXXX'
14:27:33.335 MQT: TLS connected in 2528 ms, stack low mark 1108
14:27:33.340 MQT: MFLN not supported by TLS server
14:27:33.344 MQT: Server fingerprint: F1 6C DE FE 8F B2 81 81 70 7B 02 F5 A2 8B E0 E3 AF 5B XX XX 
14:27:33.345 MQT: Connected
14:27:33.360 MQT: Invalid JSON for topic 'Integrar_Energia_Piloto/LWT', not sending to Azure IoT Hub
14:27:33.361 MQT: Integrar_Energia_Piloto/LWT = Online (retained)
14:27:33.376 MQT: Invalid JSON for topic 'Integrar_Energia_Piloto/cmnd/POWER', not sending to Azure IoT Hub
14:27:33.377 MQT: Integrar_Energia_Piloto/cmnd/POWER = 
14:27:33.392 MQT: Subscribe to Integrar_Energia_Piloto/cmnd/#
14:27:33.400 CFG: CR 515/699, Busy 0
14:27:33.400 CFG: CR 515/699, Busy 0
14:27:33.452 MQT: tasmota/discovery/84CCA82C18A0/config = {"ip":"192.168.1.144","dn":"Tasmota","fn":["Tasmota","Tasmota2","Tasmota3","Tasmota4",null,null,null,null],"hn":"Integrar-Energia-Piloto-6304","mac":"84CCA82C18A0","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"11.1.0.3","t":"Integrar_Energia_Piloto","ft":"Integrar_Energia_Piloto","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":1,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
14:27:33.539 MQT: tasmota/discovery/84CCA82C18A0/sensors = {"sn":{"Time":"2022-06-06T14:27:33","ESP32":{"Temperature":53.3},"TempUnit":"C"},"ver":1} (retained)

```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP32
- [ ] Tasmota binary firmware version number used: 11.1.0.3
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: 
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
14:57:59.222 CMD: Backlog Template; Module; GPIO 255
14:57:59.281 MQT: Integrar_Energia_Piloto/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
14:57:59.514 MQT: Integrar_Energia_Piloto/RESULT = {"Module":{"1":"ESP32-DevKit"}}
14:57:59.782 MQT: Integrar_Energia_Piloto/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"0":"None"},"GPIO14":{"33":"Button2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"226":"Relay3"},"GPIO23":{"227":"Relay4"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"35":"Button4"},"GPIO27":{"34":"Button3"},"GPIO32":{"224":"Relay1"},"GPIO33":{"225":"Relay2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
14:58:00.924 MQT: Attempting connection...
[... 56 lines cut ...]
14:27:33.361 MQT: Integrar_Energia_Piloto/LWT = Online (retained)
14:27:33.376 MQT: Invalid JSON for topic 'Integrar_Energia_Piloto/cmnd/POWER', not sending to Azure IoT Hub
14:27:33.377 MQT: Integrar_Energia_Piloto/cmnd/POWER = 
14:27:33.392 MQT: Subscribe to Integrar_Energia_Piloto/cmnd/#
14:27:33.400 CFG: CR 515/699, Busy 0
14:27:33.400 CFG: CR 515/699, Busy 0
14:27:33.452 MQT: tasmota/discovery/84CCA82C18A0/config = {"ip":"192.168.1.144","dn":"Tasmota","fn":["Tasmota","Tasmota2","Tasmota3","Tasmota4",null,null,null,null],"hn":"Integrar-Energia-Piloto-6304","mac":"84CCA82C18A0","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"11.1.0.3","t":"Integrar_Energia_Piloto","ft":"Integrar_Energia_Piloto","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":1,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
14:27:33.539 MQT: tasmota/discovery/84CCA82C18A0/sensors = {"sn":{"Time":"2022-06-06T14:27:33","ESP32":{"Temperature":53.3},"TempUnit":"C"},"ver":1} (retained)

```

### TO REPRODUCE

Connect to Azure IoT Hub using TLS, Setoption132 to ON, and Fingerprint to FF FF FF...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-06-06:
Hi,

Seems that your issue is related to the open bug report:
https://github.com/arendst/Tasmota/issues/14624
Which is unresolved so far.

martinren87, 2022-06-06:
If I'm not wrong, in bug #14624 you can't even establish the connection due to a handshake problem. In this case I am able to establish the connection, and I even see the published messages with the Azure IoT Explorer... the problem is in the stability of the connection.
In any case, if there is any test that I can do to verify if it is the same problem, please tell me and I will do it.

Jason2866, 2022-06-06:
The Azure IoT Hub is not used from any of the main contributers.

Jason2866, 2022-06-24:
@martinren87 Tasmota docu is Open Source too. Since you know now best what is needed to get it going, could you provide a PR for the docu?

closed 2022-07-24 by github-actions[bot]