arendst/Tasmota #15780: Sonoff Zigbee motion sensor UNSUPPORTED_ATTRIBUTE

amaximus, 2022-06-10:
### PROBLEM DESCRIPTION
Brand new Sonoff Zigbee motion sensor throws UNSUPPORTED_ATTRIBUTE at joining.
Join operation is successful, battery state is reported, but motion states are not reported.
Sensor FCC ID: 2APN5SNZB-03

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic):  Sonoff Zigbee Bridge
- [X] Tasmota binary firmware version number used: 11.1.0 (the same was on 9.5.0)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:54:16.224 CMD: Backlog Template; Module; GPIO 255
14:54:16.289 MQT: sonoff_zigbee/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:54:16.497 MQT: sonoff_zigbee/stat/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
14:54:16.750 MQT: sonoff_zigbee/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:55:25.702 CMD: Backlog Rule1; Rule2; Rule3
14:55:25.765 MQT: sonoff_zigbee/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:55:25.977 MQT: sonoff_zigbee/stat/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:55:26.226 MQT: sonoff_zigbee/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:56:02.793 CMD: Status 0
14:56:02.798 MQT: sonoff_zigbee/stat/STATUS = {"Status":{"Module":75,"DeviceName":"ZigbeeBridge","FriendlyName":["Zigbee Bridge"],"Topic":"sonoff_zigbee","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:56:02.804 MQT: sonoff_zigbee/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:33:06","StartupUTC":"2022-06-09T13:22:56","Sleep":50,"CfgHolder":4617,"BootCount":147,"BCResetTime":"2021-02-09T21:28:20","SaveCount":208,"SaveAddress":"FC000"}}
14:56:02.808 MQT: sonoff_zigbee/stat/STATUS2 = {"StatusFWR":{"Version":"11.1.0(zbbridge)","BuildDateTime":"2022-04-13T06:40:30","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"421/699"}}
14:56:02.813 MQT: sonoff_zigbee/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Rozsaek",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
14:56:02.822 MQT: sonoff_zigbee/stat/STATUS4 = {"StatusMEM":{"ProgramSize":678,"Free":1112,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","04000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50,56,62","Sensors":"1"}}
14:56:02.828 MQT: sonoff_zigbee/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-zigbee-4331","IPAddress":"192.168.1.51","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"192.168.1.1","Mac":"98:F4:AB:E3:70:EB","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:56:02.832 MQT: sonoff_zigbee/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.156","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E370EB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:56:02.838 MQT: sonoff_zigbee/stat/STATUS7 = {"StatusTIM":{"UTC":"2022-06-09T13:56:02","Local":"2022-06-09T14:56:02","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:51"}}
14:56:02.842 MQT: sonoff_zigbee/stat/STATUS10 = {"StatusSNS":{"Time":"2022-06-09T14:56:02"}}
14:56:02.848 MQT: sonoff_zigbee/stat/STATUS11 = {"StatusSTS":{"Time":"2022-06-09T14:56:02","Uptime":"0T00:33:06","UptimeSec":1986,"Vcc":3.462,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Rozsaek","BSSId":"0C:80:63:91:B0:FC","Channel":2,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:57:29.779 CMD: ZbPermitJoin 1
14:57:29.780 SRC: WebConsole from 192.168.1.44
14:57:29.782 CMD: Grp 0, Cmd 'ZBPERMITJOIN', Idx 1, Len 1, Pld 1, Data '1'
14:57:29.783 ZIG: ZbEZSPSend 22003C
14:57:29.784 ZIG: adding packet to_send, to_ack:3, to_send:3, to_end:3
14:57:29.785 ZIG: ZbEZSPSend 3600FCFF00003600000000000000061E0103063C01
14:57:29.786 ZIG: adding packet to_send, to_ack:3, to_send:3, to_end:4
14:57:29.789 MQT: sonoff_zigbee/tele/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
14:57:29.793 MQT: sonoff_zigbee/stat/RESULT = {"ZbPermitJoin":"Done"}
14:57:29.856 ZIG: Something to_send, to_ack:3, to_send:3, to_end:5
14:57:29.857 ZIG: ZbEZSPSentRaw 3033000122003C
14:57:29.860 ZIG: Something to_send, to_ack:3, to_send:4, to_end:5
14:57:29.861 ZIG: ZbEZSPSentRaw 403400013600FCFF00003600000000000000061E0103063C01
14:57:29.863 ZIG: {"ZbEZSPReceived2":"04338001220000"}
14:57:29.864 ZIG: new ack/data received, was 3 now 4
14:57:29.865 ZIG: freeing packet 3 from memory
14:57:29.866 ZIG: ZbEZSPSentRaw 81
14:57:29.867 ZIG: {"ZbEZSPReceived":"220000"}
[... 273 lines cut ...]
14:58:00.764 ZIG: {"ZbEZSPReceived2":"203F90013F0000A9490401000501014001000048010000"}
14:58:00.765 ZIG: ZbEZSPSentRaw 83
14:58:00.766 ZIG: {"ZbEZSPReceived":"3F0000A9490401000501014001000048010000"}
14:58:00.777 ZIG: {"ZbEZSPReceived2":"303F9001450000040100050101000100000D50B0A949FFFF07180C078600020004"}
14:58:00.778 ZIG: ZbEZSPSentRaw 84
14:58:00.779 ZIG: {"ZbEZSPReceived":"450000040100050101000100000D50B0A949FFFF07180C078600020004"}
14:58:00.781 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0500","srcaddr":"0x49A9","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":19,"securityuse":0,"seqnumber":13,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":12,"cmdid":"0x07","payload":"86000200"}}
14:58:00.784 ZIG: ZbZCLRawReceived: {"0x49A9":{"ConfigResponse":{"ZoneStatus":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":19}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Try to re-join the sensor.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I suspect that because of the UNSUPPORTED_ATTRIBUTE ZoneStatus, zones state changes are not reported.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-06-10:
Zigbee2Tasmota tries to configure attribute reporting at pairing time. Some device support it, some don't. You can ignore this.

amaximus, 2022-06-10:
Thanks for the explanation.

closed 2022-06-10 by amaximus