arendst/Tasmota #15785: Range Extender stops routing

dsteinkopf, 2022-06-12:
### PROBLEM DESCRIPTION
"Sometimes" the range extender stops routing. The Wifi is still open, clients stay connected, others can newly connect. But packets are not routed anymore.
I experienced some correlation between that problem and change of the bssid that tasmota hotspot is using: Shortly after it changed its bssid, other devices connected to this range extender lost tcp connection. But I am not able to reproduce the problem.
BTW: Number of clients connected to the range extender is less than 5-6.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 2.5
- [x] Tasmota binary firmware version number used: 11.1.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA update (minimal then mine)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
07:22:58.572 MQT: stat/kleinelampe_335E29/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
07:22:58.792 MQT: stat/kleinelampe_335E29/RESULT = {"Module":{"0":"Shelly 2.5"}}
07:22:59.048 MQT: stat/kleinelampe_335E29/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"193":"Switch_n2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"192":"Switch_n1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
07:23:37.230 MQT: stat/kleinelampe_335E29/STATUS = {"Status":{"Module":0,"DeviceName":"Kleinelampe","FriendlyName":["Kleinelampe","nan"],"Topic":"kleinelampe_335E29","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
07:23:37.238 MQT: stat/kleinelampe_335E29/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:15:58","StartupUTC":"2022-06-12T04:07:39","Sleep":50,"CfgHolder":4617,"BootCount":41,"BCResetTime":"2022-05-17T12:45:42","SaveCount":314,"SaveAddress":"F6000"}}
07:23:37.245 MQT: stat/kleinelampe_335E29/STATUS2 = {"StatusFWR":{"Version":"11.1.0.3(tasmota)","BuildDateTime":"2022-05-23T10:20:46","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"525/699"}}
07:23:37.252 MQT: stat/kleinelampe_335E29/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":2,"SysLog":2,"LogHost":"192.168.40.5","LogPort":514,"SSId":["pipwlan","pipshelly2"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00228009","2805C80001000600003C5A0A190000000000","000002EC","00006000","00004000"]}}
07:23:37.265 MQT: stat/kleinelampe_335E29/STATUS4 = {"StatusMEM":{"ProgramSize":628,"Free":372,"Heap":20,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15400B","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","24368001","000000CF","010013C0","C000F981","00004004","00001000","04000420"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,58,62","Sensors":"1,2,3,4,5,6"}}
07:23:37.273 MQT: stat/kleinelampe_335E29/STATUS5 = {"StatusNET":{"Hostname":"kleinelampe-335E29-7721","IPAddress":"192.168.40.207","Gateway":"192.168.40.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.40.1","DNSServer2":"0.0.0.0","Mac":"4C:75:25:33:5E:29","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
07:23:37.280 MQT: stat/kleinelampe_335E29/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.40.227","MqttPort":1883,"MqttClientMask":"kleinelampe_%06X","MqttClient":"kleinelampe_335E29","MqttUser":"tasmota_device","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:23:37.290 MQT: stat/kleinelampe_335E29/STATUS7 = {"StatusTIM":{"UTC":"2022-06-12T05:23:37","Local":"2022-06-12T07:23:37","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:46","Sunset":"21:53"}}
07:23:37.296 MQT: stat/kleinelampe_335E29/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:23:37.311 MQT: stat/kleinelampe_335E29/STATUS10 = {"StatusSNS":{"Time":"2022-06-12T07:23:37+02:00","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":55.8},"ENERGY":{"TotalStartTime":"2022-05-17T12:45:42","Total":0.010,"Yesterday":0.002,"Today":0.000,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":235,"Current":[0.000,0.000]},"TempUnit":"C"}}
07:23:37.320 MQT: stat/kleinelampe_335E29/STATUS11 = {"StatusSTS":{"Time":"2022-06-12T07:23:37+02:00","Uptime":"0T01:15:58","UptimeSec":4558,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"pipwlan","BSSId":"8A:DA:88:BE:9A:60","Channel":6,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
I am not able to reproduce the problem. It only happens "sometimes".

### EXPECTED BEHAVIOUR
Range extender continues to do routing.

### SCREENSHOTS
.

### ADDITIONAL CONTEXT
My Range extender config
```
Backlog
RgxSSID pipshelly1;
RgxPassword xxx;
RgxAddress 192.168.37.1;
RgxSubnet 255.255.255.192;
RgxNAPT 0;
RgxState ON
```
**(Please, remember to close the issue when the problem has been addressed)**

dsteinkopf, 2022-06-12:
Inbetween the problem re-occurred while having syslog level set to 4.
Output from `grep kleinelampe /var/log/syslog | grep -v 'SENSOR\|STATE\|STATUS\|RESULT' | less`:
(Shortened because it was very long. If you need full output, please tell me.)
```
Jun 12 08:15:55 kleinelampe-335E29-7721 ESP-ADE: U 6174243, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:15:56 kleinelampe-335E29-7721 ESP-WIF: Checking connection...
Jun 12 08:15:56 kleinelampe-335E29-7721 ESP-ADE: 6173307, 4477, [1773, -14, 651, 3], [1773, -22, 651, 3]
Jun 12 08:15:56 kleinelampe-335E29-7721 ESP-ADE: U 6173307, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:15:57 kleinelampe-335E29-7721 ESP-ADE: 6174535, 4477, [1773, -71, 651, 64], [1773, 17, 651, -5]
Jun 12 08:15:57 kleinelampe-335E29-7721 ESP-ADE: U 6174535, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:15:58 kleinelampe-335E29-7721 ESP-ADE: 6172722, 4477, [1773, 15, 651, 71], [1773, -12, 651, -11]
Jun 12 08:15:58 kleinelampe-335E29-7721 ESP-ADE: U 6172722, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:15:59 kleinelampe-335E29-7721 ESP-ADE: 6172244, 4478, [1773, -38, 651, 100], [1773, 22, 651, -4]
Jun 12 08:15:59 kleinelampe-335E29-7721 ESP-ADE: U 6172244, C 4478, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:00 kleinelampe-335E29-7721 ESP-ADE: 6172829, 4477, [1773, -53, 652, 13], [1773, -1, 651, 9]
Jun 12 08:16:00 kleinelampe-335E29-7721 ESP-ADE: U 6172829, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:01 kleinelampe-335E29-7721 ESP-ADE: 6171288, 4477, [1773, -6, 652, 71], [1773, 11, 651, 1]
Jun 12 08:16:01 kleinelampe-335E29-7721 ESP-ADE: U 6171288, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:02 kleinelampe-335E29-7721 ESP-ADE: 6167333, 4477, [1773, -70, 651, 34], [1773, 2, 651, -6]
Jun 12 08:16:02 kleinelampe-335E29-7721 ESP-ADE: U 6167333, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:03 kleinelampe-335E29-7721 ESP-ADE: 6166639, 4477, [1773, 3, 651, -18], [1773, -13, 651, -13]
Jun 12 08:16:03 kleinelampe-335E29-7721 ESP-ADE: U 6166639, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:04 kleinelampe-335E29-7721 ESP-ADE: 6168391, 4477, [1773, -20, 651, 11], [1773, 4, 651, 0]
Jun 12 08:16:04 kleinelampe-335E29-7721 ESP-ADE: U 6168391, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:05 kleinelampe-335E29-7721 ESP-ADE: 6169396, 4477, [1773, 9, 651, 41], [1773, -5, 651, -1]
Jun 12 08:16:05 kleinelampe-335E29-7721 ESP-ADE: U 6169396, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:06 kleinelampe-335E29-7721 ESP-ADE: 6174459, 4477, [1773, -46, 652, 56], [1773, -3, 652, 7]
Jun 12 08:16:06 kleinelampe-335E29-7721 ESP-ADE: U 6174459, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:07 kleinelampe-335E29-7721 ESP-WIF: Sending Gratuitous ARP
Jun 12 08:16:07 kleinelampe-335E29-7721 ESP-WIF: Network (re)scan started...
Jun 12 08:16:07 kleinelampe-335E29-7721 ESP-ADE: 6173217, 4477, [1773, -10, 652, 26], [1773, 8, 652, 2]
Jun 12 08:16:07 kleinelampe-335E29-7721 ESP-ADE: U 6173217, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:08 kleinelampe-335E29-7721 ESP-ADE: 6172731, 4477, [1773, -17, 651, 12], [1773, 0, 652, 3]
Jun 12 08:16:08 kleinelampe-335E29-7721 ESP-ADE: U 6172731, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:10 kleinelampe-335E29-7721 ESP-WIF: Network 4, AP1, SSId pipwlan, Channel 6, BSSId C0:06:C3:25:85:88, RSSI -73, Encryption 1
Jun 12 08:16:10 kleinelampe-335E29-7721 ESP-WIF: Network 5, AP-, SSId pipwlan gast, Channel 6, BSSId C6:06:C3:25:85:88, RSSI -73, Encryption 1
Jun 12 08:16:10 kleinelampe-335E29-7721 ESP-WIF: Network 6, AP-, SSId pip101, Channel 6, BSSId CE:06:C3:25:85:88, RSSI -74, Encryption 1
Jun 12 08:16:10 kleinelampe-335E29-7721 ESP-WIF: Network 7, AP-, SSId DIRECT-aa-HP M148f LaserJet, Channel 6, BSSId 76:12:B3:8A:38:AA, RSSI -80, Encryption 1
Jun 12 08:16:10 kleinelampe-335E29-7721 ESP-WIF: Network 8, AP-, SSId RIKA_FRITZ!Box, Channel 6, BSSId 64:66:B3:4F:B4:40, RSSI -91, Encryption 1
Jun 12 08:16:12 kleinelampe-335E29-7721 ESP-RGX: Link count now: 3, WiFi.getMode(): 1, unconfigure...
Jun 12 08:16:12 kleinelampe-335E29-7721 ESP-RGX: WiFi Extender AP Enabled with SSID: pipshelly1
Jun 12 08:16:12 kleinelampe-335E29-7721 ESP-ADE: 6174568, 4477, [1773, 19, 651, 24], [1773, -37, 652, -4]
Jun 12 08:16:12 kleinelampe-335E29-7721 ESP-ADE: U 6174568, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:12 kleinelampe-335E29-7721 ESP-ADE: 6170659, 4477, [1773, -48, 652, 54], [1773, 25, 651, -15]
Jun 12 08:16:12 kleinelampe-335E29-7721 ESP-ADE: U 6170659, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-MQT: Attempting connection...
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-MQT: Connected
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-MQT: tele/kleinelampe_335E29/LWT = Online (retained)
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-MQT: cmnd/kleinelampe_335E29/POWER = 
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-MQT: Subscribe to cmnd/kleinelampe_335E29/#
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-MQT: Subscribe to cmnd/tasmotas/#
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-MQT: Subscribe to cmnd/kleinelampe_335E29_fb/#
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-ADE: 6170859, 4478, [1773, 15, 652, -6], [1773, -9, 651, 24]
Jun 12 08:16:13 kleinelampe-335E29-7721 ESP-ADE: U 6170859, C 4478, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:14 kleinelampe-335E29-7721 ESP-MQT: tasmota/discovery/4C7525335E29/config = {"ip":"192.168.40.207","dn":"Kleinelampe","fn":["Kleinelampe","nan",null,null,null,null,null,null],"hn":"kleinelampe-335E29-7721","mac":"4C7525335E29","md":"Shelly 2.5","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"]," [... cut ...]
Jun 12 08:16:17 kleinelampe-335E29-7721 ESP-ADE: 6169415, 4477, [1773, -5, 651, 15], [1773, -9, 651, 26]
Jun 12 08:16:17 kleinelampe-335E29-7721 ESP-ADE: U 6169415, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:18 kleinelampe-335E29-7721 ESP-ADE: 6169088, 4477, [1773, -67, 651, 84], [1773, -6, 651, -8]
Jun 12 08:16:18 kleinelampe-335E29-7721 ESP-ADE: U 6169088, C 4477, I 0 + 0 = 0, P 0 + 0
Jun 12 08:16:18 kleinelampe-335E29-7721 ESP-WIF: Checking connection...
Jun 12 08:16:19 kleinelampe-335E29-7721 ESP-ADE: 6167887, 4477, [1773, -33, 651, 93], [1773, 5, 651, 16]
```

After that routing had stopped.

dsteinkopf, 2022-06-12:
Also After Power loss, the hotspot is not always working.

reboot helps in any case.

Jason2866, 2022-06-12:
Which Router/Ap do you have?

dsteinkopf, 2022-06-12:
I have a fritzbox and some TP-Link EAP access points controlled by an "omada" controller. All using the same SSID(s).

Jason2866, 2022-06-12:
Tbh, i think we hardly can help to find the reason for the issue. The setup has to many components involved. FritzBox is known to make problems sometimes in some situations. The APs can be involved too.
First step i would do is to find a minimalistic setup where the problem occurs (reproducable).

dsteinkopf, 2022-06-12:
Ok. Thank you anyway. Your answer contained an important detail: It's working in other environments and it should (could) work for me, too.

Also my idea was to put the range extender device into a different, more simple wifi (I have one in my mind...).

I'll come back when I have more info or solved the problem. Thank you again!

dsteinkopf, 2022-06-13:
I am now able to reproduce the problem reliably: I move the range extender tasmota device to a different wifi ("pipiot") and reduced this wifi to only 1 access point. Then, rebooting this access point makes the problem occur: Tasmota retries some times to reconnect, then succeeds, enables the wifi extender AP - but client packets are not routed anymore.
Restarting tasmota resolved the problem.

What can I do to further narrow down the problem? I have logs at loglevel 4 if that helps.

dsteinkopf, 2022-06-13:
In between, two more outages happened, but without loss of the wifi connection. I can't see any special in the logs around that time.
So the loss of the wifi connection + search + reconnect seems not to be the root cause of the problem.

dsteinkopf, 2022-06-14:
For the last 20 hours, I tried a separate device running https://github.com/martin-ger/esp_wifi_repeater instead of my Tasmota range extender. No problems there, so it seems not to be a problem with my infrastructure. - But it's not Tasmota...

How could I get further helpful debugging info?

Jason2866, 2022-06-14:
Good to know the network does work in general. Since the wifi repeater sketch (martinger) use a different platform as underlying layer (dont know which?) it does not really help to track the issue. Tasmota is based on Arduino Esp8266 v2.7.4.9 (modified 2.7.4) which is hooked up on NonOSDK v2.2.2. This combination is known to have problems with some network setups/configs. See https://github.com/arendst/Tasmota/discussions/13623 to get some infos which could help to find the reason.

Tomps1, 2022-06-14:
Hi dsteinkopf, I obviously having similar issue with D1 Mini Pro/ESP8266. 
I am having this connected to an MQTT Broker (resided in core network)
I am constantly pinging the ESP from a device in core network
I can connect the Webserver (config pages) on the ESP from core network

The moment I active the range extender:
ESP starts complaining to not being able to connect to MQTT Broker
I am losing ONE ping (thats ok), the ESP is still pingable FROM core network devices 
I cannot connect the ESP webserver FROM device on core network
I can connect to the rangeextender via WiFi now
From the rangeextender-network (WiFi) I can connect to the webserver on the ESP
From rangeextender-network I cannot ping any device on the core network (Host not reachable)

This is reproducable here - but I am an absolute noob (thats my first project, ESP arrived last Saturday, I was able to flash it first time without problems last Sunday...)

dsteinkopf, 2022-06-17:
> Good to know the network does work in general. Since the wifi repeater sketch (martinger) use a different platform as underlying layer (dont know which?) it does not really help to track the issue. Tasmota is based on Arduino Esp8266 v2.7.4.9 (modified 2.7.4) which is hooked up on NonOSDK v2.2.2. This combination (Closed Source) is known to have problems with some network setups/configs. See #13623 to get some infos which could help to find the reason.

I tried to read (fly through) #13623 and it seems, those problems are different to mine: In my case, the connection to the main AP is always ok, the tasmota device is reachable via ping at any time (when the AP is available). In the case of the problem, the clients of the range extender stay connected, keep their IP and are able to ping the range extender. The "only" problem is that packets seem not to be forwarded.

dsteinkopf, 2022-06-17:
Maybe also worth mentioning: When the range extender is in the "problem state" and I watch packets passing its AP using tcpdump, then I see packets coming from the client devices (connected to the range extender), but no packets in the other direction. Still, a reboot of the range extender resolves the problem and I see packets correctly in both directions.
(BTW: During the problem state, packet from "outside" to non existing IPs of the range extender's range still pass the AP as expected - but get no response, of course.)

closed 2022-07-17 by github-actions[bot]

dsteinkopf, 2022-07-18:
I gave up on debugging this problem, but it still exists. So I am currently not able to use the range extender feature.
(But still, Tasmota is a great piece of software)

joba-1, 2022-11-16:
I think I found your problem in [support_wifi.ino](https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_support/support_wifi.ino#L361)

```c
    // If bssid changed then (re)connect wifi
    for (uint32_t i = 0; i < sizeof(Wifi.bssid); i++) {
      if (last_bssid[i] != Wifi.bssid[i]) {
        WifiBegin(ap, channel);                     // 0 (AP1), 1 (AP2) or 3 (default AP)
        break;
      }
    }
```

Currently WifiBegin() always sets wifi mode to STA, and as you can see above it is called on BSSID changes.

I have PR #17103 open (to protect my NAT port forwardings) which should fix this too. 

To be merged soon (I hope)...

dsteinkopf, 2022-11-17:
Sounds very plausible and interesting. Thank you for the fix and for keeping your eyes open :-)

joba-1, 2022-11-17:
No problem - they are open anyways most of the day :). 
Should you still need/want it: fix is merged now. And so far my "strange" connection and port forwarding issues are all gone...

dsteinkopf, 2023-01-16:
It took some time, but now I tried the new version: I've used tasmota-rangeextender.bin.gz ("Tasmota 12.3.1.4 - Based on 82b6d1ee9102b350bdfe545f0033527def1b18f4" from unofficial precompiled: https://github.com/tasmota/install/tree/main/firmware/unofficial).

But now success: After about an hour, the tasmota AP again stopped routing: Again, it was still possible to connect, but no packets were transported - even not to the AP itself.

So I can not see any change in the problem. Could I have done something wrong?

joba-1, 2023-01-17:
Sorry to hear that. Uptime here is >10 days and still routing fine. 
Before that it was the same. Rock solid. I don’t know if it matters, but my version is (still) 12.2.0.5 and iirc a normal esp32 build.

LngWork, 2023-07-24:
I have a similar issue. Not sure it is exactly the same.
My config :  Tasmota 1 (on the extended network) <> Tasmota RE (Range extender) <> Main Router <> Linux Server

Tasmota 1 :
Program Version	9.5.0(lite)
Build Date & Time	2021-06-17T08:26:41
Core/SDK Version	2_7_4_9/2.2.2-dev(38a443e)à

Tasmota RE :
Program Version	13.0.0.2(tasmota32)
Build Date & Time	2023-07-20T12:49:41
Core/SDK Version	2_0_10/4.4.5.230614

After initial setup, all is OK and works fine.

**PROBLEM DESCRIPTION**
After a while (30 minutes or 1 hour), "Tasmota 1" cannot reach "Linux Server". If I Power OFF "Tasmota 1", then POWER ON it can reconnect to Extended AP easily but still cannot reach "Linux Server".

**Additional Information**
When "Tasmota 1" is not able to reach "Linux Server", if I try to join "Tasmota 1" from "Linux Server" (ping or http), I get an answer from "Tasmota 1". Then "Tasmota 1" is able to communicate for a while.

dsteinkopf, 2023-07-24:
Does restarting Tasmota RE also solve the problem for some time?

And what device type are you using? Maybe this is one important difference: I am using ESP8266 in Shellys while @joba-1 (where RE is working fine) is using an ESP32 (see above).

LngWork, 2023-07-24:
Does restarting Tasmota RE also solve the problem for some time? >> Yes
And what device type are you using?
>> "Tasmota 1 " is a SonOff MINI R2, so ESP8266
>> "Tasmota RE" is a ESP32-WROVER 16Mb flash + 8Mb PSRAM

dsteinkopf, 2023-07-24:
ok. So the ESP32/8266 is not the important difference that makes it run or not :-(

joba-1, 2023-07-25:
Sorry, don't have much value to add.

I use only port forwarding, not routing. And with that I still see no issues over long uptimes:

Program Version: 12.3.1.6(tasmota)
Build Date & Time: 2023-02-10T19:30:52
Core/SDK Version: 2_0_6/4.4.3.20230129
Uptime: 28T03:05:59