arendst/Tasmota #15790: Incorrect values for SHT1x

WGrobler, 2022-06-13:
### PROBLEM DESCRIPTION
I connected a SHT10 temperature and humidity sensor to a D1 ESP32 Dev Board and set the I2C SCL and SDA pins. When using the default tasmota-sensors firmware (installed via Web Flasher), I get incorrect readings for the sensor.  The temperature reading was around -30 when it should be around +25 and the humidity was -3. 

When I create my own firmware using the default library for the sensor in Arduino IDE, the values are correct, therefore I concluded that there must be some differences between the SHT1x library that form part of Tasmota and the library recommended by the hardware supplier. 

After comparing the code for the two libraries, I found that the default sensor library included a custom function for shiftIn, where Tasmota uses the default shiftIn function. I modified the tasmota_xsns_sensor/xsns_07_sht1x.ino file in the development branch of Tasmota and compiled my own firmware. Upon uploading the firmware to my device, I managed to get the correct values.

Since I am not a C developer, I'm not familiar with the working of the shiftIn function or why the custom function works. But I decided to create an issue to provide the code I used to get the correct output. Perhaps someone with more knowledge can determine if a change to the original code is required or if the specific sensor I am using requires this change. 

Link to supplier of the sensor: https://www.robotics.org.za/SHT10-5M
Link to supplier of the board: https://www.robotics.org.za/D1-ESP32
Link to default library: https://github.com/practicalarduino/SHT1x

Link to similar (unresolved) issue: https://github.com/arendst/Tasmota/issues/5768

Custom function required to perform fix (this was added to  tasmota_xsns_sensor/xsns_07_sht1x.ino file):

`int shiftIn(int _dataPin, int _clockPin, int _numBits)
{
  int ret = 0;
  for (int i=0; i<_numBits; ++i)
  {
     digitalWrite(_clockPin, HIGH);
     delay(10);  // I don't know why I need this, but without it I don't get my 8 lsb of temp
     ret = ret*2 + digitalRead(_dataPin);
     digitalWrite(_clockPin, LOW);
  }
  return(ret);
}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): D1 ESP32 Dev Board
- [ X] Tasmota binary firmware version number used: latest web installer
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: upload of self-compiled binary was done by uploading file to device via firmware upgrade tool on a Web installed base firmware
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:57:25.336 MQT: stat/tasmota_0642B8/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
09:57:25.572 MQT: stat/tasmota_0642B8/RESULT = {"Module":{"1":"ESP32-DevKit"}}
09:57:25.827 MQT: stat/tasmota_0642B8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"608":"I2C SCL1"},"GPIO17":{"640":"I2C SDA1"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
09:57:26.123 MQT: tele/tasmota_0642B8/STATE = {"Time":"2022-06-13T09:57:26","Uptime":"0T00:10:11","UptimeSec":611,"Heap":152,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"#redacted#","BSSId":"#redacted#","Channel":11,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
09:57:26.142 MQT: tele/tasmota_0642B8/SENSOR = {"Time":"2022-06-13T09:57:26","SHT1X":{"Temperature":14.7,"Humidity":53.3,"DewPoint":5.3},"ESP32":{"Temperature":35.0},"TempUnit":"C"}

```
- [ X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
09:58:42.549 MQT: stat/tasmota_0642B8/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:58:42.778 MQT: stat/tasmota_0642B8/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:58:43.027 MQT: stat/tasmota_0642B8/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:59:06.999 MQT: stat/tasmota_0642B8/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_0642B8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:59:07.017 MQT: stat/tasmota_0642B8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:11:52","StartupUTC":"2022-06-13T08:47:15","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2022-06-10T11:36:04","SaveCount":20}}
09:59:07.033 MQT: stat/tasmota_0642B8/STATUS2 = {"StatusFWR":{"Version":"11.1.0.4(tasmota)","BuildDateTime":"2022-06-13T07:51:22","Core":"2_0_4dev","SDK":"v.4.4.0-264-gd327997163","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"415/699"}}
09:59:07.055 MQT: stat/tasmota_0642B8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Joulco2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000","00000000"]}}
09:59:07.087 MQT: stat/tasmota_0642B8/STATUS4 = {"StatusMEM":{"ProgramSize":1396,"Free":1856,"Heap":150,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9F9AC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","1434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
09:59:07.105 MQT: stat/tasmota_0642B8/STATUS5 = {"StatusNET":{"Hostname":"tasmota-0642B8-0696","IPAddress":"#redacted#","Gateway":"#redacted#","Subnetmask":"255.255.255.0","DNSServer1":"#redacted#","DNSServer2":"0.0.0.0","Mac":"#redacted#","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
[... 5 lines cut ...]
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
No error in logs

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Unsure if it can be reproduced with any SHT10 sensor or if mine is different.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Correct values to be displayed as was achieved when I made the code changes.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Not applicable since already explained in description

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Not applicable since already explained in description

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-06-13:
Thx for your research.

As you compile yourself could you pls try if the available ``TasShiftIn()`` and ``TasShiftOut()`` functions do solve your issue too? 

Just replace the current ``shiftIn()`` and ``shiftOut()`` functions and report back pls.

arendst, 2022-06-13:
Just tested it with command ``cpufrequency 160``. In that case the original driver indeed reports strange values.

With the suggested changes it works fine. Currently testing on ESP32S3 at 240MHz and see if works there too....

arendst, 2022-06-13:
Yes works too (once) but seems to hang the device with a devidebyzero exception. Probably caused by the 240MHz clock of the ESP32S3 and the crappy SHT1X device/driver heavily depending on timely bitshifting.

Time to choose a real I2C device instead of the malfunctioning SHT1X

Anyway I will update the driver with TasShifts and release in dev ASAP.

WGrobler, 2022-06-13:
I can confirm that changing shiftIn to TasShiftIn fixed the problem

arendst, 2022-06-13:
Thx for your reply.

Never fixed an issue as quick as this one ;-)

WGrobler, 2022-06-13:
> Thx for your reply.
> 
> Never fixed an issue as quick as this one ;-)

Thanks for your help

closed 2022-06-13 by WGrobler