arendst/Tasmota #15794: ESP32 Sendmail not working with Ethernet mode

enesbcs, 2022-06-14:
### PROBLEM DESCRIPTION
With an Ethernet capable ESP32 (WT32-ETH01) when disabling Wifi and using only wired network, Sendmail always fails with "connection lost".
When enabling Wifi with "wifi 1" sendmail works, after "wifi 0" sendmail fails.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WT32-ETH01
- [ ] Tasmota binary firmware version number used: 2022.06.07
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:17:04.299 CMD: STATUS 0
19:17:04.306 RSL: STATUS = {"Status":{"Module":5,"DeviceName":"pwatch1","FriendlyName":["pwatch1"],"Topic":"tasmota_A153E4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:17:04.321 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:12:16","StartupUTC":"2022-06-14T18:04:48","Sleep":50,"CfgHolder":4617,"BootCount":39,"BCResetTime":"2022-05-29T16:20:14","SaveCount":103}}
19:17:04.334 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022.06.07 09:30:28","Core":"2_0_3","SDK":"v.4.4.0","CpuFrequency":80,"Hardware":"ESP32-D0WD","CR":"378/699"}}
19:17:04.341 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wolfcave2",""],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00004000","00004000"]}}
19:17:04.368 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1526,"Free":1856,"Heap":142,"StackLowMark":8,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["0000040E","97DA079E","0019A001","B7F7BFCF","05DA9B00","E0360D47","400880D2","20200000","0434482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,27,29,34,35,38,40,50,52,56,59,62,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
19:17:04.386 RSL: STATUS5 = {"StatusNET":{"Hostname":"pwatch1","IPAddress":"0.0.0.0","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"A8:03:2A:A1:53:E4","Ethernet":{"Hostname":"pwatch1-eth","IPAddress":"192.168.2.90","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"A8:03:2A:A1:53:E7"},"Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":17.0}}
19:17:04.402 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A153E4","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:17:04.412 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-06-14T18:17:04","Local":"2022-06-14T19:17:04","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:54"}}
19:17:04.427 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-06-14T19:17:04","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
19:17:04.435 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-06-14T19:17:04","Uptime":"0T00:12:16","UptimeSec":736,"Heap":143,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":37}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just try to send an email with gmail smtp without wifi

### EXPECTED BEHAVIOUR

WIFI 1
19:05:39.042 CMD: sendmail [smtp.gmail.com:465:sender@gmail.com:apppassword:sender@gmail.com:receiver@gmail.com:Powerwatch1 alarm] Test
19:05:39.057 heap: 158916
19:05:39.058 mailsize: 112
19:05:39.059 smtp.gmail.com - 465 - sender@gmail.com - apppassword
19:05:39.060 sender@gmail.com - receiver@gmail.com - Powerwatch1 alarm -  Test
19:05:39.072 ESP32mail: > C: ESP Mail Client v1.2.0
19:05:39.074 ESP32mail: > C: connect to SMTP server
19:05:39.075 ESP32mail: > C: host > smtp.gmail.com
19:05:39.086 ESP32mail: > C: port > 465
19:05:39.088 ESP32mail: > C: starting socket
19:05:39.106 ESP32mail: > C: connecting to Server
19:05:39.143 ESP32mail: > C: seeding the random number generator
19:05:39.147 ESP32mail: > C: setting up the SSL/TLS structure
19:05:39.151 ESP32mail: ! W: Skipping SSL Verification. INSECURE!
19:05:39.152 ESP32mail: > C: setting hostname for TLS session
19:05:39.165 ESP32mail: > C: performing the SSL/TLS handshake
19:05:42.582 ESP32mail: > C: verifying peer X.509 certificate
19:05:42.586 ESP32mail: > C: smtp server connected
19:05:42.591 ESP32mail: < S: 220 smtp.gmail.com ESMTP b11-20020a056402350b00b0042de8155fa1sm7825220edd.0 - gsmtp
19:05:42.592 ESP32mail: > C: send smtp command, HELO
19:05:42.654 ESP32mail: < S: 250-smtp.gmail.com at your service, [213.157.97.108]
19:05:42.660 ESP32mail: < S: 250-SIZE 35882577
19:05:42.663 ESP32mail: < S: 250-8BITMIME
19:05:42.668 ESP32mail: < S: 250-AUTH LOGIN PLAIN XOAUTH2 PLAIN-CLIENTTOKEN OAUTHBEARER XOAUTH
19:05:42.674 ESP32mail: < S: 250-ENHANCEDSTATUSCODES
19:05:42.678 ESP32mail: < S: 250-PIPELINING
19:05:42.682 ESP32mail: < S: 250-CHUNKING
19:05:42.687 ESP32mail: < S: 250 SMTPUTF8
19:05:42.688 ESP32mail: > C: send smtp command, AUTH PLAIN
19:05:42.690 ESP32mail: > C: sender@gmail.com
19:05:42.704 ESP32mail: > C: sender@gmail.com ****************
19:05:43.076 ESP32mail: < S: 235 2.7.0 Accepted
19:05:43.079 ESP32mail: > C: send Email
19:05:43.080 ESP32mail: > C: send message header
19:05:43.126 ESP32mail: < S: 250 2.1.0 OK b11-20020a056402350b00b0042de8155fa1sm7825220edd.0 - gsmtp
19:05:43.176 ESP32mail: < S: 250 2.1.5 OK b11-20020a056402350b00b0042de8155fa1sm7825220edd.0 - gsmtp
19:05:43.178 ESP32mail: > C: send message body
19:05:43.207 ESP32mail: > C: Finish the message sending
19:05:43.698 ESP32mail: < S: 250 2.0.0 OK b11-20020a056402350b00b0042de8155fa1sm7825220edd.0 - gsmtp
19:05:43.703 ESP32mail: < S: 250 2.0.0 OK b11-20020a056402350b00b0042de8155fa1sm7825220edd.0 - gsmtp
19:05:43.719 ESP32mail: < S: 250 2.0.0 OK b11-20020a056402350b00b0042de8155fa1sm7825220edd.0 - gsmtp
19:05:43.722 ESP32mail: > C: terminate the SMTP session
19:05:44.106 ESP32mail: < S: 250 2.0.0 OK b11-20020a056402350b00b0042de8155fa1sm7825220edd.0 - gsmtp
19:05:44.109 ESP32mail: > C: Message sent successfully
19:05:44.126 ESP32mail: > C: cleaning SSL connection
19:05:44.129 ESP32mail: > C: cleaning SSL connection
19:05:44.131 ESP32mail: > C: cleaning SSL connection
19:05:44.135 RSL: RESULT = {"Sendmail":"Done"}

------------------

19:07:41.975 CMD: wifi 0
19:07:41.982 RSL: RESULT = {"Wifi":"OFF","Mode":"11n"}

19:08:06.349 CMD: ping smtp.gmail.com
19:08:06.361 RSL: RESULT = {"Ping":"Done"}
19:08:07.503 RSL: RESULT = {"Ping":{"smtp.gmail.com":{"Reachable":true,"IP":"142.250.27.109","Success":1,"Timeout":0,"MinTime":27,"MaxTime":27,"AvgTime":27}}}

19:08:39.819 CMD: sendmail [smtp.gmail.com:465:sender@gmail.com:apppassword:sender@gmail.com:receiver@gmail.com:Powerwatch1 alarm] Test
19:08:39.835 heap: 180140
19:08:39.837 mailsize: 112
19:08:39.840 smtp.gmail.com - 465 - sender@gmail.com - apppassword
19:08:39.843 sender@gmail.com - receiver@gmail.com - Powerwatch1 alarm -  Test
19:08:39.850 ESP32mail: > E: connection lost
19:08:39.855 RSL: RESULT = {"Sendmail":"Wrong parameters"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

enesbcs, 2022-06-15:
FYI: I've downgraded the unit to Tasmota 9.4.0.4 and e-mail sending works with Wifi off, and Ethernet on.

Jason2866, 2022-06-15:
Please try latest Tasmota development version. There was a fix regarding Ethernet.

enesbcs, 2022-06-19:
Compiled new development build:

Program verzió	12.0.1.2(tasmota)
Build ideje	2022.06.19 19:03:24
Core/SDK verzió	2_0_3/v4.4.1

Hosztnév	tasmota-2A1B54-6996-eth
MAC cím	24:0A:C4:2A:1B:57
IP cím (eth)	192.168.2.42
Átjáró	192.168.2.1
Alhálózati maszk	255.255.255.0
DNS szerver1	192.168.2.1
DNS szerver2	0.0.0.0

When Wifi enabled, sendmail works. When Wifi disabled and Ethernet enabled, unit accessible on the network, it can ping smtp.google.com so internet is reachable in fact.

"Ping":{"smtp.google.com":{"Reachable":true,"IP":"142.250.27.27","Success":1,"Timeout":0,"MinTime":28,"MaxTime":28,"AvgTime":28}}}

->But sendmail still not works in ethernet mode!

18:27:25.822 ESP32mail: > E: connection lost
18:27:25.837 RSL: RESULT = {"Sendmail":"Wrong parameters"}

I guess there are some problems inside the integrated ESP Mail Client v1.2.0 which is currently at [v2.4.1](https://github.com/mobizt/ESP-Mail-Client) at these days. Either it still tries to reach disabled wifi interface despite ethernet is available OR esp-mail-client has some ethernet bug.

Which is very interesting, as Tasmota 9.4.0.4 was the [ first version which has ESP Mail Client v.1.2.0](https://github.com/arendst/Tasmota/tree/acd545f1bfa3a726e276aac287ed62848fa5aa33) but this is the only version i found that works with sendmail feature through Ethernet... downgraded back to 9.4.0.4.

Jason2866, 2022-06-19:
@enesbcs Have you tried to replace the Mail lib with the latest version?
Tasmota v9.4.0.4 use a older Arduino core. A lot changed under the hood since that time...

enesbcs, 2022-06-20:
Yes i've tried esp-mail-client 1.5.4, 1.5.10, 2.2.4 and 2.4.1. Although it is not a drop-in replacement, as this lib integrated inside Tasmota was heavily modified.
Tried to comment out SD card an file system routines in mail lib, but when finally got Tasmota compiled with these newer libs, all i got was an immediate reboot after issueing sendmail command.

arendst, 2022-06-20:
I have it working. The test for ethernet connection fails for some reason. Forcing connected makes it work.

I'll dive in to find a solid solution and will update file ``ESP_Mail_Client.cpp`` accordingly. Hold your breath....

Jason2866, 2022-06-21:
Closing since the issue is fixed. Feel free to reopen if the fix is not working for you

closed 2022-06-21 by Jason2866