arendst/Tasmota #15809: Mqtt failed connection rc -4

sercio, 2022-06-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
on Sonoff (simple), after upgrade to 12.0 version, i received a connection error with mqttserver rc -4

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

tasmota info:
<html><body>

rogram Version | 12.0.0(tasmota)
-- | --
2022-06-16T11:07:55
2_7_4_9/2.2.2-dev(38a443e)
0T00:00:39
551 at 0xFA000
287
Software/System restart
Sonoff
 
famsd-ve (82%, -59 dBm) 11n
sonoff1-0421
80:7D:3A:67:61:A5
192.168.2.20
192.168.2.1
255.255.255.0
151.5.216.150
151.5.216.15
 
Enabled
 
192.168.2.19
1883
mqttuser
sonoff1
sonoff1
cmnd/sonoffs/
cmnd/sonoff1/
cmnd/sonoff1_fb/
Disabled
 
None
 
6775205 (ESP8266EX)
0x146085
1024 KB
1024 KB
618 KB
384 KB
27.1 KB

</body>
</html>

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:22:21.380 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,4704],"FLAG":0,"BASE":18}
18:22:21.587 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
18:22:21.841 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:23:29.556 CMD: status 0
18:23:29.562 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Sonoff","FriendlyName":["Sonoff"],"Topic":"sonoff1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:23:29.568 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:43:36","StartupUTC":"2022-06-16T16:39:53","Sleep":50,"CfgHolder":4617,"BootCount":287,"BCResetTime":"2020-06-05T12:15:46","SaveCount":551,"SaveAddress":"FA000"}}
18:23:29.572 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.0(tasmota)","BuildDateTime":"2022-06-16T11:07:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}}
18:23:29.577 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["famsd-ve",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000","00000000"]}}
18:23:29.587 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":618,"Free":384,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
18:23:29.594 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff1-0421","IPAddress":"192.168.2.20","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"151.5.216.150","DNSServer2":"151.5.216.15","Mac":"80:7D:3A:67:61:A5","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
18:23:29.597 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.19","MqttPort":1883,"MqttClientMask":"sonoff1","MqttClient":"sonoff1","MqttUser":"mqttuser","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:23:29.603 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-06-16T17:23:29","Local":"2022-06-16T18:23:29","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:55"}}
18:23:29.608 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-06-16T18:23:29"}}
18:23:29.613 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-06-16T18:23:29","Uptime":"0T00:43:36","UptimeSec":2616,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"famsd-ve","BSSId":"0C:B6:D2:5F:49:20","Channel":11,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
18:24:24.072 RSL: RESULT = {"WebLog":4}
18:24:24.075 SRC: Backlog
18:24:24.078 CMD: Grp 0, Cmd 'MQTTLOG', Idx 1, Len 1, Pld 0, Data '0'
18:24:24.082 RSL: RESULT = {"MqttLog":0}
18:24:24.085 SRC: Backlog
18:24:24.087 CMD: Grp 0, Cmd 'SYSLOG', Idx 1, Len 1, Pld 0, Data '0'
18:24:24.090 RSL: RESULT = {"SysLog":{"0":{"Active":"0"}}}
18:24:24.094 SRC: Backlog
18:24:24.096 CMD: Grp 0, Cmd 'LOGHOST', Idx 1, Len 1, Pld 1, Data '1'
18:24:24.098 CFG: CR 372/699, Busy 0
18:24:24.101 RSL: RESULT = {"LogHost":""}
18:24:24.105 SRC: Backlog
18:24:24.108 CMD: Grp 0, Cmd 'LOGPORT', Idx 1, Len 3, Pld 514, Data '514'
18:24:24.111 RSL: RESULT = {"LogPort":514}
18:24:24.115 SRC: Backlog
18:24:24.117 CMD: Grp 0, Cmd 'TELEPERIOD', Idx 1, Len 3, Pld 300, Data '300'
18:24:24.120 RSL: RESULT = {"TelePeriod":300}
18:24:24.652 CFG: Saved to flash at F9, Count 552, Bytes 4096
18:24:24.814 RSL: STATE = {"Time":"2022-06-16T18:24:24","Uptime":"0T00:44:31","UptimeSec":2671,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":47,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"famsd-ve","BSSId":"0C:B6:D2:5F:49:20","Channel":11,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:24:27.667 WIF: Checking connection...
18:24:47.680 WIF: Checking connection...
18:24:52.511 HTP: Main Menu
18:24:53.735 HTP: Console
18:24:54.415 WIF: Sending Gratuitous ARP
18:25:08.207 WIF: Checking connection...
18:25:16.445 MQT: Attempting connection...
18:25:20.761 MQT: Connect failed to 192.168.2.19:1883, rc -4. Retry in 120 sec
18:25:32.422 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
 I have update to 12.0 also a Shelly 1 and it work fine....

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-06-16:
Do not waste time opening issues just because you got an error code. As [explained](https://tasmota.github.io/docs/MQTT/#return-codes-rc) in the docs, return code -4 [means](https://pubsubclient.knolleary.net/api.html#state) timeout, which is outside the control of Tasmota.

sercio, 2022-06-16:
Hello,
Yes, but is not a simple timeout.... in the mosquitto log i can see:

1655412211: New client connected from 192.168.2.20:59712 as sonoff1 (p2, c1, k30, u'mqttuser').
1655412211: Client sonoff1 closed its connection.

So, it seems that is sonoff that close the connection...

Ottenere Outlook per Android<https://aka.ms/AAb9ysg>

arendst, 2022-06-16:
I suggest you scan your moquitto log for "sonoff1" having different IP addresses. This tells you where to look (unique client ids) and what to change.

sercio, 2022-06-17:
Hi,
I have scanned the log.... sonoff1 has fixed IP, and nothing is changed. Now also the shelly with tasmota v12.0 has the same problem.... the other devices using mosquitto have no problem. in the mosquitto log I find:

1655448633: New connection from 192.168.2.38:62635 on port 1883.
1655448637: New client connected from 192.168.2.38:62635 as shelly-c (p2, c1, k30, u'mqttuser').
1655448637: Client shelly-c closed its connection.

sercio, 2022-06-17:
Hi,
just another detail: withe the tasmota minimal v12 the connection to mosquitto works fine and it's stable.
The problem is with the full v12 (both English and IT).

iointerrupt, 2022-06-17:
I have having the same issue, except I am failing with an rc = -5 (which is more due to credentials).  But nothing is different.  I upgraded from 11.1 (via minimal) and MQTT stopped working with rc = -5.  Then I reverted back to 11.1 and MQTT is working again.  I am using a Sonoff RF Bridge v2.2 (white one) using Tasmota Sensors builds.  Tried with the 12.0 regular Tasmota and same thing happens.  Have NOT tried MQTT on minimal though since I reverted back to 11.1 after multiple attempts on 12.

iointerrupt, 2022-06-17:
> The -4 is another issue than -5.
> 
> Anyway. Release v12.0.1 is on the brink.

Thank you guys.  Appreciate this project and all the hard work the devs put into it.

arendst, 2022-06-17:
-5 fixed in latest release https://github.com/arendst/Tasmota/releases/tag/v12.0.1

-4 pls try latest release and report back if it fixed your issue too (not sure it does).

a-baturin, 2022-06-17:
-4 is not fixed with 12.0.1

a-baturin, 2022-06-17:
One out of nine devices faced it. I don't see anything special with this device

a-baturin, 2022-06-17:
Connected after multiple attempts

sercio, 2022-06-17:
Hi,
no, sorry. it's no fixed -4 error

barbudor, 2022-06-17:
@sercio
Try MqttWifiTimeout with a value greater than 200
Start with `MqttWifiTimeout 1000` for example

sercio, 2022-06-18:
Hi,
with MqttWifiTimeout 1000 don't works, it's the same.
Sorry

sercio, 2022-06-18:
Hi,

with MqttWifiTimeout 2000 is the same. I think it's not a real timeout.

Everytime rc -4

Hi,

barbudor, 2022-06-18:
Can you check your options regarding MQTT over TLS? 
I'm not 100% sure but I would not be surprised if trying TLS on a broker where it is not enabled or the opposite could generate a timeout

adrianmoisey, 2022-06-18:
> One out of nine devices faced it. I don't see anything special with this device

I'm getting the same thing, 1 of about 12 devices is facing this issue.
I've tried a few options, but nothing seems to work.

Is there any debugging information that I can provide for this issue?

sercio, 2022-06-18:
> Can you check your options regarding MQTT over TLS? I'm not 100% sure but I would not be surprised if trying TLS on a broker where it is not enabled or the opposite could generate a timeout

Hi, 
my mosquitto is not configured to use TLS, and also the Sonoff with Tasmota v12.0.

barbudor, 2022-06-18:
Are you using stndard tasmota.bin(.gz) from github or ota.tasmota.com/tasmota/release ?
If you are compiling your own, can you please share what is your user_config_override.h or any other changes you may have done

sercio, 2022-06-18:
I use the standard tasmota.bin(.gz) from github.

barbudor, 2022-06-18:
I can't reproduce. I tried to twek my settings in every direction but MQTT always connect seamlessly

The only change in MQTT between 11.1.0 and 12.0.1 (now the DNS stuff has been reverted) is the introduction of SetOption140 to enable persistent MQTT sessions.
But even if it was inadvertedly set ON it should break the connection

I tried to compare minimal vs standard and I can't see anything obvious which could explain your case.

arendst, 2022-06-18:
Is still believe user has two devices with same clientid:

> Has another client started that used the same ClientIdentifier?
If a second client is started, or the same client is restarted, using the same ClientIdentifier, the first connection to the first client is dropped.

Not so strange considering a clientid of "sonoff1"....

adrianmoisey, 2022-06-18:
> Just for test execute command `MqttClient 1` and see what happens

I gave this a try, and I get the same result:

```
mosquito  | 1655555525: New client connected from 192.168.1.117 as DVES_3EC08D (c1, k30, u'homeassistant').
mosquito  | 1655555525: Socket error on client DVES_3EC08D, disconnecting.
```

I also tried making the client  the string `random` and I still get the same thing.

adrianmoisey, 2022-06-18:
I don't know if this is useful in anyway, but I've noticed that I can't ping the device when it's trying to connect to MQTT.

These two messages seem to be about 3 seconds apart:
```
mosquito  | 1655556108: New client connected from 192.168.1.117 as random (c1, k30, u'homeassistant').
mosquito  | 1655556108: Socket error on client random, disconnecting.
```

And during that time, I get packet loss when pinging the device:
```
64 bytes from 192.168.1.117: icmp_seq=44 ttl=255 time=13.918 ms
64 bytes from 192.168.1.117: icmp_seq=45 ttl=255 time=10.518 ms
64 bytes from 192.168.1.117: icmp_seq=46 ttl=255 time=4.134 ms
Request timeout for icmp_seq 47
Request timeout for icmp_seq 48
Request timeout for icmp_seq 49
64 bytes from 192.168.1.117: icmp_seq=48 ttl=255 time=2564.675 ms
64 bytes from 192.168.1.117: icmp_seq=49 ttl=255 time=1573.008 ms
64 bytes from 192.168.1.117: icmp_seq=50 ttl=255 time=581.271 ms
```

Again, unsure if this is useful for debugging, but it's something I noticed.

sercio, 2022-06-18:
Hi,
the situation is this:

I have two device with tasmota, sonoff basic and shelly, both with 12.0 the first with the error rc -4 the second now connected.  I haven't other device with id sonoff1 (I have just another sonoff but with mqtt not configured, it's managed by the eWelink). I have changed the clientid (now is websalotto) with the same result (error rc -4). If I use the minimal firmware the device works with both clientid (sonoff1 and websalotto).

sercio, 2022-06-18:
Hi,
with the command  MqttClient 1  my sonoff reboot.

adrianmoisey, 2022-06-18:
FWIW, I downgraded back to 11.1.0, with the same settings, and the MQTT is working fine now

sercio, 2022-06-18:
It's the same for me. After downgrade back to 11.1 mqtt is working fine, with the same settings.

ahmaddxb, 2022-06-18:
I was having rc -4 issue with one of my devices and I think I fixed it by changing one of my rules. 

Not sure if that was the cause but its the only thing I changed before it started working again. Perhaps take a look if you have used rules on your device, try removing them and restart and see if its fixed.

before
rule2 on system#boot do power3 on ENDON
after
rule2 on System#Boot do power3 on ENDON

a-baturin, 2022-06-19:
No rules were set on my devices

sercio, 2022-06-20:
Also for me it's ok with 12.0.2.
Thanks