arendst/Tasmota #15816: sunrise/sunset time in "Timer" section swapped

steinthomas, 2022-06-17:
### PROBLEM DESCRIPTION
The sunrise ("Sonnenaufgang") and sunset ("Sonnenuntergang") time in the "timer" section in v12.0.0 is swapped and displayed wrong.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 12.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
screenshot
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-06-17:
Not seeing that with version 12. What are your settings for TimeDST/TimeSTD/Timezone?
[image]

barbudor, 2022-06-17:
I would bet your Tasmota doesn't get a proper time from NTP server
Check in the console that the logs have the expect date and time

sfromis, 2022-06-17:
Saw that there may be an issue with DNS lookup for NTP/MQTT domain names, maybe this is the case here....

barbudor, 2022-06-17:
Was getting to the same conclusion as I when using a name for the server, I don't get time. Works fine with IP address (which is my default setup)

arendst, 2022-06-17:
I'll do a re-release today onve I verified the solution.

Jason2866, 2022-06-17:
related to #15809

steinthomas, 2022-06-17:
THX!!! Problem fixed...sunrise/sunset time now correct...
[image]
.

closed 2022-06-17 by steinthomas