arendst/Tasmota #15825: Exception using physical shutter buttons

epandi, 2022-06-19:
### PROBLEM DESCRIPTION
In Tasmota 11.0.0 the physical buttons of my shutters (using the default DS-102 2 button template) the shutters would go up / down using a custom configuration ({"ShutterButton1":"1 100 50 74 100 - - - 1 1|2 0 50 74 100 - - - 1 1"}). Yesterday I've upgraded all my Tasmota devices to 12.0.1 and noticed the physical shutter buttons would not function anymore. Pressing the "close" button, the shutter whould move for a tiny bit of a second, then the led starts blinking blue/red and the device is restarted. Pressing the "open" button would not do anything at all.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): DS-102 2 Gang Switch (https://templates.blakadder.com/DS-102_2.html)
- [x] Tasmota binary firmware version number used: 12.0.0
  - [x] Pre-compiled
- [x] Flashing tools used: web upgrade
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:26:09.798 MQT: stat/hobbyblinds/RESULT = {"NAME":"DS-102 2 Gang","GPIO":[576,321,0,32,225,33,0,0,0,224,320,1,0,0],"FLAG":0,"BASE":18}
16:26:10.012 MQT: stat/hobbyblinds/RESULT = {"Module":{"0":"DS-102 2 Gang"}}
16:26:10.265 MQT: stat/hobbyblinds/RESULT = {"GPIO0":{"576":"LedLink_i"},"GPIO1":{"321":"Led_i2"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"225":"Relay2"},"GPIO5":{"33":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"224":"Relay1"},"GPIO14":{"320":"Led_i1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
16:26:37.683 MQT: stat/hobbyblinds/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
16:26:37.900 MQT: stat/hobbyblinds/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":40,"Free":471,"Rules":"ON System#Boot DO Shutterposition1 ENDON"}}
16:26:38.151 MQT: stat/hobbyblinds/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":67,"Free":444,"Rules":"ON Shutter1#Position DO Publish stat/%topic%/SHUTTER1 %value% ENDON"}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:27:53.111 MQT: stat/hobbyblinds/STATUS = {"Status":{"Module":0,"DeviceName":"HobbyBlinds","FriendlyName":["HobbyBlinds","HobbyBlinds"],"Topic":"hobbyblinds","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:27:53.151 MQT: stat/hobbyblinds/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:00:31","StartupUTC":"2022-06-19T14:27:22","Sleep":50,"CfgHolder":4617,"BootCount":108,"BCResetTime":"2020-04-30T22:20:10","SaveCount":1891,"SaveAddress":"FB000"}}
16:27:53.185 MQT: stat/hobbyblinds/STATUS2 = {"StatusFWR":{"Version":"12.0.1(tasmota)","BuildDateTime":"2022-06-17T14:38:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}
16:27:53.207 MQT: stat/hobbyblinds/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxx","xxxxxxxxxx"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C809","1405C80001000600003C5A0A000000000000","400000C8","00006000","00004000","00000000"]}}
16:27:53.243 MQT: stat/hobbyblinds/STATUS4 = {"StatusMEM":{"ProgramSize":618,"Free":384,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
16:27:53.284 MQT: stat/hobbyblinds/STATUS5 = {"StatusNET":{"Hostname":"hobbyblinds-4319","IPAddress":"192.168.1.31","Gateway":"192.168.0.1","Subnetmask":"255.255.252.0","DNSServer1":"8.8.4.4","DNSServer2":"0.0.0.0","Mac":"EC:FA:BC:4B:10:DF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:27:53.312 MQT: stat/hobbyblinds/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B10DF","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:27:53.338 MQT: stat/hobbyblinds/STATUS7 = {"StatusTIM":{"UTC":"2022-06-19T14:27:53","Local":"2022-06-19T16:27:53","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:46","Sunset":"21:56"}}
16:27:53.359 MQT: stat/hobbyblinds/STATUS10 = {"StatusSNS":{"Time":"2022-06-19T16:27:53","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}}
16:27:53.376 MQT: stat/hobbyblinds/STATUS11 = {"StatusSTS":{"Time":"2022-06-19T16:27:53","Uptime":"0T00:00:31","UptimeSec":31,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","BSSId":"28:6C:07:98:E3:95","Channel":6,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:07"}}}
16:27:53.418 MQT: stat/hobbyblinds/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["40252be6","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40230fd8","4023dd93","4023ea6c","401059a3","401026f0","4025269c","40101917","40101b9f","4028e69b","402488f2","40248c9d","40248c9d","4026d400","4024b81c","40101b9f","4023eedc","40101206","4021e866","4023a1b8","402285ac","40101206","40101206","4023a226","4023d3d4","4023d48c","40100a7c","4024c7b0","40101e95"]}}
16:27:53.464 MQT: stat/hobbyblinds/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":330,"Close":330,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
:00:00.001 HDW: ESP8266EX
00:00:00.046 CFG: Loaded from flash at FB, Count 1891
00:00:00.051 QPC: Count 1
00:00:00.069 Project tasmota - HobbyBlinds Version 12.0.1(tasmota)-2_7_4_9(2022-06-17T14:38:38)
00:00:04.513 WIF: Connecting to AP1 xxxxxxxxxx Channel 6 BSSId 28:6C:07:98:E3:95 in mode 11n as hobbyblinds-4319...
00:00:05.756 WIF: Connected
00:00:06.009 HTP: Web server active on hobbyblinds-4319 with IP address 192.168.1.31
00:00:06.508 QPC: Reset
16:29:07.014 MQT: Attempting connection...
16:29:07.030 MQT: Connected
16:29:07.035 MQT: tele/hobbyblinds/LWT = Online (retained)
16:29:07.037 MQT: cmnd/hobbyblinds/POWER = 
16:29:07.050 MQT: tele/hobbyblinds/INFO1 = {"Info1":{"Module":"DS-102 2 Gang","Version":"12.0.1(tasmota)","FallbackTopic":"cmnd/DVES_4B10DF_fb/","GroupTopic":"cmnd/tasmotas/"}}
16:29:07.062 MQT: tele/hobbyblinds/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"hobbyblinds-4319","IPAddress":"192.168.1.31"}}
16:29:07.082 MQT: tele/hobbyblinds/INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40252be6","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40230fd8","4023dd93","4020e228","4023ea6c","4025269c","40101917","40101b9f","4028e69b","402488f2","40248c9d","40248c9d","4026d400","4024b81c","40101b9f","4023eedc","40101206","4021e866","4023a1b8","402285ac","40101206","40101206","4023a226","4023d3d4","4023d48c","40100a7c","4024c7b0","40101e95"]},"BootCount":108}}
16:29:07.126 MQT: stat/hobbyblinds/RESULT = {"POWER1":"OFF"}
16:29:07.129 MQT: stat/hobbyblinds/POWER1 = OFF
16:29:07.135 MQT: stat/hobbyblinds/RESULT = {"POWER2":"OFF"}
16:29:07.140 MQT: stat/hobbyblinds/POWER2 = OFF
16:29:07.156 RUL: SYSTEM#BOOT performs "Shutterposition1"
[... cut ...]
16:29:11.502 MQT: tele/hobbyblinds/SENSOR = {"Time":"2022-06-19T16:29:11","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
```

### TO REPRODUCE
Make sure the sutterbuttons are setup as:
{"ShutterButton1":"1 100 50 74 100 - - - 1 1|2 0 50 74 100 - - - 1 1"}

### EXPECTED BEHAVIOUR
The shutter moving up or down without an exception.
The issue is resolved settting up the buttons as follow:
Backlog Shutterbutton1 up; Shutterbutton2 down;

**(Please, remember to close the issue when the problem has been addressed)**

michal88sjz, 2022-06-22:
Hello,
Same issue here:
Backlog Template; Module; GPIO 255
`07:45:03.611 MQT: stat/tasmota_24A140/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,33,34,0,0,0,0,0,0],"FLAG":0,"BASE":1}
07:45:03.845 MQT: stat/tasmota_24A140/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
07:45:04.106 MQT: stat/tasmota_24A140/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"33":"Button2"},"GPIO33":{"34":"Button3"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}`

Backlog Rule1; Rule2; Rule3
`07:47:02.716 MQT: stat/tasmota_24A140/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
07:47:02.946 MQT: stat/tasmota_24A140/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
07:47:03.196 MQT: stat/tasmota_24A140/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}`

Usage of pysical button (with toggle funbction)
`00:00:00.003 HDW: ESP32-D0WD-V3 
00:00:00.071 UFS: FlashFS mounted with 304 kB free
00:00:00.141 CFG: Loaded from File, Count 2238
00:00:00.157 QPC: Count 1
00:00:00.345 BRY: Berry initialized, RAM used=3731 bytes
00:00:00.378 Project tasmota - Drzwi_Astra Version 12.0.2(tasmota)-2_0_3(2022-06-20T12:41:24)
00:00:01.054 WIF: Connecting to AP2 Duga Channel 6 BSSId 00:31:92:1B:3F:DD in mode 11n as tasmota-24A140-0320...
00:00:03.202 WIF: Connected
00:00:03.467 HTP: Web server active on tasmota-24A140-0320 with IP address 192.168.0.151
07:56:27.506 MQT: Attempting connection...
07:56:27.559 MQT: Connected
07:56:27.565 MQT: tele/tasmota_24A140/LWT = Online (retained)
07:56:27.569 MQT: cmnd/tasmota_24A140/POWER = 
07:56:27.578 MQT: tele/tasmota_24A140/INFO1 = {"Info1":{"Module":"Sonoff Dual R3","Version":"12.0.2(tasmota)","FallbackTopic":"cmnd/Sonoff_Drz_Astra_fb/","GroupTopic":"cmnd/tasmotas/"}}
07:56:27.595 MQT: tele/tasmota_24A140/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-24A140-0320","IPAddress":"192.168.0.151"}}
07:56:27.614 MQT: tele/tasmota_24A140/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":62}}
07:56:27.619 MQT: stat/tasmota_24A140/RESULT = {"POWER1":"OFF"}
07:56:27.623 MQT: stat/tasmota_24A140/POWER1 = OFF
07:56:27.632 MQT: stat/tasmota_24A140/RESULT = {"POWER2":"OFF"}
07:56:27.635 MQT: stat/tasmota_24A140/POWER2 = OFF
07:56:28.747 MQT: stat/tasmota_24A140/STATUS10 = {"StatusSNS":{"Time":"2022-06-22T07:56:28","ENERGY":{"TotalStartTime":"2021-11-05T13:26:40","Total":1.380,"Yesterday":0.002,"Today":0.007,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000]},"ESP32":{"Temperature":78.3},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"}}
07:56:29.065 C61: CSE7761 found
07:56:29.796 QPC: Reset
07:56:31.798 MQT: tele/tasmota_24A140/STATE = {"Time":"2022-06-22T07:56:31","Uptime":"0T00:00:10","UptimeSec":10,"Heap":155,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"Duga","BSSId":"00:31:92:1B:3F:DD","Channel":6,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:56:31.849 MQT: tele/tasmota_24A140/SENSOR = {"Time":"2022-06-22T07:56:31","ENERGY":{"TotalStartTime":"2021-11-05T13:26:40","Total":1.380,"Yesterday":0.002,"Today":0.007,"Period":[ 0, 0],"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000]},"ESP32":{"Temperature":77.2},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"}
00:00:00.379 Project tasmota - Drzwi_Astra Version 12.0.2(tasmota)-2_0_3(2022-06-20T12:41:24)
00:00:01.188 WIF: Connecting to AP2 Duga Channel 6 BSSId 00:31:92:1B:3F:DD in mode 11n as tasmota-24A140-0320...
00:00:03.203 WIF: Connected
00:00:03.468 HTP: Web server active on tasmota-24A140-0320 with IP address 192.168.0.151
07:57:30.516 MQT: Attempting connection...
07:57:30.553 MQT: Connected
07:57:30.559 MQT: tele/tasmota_24A140/LWT = Online (retained)
07:57:30.563 MQT: cmnd/tasmota_24A140/POWER = 
07:57:30.572 MQT: tele/tasmota_24A140/INFO1 = {"Info1":{"Module":"Sonoff Dual R3","Version":"12.0.2(tasmota)","FallbackTopic":"cmnd/Sonoff_Drz_Astra_fb/","GroupTopic":"cmnd/tasmotas/"}}
07:57:30.589 MQT: tele/tasmota_24A140/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-24A140-0320","IPAddress":"192.168.0.151"}}
07:57:30.608 MQT: tele/tasmota_24A140/INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"StoreProhibited","EPC":"4008ce2e","EXCVADDR":"00000000","CallChain":["4008ce2b","40129f45","4012a30d","4012a33d","40122011","4010759d","4014f813","400e7197"]},"BootCount":63}}
[... cut ...] "Downtime":"0T00:00:05"}}
07:59:35.711 MQT: tele/tasmota_24A140/SENSOR = {"Time":"2022-06-22T07:59:35","ENERGY":{"TotalStartTime":"2021-11-05T13:26:40","Total":1.380,"Yesterday":0.002,"Today":0.007,"Period":[ 0, 0],"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000]},"ESP32":{"Temperature":77.8},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"}`

Opening/Closing from webgui or via HA works as usually.

Jason2866, 2022-06-23:
@stefanbode Can you help?

001oc, 2022-06-24:
Hello,

same here.

as a quick and very dirty workarround I have done the following:

unconfig shutterbutton1
config a "virtual" button instead of the former button - in my config it is button #8

this are my other important configs so fare:
```
SetOption80 1 // Enable shutters support.
ShutterMode 4 // Enable "Shutter mode 4".
ShutterRelay1 1 // for relay Relay1 and Relay2.
SetOption73 1 // Enable detach buttons from relays.

```
After that you will have an approximately same function as with shutterbutton when you setup rules like the following:

 on button8#state=10 do shutteropen1 endon
 on button8#state=11 do shutterclose1 endon

Whats missing so fare is, that shutter stops when pressing the button while shutter is moving. That could be implemented within rules too when using vars, but I hope the error is fixed soon in tasmota and for the time being the "dirty workarround" will work for me.

stefanbode, 2022-06-24:
Hi gents. I'm currently On Holiday and back in July. There is no visible change that this function is now broken.

stefanbode, 2022-06-24:
I already pinged @arendst what has changed on 12 that may cause the exception. The driver itself was not changed between 11 and 12 as far as I know.

Mark-Muc, 2022-06-24:
have the same problem described in #15848
didn't see this thread .. #15848 now can be closed!!
Mark_Muc

001oc, 2022-06-24:
For me the problem was existing in 11 too. I updated from 10 to 11 the other day and didn't found the error directly because the button in my config is for backup reason and I hadn't test it after the update directly. Some days ago I found the error in 11 and tried to update to 12. The error was still there. A downgrade to 11 doesn't helped, a downgrade to 10 makes functioning the button again as expected. I will stick with 12 because of some changes that I am happy about - an my workarround is working goog enough for something I use for backup reason...

arendst, 2022-06-24:
Sorry, I have no shutters. 

Form the entries here one could conclude the error occurs on both ESP8266 and ESP32 ruling out SDK and Core software. It's also intermittent but alway results in exception 29 on different addresses. The ESP32 one could shine more light on the cause: ```07:57:30.608 MQT: tele/tasmota_24A140/INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"StoreProhibited","EPC":"4008ce2e","EXCVADDR":"00000000","CallChain":["4008ce2b","40129f45","4012a30d","4012a33d","40122011","4010759d","4014f813","400e7197"]},"BootCount":63}}
```
where
40129f45 = ShutterStartInit
4008ce2b = strcpy

For the ESP8266:
40231020 = ExecuteCommandPowerShutter

So I fear the issue is in the shuttercode.

As Stefan is on holiday I'll have a look in the driver and try to understand what's going on...

arendst, 2022-06-24:
I see some code trickery I don't like:
```
  while (uptime_Local+10 > TasmotaGlobal.uptime && (String)rules_vars[i] == "99") {
    loop();
  }
```
I guess calling ``loop()`` like here is bound to fail someday.

I need to simulate shutters.....

001oc, 2022-06-24:
maybe helpful is, that I have seen, that self compiled versions of tasmota, for me, seems to be more vulnerable concerning the button problem. I have tried precompiled versions from tasmotas ota section and have seen at first no error. When i pressed the button at least 20 times the error pop up sometimes. With a self compiled version I get the error directly and everytime I press the button. For testing purpose I have only changed the platformio_override.ini -> default_envs from "tasmota" to "tasmota32".

arendst, 2022-06-24:
[image]

My shutter testbed :-)

Jason2866, 2022-06-24:
Nice small motors 😁

001Tycoon, 2022-06-24:
Thanks, it works for me as it should

```
19:14:53.891 CMD: status 0
19:14:53.893 SRC: WebConsole from 172.16.1.8
19:14:53.895 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
19:14:53.901 MQT: stat/wz-rl-01/STATUS = {"Status":{"Module":0,"DeviceName":"Wohnzimmer Rolladen","FriendlyName":["Wohnzimmer Rolladen","Wohnzimmer Rolladen down","Wohnzimmer Rolladen stop"],"Topic":"wz-rl-01","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,4,4,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
19:14:53.935 MQT: stat/wz-rl-01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://172.16.0.254:9999/data/firmwares/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:53","StartupUTC":"2022-06-24T17:12:00","Sleep":50,"CfgHolder":4617,"BootCount":250,"BCResetTime":"2020-03-19T16:55:16","SaveCount":4080,"SaveAddress":"F4000"}}
19:14:53.967 MQT: stat/wz-rl-01/STATUS2 = {"StatusFWR":{"Version":"12.0.2.2(tasmota)","BuildDateTime":"2022.06.24 17:06:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"461/699"}}
19:14:53.995 MQT: stat/wz-rl-01/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":4,"MqttLog":1,"SysLog":4,"LogHost":"bananapi-m2.home.pri","LogPort":514,"SSId":["HOMEWLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["8402E02B","2805C80001000600003C5A0A000000000000","400002C9","00006000","00004000","00000000"]}}
19:14:54.032 MQT: stat/wz-rl-01/STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
19:14:54.059 MQT: stat/wz-rl-01/STATUS5 = {"StatusNET":{"Hostname":"wz-rl-01","IPAddress":"172.16.6.18","Gateway":"172.16.0.1","Subnetmask":"255.255.0.0","DNSServer1":"172.16.0.1","DNSServer2":"0.0.0.0","Mac":"A4:CF:12:EA:FC:B6","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:14:54.086 MQT: stat/wz-rl-01/STATUS6 = {"StatusMQT":{"MqttHost":"openhabianpi.home.pri","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EAFCB6","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:14:54.115 MQT: stat/wz-rl-01/STATUS7 = {"StatusTIM":{"UTC":"2022-06-24T17:14:54","Local":"2022-06-24T19:14:54","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:47","Sunset":"21:57"}}
19:14:54.145 MQT: stat/wz-rl-01/STATUS10 = {"StatusSNS":{"Time":"2022-06-24T19:14:54","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}}
19:14:54.175 MQT: stat/wz-rl-01/STATUS11 = {"StatusSTS":{"Time":"2022-06-24T19:14:54","Uptime":"0T00:02:54","UptimeSec":174,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"HOMEWLAN","BSSId":"E0:28:6D:9D:7C:BA","Channel":13,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:07"}}}
19:14:54.204 MQT: stat/wz-rl-01/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":260,"Close":250,"50perc":50,"Delay":0,"Opt":"0100","Calib":[111,333,556,778,1000],"Mode":"0"}}}
19:15:01.464 WIF: Sending Gratuitous ARP
```
```
19:12:31.218 HTP: Konsole
19:12:34.044 CMD: weblog 4
19:12:34.046 SRC: WebConsole from 172.16.1.8
19:12:34.048 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 4, Data '4'
19:12:34.053 MQT: stat/wz-rl-01/RESULT = {"WebLog":4}
19:12:40.829 SHT: Shtr1, Button 2 = 1 (single=1, double=2, tripple=3, hold=4)
19:12:40.831 SHT: Pos. in: payload  (0), payload 0, idx 1, src 3
19:12:40.833 SHT: lastsource 3:, real 520000, target 0, tiltreal: 0, tilttarget: 0, payload 0
19:12:40.835 SHT: Delay Start? var1 <99>=<>, max10s?
19:12:40.837 SRC: Shutter
19:12:40.839 SHT: Switched relay 2 by Shutter
19:12:40.851 MQT: tele/wz-rl-01/STATE = {"Time":"2022-06-24T19:12:40","Uptime":"0T00:00:40","UptimeSec":40,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"HOMEWLAN","BSSId":"E0:28:6D:9D:7C:BA","Channel":13,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:07"}}
19:12:40.859 MQT: stat/wz-rl-01/RESULT = {"POWER2":"ON"}
19:12:40.862 MQT: stat/wz-rl-01/POWER2 = ON (beibehalten)
19:12:40.865 MQT: stat/wz-rl-01/RESULT = {"Shutter1":{"Button2":1}}
19:12:40.909 RUL: POWER2#STATE=1 performs "LEDPower2 1"
19:12:40.911 SRC: Rule
19:12:40.913 CMD: Grp 0, Cmd 'LEDPOWER', Idx 2, Len 1, Pld 1, Data '1'
19:12:40.920 MQT: stat/wz-rl-01/RESULT = {"LedPower2":"ON"}
19:12:40.930 SHT: Shtr1 Real 520000, Start 520000, Stop -20000, Dir -1, Delay 0, Rtc 0.05 [s], Freq 0, PWM 0, Tilt 0
19:12:40.937 MQT: stat/wz-rl-01/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0,"Tilt":0}}
19:12:40.960 SHT: Time 1(1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 520000, nStop 0, Trgt -20000, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
[... cut ...]
19:12:55.397 MQT: stat/wz-rl-01/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
19:12:55.490 CFG: in Flash gespeichert an F9, Anzahl 4075, Bytes 4096
19:12:55.551 RUL: POWER1#STATE=0 performs "LEDPower1 0"
19:12:55.553 SRC: Rule
19:12:55.556 CMD: Grp 0, Cmd 'LEDPOWER', Idx 1, Len 1, Pld 0, Data '0'
19:12:55.562 MQT: stat/wz-rl-01/RESULT = {"LedPower1":"OFF"}
19:12:56.353 CFG: in Flash gespeichert an F8, Anzahl 4076, Bytes 4096
19:13:01.468 WIF: Sending Gratuitous ARP
```

many thanks for all the work

epandi, 2022-06-25:
Issue is resolved. Thanks for your awesome work Theo. And well done buying those shutters on such short amount of time 😉

closed 2022-06-25 by epandi

Mark-Muc, 2022-06-25:
.... my thanks for the quick fix and work. I just fllshed the device in question and it works like charm!!
Again thank you!!
Mark_Muc

```
08:43:34.860 MQT: stat/Rollo_OG_Gal_Sued/POWER2 = ON
08:43:34.866 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Button1":1}}
08:43:34.885 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":64,"Direction":-1,"Target":0,"Tilt":0}}
08:43:35.417 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":61,"Direction":-1,"Target":0,"Tilt":0}}
08:43:36.405 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":54,"Direction":-1,"Target":0,"Tilt":0}}
08:43:37.407 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":48,"Direction":-1,"Target":0,"Tilt":0}}
08:43:38.415 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":41,"Direction":-1,"Target":0,"Tilt":0}}
08:43:39.384 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":35,"Direction":-1,"Target":0,"Tilt":0}}
08:43:39.683 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"POWER2":"OFF"}
08:43:39.689 MQT: stat/Rollo_OG_Gal_Sued/POWER2 = OFF
08:43:40.445 MQT: stat/Rollo_OG_Gal_Sued/SHUTTER1 = 33
08:43:40.450 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":33,"Direction":0,"Target":33,"Tilt":0}}
08:43:40.463 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Button1":1}}
08:43:41.982 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"POWER1":"ON"}
08:43:41.988 MQT: stat/Rollo_OG_Gal_Sued/POWER1 = ON
08:43:41.994 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Button1":1}}
08:43:42.011 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":33,"Direction":1,"Target":106,"Tilt":0}}
08:43:42.382 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":35,"Direction":1,"Target":106,"Tilt":0}}
08:43:43.387 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":42,"Direction":1,"Target":106,"Tilt":0}}
08:43:44.398 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":48,"Direction":1,"Target":106,"Tilt":0}}
08:43:45.403 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":55,"Direction":1,"Target":106,"Tilt":0}}
08:43:46.402 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":61,"Direction":1,"Target":106,"Tilt":0}}
08:43:47.418 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":68,"Direction":1,"Target":106,"Tilt":0}}
08:43:48.418 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":74,"Direction":1,"Target":106,"Tilt":0}}
08:43:49.418 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":81,"Direction":1,"Target":106,"Tilt":0}}
08:43:50.417 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":87,"Direction":1,"Target":106,"Tilt":0}}
08:43:51.417 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":94,"Direction":1,"Target":106,"Tilt":0}}
08:43:52.421 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":106,"Tilt":0}}
08:43:53.415 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"POWER1":"OFF"}
08:43:53.421 MQT: stat/Rollo_OG_Gal_Sued/POWER1 = OFF
08:43:54.178 MQT: stat/Rollo_OG_Gal_Sued/SHUTTER1 = 100
08:43:54.184 MQT: stat/Rollo_OG_Gal_Sued/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
```

stefanbode, 2022-06-25:
Thanks Theo for fixing. For me it looks like I never have the chance to find it on my own. Regarding the loop() I will take care after holiday.

stefanbode, 2022-06-25:
The prestart is a requested option to execute a rule before a shutter operates. E.g. release a lock or power up additional hardware. The rule should execute always before a shutter start moving or change direction.

arendst, 2022-06-25:
As rules are processed async the current solution seems to be right. The problem is that a lot is happening in the loop other than rule processing so havoc is near. We have to find a better way to directly handle these limited preprocess actions. To start off document all known preprocess actions.

arendst, 2022-06-25:
Having thought it over I think if a user needs to do stuff before the shutter can move he will need to put the whole action into a rule instead of allowing the rule to act within the shutter code. That way Tasmota will stay in a safe mode without unexpected things that can happen.

So de-couple the up/down buttons and let it execute a rule event ending in the shutter move. The current 10 seconds delay (doing the loop()) is then also gone.

bennybubble, 2022-07-09:
thanks for solving this issue Arend, took my a couple of hours to find out .. :-)