arendst/Tasmota #15834: TasmotaGlobal.temperature_celsius includes ESP32 core temperature --> wrong compensation for Sensores

Links2004, 2022-06-20:
### PROBLEM DESCRIPTION

`TasmotaGlobal.temperature_celsius` includes `ESP32 Temperature` resulting in wrong compensation for Sensores.

for example the Absolute Humidity calculated here:

https://github.com/arendst/Tasmota/blob/c08561f67c722556655da7b8cbe1482d98aecd90/tasmota/tasmota_xsns_sensor/xsns_21_sgp30.ino#L86-L90

jumps from 8.8 g/m3 to 32.9 g/m3 and back, based on if the ESP32 Temperature where used or the connected Temperature sensor.
since there is a big difference in Temperature.

```
ESP32-DevKit
Air Sensor
MQ-2	1.02 ppm
MQ-3	null ppm
MQ-6	876.90 ppm
MQ-7	0.18 ppm
BME280 Temperature	24.0 °C
BME280 Humidity	36.6 %
BME280 Dew point	8.2 °C
BME280 Pressure	994.8 hPa
MHZ19B Carbon dioxide	410 ppm
MHZ19B Temperature	26.0 °C
SGP30 eCO₂	400 ppm
SGP30 TVOC	0 ppb
SGP30 Abs Humidity	35.2803 g/m3
ESP32 Temperature	52.8 °C
```

I found https://github.com/arendst/Tasmota/issues/12630 but my firmware is form this year.

tested with 
```
Program Version	11.1.0.3(tasmota)
Build Date & Time	2022-05-31T15:54:53
```
and
```
Program Version	12.0.1.2(tasmota)
Build Date & Time	2022-06-20T13:26:23
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32-DevKit
- [ ] Tasmota binary firmware version number used: 12.0.1.2(tasmota) 
  - [ ] Pre-compiled
  - [X] Self-compiled from master
- [X] Flashing tools used: VScode (PlatformIO) and update via Web Page
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:33:53.181 MQT: stat/tasmota_14FBDC/RESULT = {"NAME":"Sonoff Dual R3 v2","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
12:33:53.383 MQT: stat/tasmota_14FBDC/RESULT = {"Module":{"1":"ESP32-DevKit"}}
12:33:53.606 MQT: stat/tasmota_14FBDC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"544":"LedLink"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"1376":"WS28121"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"1440":"MHZ Rx"},"GPIO17":{"1408":"MHZ Tx"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"8546":"ADC MQ3"},"GPIO35":{"8547":"ADC MQ4"},"GPIO36":{"8544":"ADC MQ1"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"8545":"ADC MQ2"}}
12:34:01.481 MQT: tele/tasmota_14FBDC/STATE = {"Time":"2022-06-20T12:34:01","Uptime":"0T00:04:11","UptimeSec":251,"Heap":150,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"ON","Dimmer":79,"Color":"0000C8","HSBColor":"240,100,79","Channel":[0,0,79],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"133744445678785154IoT","BSSId":"7C:5A:1C:44:B5:11","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:34:01.545 MQT: tele/tasmota_14FBDC/SENSOR = {"Time":"2022-06-20T12:34:01","ANALOG":{"MQ2_1":1.02,"MQ3_2":null,"MQ6_3":858.16,"MQ7_4":0.17},"BME280":{"Temperature":23.9,"Humidity":36.6,"DewPoint":8.1,"Pressure":995.2},"MHZ19B":{"Model":"B","CarbonDioxide":418,"Temperature":26.0},"SGP30":{"eCO2":419,"TVOC":0,"aHumidity":7.7963},"ESP32":{"Temperature":52.8},"PressureUnit":"hPa","TempUnit":"C"}
12:34:01.652 MQT: stat/tasmota_14FBDC/RESULT = {"POWER":"ON","Dimmer":79,"Color":"0000C8","HSBColor":"240,100,79","Channel":[0,0,79]}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
[... 16 lines cut ...]
12:35:03.361 MQT: stat/tasmota_14FBDC/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.iot.ext.k8s.lan.sattlerit.de","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_14FBDC","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:35:03.392 MQT: stat/tasmota_14FBDC/STATUS7 = {"StatusTIM":{"UTC":"2022-06-20T11:35:03","Local":"2022-06-20T12:35:03","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:56"}}
12:35:03.427 MQT: stat/tasmota_14FBDC/STATUS10 = {"StatusSNS":{"Time":"2022-06-20T12:35:03","ANALOG":{"MQ2_1":1.03,"MQ3_2":null,"MQ6_3":858.28,"MQ7_4":0.17},"BME280":{"Temperature":23.9,"Humidity":36.3,"DewPoint":8.0,"Pressure":995.2},"MHZ19B":{"Model":"B","CarbonDioxide":426,"Temperature":26.0},"SGP30":{"eCO2":811,"TVOC":333,"aHumidity":7.7791},"ESP32":{"Temperature":53.3},"PressureUnit":"hPa","TempUnit":"C"}}
12:35:03.476 MQT: stat/tasmota_14FBDC/STATUS11 = {"StatusSTS":{"Time":"2022-06-20T12:35:03","Uptime":"0T00:05:13","UptimeSec":313,"Heap":147,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"ON","Dimmer":79,"Color":"0000C8","HSBColor":"240,100,79","Channel":[0,0,79],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"y","BSSId":"y","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

use anything that uses `TasmotaGlobal.temperature_celsius` with the ESP32

### EXPECTED BEHAVIOUR

ESP32 temperature is not included in `TasmotaGlobal.temperature_celsius`

### SCREENSHOTS

[image]

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-06-20:
Disable ESP32 temperature with command ``setsensor127 0``

Links2004, 2022-06-20:
yes this works a a workaround. 
But not everyone will notice this so excluding it from `TasmotaGlobal.temperature_celsius` sounds like a good idea.

arendst, 2022-06-20:
What works for you doesn't work for energy monitoring devices using the single present temp sensor to monitor overtemp.

So just execute the above command and you're good to go.

sfromis, 2022-06-20:
Anyway, that ESP32 Temperature cannot generally be expected to have any relationship to reality, like actual chip temperature. May depend on chip variant.

Links2004, 2022-06-20:
yes the MHZ19B shows to high temperatures too.
Resulting in the same problem with the ESP32 but less extrem in the Abs Humidity calculation.
the bad part is that is practically random which temperature is stored in `TasmotaGlobal`.
I ended up adding some defines to disable the Global update.
if you like I can open a PR.

It where unexpected to find that the `Convert` functions are updating the `TasmotaGlobal`, 
but I added a optional parameter to disable the update for `ConvertTemp`.

```diff
diff --git a/tasmota/tasmota_support/support.ino b/tasmota/tasmota_support/support.ino
index 0cb2d6af2..a1d5dac64 100755
--- a/tasmota/tasmota_support/support.ino
+++ b/tasmota/tasmota_support/support.ino
@@ -756,8 +756,10 @@ void UpdateGlobalTemperature(float c) {
   TasmotaGlobal.temperature_celsius = c;
 }
 
-float ConvertTemp(float c) {
-  UpdateGlobalTemperature(c);
+float ConvertTemp(float c, bool update = true) {
+  if (update) {
+    UpdateGlobalTemperature(c);
+  }
 
   return ConvertTempToFahrenheit(c);
 }
diff --git a/tasmota/tasmota_xsns_sensor/xsns_127_esp32_sensors.ino b/tasmota/tasmota_xsns_sensor/xsns_127_esp32_sensors.ino
index 8ac2ee7f6..b7559307b 100644
--- a/tasmota/tasmota_xsns_sensor/xsns_127_esp32_sensors.ino
+++ b/tasmota/tasmota_xsns_sensor/xsns_127_esp32_sensors.ino
@@ -61,9 +61,11 @@ void Esp32SensorShow(bool json) {
     add_global_temp = !ResponseContains_P(PSTR(D_JSON_TEMPERATURE));
   }
   float c = CpuTemperature();  // in Celsius
+#ifndef ESP32_SENSORS_TEMP_NO_GLOBAL
   if (add_global_temp) {
     UpdateGlobalTemperature(c);
   }
+#endif
   float t = ConvertTempToFahrenheit(c);
 
 #if CONFIG_IDF_TARGET_ESP32
diff --git a/tasmota/tasmota_xsns_sensor/xsns_15_mhz19.ino b/tasmota/tasmota_xsns_sensor/xsns_15_mhz19.ino
index 67745ddb4..2d61cd9fd 100644
--- a/tasmota/tasmota_xsns_sensor/xsns_15_mhz19.ino
+++ b/tasmota/tasmota_xsns_sensor/xsns_15_mhz19.ino
@@ -17,6 +17,7 @@
   along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */
 
+
 #ifdef USE_MHZ19
 /*********************************************************************************************\
  * MH-Z19 - CO2 sensor
@@ -225,7 +226,11 @@ void MhzEverySecond(void)
       }
     } else {
       uint16_t ppm = (mhz_response[2] << 8) | mhz_response[3];
+#ifdef USE_MHZ19_TEMP_NO_GLOBAL
+      mhz_temperature = ConvertTemp((float)mhz_response[4] - 40, false);
+#else
       mhz_temperature = ConvertTemp((float)mhz_response[4] - 40);
+#endif
       uint8_t s = mhz_response[5];
       mhz_type = (s) ? 1 : 2;
       if (MhzCheckAndApplyFilter(ppm, s)) {
```

arendst, 2022-06-21:
Thx for you input.

Now consider the fact that there are many more calls to ConvertTemp. I challenge you to come up with a solution workable for all of them, not just the ones you feel you need, and small code change without many #defines but on the fly control.

Links2004, 2022-06-21:
If I think big with full config options, I propose:

1. remove the global update from all the `Convert` functions (its misleading naming in my opinion)
2. add a new function to do the global update that includes the ID of the sensor
3. add settings for all the different global values to set which sensor ID to use (and use this in combination with the ID parameter of the function)

May have the old behavior (no filtering) as default aka setting ID as 0
and if ID setting > 0 then only allow this sensor for this global parameter.

with this this will be flexible for any use case, and the User has full control during runtime.
but i am currently not familiar enough to say how to do this in the code base or how easy it is.

arendst, 2022-06-22:
I'll sit and think about a solution for user selection of global temperature/humidity/pressure selection.

Links2004, 2022-06-22:
with two assumptions this sounds fin for me use case

- the order in the JSON never changes (only when new sensors are added its ok)
- all sensors use the same key ("Temperature", "Humidity", ..... ) , so no sensor where other or similar names are used.

but the 300 sec Teleperiod default sounds to high for the over temperature use case we had in a earlier discussion.

arendst, 2022-06-22:
- The order is mostly OK.
- The names are Tasmota standardized.
- Overtemp is an issue if it rises quick within a minute (if I decide to poll the sensors more often separated from teleperiod).

I'll implement this as an option to the current situation and see what the future brings.

Thx so far.

Links2004, 2022-06-24:
Firmware updated and used:
```
GlobalTemp2 1
GlobalHum2 1
GlobalPress2 1
```

first look at the web interface is good, no jumping there.
will let it run a bit and check my Grafana to see it over time.

Links2004, 2022-06-24:
looks like that the current implementation has some race condition 
[image]

Links2004, 2022-06-25:
did some dissecting and the breaking change is the 10sec update, but no idea how to fix it.

```diff
diff --git a/tasmota/tasmota_support/support_tasmota.ino b/tasmota/tasmota_support/support_tasmota.ino
index 9c70c4890..18694c0e9 100644
--- a/tasmota/tasmota_support/support_tasmota.ino
+++ b/tasmota/tasmota_support/support_tasmota.ino
@@ -1086,7 +1086,7 @@ void PerformEverySecond(void)
   if (!(TasmotaGlobal.uptime % 10)) {
     for (uint32_t type = 0; type < 3; type++) {
       if (!Settings->global_sensor_index[type] || TasmotaGlobal.user_globals[type]) { continue; }
-      GetSensorValues();
+      //GetSensorValues();
       break;
     }
   }
```

arendst, 2022-06-25:
I'm out the next days.

Now that you have commented out the 10 minute update global values will only be updated at teleperiod time. From your status 0 output it is set to 30 seconds which is ok as then it will use the global values from 30 seconds ago for the SGP calculation.

When I'm back I'll try to think of a more teleperiod synced approach but in your case (having already a small teleperiod set) it wouldn't make it must better.

Links2004, 2022-06-25:
yes for my configuration the teleperiod update all 30sec is fin, since its only monitoring.

Jason2866, 2022-07-09:
Closing, feature has been added

closed 2022-07-09 by Jason2866