arendst/Tasmota #15837: Protocol ELECTRA_AC for IR seems to be broken

eta4ever, 2022-06-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Electrolux AC, captured code from remote on YTF IR clone

19:51:49.764 RSL: RESULT = {"IrReceived":{"Protocol":"ELECTRA_AC","Bits":104,"Data":"0xC397E00020002000002000059F","Repeat":0,"IRHVAC":{"Vendor":"ELECTRA_AC","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

Trying to reproduce.  No output at all, checked with camera. But works nice in raw data mode (output is visible, AC is controlled).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [v] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [v] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [v] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [v] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [v] Device used (e.g., Sonoff Basic): Generic YTF IR clone
- [v] Tasmota binary firmware version number used: 12.0.2
  - [v] Pre-compiled
  - [ ] Self-compiled
- [v] Flashing tools used: Tasmotizer, OTA upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
01:22:13.661 MQT: stat/ir_bridge/RESULT = {"NAME":"IR","GPIO":[0,0,0,0,320,1088,0,0,0,32,1056,0,0,0],"FLAG":0,"BASE":62}
01:22:13.860 MQT: stat/ir_bridge/RESULT = {"Module":{"62":"YTF IR Bridge"}}
01:22:14.065 MQT: stat/ir_bridge/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"320":"Led_i1"},"GPIO5":{"1088":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"32":"Button1"},"GPIO14":{"1056":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
01:22:50.701 MQT: stat/ir_bridge/STATUS = {"Status":{"Module":62,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"ir_bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
01:22:50.708 MQT: stat/ir_bridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-ir.bin.gz","RestartReason":"External System","Uptime":"0T00:01:10","StartupUTC":"2022-06-20T22:21:40","Sleep":50,"CfgHolder":4617,"BootCount":33,"BCResetTime":"2022-06-20T17:46:58","SaveCount":90,"SaveAddress":"F9000"}}
01:22:50.713 MQT: stat/ir_bridge/STATUS2 = {"StatusFWR":{"Version":"12.0.2(ir)","BuildDateTime":"2022-06-20T12:43:45","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"410/699"}}
01:22:50.719 MQT: stat/ir_bridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["etanet2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00190000000000","00000180","00006000","00004000","00000000"]}}
01:22:50.730 MQT: stat/ir_bridge/STATUS4 = {"StatusMEM":{"ProgramSize":623,"Free":380,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12,62","Sensors":"1"}}
01:22:50.737 MQT: stat/ir_bridge/STATUS5 = {"StatusNET":{"Hostname":"ir-bridge-6072","IPAddress":"192.168.2.107","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"109.195.160.1","Mac":"E8:68:E7:57:37:B8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
01:22:50.743 MQT: stat/ir_bridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.104","MqttPort":1883,"MqttClientMask":"ir_bridge","MqttClient":"ir_bridge","MqttUser":"ir_bridge","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
01:22:50.753 MQT: stat/ir_bridge/STATUS7 = {"StatusTIM":{"UTC":"2022-06-20T22:22:50","Local":"2022-06-21T01:22:50","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+03:00","Sunrise":"06:46","Sunset":"22:57"}}
01:22:50.760 MQT: stat/ir_bridge/STATUS10 = {"StatusSNS":{"Time":"2022-06-21T01:22:50"}}
01:22:50.770 MQT: stat/ir_bridge/STATUS11 = {"StatusSTS":{"Time":"2022-06-21T01:22:50","Uptime":"0T00:01:10","UptimeSec":70,"Vcc":3.529,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":1,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

irsend {"Protocol":"ELECTRA_AC","Bits":104,"Data":"0xC397E00020002000002000059F","Repeat":0,"IRHVAC":{"Vendor":"ELECTRA_AC","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":26,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The output is seen through camera; the AC is controlled

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-06-21:
I think your command is not correct example from the docu https://tasmota.github.io/docs/Tasmota-IR/#sending-ir-commands
`IRhvac {"Vendor":"Mitsubishi_Heavy_152", "Power":"On","Mode":"Hot","FanSpeed":3,"Temp":22.5}`

eta4ever, 2022-06-21:
> I think your command is not correct example from the docu https://tasmota.github.io/docs/Tasmota-IR/#sending-ir-commands `IRhvac {"Vendor":"Mitsubishi_Heavy_152", "Power":"On","Mode":"Hot","FanSpeed":3,"Temp":22.5}`

Well, maybe I mixed the things. But this does not work too.
irsend {"Protocol":"ELECTRA_AC","Bits":104,"Data":"0xC397E00020002000002000059F","Repeat":0}

eta4ever, 2022-06-21:
Sorry, I really mixed the protocols for irsend and irhvac.

closed 2022-06-21 by eta4ever