arendst/Tasmota #15851: Hardware-Buttons 5 through 8 don't work with shutters 3 and 4 (Device restarts)

baennjer, 2022-06-23:
### PROBLEM DESCRIPTION
The Hardware-Buttons 5 through 8 (hooked up to GPIOs) don't work with shutters 3 and 4.
When pressing buton 5 or 6 the Device restarts.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Olimex ESP32 ETH
- [ ] Tasmota binary firmware version number used: v12.0.2
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: ESP-Flasher.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:02:14.970 CMD: Backlog Template; Module; GPIO 255
14:02:15.001 RSL: RESULT = {"NAME":"Olimex ESP32-PoE","GPIO":[1,1,1,1,1,1,0,0,5536,1,1,1,1,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
14:02:15.221 RSL: RESULT = {"Module":{"0":"Olimex ESP32-PoE"}}
14:02:15.488 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"33":"Button2"},"GPIO2":{"34":"Button3"},"GPIO3":{"35":"Button4"},"GPIO4":{"164":"Switch5"},"GPIO5":{"608":"I2C SCL1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"5536":"ETH POWER"},"GPIO13":{"640":"I2C SDA1"},"GPIO14":{"165":"Switch6"},"GPIO15":{"166":"Switch7"},"GPIO16":{"167":"Switch8"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:03:23.829 CMD: status 0
14:03:23.839 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","","","","","","",""],"Topic":"tasmota_D0F75C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,3,3,3,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:03:23.848 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:30:21","StartupUTC":"2022-06-23T11:33:02","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2022-06-23T08:47:42","SaveCount":56}}
14:03:23.854 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2(TasmoCompiler-esp32generic)","BuildDateTime":"2022.06.23 07:43:22","Core":"2_0_3","SDK":"v4.4.1","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"376/699"}}
14:03:23.862 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WLANUG",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A190000000000","C0000080","00004000","00004000","00000000"]}}
14:03:23.876 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1188,"Free":1856,"Heap":172,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000407","0F8045CE","00040001","00000003","00000000","C0000B81","40000080","00200000","14000008","00000000"],"Drivers":"1,2,4,8,9,10,12,24,26,27,28,29,35,38,50,52,62,82","Sensors":"1,2,127"}}
14:03:23.886 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D0F75C-5980","IPAddress":"0.0.0.0","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"E8:9F:6D:D0:F7:5C","Ethernet":{"Hostname":"tasmota-D0F75C-5980-eth","IPAddress":"192.168.178.203","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"E8:9F:6D:D0:F7:5F"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:03:23.892 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D0F75C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:03:23.900 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-06-23T12:03:23","Local":"2022-06-23T14:03:23","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:47","Sunset":"21:57"}}
14:03:23.915 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-06-23T14:03:23","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"ON","ESP32":{"Temperature":53.3},"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"TempUnit":"C"}}
14:03:23.926 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-06-23T14:03:23","Uptime":"0T00:30:21","UptimeSec":1821,"Heap":173,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":32},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF"}}
14:03:23.943 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"},"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"},"SHT2":{"Relay1":5,"Relay2":6,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"},"SHT3":{"Relay1":7,"Relay2":8,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Setup a ESP32 Device with 8 inputs configured as Buttons 1-8 and 8 outputs configured as Relay 1-8
SO80 1
Backlog PulseTime1 0; PulseTime2 0; PulseTime3 0; PulseTime4 0; PulseTime5 0; PulseTime6 0; PulseTime7 0; PulseTime8 0
Backlog Interlock 1,2 3,4 5,6 7,8; Interlock ON
SHUTTERRELAY2 3
SHUTTERRELAY3 5
SHUTTERRELAY3 7

### EXPECTED BEHAVIOUR
Shutters can be controlled by physical buttons hooked up to the ESP32
But only buttons 1-4 will work, pressing 5 or 6 will let the device restart
Web Buttons 5+6 work

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
This issue seems to be related to the historical fact that there once only 4 buttons were assignable. When more buttons were nedded the GPIOs had to be setup es Switch with Switchmode 3.

Finally when I setup Buttons 1-4 and Switches 5-8 in Switchmode 3 alle hardware buttons work fine.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-06-23:
your screenshot shows an incorrect config. Mixed switches and buttons with wrong numberings

baennjer, 2022-06-23:
> your screenshot shows an incorrect config. Mixed switches and buttons with wrong numberings

What should there be wrong?
I have buttons 1-4 and switches 5-8
They are corrosponding to realys 1-8

As I mentioned above when I use buttons 1-8 the device crashes and restarts when pressing hardware buttons 5 and 6

meingraham, 2022-06-23:
You have four buttons enumerated 1-4. You have four switches, enumerated 1-4. They, in combination, are attached to the eight hardware pins.

baennjer, 2022-06-23:
> You have four buttons enumerated 1-4. You have four switches, enumerated 1-4. They, in combination, are attached to the eight hardware pins.

Sorry in my screenshot it is clearly that I have buttons 1-4 and switches 5-8, I cant see switches 1-4
If having a button 1 and a switch 1 they both would operate relay1...

Jason2866, 2022-06-23:
Yes, and that way it is wrong and does not work. You have to decide. All switches or all buttons.

Jason2866, 2022-06-23:
Closing since behaviour is caused from wrong config

baennjer, 2022-06-23:
> Yes, and that way it is wrong and does not work. You have to decide. All switches or all buttons.

I have the test devices here on the table, tested with 2 diferent types
When I configure 8 buttons, then a hardware button connected to 5 + 6 let the device crash and restart, web buttons work.
I have to configure buttons 1-4 and switches 5-8 (switchmode 3 ) to have it work

So I stick with my work around

baennjer, 2022-06-23:
Thanks Jason2866 for not even looking into it !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Jason2866, 2022-06-23:
So why do you not describe a correct config which is not working?
A workaround solution is out of interest here.

baennjer, 2022-06-23:
> So why do you not describe a correct config which is not working? A (wrong) workaround solution is out of interest here.

I have updated the issue  above and included the config screenshot that doesn't work.

Jason2866, 2022-06-24:
Please try latest dev build.

barbudor, 2022-06-24:
Yes, if you select Development you get the lastest development branch

baennjer, 2022-06-25:
> Please try latest dev build. Theo did fixes for Shutter

Unfortunately the same behavior. 
In log i.e. I get the message "shutter4 button6" when pressing button6 although shutter 4 should be related to buttons 7+8

[image]

Jason2866, 2022-06-25:
Hmm
```
SHUTTERRELAY2 3
SHUTTERRELAY3 5
SHUTTERRELAY3 7
```

baennjer, 2022-06-25:
> Hmm
> 
> ```
> SHUTTERRELAY2 3
> SHUTTERRELAY3 5
> SHUTTERRELAY3 7
> ```

Ja alles was Shutter benötigt ist drin
SO80 1
Backlog Interlock 1,2 3,4 5,6 7,8; Interlock ON
SHUTTERRELAY2 3
SHUTTERRELAY2 3

pawol, 2022-07-05:
Looks loke similar issue to mine #15929

stefanbode, 2022-08-03:
I will take a close look at this. to directly operate the relay with buttons IS supported but it is not the "fine" way of doing it. Normally you should assign to your buttons "shutterbuttons" there you can configure what should happen. Also long press and double press events are possible here

stefanbode, 2022-08-03:
Button 7 and 8 on shutter4 operate normal. On ESP32 no problem seen

```
13:18:00.398 MQT: hm/status/ESP_F51C40/RESULT = {"POWER7":"ON"}
13:18:00.408 MQT: hm/status/ESP_F51C40/POWER7 = ON
13:18:00.420 MQT: hm/setting/tasmotas/ShutterPosition1 = 124
13:18:00.432 MQT: hm/setting/tasmotas/ShutterPosition2 = 124
13:18:00.438 MQT: hm/setting/tasmotas/ShutterPosition3 = 124
13:18:00.442 MQT: hm/setting/tasmotas/ShutterPosition4 = 124
13:18:00.447 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter4":{"Button7":1}}
13:18:00.478 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
13:18:00.485 MQT: hm/status/ESP_F51C40/RESULT = {"Position1":100}
13:18:00.498 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter4":{"Position":0,"Direction":1,"Target":100,"Tilt":0}}
13:18:00.526 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":1,"Direction":1,"Target":100,"Tilt":0}}
13:18:00.659 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
13:18:00.669 MQT: hm/status/ESP_F51C40/RESULT = {"Position2":100}
13:18:00.727 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}
13:18:00.739 MQT: hm/status/ESP_F51C40/RESULT = {"Position3":100}
13:18:00.792 MQT: hm/status/ESP_F51C40/RESULT = {"Position4":100}
13:18:01.497 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":10,"Direction":1,"Target":100,"Tilt":0}}
13:18:02.467 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":20,"Direction":1,"Target":100,"Tilt":0}}
13:18:03.467 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":30,"Direction":1,"Target":100,"Tilt":0}}
13:18:04.486 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":40,"Direction":1,"Target":100,"Tilt":0}}
13:18:05.501 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":50,"Direction":1,"Target":100,"Tilt":0}}
13:18:06.498 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":60,"Direction":1,"Target":100,"Tilt":0}}
13:18:07.465 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":70,"Direction":1,"Target":100,"Tilt":0}}
13:18:08.464 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":80,"Direction":1,"Target":100,"Tilt":0}}
13:18:09.490 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":90,"Direction":1,"Target":100,"Tilt":0}}
13:18:09.606 MQT: hm/status/ESP_F51C40/RESULT = {"POWER7":"OFF"}
13:18:09.616 MQT: hm/status/ESP_F51C40/POWER7 = OFF
13:18:10.129 MQT: hm/status/ESP_F51C40/SHUTTER4 = 101
13:18:10.139 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter4":{"Position":101,"Direction":0,"Target":100,"Tilt":0}}
13:18:37.878 MQT: hm/status/ESP_F51C40/RESULT = {"POWER8":"ON"}
13:18:37.883 MQT: hm/status/ESP_F51C40/POWER8 = ON
13:18:37.895 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter4":{"Position":101,"Direction":-1,"Target":0,"Tilt":0}}
13:18:38.480 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":86,"Direction":-1,"Target":0,"Tilt":0}}
13:18:39.505 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":76,"Direction":-1,"Target":0,"Tilt":0}}
[... cut ...]
13:18:46.467 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter4":{"Position":6,"Direction":-1,"Target":0,"Tilt":0}}
13:18:47.033 MQT: hm/status/ESP_F51C40/RESULT = {"POWER8":"OFF"}
13:18:47.045 MQT: hm/status/ESP_F51C40/POWER8 = OFF
13:18:47.563 MQT: hm/status/ESP_F51C40/SHUTTER4 = 0
13:18:47.573 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}
```

baennjer, 2022-08-03:
@stefanbode I am not quiet sure no more but I think I remember buttons 7+8 worked, but not buttons 5+6.

stefanbode, 2022-08-03:
Button 6 behave strange. It switches on relay7 and not relay6. Additionally all this .../setting/... is more than wrong. I posts on the group topic to open all shutters. Create some strange movements in my house. Now changed the grouptopic to "test". just ensure the rest of the devices do no assume this as a command. I will definitly check and do a fix. Anyhow there is no reboot.
```
17:08:18.950 MQT: hm/status/ESP_F51C40/RESULT = {"POWER7":"ON"}
17:08:18.954 MQT: hm/status/ESP_F51C40/POWER7 = ON
17:08:18.961 MQT: hm/setting/test/ShutterPosition1 = 124
17:08:18.965 MQT: hm/setting/test/ShutterPosition2 = 124
17:08:18.969 MQT: hm/setting/test/ShutterPosition3 = 124
17:08:18.975 MQT: hm/setting/test/ShutterPosition4 = 124
17:08:18.979 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter4":{"Button6":1}}
17:08:19.006 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter4":{"Position":0,"Direction":1,"Target":100,"Tilt":0}}
17:08:19.074 MQT: hm/status/ESP_F51C40/RESULT = {"POWER1":"ON"}
17:08:19.085 MQT: hm/status/ESP_F51C40/POWER1 = ON
17:08:19.096 MQT: hm/status/ESP_F51C40/RESULT = {"Position1":100}
17:08:19.121 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":0,"Direction":1,"Target":100,"Tilt":0}}
17:08:19.194 MQT: hm/status/ESP_F51C40/RESULT = {"POWER3":"ON"}
17:08:19.203 MQT: hm/status/ESP_F51C40/POWER3 = ON
17:08:19.212 MQT: hm/status/ESP_F51C40/RESULT = {"Position2":100}
17:08:19.230 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":0,"Direction":1,"Target":100,"Tilt":0}}
17:08:19.312 MQT: hm/status/ESP_F51C40/RESULT = {"POWER5":"ON"}
17:08:19.319 MQT: hm/status/ESP_F51C40/POWER5 = ON
17:08:19.325 MQT: hm/status/ESP_F51C40/RESULT = {"Position3":100}
17:08:19.342 MQT: hm/status/ESP_F51C40/RESULT = {"Shutter1":{"Position":0,"Direction":1,"Target":100,"Tilt":0}}
17:08:19.419 MQT: hm/status/ESP_F51C40/RESULT = {"Position4":100}

```

stefanbode, 2022-08-03:
The error comes from the ShutterButtonHandler(void). I assume this is because the author of this functioned never thought about assigning buttons but then not using the "shutterbuttonX" command to assign functions to the buttons. For sure also the crash could be a result of this problem.

stefanbode, 2022-08-03:
The shutterbuttons only define 4 buttons. If you came in with more than 4 everything could happen. In your special use case you only use the buttons without the functionality of the shutterbuttons. This was not tested before. Fix should work.

PR #16147 (2022-08-03): fix #15851
buffer overflow

## Description:

**Related issue (if applicable):** fixes #<Tasmota issue number goes here>
fix #15851
## Checklist:
  - [ x] The pull request is done against the latest development branch
  - [x ] Only relevant files were touched
  - [ x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x ] The code change is tested and works with Tasmota core ESP32 V.2.0.4.1
  - [ x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBU [... cut]
closed 2022-08-03 by arendst