arendst/Tasmota #15856: Support for SONOFF Smart Power Meter Switch | POW Elite

nobodyman1, 2022-06-23:
### PROBLEM DESCRIPTION
Is it possible to use Tasmota on the new [SONOFF POW Elite](https://sonoff.tech/product/diy-smart-switch/pow-elite/)?
I found some Pictures [here](https://notenoughtech.com/home-automation-review/elite-measurements-sonoff-pow-elite/).
It´s based on an ESP32 and the power measuring is done with a CSE7759B (same as used in SONOFF POW R2).
But what´s about the LCD driver?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [] Device used (e.g., Sonoff Basic): _____
- [] Tasmota binary firmware version number used: _____
  - [] Pre-compiled
  - [] Self-compiled
- [] Flashing tools used: _____
- [] Provide the output of command: `Backlog Template; Module; GPIO 255`:
### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Erling55, 2022-06-28:
Maybe it's the wrong place here, but I couldn't get my Pow 3 in flashing mode. The Flashing knob doesn't seem to work. I wonder if there is another place where I can connect the GPIO 0 to Ground. Many thanks

sfromis, 2022-06-28:
When you know that it is the wrong place, do not make off topic posts in existing threads.

Erling55, 2022-06-28:
Do I have to say “sorry” or could you tell me the right place?

Verstuurd vanaf mijn iPad

dervomsee, 2022-07-01:
Hmm, good Idea. I started with my own firmware to play with the IOs and the LCD.
I will try the default image later..

arendst, 2022-07-02:
GPIO00 = Button1
GPIO05 = LedLink_i
GPIO13 = Relay1
GPIO18 = Led_i1

barbudor, 2022-07-02:
Lcd driver datasheet
[Titan Micro TM1621 Lcd driver.pdf](https://github.com/arendst/Tasmota/files/9033220/Titan.Micro.TM1621.Lcd.driver.pdf)
Chinese but seems understandable. Otherwise Google translate is our friend 😊

sfromis, 2022-07-02:
Some driver code found here: https://github.com/emsyscode/TM1621
```
/*
This code is not clean and also is not perfect, this is only 
a reference to extract ideas and adapte to your solution.
*/
```

nobodyman1, 2022-07-03:
> Lcd driver datasheet [Titan Micro TM1621 Lcd driver.pdf](https://github.com/arendst/Tasmota/files/9033220/Titan.Micro.TM1621.Lcd.driver.pdf) Chinese but seems understandable. Otherwise Google translate is our friend 😊

I found the [HT1621](https://datasheetspdf.com/pdf-file/170416/HoltekSemiconductorInc/HT1621/1) which looks very similar to me.

Nuffic, 2022-07-03:
TH Elite seems to have the same base as POW Elite (by the looks at least). Would be nice to get support for it as well. I have few of them here + DS18B20, THS01 sensors. I can share info if needed, willing to help any way I can.

sfromis, 2022-07-03:
Looks like support for both is in the works.

dervomsee, 2022-07-04:
@arendst Are you sure with the 320D pins?
My bistable relay uses the following pins: on=GPIO2, off=GPIO4.
The LEDs and the CSE pins are the same.

My config is currently the following:
`{"NAME":"Sonoff POWR320D","GPIO": 32,0,224,0,225,576,0,0,0,0,0,0,3104,0,320,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}`

Does the 316D not have a bistable relay?

arendst, 2022-07-04:
Not sure about gpio15. As you noticed it's gpio4 it might well be ok.

My 316D has no bistable relay and is controlled by gpio13.

In the meantime I managed to logic analyse the sonoff lcd init sequence and 5 second display update sequence (on the thr316d to stay away from AC voltage). Currently need to find out which bit controls which part of the lcd. Init sequence is clear. 

The ht1621 docs are indeed equal to my google translated tm1621 docs.

Nuffic, 2022-07-04:
@Philippe12 big thanks, I've managed to get everything working with this config, but the temperature sensor. Have you managed to use the temperature sensor there?

arendst, 2022-07-04:
By adding define ``USE_DISPLAY_TM1621_SONOFF`` you should be able to see Sonoff POWR3xxD alternating Voltage/Current and TotalEnergy/Power on the LCD display.

That's it for today (00:11)

Philippe12, 2022-07-05:
@Nuffic THR sensor (Sonoff Si7021) is a dificulty, it use the same protocole one wire than DHT11 but not with the same timming.

You can find mi first modification at : https://github.com/Philippe12/Tasmota/commit/0d9f6749f605c61473bd4413445b25117ed74a0f

I know all this modifications are dirty that it for test, but if can help you.

BangerTech, 2022-07-05:
hey @Philippe12 thanks for your modification i´m testing it right now ;-) could you please post your template?
Thanks

Philippe12, 2022-07-05:
@BangerTech 

{"NAME":"THR316D","GPIO":[32,3200,0,3232,225,9280,0,0,0,8224,0,544,3872,9184,9216,0,0,224,0,9248,0,1248,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}

I add TM1621 pin

Nuffic, 2022-07-05:
@Philippe12 thanks for all the info. We even got the DS18b20 working with it.

arendst, 2022-07-06:
Try latest dev.

BangerTech, 2022-07-06:
@arendst it does show some temps but some weird ones ;-) and the Tasmota Interface shows no Temp/Hum at all. See Pic. 
so i think one of the temps on the display is correct it is the esp32 temp. 
@Philippe12 build worked well with the status infos in the tasmota interface but nothing on the display

arendst, 2022-07-06:
This is expected. The SI7021 driver is not available on the ESP32 (yet). The @Philippe12 patched version is not stable either considering the ``DHT: Invalid reading`` logging messages.

The display driver selects the current global-temp for being displayed. As the SI7021 is invalid it displays the ESP32 temp. To remove the ESP32 from being fed to global-temp you can either remove the ESP32 temp with command ``SetSensor127 0`` or use the recently released command ``GlobalTemp2 1`` to select the first JSON temperature being the SI7021 one which in turn is invalid and therefor does not show as a temp on the LCD display.

Also when no valid humidity is read it won't show anything on the LCD. Using command ``GlobalHum2 1`` would select the SI7021 humidity.

BangerTech, 2022-07-06:
oh ok didn´t know that the THS01 / SI7021 isn´t supported yet. 
Thanks for all the informations!!

BangerTech, 2022-07-06:
Just tried out the new build, but only get the ESP Temp. See Pic. 
[image]

arendst, 2022-07-06:
I suppose you have selected the correct sensor for gpio25 using config module??

BangerTech, 2022-07-06:
ahhh ;-) thaaanks!!
so now i get everything in the web interface. 
so i get temp and hum from si7021 and esp32 temp. but i get one more temp how do i find out which one is that? 
for example i got si7021 temp - 26,7 , hum - 38,4 , dew point - 11,3 and esp32 temp - 30. 
then i get one more with 53,3 on the display but not in the console or somewhere in the web interface. 
on the template side of things are there any differences between THR320D and THR316D?
Thanks again!!

nobodyman1, 2022-07-06:
@arendst 
Today I flashed my SONOFF POW Elite. Everything seems to be okay.
Great job! Thanks a lot!
When will be the next official release which includes the changes?
Should I close the issue or wait until the THR316D implementation is finished?

closed 2022-07-07 by arendst

nobodyman1, 2022-07-10:
Is it possible to enable the [Smart Meter Interface](https://tasmota.github.io/docs/Smart-Meter-Interface/)?
It should be very simple to connect to the device!

arendst, 2022-07-10:
Smart Meter Interface as an energy source is not planned. Use the already available MQTT messages as available for over 4 years!

POWR320D template update.

As bistable relays are supported since 20220709 the template for the POWR320D has changed to:

```
{"NAME":"Sonoff POWR320D","GPIO":[32,0,9312,0,9313,576,0,0,0,0,9280,0,3104,0,320,0,0,0,0,0,0,9184,9248,9216,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

This will show only one web button toggling the relay as designed.

nobodyman1, 2022-07-10:
I think I have to explain my intention:
I want to use the Smart Meter Interface as an additional input device.
I plan to use the Sonoff for measuring the delivered energy from a photovoltaic system. Therefore I think I place the Sonoff device near to my Smart Meter and connect it via an infrared reader to the Sonoff. Then I´m able to get delivered and consumed energy with one device!

arendst, 2022-07-10:
Clear.

In that case it's off topic and I suggest you open a discussion.

nobodyman1, 2022-07-10:
Okay, I have opened #15955.
Could you please tell me if it's simple to implement or not?

arendst, 2022-07-10:
Latest Sonoff THRxxD templates:

```
{"NAME":"Sonoff THR316D","GPIO":[32,0,0,0,225,9280,0,0,0,321,0,576,320,9184,9216,0,0,224,0,9248,0,0,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
{"NAME":"Sonoff THR320D","GPIO":[32,0,0,0,226,9280,0,0,0,321,0,576,320,9184,9216,0,0,0,0,9248,0,9312,9313,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

BangerTech, 2022-07-11:
hey @arendst i've tried the new templates and updated firmware, for the th everything works but for the pow (20a) i don't get the toggle button. Both are gone and the one like on the th is not there.

arendst, 2022-07-11:
Ah thx. I watched your video. Nice.

I will dive into the pow PCA as I'm still waiting to receive my 20A version...

BangerTech, 2022-07-11:
hope i´ve explained everything the right way ;-)
perfect thanks buddy!

arendst, 2022-07-11:
Are you sure updated the POWR320D to the very latest dev release? With the latest template there should at least be a button in the GUI....
```
{"NAME":"Sonoff POWR320D","GPIO":[32,0,9312,0,9313,576,0,0,0,0,9280,0,3104,0,320,0,0,0,0,0,0,9184,9248,9216,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

chu11u, 2022-07-11:
Hi all, does this thread deal with both POWR3 and THR3? or there is a plan to have a separate discussion for it?

I have trouble connecting the waterproof DB18B20 sensor that came with it.  the device shows only the internal temp sensor

Thanks

arendst, 2022-07-11:
And you did configure the ds18b20 sensor on gpio25 using configure module?

chu11u, 2022-07-12:
yes. it even shoed some data for a short time, but then gone

vbokaty, 2022-07-12:
> And you did configure the ds18b20 sensor on gpio25 using configure module?

Yes. gpio25 ds18b20 sensor work.
my screen doesn't work - it's fine ???

arendst, 2022-07-12:
I think I know what goes wrong with the ds18x20 sensor.

The sonoff device doesn't use a pullup resistor. On esp8266 is an option to enable it's internal pullup resistor. This is not available in the esp32 ds18x20 driver. An attached ds18x20 with external pullup resistor therefor just works fine; the sonoff provided sensor won't work with the current tasmota driver.

I'll see what I can do to fix this.

arendst, 2022-07-12:
I just tried the sonoff sensor and it indeed works fine. So my suspection of missing pullup resistor is NOT AN ISSUE.

@vbokaty I suspect you mean tasmota32-display.bin firmware. In that case what version are you using? Only the latest development version supports both the LCD display and the optional bistable relay.

vbokaty, 2022-07-12:
> Я только что попробовал датчик sonoff, и он действительно работает нормально. Так что мое подозрение на отсутствие подтягивающего резистора НЕ ЯВЛЯЕТСЯ ПРОБЛЕМОЙ.
> 
> @vbokaty Я подозреваю, что вы имеете в виду прошивку tasmota32-display.bin. В таком случае какую версию вы используете? Только последняя версия поддерживает как ЖК-дисплей, так и дополнительное бистабильное реле.
> 
> Кстати, ЖК-дисплей в настоящее время работает только на "обычной" прошивке tasmota32.bin.

Please giv me link to download  tasmota32.bin ?  I have tried different versions and no firmware is working LCD display

arendst, 2022-07-12:
This one http://ota.tasmota.com/tasmota32/tasmota32.bin

vbokaty, 2022-07-12:
> http://ota.tasmota.com/tasmota32/tasmota32.bin

Thank you - it works!!! 

LCD display shows DS18B20 Temperature and after a few seconds shows ESP32 temperature. How to show only DS18B20 ?

arendst, 2022-07-12:
Disable ESP32 sensors with command ``SetSensor127 0`` or force to select DS18B20 temperature with command ``GlobalTemp2 1`` where you select the first temperature value from the JSON message.

vbokaty, 2022-07-12:
> Отключите датчики ESP32 с помощью команды `SetSensor127 0`или принудительно выберите температуру DS18B20 с помощью команды`GlobalTemp2 1`, в которой вы выбираете первое значение температуры из сообщения JSON.

Thanks everything works.

chu11u, 2022-07-12:
> I just tried the sonoff sensor and it indeed works fine. So my suspection of missing pullup resistor is NOT AN ISSUE.
> 
Hi, this means I need to but a sensor with a built-in pullup resistor? this is weird because the standard firmware shows the temp. How do they do it?

vbokaty, 2022-07-12:
> > Я только что попробовал датчик sonoff, и он действительно работает нормально. Так что мое подозрение на отсутствие подтягивающего резистора НЕ ЯВЛЯЕТСЯ ПРОБЛЕМОЙ.
> 
> Привет, это означает, что мне нужен датчик со встроенным подтягивающим резистором? это странно, потому что стандартная прошивка показывает температуру. Как они это делают?
I bought this
https://aliexpress.ru/item/1005003674650332.html?gatewayAdapt=glo2rus&sku_id=12000026753220145&spm=a2g0o.order_list.0.0.21efa3965Ootqx

freddysherman, 2022-07-12:
Just received Sonoff POW Elite, managed to flash it and get the functionality using the latest development release of tasmota23.bin and the following template 

{"NAME":"Sonoff POWEL20","GPIO":[32,0,9344,0,9345,576,0,0,0,0,9280,0,3104,0,320,0,0,0,0,0,0,9184,9248,9216,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}

paulinchen14, 2022-07-12:
Hallo,
ich habe heute einen Sonoff POWR316D erhalten und mit dem letzten development release of tasmota32-DE.bin geflasht.
Leider finde ich kein Template, wo die Powermessung funktioniert. Auf dem Web-IF werden keine gültigen Werte angezeigt?

Was ich herausgefunden habe:
- das Relais funktioniert nur auf GPIO21
- der angeschlossene DS18B20 funktioniert auf GPIO25
- wenn ich den GPIO16 auf "CSE7766Rx" stelle, erscheinen im Web-IF von Tasmota die Powerwerte. Diese stehen allerdings immer auf 0, trotz angeschlossener und eingeschalteter Last.

Woran könnte das liegen?

{"NAME":"Sonoff POWR316D","GPIO":[32,0,0,0,0,576,0,0,0,0,9280,0,3104,0,320,0,0,224,0,0,0,9184,9248,9216,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}

blacknell, 2022-07-13:
Is there anyway to connect up I2C by repurposing GPIO25 and one other? I can't see any other 'User' GPIO's listed but wondered if one more is accessible via a board hack? I'd forgo the display or an LED if there was. I can't find a schematic or board layout yet.

arendst, 2022-07-13:
@paulinchen14 gpio21 is a relay in THR316D. You do not have a POWR316D.

@blacknell On the THR3xxD you can try to use the serial flash connector. On the POWR3xxD we do not support ANY ATTACHMENT AS IT WILL BE CONNECTED TO AC VOLTAGE.

blacknell, 2022-07-13:
I should have been more explicit - I'm looking at THR3xxD, so I believe it'll BE GPIO 1&3 (?) although this is not clear. I'll buy one and have a look

paulinchen14, 2022-07-13:
@arendst: Du hast Recht, ich habe einen THR316D. Ich werde mir jetzt noch einen POWR316D kaufen.

stone17, 2022-07-16:
I notice that my Powr320d gets considerably warmer with this FW than with the standard Sonoff FW. Is that normal? Any way to lower the temp?

arendst, 2022-07-16:
Make sure to use the correct supported template:
```
{"NAME":"Sonoff POWR320D","GPIO":[32,0,9313,0,9312,576,0,0,0,0,9280,0,3104,0,320,0,0,0,0,0,0,9184,9248,9216,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```
There are user templates floating around using the wrong bistable relay gpio config. The wrong ones keep supplying power to the relay instead of providing only a pulse. This heats up the device and breaks the relay very soon.

So pls check your template and report back.

stone17, 2022-07-16:
> Make sure to use the correct supported template:
>
> ```
> {"NAME":"Sonoff [...]

Thanks a lot for quick answer. Only value 3 and 5 differed slightly to the config that I used, but the temperature is already dropping several degrees with your config. Was at 61C, now falling steadily.

arendst, 2022-07-16:
Great!! Yes those are the two pins driving the relay and should be set like in the latest template.

CyrielRct, 2022-07-18:
Really I want to say thank you for all the work done here. My THR320D is working perfectly. 
It took me a bit to figure out how to enable the gpio25 for my DS18b20 and force the display to show only the temperature of the sensor and not the ESP using 'GlobalTemp2 1'

Again thanks guys

[433 more replies]