arendst/Tasmota #15864: Shutter Mode 2 (Circuit Safe) Problem

cyberdomotik, 2022-06-24:
### PROBLEM DESCRIPTION
Activation of the run relay (relay 1 on) before the definition of the opening/closing relay  (relay 2 direction).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP-01
- [X] Tasmota binary firmware version number used: 12.0.2
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: TASMOTIZER
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
  Configuration output here:
= {"NAME":"LC-Tech_4CH ","GPIO":[288,1,32,1,0,0,0,0,224,225,226,227,0,0],"FLAG":0,"BASE":18}
= {"Module":{"0":"LC-Tech_4CH "}}
 = {"GPIO0":{"288":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"225":"Relay2"},"GPIO14":{"226":"Relay3"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
= {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":383,"Free":128,"Rules":"on Power1#State=1 do SerialSend5 A00101A2 endon on Power1#State=0 do SerialSend5 A00100A1 endon on Power2#State=1 do SerialSend5 A00201A3 endon on Power2#State=0 do SerialSend5 A00200A2 endon on Power3#State=1 do SerialSend5 A00301A4 endon on Power3#State=0 do SerialSend5 A00300A3 endon on Power4#State=1 do SerialSend5 A00401A5 endon on Power4#State=0 do SerialSend5 A00400A4 endon"}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
= {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","","",""],"Topic":"tasmota_166A19","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
02:08:28.818 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:23:07","StartupUTC":"2022-06-22T21:45:21","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-06-22T08:57:26","SaveCount":58,"SaveAddress":"F6000"}}
02:08:28.846 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"347/699"}}
02:08:28.872 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUS_AX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C0C009","2805C80001000600003C5A0A000000000000","40000080","00006000","00004000","00000000"]}}
02:08:28.905 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
02:08:28.934 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-166A19-2585","IPAddress":"192.168.1.107","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"98:CD:AC:16:6A:19","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
02:08:28.960 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_166A19","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
02:08:28.987 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-06-23T00:08:28","Local":"2022-06-23T02:08:28","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+02:00","Sunrise":"05:47","Sunset":"21:57"}}
02:08:29.026 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-06-23T02:08:29","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter2":{"Position":30,"Direction":0,"Target":30,"Tilt":0}}}
02:08:29.054 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-06-23T02:08:29","Uptime":"0T02:23:08","UptimeSec":8588,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"ASUS_AX","BSSId":"7C:10:C9:2B:29:F0","Channel":9,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:04:28"}}}
02:08:29.084 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1"},"SHT1":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"1000","Calib":[300,500,700,900,1000],"Mode":"1"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
ShutterMode 2 (Circuit safe)

### EXPECTED BEHAVIOUR
The on-relay must be activated when the direction of movement of the blind has been defined.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
In shuttermode 2 (circuit safe ) I realise that sometimes when I press web button to close the shutter, first it`s activated relay 1 (ON) and then relay 2 change to close position, so this means that when relay 1 is activated, the shutter starts to open and a few moments later when relay 2 is activated, the shutter closes, causing a rapid change of rotation in the motor.

Also when it reaches the target position it happens that the de-energisation of the relay (open/close) occurs moments before the de-energisation of relay 1 (off) causing the motor to change direction before it stops.

The on/off relay should be activated after the direction of movement of the motor is defined. And when the target is reached, the on/off relay should be activated before the direction of rotation of the motor changes.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-06-26:
Hi,

The connection explained in https://github.com/arendst/Tasmota/discussions/15835#discussioncomment-2986806 seems to be better for this. Are you able to try it?

rapi3, 2022-06-26:
There is a moment of delay until the engine stops rotating permanently due to inertia - the laws of physics.
It is necessary to introduce a small delay in the sw if the direction of rotation changes quickly.

cyberdomotik, 2022-06-26:
> Hi,
> 
> The connection explained in [#15835 (comment)](https://github.com/arendst/Tasmota/discussions/15835#discussioncomment-2986806) seems to be better for this. Are you able to try it?

Yes it's another option! But I think safe mode would work better if we could fix this problem.

cyberdomotik, 2022-06-26:
> There is a moment of delay until the engine stops rotating permanently due to inertia - the laws of physics. ( this depend of the motor also ) It is necessary to introduce a small delay in the sw if the direction of rotation changes quickly.

Introducing a delay in the switch is not the solution. You can reproduce this issue starting with the motor stopped.

When the shutter is requested to close, relay 1 can be energised moments before relay 2 and can request an opening and closing with a consequent sudden change of the motor rotation.

The same would occur when the target position is reached if relay 2 is de-energised before relay 1 is de-energised.

cyberdomotik, 2022-07-02:
This is an example of my log:

08:24:25.683 CMD: shutterclose1
08:24:25.711 RSL: RESULT = {"POWER2":"ON"}
08:24:25.713 RSL: POWER2 = ON
08:24:25.816 RSL: RESULT = {"POWER1":"ON"}
08:24:25.819 RSL: POWER1 = ON
08:24:25.821 RSL: RESULT = {"Position1":0}
08:24:25.953 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"		FIRST RELAY ONE IS SWITCHED ON
08:24:25.957 RSL: RESULT = {"SerialSend":"Done"}
08:24:25.996 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"		THEN RELAY TWO COMES ON
08:24:25.000 RSL: RESULT = {"SerialSend":"Done"}

stefanbode, 2022-07-21:
I do not understand the problem. If the shutter in in shutter_mode = 2, then the Relay 2 will be switched in the right direction FIRST. Then the power relays will be switched on. Additionally there is a defined short delay to ensure Relay2 is in the correct direction:
line 1213ff:
```
            case SHT_TIME_UP_DOWN:
              if (!ShutterGlobal.skip_relay_change) {
                // Code for shutters with circuit safe configuration, switch the direction Relay
                ExecuteCommandPowerShutter(Settings->shutter_startrelay[index] +1, new_shutterdirection == 1 ? 0 : 1, SRC_SHUTTER);
                delay(SHUTTER_RELAY_OPERATION_TIME);
                // power on
                ExecuteCommandPowerShutter(Settings->shutter_startrelay[index], 1, SRC_SHUTTER);
              }
```

I cannot influence the rule execution. This is outside the shutter driver. Additionally there is no option to stop the code after the relay2 to let rules run and then switch relay1. The root cause for your problem is the rule stuff. Maybe it works if you change the order in the rules

stefanbode, 2022-08-03:
Please close with: No plans to change.

ascillato, 2022-08-03:
I still believe that is better the safe hardware solution explained in https://github.com/arendst/Tasmota/discussions/15835#discussioncomment-2986806

stefanbode, 2022-08-05:
@cyberdomotik Have you checked if reordering your rules solves your problem? As mentioned above the execution is in the right order. It messes up later in rules.

cyberdomotik, 2022-08-05:
@stefanbode I have changed the order in rule1 to try to get first relay2 to energize and then relay1, as follow:

19:28:14.312 CMD: rule1
19:28:14.318 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":384,"Free":127,"Rules":"on Power2#State=1 do SerialSend5 A00201A3 endon on Power1#State=1 do SerialSend5 A00101A2 endon on Power1#State=0 do SerialSend5 A00100A1 endon on Power2#State=0 do SerialSend5 A00200A2 endon  on Power4#State=1 do SerialSend5 A00401A5 endon on Power3#State=1 do SerialSend5 A00301A4 endon on Power3#State=0 do SerialSend5 A00300A3 endon on Power4#State=0 do SerialSend5 A00400A4 endon"}}

However, relay 1 is energised first and then relay 2,  I don't understand why

19:57:41.246 CMD: shutterclose1
19:57:41.274 RSL: RESULT = {"POWER2":"ON"}
19:57:41.277 RSL: POWER2 = ON
19:57:41.381 RSL: RESULT = {"POWER1":"ON"}
19:57:41.384 RSL: POWER1 = ON
19:57:41.386 RSL: RESULT = {"Position1":0}
19:57:41.533 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
19:57:41.538 RSL: RESULT = {"SerialSend":"Done"}
19:57:41.569 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"
19:57:41.574 RSL: RESULT = {"SerialSend":"Done"}

cyberdomotik, 2022-08-05:
Any suggestions on how to order the rule?

@ascillato Maybe you are right and I will have to discard using shutter mode 2.

stefanbode, 2022-08-07:
How does your rule currently? I would just reorder

stefanbode, 2022-08-07:
Maybe we can remove shuttermode 2 from the automatic detection. But I have never heard about the problem with other users. This is the first short circuit we had.

cyberdomotik, 2022-08-07:
This is my rule1. Remember my device is a ESP-01 with 4Ch-Relay. 

10:52:55.609 CMD: rule1
10:52:55.615 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":384,"Free":127,"Rules":"on Power2#State=1 do SerialSend5 A00201A3 endon on Power1#State=1 do SerialSend5 A00101A2 endon on Power1#State=0 do SerialSend5 A00100A1 endon on Power2#State=0 do SerialSend5 A00200A2 endon  on Power4#State=1 do SerialSend5 A00401A5 endon on Power3#State=1 do SerialSend5 A00301A4 endon on Power3#State=0 do SerialSend5 A00300A3 endon on Power4#State=0 do SerialSend5 A00400A4 endon"}}

stefanbode, 2022-08-07:
@arendst: hi Theo,is there any option in rules to enforce an order of execution? On my shutter I trigger relay2 and then relay1. Anyhow rules are called after relay1. Now the rule always execute relay1 before relay2 nomatter how you order the rules. Any idea? The only idea I have is just to trigger on relay1 and the. Use the backlog command with a delays to first send serial to relay2 and then to relay1.

stefanbode, 2022-08-07:
Maybe it helps if you have a rule1 for relay2 and relay4 and a rule2 for relay1 and relay3

arendst, 2022-08-07:
Rules are processed in the order they are added to rules. If in the meantime new triggers come along they are processed first unless no more resources are available.

There is no rule sequencing or prioritizing as it would costs a large amount of memory to store them before executing.

Perhaps it's time to stop users using rules in time concerned shutter operations.

stefanbode, 2022-08-10:
I did not get it worked to reorder the rules. But anyhow it works on my side with a rule on relay1 only than execute back lock and use if Statement to set relay2 first in the backlog command. The "if" is a compiler option need to be added manually.

stefanbode, 2022-08-22:
Another idea how to rewrite. You need to include the IF THEN ELSE. Not part of the standard compile process. Use:
```
#define USE_EXPRESSION         // Add support for expression evaluation in rules (+3k2 code, +64 bytes mem)  
#define SUPPORT_IF_STATEMENT   // Add support for IF statement in rules (+4k2 code, -332 bytes mem)  
```

```
on Power1#State=1 do backlog
if (Power2#State==1) SerialSend5 A00201A3 else SerialSend5 A00200A2 endif ; delay 10;
SerialSend5 A00101A2 endon

on Power1#State=0 do backlog
if (Power2#State==1) SerialSend5 A00201A3 else SerialSend5 A00200A2 endif ; delay 10;
SerialSend5 A00100A1 endon
```
Please close this thread if this workaround will work. As mentioned your serial stuff is the reason why the relays do not work in the correct order.

Jason2866, 2022-08-31:
If the suggested solution does not work, feel free to reopen

closed 2022-08-31 by Jason2866