arendst/Tasmota #15870: MiElHVAC - sometimes toggling power via mqtt toggles twice via SRC Switch

logon84, 2022-06-25:
### PROBLEM DESCRIPTION
Sometimes while switch power of MiElHVAC module via MQTT (I only use mqtt) a SRC:SWITCH power state appears in log, switching again the power state. This results in no power state change at all, since power toggles twice.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): ESP8266 - MielHVac
- [ ] Tasmota binary firmware version number used: 12.0.1.1
  - [ ] Pre-compiled
  - [X ] Self-compiled
- [ ] Flashing tools used: pio
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:45:57.353 MQT: stat/heatpump/RESULT = {"GPIO0":{"0":"Ninguno"},"GPIO1":{"5376":"MiEl HVAC Tx"},"GPIO2":{"0":"Ninguno"},"GPIO3":{"5408":"MiEl HVAC Rx"},"GPIO4":{"0":"Ninguno"},"GPIO5":{"0":"Ninguno"},"GPIO9":{"0":"Ninguno"},"GPIO10":{"0":"Ninguno"},"GPIO12":{"0":"Ninguno"},"GPIO13":{"0":"Ninguno"},"GPIO14":{"0":"Ninguno"},"GPIO15":{"0":"Ninguno"},"GPIO16":{"0":"Ninguno"},"GPIO17":{"0":"Ninguno"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
23:46:44.208 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
23:46:44.217 MQT: stat/heatpump/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":63,"Free":448,"Rules":"ON switch?#state==2 DO publish cmnd/heatpump/Power toggle ENDON"}}
23:46:44.225 MiElHVAC: sending FC42013010 02000000000000000000000000000000 7b
23:46:44.374 WIF: Probando conexión...
23:46:44.377 MiElHVAC: sending FC42013010 06000000000000000000000000000000 77
23:46:44.480 MiElHVAC: response data 020000010308030500000CAE00000000
23:46:44.482 SRC: Backlog
23:46:44.485 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
23:46:44.494 MQT: stat/heatpump/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:46:44.633 MiElHVAC: sending FC42013010 02000000000000000000000000000000 7b
23:46:44.735 SRC: Backlog
23:46:44.738 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
23:46:44.745 MQT: stat/heatpump/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
NOTE: I tried solving this with rule1, but I get same behaviour. Without rules the same happens too
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:48:01.984 CMD: Status 0
23:48:01.986 SRC: WebConsole from 192.168.1.2
23:48:01.989 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
23:48:01.995 MQT: stat/heatpump/STATUS = {"Status":{"Module":0,"DeviceName":"MiEl-HVAC","FriendlyName":["Tasmota"],"Topic":"heatpump","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[15,15,15,15,15,15,15,15],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":1,"StateRetain":0}}
23:48:02.002 MQT: stat/heatpump/STATUS1 = {"StatusPRM":{"Baudrate":2400,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"6T05:32:26","StartupUTC":"2022-06-19T16:15:35","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2022-06-19T17:10:04","SaveCount":112,"SaveAddress":"F4000"}}
23:48:02.013 MQT: stat/heatpump/STATUS2 = {"StatusFWR":{"Version":"12.0.1.1(tasmota)","BuildDateTime":"2022-06-19T18:01:59","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"361/699"}}
23:48:02.025 MQT: stat/heatpump/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["neologonet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C80001000600003C5A0A190000000000","00000080","00006000","00004101","00000000"]}}
23:48:02.044 MQT: stat/heatpump/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["0000040A","8F9AC787","04368001","000000CF","010013C0","C000F981","04004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,44,45,62","Sensors":"1,2,3,4,5,6"}}
23:48:02.056 MQT: stat/heatpump/STATUS5 = {"StatusNET":{"Hostname":"tasm-heatpump","IPAddress":"192.168.1.212","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"80:7D:3A:69:88:AC","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:48:02.066 MQT: stat/heatpump/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"mitsu_aircon","MqttClient":"mitsu_aircon","MqttUser":"mitsu_aircon","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:48:02.078 MQT: stat/heatpump/STATUS7 = {"StatusTIM":{"UTC":"2022-06-25T21:48:02","Local":"2022-06-25T23:48:02","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+02:00","Sunrise":"05:47","Sunset":"21:57"}}
23:48:02.092 MQT: stat/heatpump/STATUS10 = {"StatusSNS":{"Time":"2022-06-25T23:48:02","MiElHVAC":{"Temperature":24.5,"Operation":"ON","Compressor":"OFF","roomtemp":"0300000E00B4B1000000000000000000","status":"060000000107759E4200000000000000","stage":"09000000030000000000000000000000"},"TempUnit":"C"}}
23:48:02.104 MQT: stat/heatpump/STATUS11 = {"StatusSTS":{"Time":"2022-06-25T23:48:02","Uptime":"6T05:32:27","UptimeSec":538347,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"neologonet","BSSId":"14:91:82:73:67:85","Channel":1,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":2,"Downtime":"0T00:01:43"}}}
23:48:02.136 MiElHVAC: response data 09000000030000000000000000000000

```
NOTE: SwitchMode was 0 0 0 0 0 0 0 0 nd the issue happened. Tried to set it to 15 to solve the problem, but no luck.
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:17:56.203 MiElHVAC: response data 060000000000199DD400000000000000
14:17:56.258 MiElHVAC: sending FC42013010 02000000000000000000000000000000 7b
14:17:56.361 SRC: MQTT
14:17:56.364 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 2, Pld 1, Data 'ON'
14:17:56.370 MQT: stat/heatpump/RESULT = {"POWER":"ON"}
14:17:56.375 MQT: stat/heatpump/POWER = ON
14:17:56.462 MiElHVAC: sending FC41013010 01010001000000000000000000000000 7b
14:17:56.564 MiElHVAC: response data 020000000307030500000CB000000000
14:17:56.566 SRC: Switch
14:17:56.574 MQT: stat/heatpump/RESULT = {"POWER":"OFF"}
14:17:56.578 MQT: stat/heatpump/POWER = OFF

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Power ON or power OFF via MQTT sometimes works. Others a SRC SWITCH toggles power again

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
No SRC Switch toggling power when sending a mqtt command

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-06-25:
You did not provide the template asked for in the template, and thus your GPIO configuration is not visible, as other messages indicate that you with Module 0 use the template for this part of the config.

`SRC: Switch` should indicate that a gpio pin defined as a "Switch" had activity, with the default action of Tasmota being to switch the power output. What is the purpose of the Switch pin you have defined? If you have no use for it, you could consider removing this config.

logon84, 2022-06-25:
> You did not provide the template asked for in the template, and thus your GPIO configuration is not visible, as other messages indicate that you with Module 0 use the template for this part of the [...]

Is not the output of "Backlog Template; Module; GPIO 255" what you asking for? It is present in my first message. No switches defined. I have been digging in the code, seems that SRC SWITCH in this case is a interpretation of the phisical button attached to the HVAC machine, not the esp8266. So for me it seems that the code maybe needs a delay after changing the power state: it seems to me that after changing the power via mqtt, the code compares this mqtt command with the received frames too soon, and as they don't match, the power is toggled again via SRC_SWITCH

sfromis, 2022-06-26:
Yes, that was what I was missing. You only provided the output of the last of those 3 commands.

Anyway, I did also a bit later notice that misleading usage of `SRC_SWITCH` the driver, and without using that specific device, I cannot have anything to add about what the root cause of the confusion may be. You seem to be getting there :-)

closed 2022-07-26 by github-actions[bot]