arendst/Tasmota #15877: ShutterEnableEndStopTime not precise enough (no decimals supported)

T-vK, 2022-06-26:
### PROBLEM DESCRIPTION
My shutters don't open up all the way. Every time I open them, they are offset by a millimeter or so and after opening them 10 times it's already a centimeter and then when I close them, of course they are closed too far by a centimeter as well. There is nothing that can be done about that and that's okay, but there is `ShutterEnableEndStopTime`, which would fix this issue, but it only supports 0 or 1 seconds which is way too far and would break my shutters. I would need to set it to roughly 0.1 to not cause any damage. If that command could be extended to support decimal numbers that would be perfect.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 12.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Site
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:57:02.323 CMD: Backlog Template; Module; GPIO 255
17:57:02.347 MQT: stat/tasmota_CFE004/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:57:02.552 MQT: stat/tasmota_CFE004/RESULT = {"Module":{"18":"Generic"}}
17:57:02.760 MQT: stat/tasmota_CFE004/RESULT = {"GPIO0":{"352":"Counter1"},"GPIO1":{"0":"None"},"GPIO2":{"256":"Relay_i1"},"GPIO3":{"0":"None"},"GPIO4":{"225":"Relay2"},"GPIO5":{"416":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules.

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
Not necessary.

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Not necessary.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Click the shutter up/down button in the webinterface a couple of times. Using stepper motors.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Everything happened as described in the docs. No need to elaborate.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Not necessary.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Not necessary.

**(Please, remember to close the issue when the problem has been addressed)**

Gifford47, 2022-06-27:
I also have the same problem with the timer during "normal" shutter operation in connection with the shelly 2.5.
If I open or close the shutter several times in different positions, the deviation from the real position is always greater. Several seconds are not counted when opening or closing.... the same with the hardware switches.

@stefanbode is it possible that it is the same problem as time is resolved in tasmota? https://github.com/arendst/Tasmota/discussions/10443#discussioncomment-1617309

T-vK, 2022-06-27:
An endstoptime of 0.1s would in fact fix the issue for me. I dont have end stops. Once the shutters are open, I can let the motors continue spinning for roughly 0.1 more seconds without causing damage, even though it is all the way open already. If I go any longer, it puts too much strain on the mechanics. 0.1s is fine tho, once Tasmota turns the motors off they will automatically get pulled backwards by the chain a little if they have been spun too far.

001oc, 2022-07-08:
I have the same "problem" with my chicken coop. Costwise it makes no sence to me to install additional endswitches. As of that after a couple of days I move the steppermotor by hand somewhat to have it more or less at the exact position to open and close the door completely. That "correction" I have to do very coule of days. Maybe what would help alot to be more precise, at least as an second option for stepper motors, would be the possibility to substitute seconds with a specific number of steps. Assuming there are no missing steps 1000 steps up or 1000 steps down should be the same amount of movement - every time.

stefanbode, 2022-07-19:
Steppermotors are always 100% correct. The is by design no slip possible. This is the benefit of the steppers compared to normal motors that are time based. The ONLY way why you may have a skip is a too high torque at the motor. Please use a higher Shuttermotordelay to have a smoother acceleration.  For steppers the additional time or steps are not required and not implemented. My furniture door is in sync for more that a year.

stefanbode, 2022-07-19:
> An endstoptime of 0.1s would in fact fix the issue for me. I dont have end stops. Once the shutters are open, I can let the motors continue spinning for roughly 0.1 more seconds without causing [...]

Why did you just not increase the Shuttercloseduration a bit. This can be done in 0.1 steps?

T-vK, 2022-07-19:
`Shuttercloseduration` only changed the speed at which the shutters open/close. It didn't make a difference, if I set them to 5s or 50s, the latter only took ten times longer. I don't know if this is a bug, but I couldn't find a fix.

001oc, 2022-07-19:
What I have observed is that the number that is displayed at the very top of the main menu (Counter1 / Zähler1) seems to be corresponding to something like the steps the steppermotor is doing while running, or maybe the milliseconds the (stepper)motor is spinning. The value of the counter is variing abit for every turnarround. If there is a Shutteropenduration/Shuttercloseduration of 10 seconds the counter is displaying something between 10080 and 10110 (mostly it varies between 10085 and 10095) - not depending if turning left or right. For my experiments I have used a motor that has no burden at all mounted to the motor that could slow it down, it is freely spinning. As stated above, steppermotors are mostly precise by design - and for that they are typically measured by stepps and not by time or revolutions. So, why not having the (additional) possibility to even use steppermotors by design and give it a turning value by steps instead of econds? At the beginning, when I decided to use tasmota for the door of my chicken coob, I found it a bit strange to use seconds instead of steps to tell a steppermotor how long to run - of course in the meantime I am used to do it with seconds, and I understand that this is the best (only) way to do the job with motors other than steppers, but for stepper motors it still seams strange. One thumb up from me to have the possibility to use stepper motors as designed and feed them with steps ;-)

stefanbode, 2022-07-19:
the counter depends on the PWMfrequency and the duration. Its just a multiplication for the open time. You values are varying and this is not as it should be. What shuttermotordelay do you use. Can you check if it is e.g. 1sec its gets better. Please provide all your shutter settings a log file with loglevel 4. I'm intrested in the last 2 seconds before the motor stops. the rest I do not care.

stefanbode, 2022-07-19:
If the shutter is tooooo fast it can happen that it cannot hit the should be of counter 10100. This is because of the time slice of 0.05sec of Tasmota.

stefanbode, 2022-07-19:
Hi @001oc : I can confirm that this is a bug. The software is not doing what it should do. The hint to ESP32 was very important. During the stop I change the frequency of the PWM pin to decelerate the stepper motor. This works very smart on ESP8266, but it looks like that the ESP32 did not like the frequency change or the call to change the frequency kicks not in:
```
  #ifdef ESP8266
            // Convert frequency into clock cycles
            uint32_t cc = microsecondsToClockCycles(1000000UL) / Shutter[i].pwm_velocity;
            startWaveformClockCycles(Pin(GPIO_PWM1, i), cc/2, cc/2, 0, -1, 0, false);
  #endif  // ESP8266
  #ifdef ESP32
            analogWriteFreq(Shutter[i].pwm_velocity);
            analogWrite(Pin(GPIO_PWM1, i), 50);
  #endif  // ESP32
```
The decelerate does not what it should do. At then end the shutter is too fast and there is a overrun. Symptoms and Root Cause seems quite clear. I must admit I did not test this against a ESP32. Looks like time to do it now.

001oc, 2022-07-19:
@stefanbode that sounds it could be fixed quiet simple eventually. If you need some hardware support (e.a. additional ESP32 for testing purpose) I would gladly send you one.

I am still thinking that a stepper motor is happier at all with steps instead of seconds  ;-) ... what is about a "SetOption999" that indicates if the value is interpreted as seconds or steps, with a default value for seconds of course (I could imagine to send another ESP32 for such a Setoption :-)) ) .

stefanbode, 2022-07-20:
shutteropenduration is seconds for all the others and just do not fit to steps for stepper. The steps is very easy calculation: shutteropenduration x PWMfrequency = steps ( 10sec x 1000Hz = 10.000 steps).
I'm fine with hardware. It's more a problem that the fancy functions do not exist anymore on ESP32. I have a similar problem with the Zero-Cross Triac Dimmer Driver. Here I also rely on changing the frequency in very fine granulation. Also not possible for ESP32 right now.

001oc, 2022-07-20:
@stefanbode that sounds a bit odd.

Well, when I started with my chicken coob I remember that I was wondering about that start and stop delay seems to have no effect in general. I thought I havn't understood exactly what the function is good for or at least are to stupid to see connection to other fuctions that are interfearing with the delay stuff to get it working correctly. Even so I am not only using a completely overdriven ESP32 but also overdriven TMC2208 to not wake up or scare sleeping chicken for the project and was not sure if maybe the combination will cause missing functionality concerning delays. Well, at that time I decided to simply ignore it because the whole project seems to be working good enough - except the little glitch causing the fading of start- and/or endpoints. Now it seems that even the missing effect of start- and stopdelays at that time points to the same problems you found out concerning ESP32s.

For me the bug is not very important to be fixed soon, but I am willing to support as much as I can if there is something I cann be supportive.

arendst, 2022-07-20:
I think in order to use ANY ``analogWrite()`` on the ESP32 you will need to execute ``analogAttach()`` first to make a link between the GPIO and the analog channel.

I suggest to search for ``analogAttach(`` how to use this.

stefanbode, 2022-07-21:
Thanks for the help. Code is changed and now will be tested. Hopefully release-able today. Looks promising.

stefanbode, 2022-07-21:
@arendst : There is a small missimplementation of the ESP32 _analogWriteFreqRange. In the new ESP32 the frequency change will applied on all channels directly after calling the analogWriteFreq. In the ESP8266 the call analogWriteFreq just load the new freq into a buffer. It will then be transferred to exactly ONE pin with the next analogWrite(). All other pins maintain their frequency. Can I make changes to esp8266toESP32.cpp ?

arendst, 2022-07-21:
Yes as long as you keep the current functionality.

So I suggest you add your required changes in a seperate function which doesn't change the current ones.

arendst, 2022-07-21:
I would suggest something like this:
```
void analogWriteFreq(uint32_t freq, uint32_t pin = -1);
void analogWriteFreq(uint32_t freq, uint32_t pin) {
  _analogInit();      // make sure the mapping array is initialized
  if (pin > -1) {
    int32_t chan = _analog_pin2chan(pin);
    if (chan >= 0) {
      ledcSetup(chan, freq, pwm_bit_num);
    }
  } else {
    pwm_frequency = freq;
    _analogWriteFreqRange();
  }
}

/*
void analogWriteFreq(uint32_t freq) {
  pwm_frequency = freq;
  _analogWriteFreqRange();
}
*/
```
where you call like ``analogWriteFreq(Shutter[i].pwm_velocity, Pin(GPIO_PWM1, i));

stefanbode, 2022-07-21:
Hmm, even with the analogWriteFreq I do not get a change on the frequency. it is still 977 or 1000. Something like this.... Investigating.....

stefanbode, 2022-07-21:
The decelleration process is working. but I have no idea why the accellerate did not do it. My current binary:

https://github.com/stefanbode/Sonoff-Tasmota/blob/development/firmware.bin

Limitations after power on it could take some up and down until it works. The start is always delayed by shuttermotordelay. e.g. suddenly after 1 second it starts. ramp up does not work, but the ramp down. Position is ok, but this is no version to release. Need more time to understand the side effects.

arendst, 2022-07-22:
Looking over the ESP32 analogWrite code the behaviour you experience might well be a result of how the code is implemnted.

For your use-case I suggest you take a look in platformio ``framework-arduinoespressif32`` file ``cores/esp32/esp32-hal-leds.c``. There you'll see that initial call to analogWrite will reset the frequency to 1000Hz and 8-bit resolution.

You also find a useful function ``ledcChangeFrequency()``. It calls IDF function ``ledc_timer_config()`` which in turn changes frequency for probably all channels at once.

So you'll probably have to find a way to change frequencies before every call to ``analogWrite()``.

arendst, 2022-07-22:
OK let's try this. Add the following to file ``lib\libesp32\ESP32-to-ESP8266-compat\esp8266toEsp32.cpp``:

```
void analogWriteFreqPin(uint32_t freq, int pin) {
  _analogInit();      // make sure the mapping array is initialized
  if (pin > -1) {
    int32_t chan = _analog_pin2chan(pin);
    if (chan >= 0) {
//      ledcSetup(chan, freq, pwm_bit_num);
      ledcChangeFrequency(chan, freq, pwm_bit_num);
    }
  } else {
    pwm_frequency = freq;
    _analogWriteFreqRange();
  }
}

void analogWritePin(uint8_t pin, int value) {
  _analogInit();      // make sure the mapping array is initialized
  int chan = _analog_pin2chan(pin);
//  if (chan < 0) {    // not yet allocated, try to allocate
//    chan = analogAttach(pin);
    if (chan < 0) { return; }   // failed
//  }
  ledcWrite(chan, value);
}
```
and add specific code for ESP32 for all ``analogWrite()`` and ``analogWriteFreq()`` calls in shutter.ino calling ``analogWritePin()`` and ``analogWriteFreqPin()``.

stefanbode, 2022-07-22:
Hi Theo, thanks for the help. But the reason at the end was not to use the analogWrite with ESP32. I went back to the functions you use on "pwmfrequency". This finally made it. The attach is already working in the default. At the end the change is not that big. Anyhow I see, even if I change only frequency for one channel the frequency is also changes on the other channel. This will be a future fix. At least it now works for ONE shutter with steppers and ESP32

stefanbode, 2022-07-24:
@001oc : finally we make it. At least for one shutter it works now on esp32. You will see that the shuttermotordelay now has an impact. From my testing I would guess your positioning Probleme is gone. Please use the latest development version and provide feedback.

001oc, 2022-07-24:
@stefanbode that sounds great. Currently I am on a business trip and I have forgotten to take my testkit with me (what I would do). So I can't test right now. At the end of the week I am back and will try it then asap.

stefanbode, 2022-07-25:
[image]
it looks like the ESP32 with the LED driver that we missuse does not support different frequencies on the different channels. Looks like all channels have the same frequency. Therefore stopping here investigations. 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html

stefanbode, 2022-07-25:
@arendst : I do see two major roadblocks with the ESP32 compared to the flexibility we had in ESP8266. One is the mentioned above shared frequency over all channels and secondly I really struggle that you just can provide the Hz an an integer and not as a float variable. This makes it quite "chunky" also on lower frequency. Same with my Zero-Cross driver. This will also never work on ESP32 is not e.g. 50.00012 Hz is possible.

stefanbode, 2022-07-25:
[image]
There is a bind timer call that allows to bind to different timers I will give it a try later today. Looks like normally the timer is shared over all instances but you can change it.

stefanbode, 2022-07-25:
In the anallogAttach there is this little nasty line.....      

```
// ledcAttachPin(pin, channel);  -- replicating here because we want the default duty
      uint8_t group=(chan/8), channel=(chan%8), timer=((chan/2)%4);
```
The timer here is set in the way that channel 0 and 1 are on time 0 and then 2 and 3 on timer 2 and so on. Interesting idea, but this causes my shutter 2 not operate independent from shutter1.

stefanbode, 2022-07-26:
@s-hadinger : Hi, I'm deep diving the esp8266toESp32.cpp file and found a line of code that first give me some problems with the shutter and second is not easy to understand why it is implemented this way. The line is:
`uint8_t group=(chan/8), channel=(chan%8), timer=((chan/2)%4);`

And I do not understand the timer calculation. channel 0 and 1 are on timer0 and then channel 2 and 3 on timer1 and so on. So PWM1 and PWM2 do have the same frequency. I agree there are only 4 timers but why grouping in 2. For me it would be easier if PWM1..4 have own frequencies and behave like ESP8266 and the PWM5 is on timer0 again. Maybe strange side effect but this is the way.
`uint8_t group=(chan/8), channel=(chan%8), timer=(chan%4);`
I'm testing my shutter code with the modification above, but this would be a breaking change therefore asking here for some guidance.

s-hadinger, 2022-07-28:
This piece of code was taken from Arduino.

I'm not against this change but it would also require to change `ledcSetup()` from Arduino.

https://github.com/espressif/arduino-esp32/blob/943216308d0425091498a33ba73e375cbd76287a/cores/esp32/esp32-hal-ledc.c#L64

```
uint32_t ledcSetup(uint8_t chan, uint32_t freq, uint8_t bit_num)
{
    if(chan >= LEDC_CHANNELS || bit_num > LEDC_MAX_BIT_WIDTH){
        log_e("No more LEDC channels available! (maximum %u) or bit width too big (maximum %u)", LEDC_CHANNELS, LEDC_MAX_BIT_WIDTH);
        return 0;
    }

    uint8_t group=(chan/8), timer=((chan/2)%4);

    ledc_timer_config_t ledc_timer = {
        .speed_mode       = group,
        .timer_num        = timer,
        .duty_resolution  = bit_num,
        .freq_hz          = freq,
        .clk_cfg          = LEDC_DEFAULT_CLK
    };
    if(ledc_timer_config(&ledc_timer) != ESP_OK)
    {
        log_e("ledc setup failed!");
        return 0;
    }
    channels_resolution[chan] = bit_num;
    return ledc_get_freq(group,timer);
}
```

s-hadinger, 2022-07-28:
I suppose we need to patch even more the Arduino framework to have granular control of timers. For example all LED PWM can be attached to a single timer, they don't need different frequencies

stefanbode, 2022-08-03:
There is a pull request on the way: https://github.com/arendst/Tasmota/pull/16131
This will finally solve the problem for the ESP32. Now again up to 4 shutters with steppers are supported on one device.

Jason2866, 2022-08-05:
@T-vK  Please test with latest development version.

ascillato2, 2022-08-07:
@T-vK Please test with latest development version. Thanks.

closed 2022-08-07 by ascillato2