arendst/Tasmota #15894: Feature Request:  Expose Pin Touch_Threshold as setoption

jgoldwein, 2022-06-29:
### PROBLEM DESCRIPTION
With use of touch pins on ESP32-based devices, pin touch sensitivity and hit count can only be set using the calibration module and cannot be modified afterwards unless re-calibration is performed.  Alternatively, it appears these thresholds can be modified directly in the support_button.ino file by modifying the TOUCH_PIN_THRESHOLD and/or the TOUCH_HIT_THRESHOLD parameters, and then compiled into the firmware.   

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ESP32-DevKit,LILO ESP32] Device used (e.g., Sonoff Basic): _____
- [12.0.2.2 ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
- [OTA] Flashing tools used: _____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
e.g. setoptionX 12
      setoptionY  3

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-06-29:
First, opening an issue is not the way to make suggestions, especially not without having read the docs. There seems to already exists commands for what you ask for, [TouchThres and TouchNum](https://tasmota.github.io/docs/TouchPin/#commands) :grin:

jgoldwein, 2022-06-29:
I did indeed see those, but they do not seem to work.

≈=======≈====================
PDA Brief
Joel Goldwein, M D.