arendst/Tasmota #15895: SaveData reset to OFF using Shutters

epandi, 2022-06-29:
### PROBLEM DESCRIPTION
When using the physical buttons of the my shutters (switch DS-102 2 gang using standard template from repo), the shutterposition is not saved to memory. When my shutters  go to closed state and I power cycle my device (replug power cord from wall socket) tasmota  thinks my shutters are in open position. I have to press the close button (although physically closed) and wait the shutterclose time until I can finally open my shutters  again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): DS-102 2-gang (https://templates.blakadder.com/DS-102_2.html)
- [X] Tasmota binary firmware version number used: 12.0.2.2(tasmota) 2022-06-24T15:44:48
  - [X] Pre-compiled
- [X] Flashing tools used: Webupload
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:17:36.692 MQT: stat/fleurblinds/RESULT = {"NAME":"DS-102 2 Gang","GPIO":[576,321,0,32,225,33,0,0,0,224,320,1,0,0],"FLAG":0,"BASE":18}
19:17:36.903 MQT: stat/fleurblinds/RESULT = {"Module":{"0":"DS-102 2 Gang"}}
19:17:37.158 MQT: stat/fleurblinds/RESULT = {"GPIO0":{"576":"LedLink_i"},"GPIO1":{"321":"Led_i2"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"225":"Relay2"},"GPIO5":{"33":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"224":"Relay1"},"GPIO14":{"320":"Led_i1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
19:18:01.654 MQT: stat/fleurblinds/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
19:18:01.869 MQT: stat/fleurblinds/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":67,"Free":444,"Rules":"ON Clock#Timer=2 DO Rule1 0 ENDON ON Clock#Timer=3 DO Rule1 1 ENDON"}}
19:18:02.119 MQT: stat/fleurblinds/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":108,"Free":403,"Rules":"ON System#Boot DO Shutterposition1 ENDON ON Shutter1#Position DO Publish stat/%topic%/SHUTTER1 %value% ENDON"}}
```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:18:18.228 MQT: stat/fleurblinds/STATUS = {"Status":{"Module":0,"DeviceName":"FleurBlinds","FriendlyName":["FleurBlinds","FleurBlinds"],"Topic":"fleurblinds","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:18:18.263 MQT: stat/fleurblinds/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:25:04","StartupUTC":"2022-06-29T16:53:14","Sleep":50,"CfgHolder":4617,"BootCount":822,"BCResetTime":"2020-05-01T08:02:39","SaveCount":2623,"SaveAddress":"F4000"}}
19:18:18.298 MQT: stat/fleurblinds/STATUS2 = {"StatusFWR":{"Version":"12.0.2.2(tasmota)","BuildDateTime":"2022-06-24T15:44:48","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"418/699"}}
19:18:18.318 MQT: stat/fleurblinds/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.0.222","LogPort":514,"SSId":["XXXXXXXXXX","Wurmhole"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C809","1405C80001000600003C5A0A000000000000","400000C9","00006000","00004000","00000000"]}}
19:18:18.353 MQT: stat/fleurblinds/STATUS4 = {"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
19:18:18.388 MQT: stat/fleurblinds/STATUS5 = {"StatusNET":{"Hostname":"fleurblinds-2933","IPAddress":"192.168.1.32","Gateway":"192.168.0.1","Subnetmask":"255.255.252.0","DNSServer1":"8.8.4.4","DNSServer2":"0.0.0.0","Mac":"EC:FA:BC:4B:0B:75","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:18:18.412 MQT: stat/fleurblinds/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B0B75","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:18:18.436 MQT: stat/fleurblinds/STATUS7 = {"StatusTIM":{"UTC":"2022-06-29T17:18:18","Local":"2022-06-29T19:18:18","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:49","Sunset":"21:57"}}
19:18:18.456 MQT: stat/fleurblinds/STATUS10 = {"StatusSNS":{"Time":"2022-06-29T19:18:18","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
19:18:18.469 MQT: stat/fleurblinds/STATUS11 = {"StatusSTS":{"Time":"2022-06-29T19:18:18","Uptime":"0T00:25:04","UptimeSec":1504,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"E4:8D:8C:D1:CC:3B","Channel":1,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:07"}}}
19:18:18.503 MQT: stat/fleurblinds/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":330,"Close":330,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Syslog output here:

See attachement

```
[savedata.txt](https://github.com/arendst/Tasmota/files/9012806/savedata.txt)

### TO REPRODUCE
1. Close the shutters
2. Physically power cycle the device
3. command: savedata, now you'll notice savedate is OFF (log: 18:52:41.412)
4. command: savedata 1, now savedata is ON (log: 18:52:48.393)
5. press button to open shutters (log: 18:53:34.490)
6. press button to close shutters (log: 18:53:37.917)
7. command: savedata, now you'll notice savedata is OFF again (log: 18:54:25.929)

### EXPECTED BEHAVIOUR
SaveData kept to ON.
ShutterPosition stored to flash so after a power cycle tasmota remembers if the blinds were last opend or closed.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-07-18:
Try latest dev release and report back pls.

epandi, 2022-07-19:
Thanks for the fix Theo. It seems to have fixed the problem.

closed 2022-07-19 by epandi