arendst/Tasmota #15900: SCD40 console commands not working

circuitmike, 2022-06-30:
### PROBLEM DESCRIPTION
The console commands for the SCD40/SCD41 sensor are not recognized (scd40cal, scd40test, etc.) I compiled my own binary with support for both the SCD30 and SCD40/SCD41 sensors. The console commands for the SCD30 sensor work as expected but those for the SCD40/SCD41 do not. Both sensors are connected simultaneously, if that might matter. I did check the source code and the commands are listed there, as is the code meant to handle them.

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Adafruit Feather HUZZAH with ESP8266 (2821)
- [X] Tasmota binary firmware version number used: 12.0.2
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: esptool.py
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:21:28.697 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:21:28.922 RSL: RESULT = {"Module":{"18":"Generic"}}
12:21:29.178 RSL: RESULT = {"GPIO0":{"64":"Button_n1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"65":"Button_n2"},"GPIO17":{"4704":"ADC Input"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
12:22:22.553 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":256,"Free":255,"Rules":"on tele-SCD30 do displaytext [i] endon on tele-SCD30#CarbonDioxide do displaytext [x0y0p19]CO2: %value% ppm endon on tele-SCD30#Temperature do displaytext [x0y11p19]Temp: %value% F endon on tele-SCD30#Humidity do displaytext [x0y22p19]RH: %value% pct endon"}}
12:22:22.781 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
12:22:23.030 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [X] Provide the output of this command: `Status 0`:
```
12:23:00.254 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"co2","FriendlyName":["co2"],"Topic":"co2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:23:00.285 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:06:49","StartupUTC":"2022-06-30T16:16:11","Sleep":50,"CfgHolder":4617,"BootCount":61,"BCResetTime":"2022-06-13T03:51:27","SaveCount":463,"SaveAddress":"F9000"}}
12:23:00.318 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-30T16:11:04","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"356/699"}}
12:23:00.335 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID1","SSID2"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["0100A109","2805C80001000600003C5A0A190000000000","000000C0","00006000","00004000","00000000"]}}
12:23:00.369 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":678,"Free":324,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FBAC787","043683A1","000000CF","014013C0","C000F981","00004024","00001000","14004020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,13,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6,42,92"}}
12:23:00.407 RSL: STATUS5 = {"StatusNET":{"Hostname":"co2","IPAddress":"192.168.0.75","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"11:22:33:44:55:66","Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":17.0}}
12:23:00.430 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F20CD","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:23:00.449 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-06-30T16:23:00","Local":"2022-06-30T12:23:00","StartDST":"2022-03-20T02:00:00","EndDST":"2022-11-13T02:00:00","Timezone":99,"Sunrise":"23:50","Sunset":"15:57"}}
12:23:00.508 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-06-30T12:23:00","ANALOG":{"A0":398},"SCD30":{"CarbonDioxide":731,"eCO2":757,"Temperature":75.6,"Humidity":42.5,"DewPoint":51.2},"SCD40":{"CarbonDioxide":896,"eCO2":922,"Temperature":71.9,"Humidity":46.6,"DewPoint":50.3},"TempUnit":"F"}}
12:23:00.535 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-06-30T12:23:00","Uptime":"0T00:06:49","UptimeSec":409,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID1","BSSId":"11:22:33:44:55:66","Channel":11,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:10"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```
12:29:31.708 CMD: scd40cal
12:29:31.710 SRC: WebConsole from 192.168.0.32
12:29:31.713 CMD: Grp 0, Cmd 'SCD40CAL', Idx 1, Len 0, Pld -99, Data ''
12:29:31.717 RSL: RESULT = {"Command":"Error"}
```

### TO REPRODUCE
Attempt to use any of the scd40 console commands (scd40test, scd40cal, etc.)

### EXPECTED BEHAVIOUR
The commands would execute as expected, as they do with the SCD30 sensor

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-06-30:
It looks like some commands including `Scd40Cal` and `Scd40Test` (but not only) requires measurment to be stopped in order to be proceed
You should use `Scd40Stop` before attempting any of these function and then use eihter `Scd40Strt` or `Scd40StLP` to restart measuments

As there doesn't seem to be any specific documentation page about this driver, a contributor who has the device and learned how to use it the hard-way would be welcomed to participate to the docs.

circuitmike, 2022-06-30:
`scd40stop` returns the same error message as the other commands, unfortunately. That's interesting, though - I didn't even notice that that command existed. I think I'll dig into the source code a bit more and see if I can figure out what's going on here. And if I do eventually get this figured out I'd be happy to contribute to the documentation!

barbudor, 2022-06-30:
Ah, that's unfortunate
Without the device in hand, I would be difficult to go further
If you are able to do some coding, this driver lacks debug messages that would be helpful to understand hte cause of the failure.
I see from your StatusSNS that the SCD40 is there, so the driver is enabled and the device is detected (driver not included, not enabled or device not detected is a common source of not working commands - not the case here apparently)

May be you could try with only the SCD40 connected (and disabling the SCD30 driver) just to narrow donw the things

sfromis, 2022-06-30:
#11329 has a little bit of discussion of the commands, including this point:
> Most commands can only be issued when no periodic (low power or regular) measurement is being done.

(And that the intention about creating documentation too, but....)

circuitmike, 2022-07-26:
I don't want the bot to close this so I'm replying. I still haven't had a chance to dig into the code, but I have the hardware sitting on my desk and I'll take a look soon.

circuitmike, 2022-07-27:
I recompiled 12.0.2 with support for the SCD40/41 but *without* support for the SCD30. Interestingly, the commands that returned "unknown command" errors in the past now seem to work fine, even with both sensors connected. I'm not sure yet why this is, but I'm thinking there's a duplicate symbol somewhere or something like that, maybe as the result of a copy-and-paste operation since the SCD40/41 and SCD30 sensor code are so similar, the latter being based on the former. I'll continue to dig and see what I come up with.

arendst, 2022-08-11:
Pls verify using latest dev release and report back.

circuitmike, 2022-08-14:
I can't even build the latest development branch. My attempt ends like this:
```
Linking .pio/build/tasmota/firmware.elf
/home/mike/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
/home/mike/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
/home/mike/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/tasmota/libFrameworkArduino.a(core_esp8266_main.cpp.o): in function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
/home/mike/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/tasmota/firmware.elf] Error 1
```
That looks suspiciously like my build environment is somehow wrong, but I can build the current master branch just fine.

circuitmike, 2022-08-14:
@arendst OK, using the latest development branch the behavior is different, but still wrong.

The SCD40 and SCD30 drivers coexist just fine now. However, the sdc40cal command (which was what sent me down this rabbit hole to begin with) is not recognized as a valid command. This is true whether or not the SCD30 is connected and/or whether the SCD30 driver is compiled in or not. All the other scd40 commands seem to work fine. Example:
```
22:20:18.872 CMD: scd40stop
22:20:18.879 RSL: RESULT = {"SCD40Stop":0}
22:20:22.459 CMD: scd40cal
22:20:22.465 RSL: RESULT = {"Command":"Unknown"}
22:20:25.045 CMD: scd40toff
22:20:25.054 RSL: RESULT = {"SCD40Toff":400}
22:20:34.154 CMD: scd40strt
22:20:34.161 RSL: RESULT = {"SCD40Strt":0}
```
In the past I was at least able to use the scd40cal command if the scd30 support wasn't compiled in.

circuitmike, 2022-08-14:
OK, after some false starts (and a couple of deleted comments as a result) I can confirm that the problem no longer exists in the latest dev version. Everything seems fine now. Thanks for everyone's help!

closed 2022-08-14 by circuitmike