arendst/Tasmota #15909: Flowratemeter not compiling

sergios100, 2022-07-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

In tasmota/my_user_config.h line 953

[tasmota/my_user_config.h](https://github.com/arendst/Tasmota/pull/15474/files#diff-f2a16bbe8989c2d8613ad0a8008616219427637e2314796a6f4acd9e14efad22)

is written #define FLOWRATEMETER

this seems not to be working. Instead it should state: #define **USE_FLOWRATEMETER**

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Generic ESP8266
- [X] Tasmota binary firmware version number used: 12.0.2.2
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: OTA-Uplaod
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  
21:34:37.406 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:34:37.624 RSL: RESULT = {"Module":{"0":"Generic"}}
21:34:37.880 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"8352":"Flowrate1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  21:35:53.710 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_C2C9BC","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:35:53.736 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:32","StartupUTC":"2022-07-01T20:34:21","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2022-06-30T21:40:17","SaveCount":119,"SaveAddress":"F9000"}}
21:35:53.769 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2.2(tasmota)","BuildDateTime":"2022-07-01T19:39:19","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"373/699"}}
21:35:53.789 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Apple Network",""],"TelePeriod":300,"Resolution":"558180C4","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000088","00006000","00004000","00000000"]}}
21:35:53.821 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004104","00001000","5C000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6,68,96"}}
21:35:53.852 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-C2C9BC-2492","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"EC:FA:BC:C2:C9:BC","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:35:53.876 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C2C9BC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:35:53.899 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-07-01T20:35:53","Local":"2022-07-01T21:35:53","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"}}
21:35:53.918 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-07-01T21:35:53","Flowrate":{"Rate":[ 0],"Value":"average","Unit":"l/min"}}}
21:35:53.929 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-07-01T21:35:53","Uptime":"0T00:01:32","UptimeSec":92,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Apple Network","BSSId":"C4:41:1E:30:3D:E4","Channel":1,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
remove the two "//" in front of #define FLOWRATEMETER in tasmota/my_user_config.h line 953 and compile it. "Flowrate" will not show up in the configuration as a GPIO input!

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

If the two "//" are removed I expect a compiled binary containing the code for the Flowratemeter.
I could solve the Problem by adding "use_" in front of FLOWRATEMETER

#define **USE_FLOWRATEMETER** than it is compiled and works

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I used GITpod to compile.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-07-02:
As you know the fix, why providing an issue instead of PR?
For such a simple one it can be done straight from Github

sergios100, 2022-07-02:
Sorry, no offence, didn't know that and was not sure if this is all it needs, if that's the final solution.
Can this "issue" be moved to a "pull request"?
If yes, how?
Still learning!

Cheers sergios

arendst, 2022-07-02:
No worries. I'll do the PR now. Thx for reporting