arendst/Tasmota #15911: Tasmota IR gives "Protocol not supported" on supported protocol

t-aus-m, 2022-07-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After upgrading my device to Tasmota 12.0.1 the IRsend command stopped working for my AV receiver. The receiver is a Technics SA-EX140 which uses Panasonic remote codes. On Tasmota 10 it was working flawlessly, on version 12 it returns `{"IRSend":"Protocol not supported"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Nivian NVS-SMARTIR-W2
- [x] Tasmota binary firmware version number used: 12.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Serial
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Nivian NVS-SMARTIR-W2","GPIO":[0,3200,0,3232,320,1088,0,0,0,32,1056,0,0,0],"FLAG":0,"BASE":62}
{"Module":{"0":"Nivian NVS-SMARTIR-W2"}}
{"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"320":"Led_i1"},"GPIO5":{"1088":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"32":"Button1"},"GPIO14":{"1056":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
13:38:21.639 MQT: stat/till/ir/STATUS = {"Status":{"Module":0,"DeviceName":"Tills-IR","FriendlyName":["Tills-IR"],"Topic":"ir","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:38:21.646 MQT: stat/till/ir/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:08:08","StartupUTC":"2022-07-02T12:30:13","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2022-05-18T20:28:02","SaveCount":51,"SaveAddress":"FA000"}}
13:38:21.653 MQT: stat/till/ir/STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"439/699"}}
13:38:21.661 MQT: stat/till/ir/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PrettyFlyForALightbulb",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00000000","00000000"]}}
13:38:21.675 MQT: stat/till/ir/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
13:38:21.683 MQT: stat/till/ir/STATUS5 = {"StatusNET":{"Hostname":"Tills-IR","IPAddress":"10.0.10.215","Gateway":"10.0.10.0","Subnetmask":"255.255.255.0","DNSServer1":"10.0.10.0","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:16:93:4D","Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
13:38:21.689 MQT: stat/till/ir/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.100.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_16934D","MqttUser":"tills-ir","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:38:21.698 MQT: stat/till/ir/STATUS7 = {"StatusTIM":{"UTC":"2022-07-02T12:38:21","Local":"2022-07-02T13:38:21","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"}}
13:38:21.704 MQT: stat/till/ir/STATUS10 = {"StatusSNS":{"Time":"2022-07-02T13:38:21"}}
13:38:21.712 MQT: stat/till/ir/STATUS11 = {"StatusSTS":{"Time":"2022-07-02T13:38:21","Uptime":"0T00:08:08","UptimeSec":488,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"PrettyFlyForALightbulb","BSSId":"52:FF:20:7F:66:28","Channel":4,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
13:39:09.041 SRC: MQTT
13:39:09.043 CMD: Grp 0, Cmd 'IRSEND', Idx 1, Len 95, Pld -99, Data '{"Protocol":"PANASONIC","Bits":48,"Data":"0x40040500BCB9","DataLSB":"0x220A0003D9D","Repeat":0}'
13:39:09.051 MQT: stat/till/ir/RESULT = {"IRSend":"Protocol not supported"}

```

### TO REPRODUCE
1. Install Tasmota IR 12.0.2 on the device 
2. Execute IRsend with the Panasonic protocol
3. It fails with return message `{"IRSend":"Protocol not supported"}`

### EXPECTED BEHAVIOUR
Tasmota should recognise the Protocol and act accordingly

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

sfromis, 2022-07-02:
To still have the extended IR protocol support, you should upgrade to `tasmota-ir.bin.gz`, instead of the default `tasmota-bin.gz` binary only having basic IR support.

t-aus-m, 2022-07-02:
Well... That was an easy fix. Maybe the different flavours should have their respective OTA files set as default and not the vanilla flavour

closed 2022-07-02 by t-aus-m