arendst/Tasmota #15920: Smart meter on ESP32C3 does not read meter data

FransOv, 2022-07-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
On an ESP32C3 the Smart Meter driver does not read the meter data.
The ESP32C3 has only 2 UART's whereas the ESP32 has three. The Smart Meter driver xns_53_sml.ino starts with an initial index for the UART at 2 and decrements this index for each successive meter. The first meter thus is assigned UART index 2 which does not exist on an ESP32C3. I solved it temporarily by changing line 2870 in the module from

  uint32_t uart_index = 2;
to

  #ifdef CONFIG_IDF_TARGET_ESP32C3 
  uint32_t uart_index = 1;
  #else
  uint32_t uart_index = 2;
  #endif
I do not know if this problem also occurs elsewhere in the Tasmota code.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos C3 Mini https://www.wemos.cc/en/latest/c3/c3_mini.html
- [x] Tasmota binary firmware version number used: 11.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: PlatfromIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:42:00.595 MQT: stat/tasmota-esmr/RESULT = {"NAME":"ESP32C3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,1,1,1,1],"FLAG":0,"BASE":1}
15:42:00.854 MQT: stat/tasmota-esmr/RESULT = {"Module":{"1":"ESP32C3"}}
15:42:01.068 MQT: stat/tasmota-esmr/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"288":"Led1"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:44:04.470 MQT: stat/tasmota-esmr/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota-esmr","FriendlyName":["Tasmota ESMR"],"Topic":"tasmota-esmr","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":6,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:44:04.475 MQT: stat/tasmota-esmr/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T01:12:15","StartupUTC":"2022-07-03T12:31:49","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2022-07-01T18:21:24","SaveCount":47}}
15:44:04.480 MQT: stat/tasmota-esmr/STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022-07-03T14:30:20","Core":"2_0_3","SDK":"v.4.4.0","CpuFrequency":160,"Hardware":"ESP32-C3","CR":"443/699"}}
15:44:04.485 MQT: stat/tasmota-esmr/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home24Automation","OverkleeftAdvies_Nomap"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
15:44:04.495 MQT: stat/tasmota-esmr/STATUS4 = {"StatusMEM":{"ProgramSize":1432,"Free":1856,"Heap":173,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":80,"FlashMode":3,"Features":["00000809","97DAC7CF","001DA001","B7F7BFCF","05DA9BC0","E0360DCF","000840D2","20200000","0434482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,56,59,62,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,53,55,56,58,59,64,66,67,74,85,92,95,127"}}
15:44:04.501 MQT: stat/tasmota-esmr/STATUS5 = {"StatusNET":{"Hostname":"tasmota-esmr","IPAddress":"192.168.2.228","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"A0:76:4E:36:99:D4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":8.5}}
15:44:04.505 MQT: stat/tasmota-esmr/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.33","MqttPort":1883,"MqttClientMask":"tasmota-esmr","MqttClient":"tasmota-esmr","MqttUser":"hamqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:44:04.514 MQT: stat/tasmota-esmr/STATUS7 = {"StatusTIM":{"UTC":"2022-07-03T13:44:04","Local":"2022-07-03T15:44:04","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:52","Sunset":"21:56"}}
15:44:04.526 MQT: stat/tasmota-esmr/STATUS10 = {"StatusSNS":{"Time":"2022-07-03T15:44:04","ESMR5":{"tariff":1,"enrg_imp_t1":60,"enrg_imp_t2":20,"enrg_exp_t1":0,"enrg_exp_t2":0,"pwr_imp":0.433,"pwr_exp":0.000,"l1_pwr_imp":0.042,"l2_pwr_imp":0.103,"l3_pwr_imp":0.278,"l1_pwr_exp":0.000,"l2_pwr_exp":0.000,"l3_pwr_exp":0.000,"volts_l1":233.0,"volts_l2":228.0,"volts_l3":231.0},"ESP32":{"Temperature":52.7},"TempUnit":"C"}}
15:44:04.532 MQT: stat/tasmota-esmr/STATUS11 = {"StatusSTS":{"Time":"2022-07-03T15:44:04","Uptime":"0T01:12:15","UptimeSec":4335,"Heap":172,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":41},"Wifi":{"AP":1,"SSId":"Home24Automation","BSSId":"C2:74:AD:13:EF:15","Channel":11,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:01"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Not applicable

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Define Smart meter on a ESP32C3 board, no information is recorded

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Information from smart meter displayed on web interface and published by means of MQTT

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

FransOv, 2022-07-03:
The meter definition in the script:
```
>D
>B
;smlj=0
->sensor53 r
>R
smlj=0
>S
if upsecs>22
then
smlj=1
endif
>M 1

+1,20,o,0,115200,ESMR5
1,=h SageCom T211 3 phase ESMR-5
1,=h<hr/>
1,0-0:96.14.0(@1,Current tariff,,tariff,0
1,=h<hr/>
1,1-0:1.8.1(@1,Energy import T1,kWh,enrg_imp_t1,0
1,1-0:1.8.2(@1,Energy import T2,kWh,enrg_imp_t2,0
1,1-0:2.8.1(@1,Energy export T1,kWh,enrg_exp_t1,0
1,1-0:2.8.2(@1,Energy export T2,kWh,enrg_exp_t2,0
1,1-0:1.7.0(@1,Power import,kW,pwr_imp,3
1,1-0:2.7.0(@1,Power export,kW,pwr_exp,3
1,=h<hr/>
1,1-0:21.7.0(@1,L1 Power import,Kw,l1_pwr_imp,3
1,1-0:41.7.0(@1,L2 Power import,Kw,l2_pwr_imp,3
1,1-0:61.7.0(@1,L3 Power import,Kw,l3_pwr_imp,3
1,1-0:22.7.0(@1,L1 Power export,Kw,l1_pwr_exp,3
1,1-0:42.7.0(@1,L2 Power export,Kw,l2_pwr_exp,3
1,1-0:62.7.0(@1,L3 Power export,Kw,l3_pwr_exp,3
1,=h<hr/>
1,1-0:32.7.0(@1,L1 Voltage,V,volts_l1,1
1,1-0:52.7.0(@1,L2 Voltage,V,volts_l2,1
1,1-0:72.7.0(@1,L3 Voltage,V,volts_l3,1
1,=h<hr/>
#
```

arendst, 2022-07-03:
No it only occurs in the SML and Script driver as they both are managed as a seperate feature from Tasmota. All other serial interfaces using TasmotaSerial are fine.

@gemu2015 pls solve.

PR #15921 (2022-07-04): fix number of esp32 uarts
## Description:

**Related issue (if applicable):** fixes #15920

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.3
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_N [... cut]
closed 2022-07-04 by arendst