arendst/Tasmota #15924: I cannot configure MQTT in Tasmota Version 12.0.2. I usually get it in version 8.5.1

Grupo-Atomos, 2022-07-04:
### PROBLEM DESCRIPTION
I cannot configure MQTT in Tasmota Version 12.0.2. I usually get it in version 8.5.1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x ] Device used (e.g., Sonoff Basic): _____
- [ x ] Tasmota binary firmware version number used: 12.0.2
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: ESP8266
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
The wrong answer from version 12.0.2:

After recording the firmware version 12.0.2, I put the address of the MQTT server (broker.mqttdashboard.com), then the topic "afpesp/jardim".
Saved the settings and MQTT does not work.
The message I receive is:
Binding failed to broker.mqttdashboard.com:1883, rc -2. Attempt in 10 s

### EXPECTED BEHAVIOUR
After recording the firmware version 8.5.1, I put the address of the MQTT server (broker.mqttdashboard.com), then the topic "afpesp/jardim".
Saved the settings and everything works normally

The correct answer from version 8.5.1:
_A clear and concise description of what you expected to happen._
18:35:27 MQT: tele/afpesp/garden/LWT = Online (Keep)
18:35:27 MQT: cmnd/afpesp/garden/POWER =
18:35:27 MQT: tele/afpesp/jardim/INFO1 = {"Module":"Sonoff Basic","Version":"8.5.1(tasmota)","FallbackTopic":"cmnd/DVES_881AE5_fb/"," GroupTopic":"cmnd/tasmotas/"}
18:35:27 MQT: tele/afpesp/jardim/INFO2 = {"WebServerMode":"Admin","Hostname":"afpesp/jardim-6885","IPAddress":"192.168.1.81"}
18:35:27 MQT: tele/afpesp/jardim/INFO3 = {"RestartReason":"External System"}
18:35:27 MQT: stat/afpesp/jardim/RESULT = {"POWER":"OFF"}
18:35:27 MQT: stat/afpesp/garden/POWER = OFF

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
With Tasmota 8.5.1:
[image]
[image]
[image]

With Tasmota 12.0.2:

[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Note that regardless of any other actions I may take, the MQTT settings are the same.
On Tasmota 8.5.1 everything works perfectly, on 12.0.2 it doesn't.
A while ago I opened a similar question with other versions here, but with no success.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-07-04:
I just tried with 12.0.2 and it works fine.

barbudor, 2022-07-04:
@Grupo-Atomos You haven't provided the logs with v12.0.2
Please edit your post and provide all the required information

barbudor, 2022-07-04:
As expected, worked imediatly:
The only thing I set was the MQTT host and it connected seamlessly
No problems with v12.0.2
```
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 10
00:00:00.055 QPC: Count 1
00:00:00.065 Project tasmota - Tasmota Version 12.0.2(tasmota)-2_7_4_9(2022-06-20T12:41:03)
00:00:00.505 WIF: Connecting to AP1 *********Channel 1 BSSId **:**:**:**:**:** in mode 11n as tasmota-2AD017-4119...
00:00:03.758 WIF: Connected
00:00:04.011 HTP: Web server active on tasmota-2AD017-4119 with IP address 192.168.168.52
20:28:12.028 MQT: Attempting connection...
20:28:12.101 MQT: Connected
20:28:12.104 MQT: tele/tasmota_2AD017/LWT = Online (retained)
20:28:12.106 MQT: cmnd/tasmota_2AD017/POWER = 
20:28:12.113 MQT: tele/tasmota_2AD017/INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"12.0.2(tasmota)","FallbackTopic":"cmnd/DVES_2AD017_fb/","GroupTopic":"cmnd/tasmotas/"}}
20:28:12.121 MQT: tele/tasmota_2AD017/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-2AD017-4119","IPAddress":"192.168.168.52"}}
20:28:12.133 MQT: tele/tasmota_2AD017/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":5}}
20:28:12.143 MQT: stat/tasmota_2AD017/RESULT = {"POWER":"OFF"}
20:28:12.148 MQT: stat/tasmota_2AD017/POWER = OFF
20:28:13.261 MQT: stat/tasmota_2AD017/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:00:08","StartupUTC":"2022-07-04T19:28:05","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2022-07-04T20:26:20","SaveCount":10,"SaveAddress":"FA000"}}
20:28:13.361 MQT: stat/tasmota_2AD017/STATUS11 = {"StatusSTS":{"Time":"2022-07-04T20:28:13","Uptime":"0T00:00:08","UptimeSec":8,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:05"}}}
20:28:13.412 MQT: stat/tasmota_2AD017/RESULT = {"Time":"2022-07-04T20:28:13","Uptime":"0T00:00:08","UptimeSec":8,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:05"}}
20:28:13.476 QPC: Reset
20:28:15.389 CMD: status 6
20:28:15.395 MQT: stat/tasmota_2AD017/STATUS6 = {"StatusMQT":{"MqttHost":"broker.mqttdashboard.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2AD017","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:28:16.441 MQT: tele/tasmota_2AD017/STATE = {"Time":"2022-07-04T20:28:16","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"******","BSSId":"**:**:**:**:**:**","Channel":1,"Mode":"11n","RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:05"}}
```