arendst/Tasmota #15929: Tasmota 12.0.2 on Sonoff T1 2CH as a shutter roller does not work when buttons pressed

pawol, 2022-07-05:
### PROBLEM DESCRIPTION
When Tasmota 12.0.2 is installed on Sonoff T1 2CH as a shutter roller does not work when touch buttons pressed.
Rollers move up about 0,5 second only and stop. Down button does not work at all.
When downgraded onto Tasmota 11 then buttons works ok again

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
11:16:52.040 CMD: Backlog Rule1; Rule2; Rule3
11:16:52.042 SRC: WebConsole from 192.168.2.9
11:16:52.044 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
11:16:52.081 SRC: Backlog
11:16:52.084 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
11:16:52.091 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:16:52.300 SRC: Backlog
11:16:52.302 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
11:16:52.308 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:16:52.549 SRC: Backlog
11:16:52.552 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
11:16:52.558 MQT: stat/roleta_w_pokoju_lewa/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:16:57.426 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-07-05T11:16:57","Uptime":"0T00:01:59","UptimeSec":119,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":4,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:16:57.433 MQT: domoticz/in = {"idx":10,"nvalue":2,"svalue":"87","Battery":100,"RSSI":4}
11:16:57.438 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-07-05T11:16:57","Shutter1":{"Position":87,"Direction":0,"Target":87,"Tilt":0}}
11:17:00.399 WIF: Sending Gratuitous ARP
11:17:01.752 WIF: Checking connection...

```
- [ ] Provide the output of this command: `Status 0`:
```lua
11:17:41.766 WIF: Checking connection...
11:17:47.406 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-07-05T11:17:47","Uptime":"0T00:02:49","UptimeSec":169,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":4,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:17:47.410 MQT: domoticz/in = {"idx":10,"nvalue":2,"svalue":"87","Battery":100,"RSSI":5}
11:17:47.414 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-07-05T11:17:47","Shutter1":{"Position":87,"Direction":0,"Target":87,"Tilt":0}}
11:17:55.446 SRC: WebConsole from 192.168.2.9
11:17:55.449 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
11:17:55.455 MQT: stat/roleta_w_pokoju_lewa/STATUS = {"Status":{"Module":1,"DeviceName":"Roleta w pokoju lewa","FriendlyName":["Roleta w pokoju lewa 1"],"Topic":"roleta_w_pokoju_lewa","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:17:55.462 MQT: stat/roleta_w_pokoju_lewa/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:02:57","StartupUTC":"2022-07-05T09:14:58","Sleep":50,"CfgHolder":4617,"BootCount":144,"BCResetTime":"2020-05-20T10:16:11","SaveCount":6363,"SaveAddress":"F5000"}}
11:17:55.471 MQT: stat/roleta_w_pokoju_lewa/STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"491/699"}}
11:17:55.480 MQT: stat/roleta_w_pokoju_lewa/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Network 4G",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["8000E009","2805C80001000600003C5A0A000000000000","40008080","00006000","00004000","00000000"]}}
11:17:55.493 MQT: stat/roleta_w_pokoju_lewa/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
11:17:55.501 MQT: stat/roleta_w_pokoju_lewa/STATUS5 = {"StatusNET":{"Hostname":"roleta-w-pokoju-lewa-5252","IPAddress":"192.168.2.222","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:C9:34:84","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:17:55.507 MQT: stat/roleta_w_pokoju_lewa/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.3","MqttPort":1883,"MqttClientMask":"DVES_roleta_w_pokoju_lewa","MqttClient":"DVES_roleta_w_pokoju_lewa","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:17:55.515 MQT: stat/roleta_w_pokoju_lewa/STATUS7 = {"StatusTIM":{"UTC":"2022-07-05T09:17:55","Local":"2022-07-05T11:17:55","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:53","Sunset":"21:55"}}
11:17:55.524 MQT: stat/roleta_w_pokoju_lewa/STATUS10 = {"StatusSNS":{"Time":"2022-07-05T11:17:55","Shutter1":{"Position":87,"Direction":0,"Target":87,"Tilt":0}}}
11:17:55.532 MQT: stat/roleta_w_pokoju_lewa/STATUS11 = {"StatusSTS":{"Time":"2022-07-05T11:17:55","Uptime":"0T00:02:57","UptimeSec":177,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":4,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}}
11:17:55.546 MQT: stat/roleta_w_pokoju_lewa/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4025305e","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40231028","4023ddf7","4020e224","4023eace","40103468","4010489b","40104898","4023009e","401037fb","4021e8da","4022b85a","4000050c","4010364c","4024de86","4026d96d","4023ef40","40101206","4021e8da","4023a224","4022869a","40101206","40101206","4023a292","4023d434","4023d4ec","40100a7c","4024cc10","40101e95"]}}
11:17:55.553 MQT: stat/roleta_w_pokoju_lewa/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":320,"Close":310,"50perc":61,"Delay":0,"Opt":"0000","Calib":[1410,499,3319,900,1000],"Mode":"0"}}}
11:17:57.395 MQT: tele/roleta_w_pokoju_lewa/STATE = {"Time":"2022-07-05T11:17:57","Uptime":"0T00:02:59","UptimeSec":179,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Network 4G","BSSId":"40:31:3C:03:A1:35","Channel":4,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:17:57.402 MQT: domoticz/in = {"idx":10,"nvalue":2,"svalue":"87","Battery":100,"RSSI":5}
11:17:57.408 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-07-05T11:17:57","Shutter1":{"Position":87,"Direction":0,"Target":87,"Tilt":0}}
11:18:00.417 WIF: Sending Gratuitous ARP

```
[... 52 lines cut ...]
11:15:37.431 MQT: domoticz/in = {"idx":10,"nvalue":2,"svalue":"87","Battery":100,"RSSI":4}
[... cut ...]
11:15:57.428 MQT: domoticz/in = {"idx":10,"nvalue":2,"svalue":"87","Battery":100,"RSSI":4}
11:15:57.433 MQT: tele/roleta_w_pokoju_lewa/SENSOR = {"Time":"2022-07-05T11:15:57","Shutter1":{"Position":87,"Direction":0,"Target":87,"Tilt":0}}

```

### TO REPRODUCE
Just update any Tasmota onto Tasmota 12.0.2 

### EXPECTED BEHAVIOUR
Normal control over shutters by touch buttons as usually

### SCREENSHOTS
no

### ADDITIONAL CONTEXT
Sonoff touch T1 2CH configured as shutter roller controller

**(Please, remember to close the issue when the problem has been addressed)**

pawol, 2022-07-05:
Looks that on the latest dev version 12.0.2.2 an issue is fixed

ascillato2, 2022-07-07:
Hi, as this issue is the same as https://github.com/arendst/Tasmota/issues/15851, please, let's continue on that thread. Thanks.