arendst/Tasmota #15931: Linkind Dimmer WS240010108 ESP32Solo - stops controlling lights after power interruption

NickLplus, 2022-07-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Linkind Switch Dimmer WS240010108 stops controlling lights after power interruption. Flashed with 12.0.2.2 tasmota32solo1.bin, starts to work properly, after power to the dimmer being interrupted LED on the device works normally, power to the lamps doesn't - always off.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Linkind Switch Dimmer WS240010108 ESP32Solo
- [x] Tasmota binary firmware version number used: 12.0.2.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: https://tasmota.github.io/install/
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:27:56.439 MQT: stat/tasmota_701B7C/RESULT = {"NAME":"Linkind Dimmer WS240010108 ESP32Solo","GPIO":[6213,8448,0,0,640,0,0,0,0,0,288,0,0,0,0,0,0,608,0,0,0,0,544,0,0,0,0,0,33,32,0,0,0,0,0,0],"FLAG":0,"BASE":1}
09:27:56.637 MQT: stat/tasmota_701B7C/RESULT = {"Module":{"0":"Linkind Dimmer WS240010108 ESP32Solo"}}
09:27:56.905 MQT: stat/tasmota_701B7C/RESULT = {"GPIO0":{"6213":"Option A6"},"GPIO1":{"8448":"Option E1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA1"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"608":"I2C SCL1"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"544":"LedLink"},"GPIO27":{"0":"None"},"GPIO32":{"33":"Button2"},"GPIO33":{"32":"Button1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:30:01.207 MQT: stat/tasmota_701B7C/STATUS = {"Status":{"Module":0,"DeviceName":"Linkind Dimmer 01","FriendlyName":["Tasmota"],"Topic":"tasmota_701B7C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:30:01.223 MQT: stat/tasmota_701B7C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:36:37","StartupUTC":"2022-07-05T12:53:24","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2022-06-19T23:45:38","SaveCount":110}}
09:30:01.239 MQT: stat/tasmota_701B7C/STATUS2 = {"StatusFWR":{"Version":"12.0.2.2(tasmota)","BuildDateTime":"2022-07-04T08:49:04","Core":"2_0_3","SDK":"v4.4.1","CpuFrequency":80,"Hardware":"ESP32-U4WDH rev.3","CR":"447/699"}}
09:30:01.248 MQT: stat/tasmota_701B7C/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mwifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000088","00006000","00004000","00000000"]}}
09:30:01.277 MQT: stat/tasmota_701B7C/STATUS4 = {"StatusMEM":{"ProgramSize":1375,"Free":1504,"Heap":165,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9F9AC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","000840D2","20200000","5434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
09:30:01.292 MQT: stat/tasmota_701B7C/STATUS5 = {"StatusNET":{"Hostname":"tasmota-701B7C-7036","IPAddress":"192.168.1.220","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"B8:F0:09:70:1B:7C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:30:01.301 MQT: stat/tasmota_701B7C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.70","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_701B7C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:30:01.312 MQT: stat/tasmota_701B7C/STATUS7 = {"StatusTIM":{"UTC":"2022-07-05T13:30:01","Local":"2022-07-05T09:30:01","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"-04:00","Sunrise":"05:48","Sunset":"20:37"}}
09:30:01.341 MQT: stat/tasmota_701B7C/STATUS10 = {"StatusSNS":{"Time":"2022-07-05T09:30:01","ESP32":{"Temperature":60.6},"TempUnit":"C"}}
09:30:01.369 MQT: stat/tasmota_701B7C/STATUS11 = {"StatusSTS":{"Time":"2022-07-05T09:30:01","Uptime":"0T00:36:37","UptimeSec":2197,"Heap":167,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":57},"POWER":"OFF","Dimmer":80,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"nlhomedeco","BSSId":"5C:A6:E6:F3:85:06","Channel":6,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:13"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash, assemble the dimmer, set the template, check the dimmer is working, interrupt power, try to control the dimmer: dimmer LED shows normal behavior, light control is absent, light is always off, no voltage on the 'load' terminal. Checked with 4 units.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Normal light control.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Remedy: used https://github.com/pcdiem/Tasmota/releases/tag/linkind-v10dev per [digiblur](https://github.com/digiblur) instructions here: https://www.digiblur.com/p/linkind-esp32-smart-switch-how-to-flash.html

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-07-06:
If you restart the device from console with `restart 1` does it work again?

NickLplus, 2022-07-06:
Unfortunately, not to my memory, but I will re-check.
Another thing: in an attempt to start from the scratch,I flashed one of the devices using esptool (erased completely, flashed, verified after flashing), and it doesn't control power, only the LED on the device indicates that the switching is requested.

Jason2866, 2022-07-06:
Without using and activating the correct template for the device this is expected.

NickLplus, 2022-07-06:
Template was activated in every case:
{"NAME":"Linkind Dimmer WS240010108 ESP32Solo","GPIO":[6213,8448,0,0,640,0,0,0,0,0,288,0,0,0,0,0,0,608,0,0,0,0,544,0,0,0,0,0,33,32,0,0,0,0,0,0],"FLAG":0,"BASE":1}
[https://templates.blakadder.com/linkind_WS2400101.html](url)

NickLplus, 2022-07-06:
Unfortunately, `restart 1` doesn't help, still no control over lights after it:
`12:38:14.523 CMD: restart 1
12:38:14.531 MQT: stat/tasmota_702130/RESULT = {"Restart":"Restarting"}
00:00:00.002 HDW: ESP32-U4WDH 
00:00:00.040 UFS: FlashFS mounted with 304 kB free
00:00:00.114 CFG: Loaded from File, Count 384
00:00:00.123 QPC: Count 1
00:00:00.299 BRY: Berry initialized, RAM used=3731 bytes
00:00:00.335 Project tasmota - Linkind Dimmer 02 Version 12.0.2.2(tasmota)-2_0_3(2022-07-04T08:49:04)
00:00:01.042 WIF: Connecting to AP1 nlhomedeco Channel 4 BSSId XX:XX:XX:XX:XX:XX in mode 11n as tasmota-702130-0304...
00:00:05.089 WIF: Connected
00:00:05.354 HTP: Web server active on tasmota-702130-0304 with IP address 192.168.1.215
12:38:22.066 MQT: Attempting connection...
12:38:22.103 MQT: Connected
12:38:22.110 MQT: tele/tasmota_702130/LWT = Online (retained)
12:38:22.128 MQT: cmnd/tasmota_702130/POWER = 
12:38:22.152 MQT: tele/tasmota_702130/INFO1 = {"Info1":{"Module":"Linkind Dimmer WS240010108 ESP32Solo","Version":"12.0.2.2(tasmota)","FallbackTopic":"cmnd/DVES_702130_fb/","GroupTopic":"cmnd/tasmotas/"}}
12:38:22.159 MQT: tele/tasmota_702130/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-702130-0304","IPAddress":"192.168.1.215"}}
12:38:22.172 MQT: tele/tasmota_702130/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":23}}
12:38:22.179 MQT: stat/tasmota_702130/RESULT = {"POWER":"OFF"}
12:38:22.182 MQT: stat/tasmota_702130/POWER = OFF
12:38:23.902 QPC: Reset
12:38:27.140 MQT: tele/tasmota_702130/STATE = {"Time":"2022-07-06T12:38:27","Uptime":"0T00:00:11","UptimeSec":11,"Heap":174,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"mwifi","BSSId":"XX:XX:XX:XX:XX:XX","Channel":4,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:38:27.173 MQT: tele/tasmota_702130/SENSOR = {"Time":"2022-07-06T12:38:27","ESP32":{"Temperature":40.0},"TempUnit":"C"}
12:42:39.334 CMD: Backlog Template; Module; GPIO 255
12:42:39.401 MQT: stat/tasmota_702130/RESULT = {"NAME":"Linkind Dimmer WS240010108 ESP32Solo","GPIO":[6213,8448,0,0,640,0,0,0,0,0,288,0,0,0,0,0,0,608,0,0,0,0,544,0,0,0,0,0,33,32,0,0,0,0,0,0],"FLAG":0,"BASE":1}
12:42:39.637 MQT: stat/tasmota_702130/RESULT = {"Module":{"0":"Linkind Dimmer WS240010108 ESP32Solo"}}
12:42:39.899 MQT: stat/tasmota_702130/RESULT = {"GPIO0":{"6213":"Option A6"},"GPIO1":{"8448":"Option E1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA1"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"608":"I2C SCL1"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"544":"LedLink"},"GPIO27":{"0":"None"},"GPIO32":{"33":"Button2"},"GPIO33":{"32":"Button1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}`

NickLplus, 2022-07-11:
> Without using and activating the correct template for the device this is expected.

It looks like you were right!
The template here [https://templates.blakadder.com/linkind_WS2400101.html](https://templates.blakadder.com/linkind_WS2400101.html) suggests 
`GPIO21 | I2C SCL 1`
But for my dimmers it is
`GPIO22 | I2C SCL 1`
I will double-check of course.

NickLplus, 2022-07-11:
Confirmed, with template
`{"NAME":"Linkind Dimmer WS240010108 ESP32Solo","GPIO":[6213,8448,0,0,640,0,0,0,0,0,288,0,0,0,0,0,0,0,608,0,0,0,544,0,0,0,0,0,33,32,0,0,0,0,0,0],"FLAG":0,"BASE":1}`
all four devices are working properly.

closed 2022-07-11 by NickLplus