arendst/Tasmota #15949: Sunrise and Sunset are in the latest firmware reversed.

CGabo, 2022-07-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Sunrise and Sunset are in the latest firmware reversed.
-> See  Sub-menu "Timer parameters"
It is working as expected, but the time for sunrise and sunset are reversed.

Example for timezone GMT+1 with Tasmota 12.0.2
Sunrise (19:48)
Sunset (05:30)

Example for timezone GMT+1 with Tasmota 11.1.0
Sunrise (05:40)
Sunset (21:27)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-07-08:
> [X] Searched the problem in issues

Did you ?
https://github.com/arendst/Tasmota/issues/15816

Check your NTP settings. Do you get a propre time ?

ascillato, 2022-07-09:
Sunrise and Sunset requires that TIMEZONE, LATITUDE, LONGITUDE to be correctly configured and that NTP servers can be reached. If your time is not correct in the console, and your configuration is not complete, those functions won't work.

CGabo, 2022-07-09:
Indeed, it was an issue with the Firewall. NTP was not working correctly.
Thanks

closed 2022-07-09 by CGabo