arendst/Tasmota #15960: PCF8574 / PCF8574A GPIO Expander. Part Input port.

shavkatzh, 2022-07-11:
### PROBLEM DESCRIPTION
The problem when there is part of the port is configuring to INPUT ports.
All ports OUTPUTS. All outputs are stable.
All ports INPUTS. All ports are stable.
Part of ports are INPUTS, and part of the ports are OUTPUTS. OUTPUTS ports work stably. The INPUTS ports are not working.
Tested with ESP32.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [] Pre-compiled
  - [*] Self-compiled
- [*] Flashing tools used: CP2104
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
#define USE_I2C                                
#define USE_PCF8574                       
#define USE_PCF8574_SENSOR        
#define USE_PCF8574_DISPLAYINPUT
#define USE_PCF8574_MQTTINPUT    

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Setup I2C
2. Configure PCF PCF8574
3. Configure INPUTS and OUTPUTS

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The inputs is active

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-07-11:
Issue template is not completed.
Please provide logs as requested when inputs are actionned

shavkatzh, 2022-07-11:
Configured 
D0,D1,D2,D3=OUTPUTS
D4,D5,D6,D7=INPUTS
**STATUS 0:**
15:41:16.971 -> 08:41:20.749 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","","","",""],"Topic":"tasmota_4A9C58","ButtonTopic":"0","Power":395,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:41:16.971 -> 08:41:20.796 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:04:19","StartupUTC":"2022-07-11T07:37:01","Sleep":50,"CfgHolder":4617,"BootCount":48,"BCResetTime":"2022-06-21T03:42:00","SaveCount":300}}
15:41:17.019 -> 08:41:20.829 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0.2(tasmota)","BuildDateTime":"2022-05-09T14:29:53","Core":"2_0_3","SDK":"v.4.4.0","CpuFrequency":80,"Hardware":"ESP32-D0WD","CR":"402/699"}}
15:41:17.019 -> 08:41:20.853 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ChinaNet-YSKS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
15:41:17.112 -> 08:41:20.903 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1371,"Free":1856,"Heap":155,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360FC7","400840D2","20200000","0434482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,28,29,34,35,38,50,52,56,59,62,82,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
15:41:17.159 -> 08:41:20.963 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-4A9C58-7256","IPAddress":"192.168.1.9","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"8C:4B:14:4A:9C:58","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:41:17.206 -> 08:41:21.014 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4A9C58","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:41:17.206 -> 08:41:21.037 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-07-11T07:41:21","Local":"2022-07-11T08:41:21","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"04:59","Sunset":"20:52"}}
15:41:17.254 -> 08:41:21.070 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-07-11T08:41:21","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","ESP32":{"Temperature":53.3},"PCF8574-1":{"D0":0,"D1":0,"D2":1,"D3":1,"D4":0,"D5":0,"D6":0,"D7":0},"TempUnit":"C"}}
15:41:17.348 -> 08:41:21.117 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-07-11T08:41:21","Uptime":"0T00:04:20","UptimeSec":260,"Heap":157,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"ON","POWER2":"ON","POWER3":"OFF","POWER4":"ON","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"ON","POWER9":"ON","Wifi":{"AP":1,"SSId":"ChinaNet-YSKS","BSSId":"54:22:F8:52:D8:1E","Channel":6,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:05"}}}

No any logs when Input in action

shavkatzh, 2022-07-11:
Problem fixed. 
Find problems with I2C IC

closed 2022-07-11 by shavkatzh