arendst/Tasmota #15977: Unreliable and slow on Tasmota v12.0.2

fabiosoft, 2022-07-14:
### PROBLEM DESCRIPTION
I just flashed a new device (XY-WFUSB) to tasmota v12.0.2 and it's extremely slow and unreliable on local network and throw home assistant, also pushing the button on it updates (and toggles the relay) very slow (tipically after 1 minute).

Anyway i downgraded to v11 and everything back to work as usual.

### REQUESTED INFORMATION

If you have the same issue please contribute to better investigate

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): XY-WFUSB
- [ ] Tasmota binary firmware version number used: v12.0.2 
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool and nodemcu pyflasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
just install the latest version

### EXPECTED BEHAVIOUR
quick response

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-07-14:
Issue template? Logs?
A slow tasmota is often the sign of problems with NTP

fabiosoft, 2022-07-14:
I always use the same configuration:

```{"NAME":"XY-WFUSB","GPIO":[1,1,0,1,32,224,0,0,0,0,320,0,544,0],"FLAG":0,"BASE":18}```

Logs are ok, just ON/OFF switch is slow

s-hadinger, 2022-07-14:
Try `SetOption127 1`

sfromis, 2022-07-14:
And do make sure that you have unimpeded real time access, typically to a NTP server