arendst/Tasmota #15983: No WIFI AP on Sonoff DUAL R3.

xoniq, 2022-07-15:
### PROBLEM DESCRIPTION
No WIFI AP on Sonoff DUAL R3.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff DUAL R3
- [X] Tasmota binary firmware version number used: 12.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [ ] Provide the output of this command: `Status 0`:

### TO REPRODUCE
**First attempt**

- `esptool.py --chip esp32 --port /dev/ttyAMA0 erase_flash`
- `esptool.py --chip esp32 --port /dev/ttyAMA0 write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x10000 tasmota32.bin`

**Second attempt**

- `esptool.py --chip esp32 --port /dev/ttyAMA0 erase_flash`
- `esptool.py --chip esp32 --port /dev/ttyAMA0 write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x00000 tasmota32.factory.bin`
- `esptool.py --chip esp32 --port /dev/ttyAMA0 write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x10000 tasmota32.bin`

### EXPECTED BEHAVIOUR
Show a WiFi AP after turning on (after flashing)

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
I have a Sonoff DUAL R3 which I used with the official app to control my sun screen without issues.
Yesterday I wanted to flash it (using a spare Pi 1 which worked fine for this). I tried these directions (this is my first Tasmota flash)

Name on the board:
**DUALR3-POW V2.0 20210427**

Identified chip according to esptool:
**ESP32-DOWD-V3 (revision 3)**

Both attemts succeeded, but no WIFI AP to connect to, not when leaving it to 3v3 and not on mains power.
Is there something I did wrong? I looked at any manual. Maybe something different with my chip as its identified via esptool?

**(Please, remember to close the issue when the problem has been addressed)**

xoniq, 2022-07-15:
Thanks!

So after erase, I only do the `tasmota32.factory.bin` and then I _should_ work? I'll try that tonight. And then I'll try both release and development version in case on doesn't work.

Also; I was confused because of the `ESP32-DOWD-V3 (revision 3)` identification, if I then perhaps need ESP32, ESP32-S2, ESP32-S3 or ESP32-C3. So I just should the regular one, right? (Just to confirm)

xoniq, 2022-07-15:
Thanks again. I'll try it out and close the task if it succeeds.

Also; I had one more follow up question, maybe you know about it. I have the choice of using tasmota32 en tasmota32-bluetooth. What's the difference?

I know it has to do with the build-in Bluetooth chip, but what can be done with that when I later on use the bluetooth version of the firmware?

sfromis, 2022-07-15:
Tasmota has some degree of support for gathering data from certain Bluetooth sensors

barbudor, 2022-07-15:
The Bluetooth feature allows to communicate with Bluetooth sensors or some thermostat valves
There's a doc page that will give you more hints: https://tasmota.github.io/docs/Bluetooth_ESP32/

However I would not recommend to use Bluetooth on your sonoff for 2 reasons
- it's CPU demanding some times and may impact usage of the main use case of the sonoff
- to get best efficiency you may want to put the ESP32 bluetooth tasmota is almost plain sight to have better coverage of bluetooth sensors around the hous while your sonoff may be hidden in the wall

So do not hesitate to dedicate a ESP32 dev board for bluetooth operations

sfromis, 2022-07-15:
Another use case for Bluetooth is to detect beacons coming in or out of range

xoniq, 2022-07-15:
Ah thanks guys, I think I don't really need it. Also because I'm planning on connecting the shutter to home assistant and to homekit, where I already do some automations. I do have plenty of other ESP boards which I can use for that purpose in the future, I'll just let this Sonoff one just handle the sun screen, for the sake of simplicity.