arendst/Tasmota #15996: Web UI saves wrong MQTT authentication data

jpicht, 2022-07-16:
### PROBLEM DESCRIPTION
Saving empty mqtt username and password in the web UI result in the defaults being saved and used.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues) There are older issues regarding mqtt anonymous authentication, but not about the web ui
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic RF R2
- [x] Tasmota binary firmware version number used: 12.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:34:40.248 CMD: Backlog Template; Module; GPIO 255
09:34:40.284 MQT: stat/tasmota_A39EA4/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:34:40.499 MQT: stat/tasmota_A39EA4/RESULT = {"Module":{"1":"Sonoff Basic"}}
09:34:40.754 MQT: stat/tasmota_A39EA4/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:35:26.230 CMD: Status 0
09:35:26.241 MQT: stat/tasmota_A39EA4/STATUS = {"Status":{"Module":1,"DeviceName":"test","FriendlyName":["test"],"Topic":"tasmota_A39EA4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:35:26.251 MQT: stat/tasmota_A39EA4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:38","StartupUTC":"2022-07-16T08:22:48","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2022-07-16T09:07:51","SaveCount":26,"SaveAddress":"FA000"}}
09:35:26.262 MQT: stat/tasmota_A39EA4/STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"337/699"}}
09:35:26.269 MQT: stat/tasmota_A39EA4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"nas.private.lan","LogPort":1514,"SSId":["hod",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000","00000000"]}}
09:35:26.282 MQT: stat/tasmota_A39EA4/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
09:35:26.291 MQT: stat/tasmota_A39EA4/STATUS5 = {"StatusNET":{"Hostname":"tasmota-A39EA4-7844","IPAddress":"172.20.0.114","Gateway":"172.20.0.1","Subnetmask":"255.255.255.0","DNSServer1":"172.20.0.1","DNSServer2":"0.0.0.0","Mac":"7C:87:CE:A3:9E:A4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:35:26.298 MQT: stat/tasmota_A39EA4/STATUS6 = {"StatusMQT":{"MqttHost":"172.20.0.129","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A39EA4","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:35:26.306 MQT: stat/tasmota_A39EA4/STATUS7 = {"StatusTIM":{"UTC":"2022-07-16T08:35:26","Local":"2022-07-16T09:35:26","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:04","Sunset":"20:48"}}
09:35:26.314 MQT: stat/tasmota_A39EA4/STATUS10 = {"StatusSNS":{"Time":"2022-07-16T09:35:26"}}
09:35:26.321 MQT: stat/tasmota_A39EA4/STATUS11 = {"StatusSTS":{"Time":"2022-07-16T09:35:26","Uptime":"0T00:12:38","UptimeSec":758,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"hod","BSSId":"08:55:31:E5:63:97","Channel":8,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
tasmota-XXXXXX-XXXX ESP-HTP: Configure MQTT
tasmota-XXXXXX-XXXX ESP-HTP: Configure MQTT
tasmota-XXXXXX-XXXX ESP-SRC: WebGui from xxx.xxx.xxx.9
tasmota-XXXXXX-XXXX ESP-CMD: Grp 0, Cmd 'BACKLOG', Idx 0, Len 131, Pld -99, Data ';MqttHost xxx.xxx.xxx.129;MqttPort 1883;MqttClient DVES_%06X;MqttUser 1;MqttPassword2 ";Topic tasmota_%06X;FullTopic %prefix%/%topic%/'
tasmota-XXXXXX-XXXX ESP-HTP: Restart
tasmota-XXXXXX-XXXX ESP-SRC: WebGui from 0.0.0.0
tasmota-XXXXXX-XXXX ESP-SRC: Backlog
tasmota-XXXXXX-XXXX ESP-CMD: Grp 0, Cmd 'MQTTHOST', Idx 1, Len 12, Pld 172, Data 'xxx.xxx.xxx.129'
tasmota-XXXXXX-XXXX ESP-CFG: CR 346/699, Busy 0
tasmota-XXXXXX-XXXX ESP-MQT: stat/tasmota_A39EA4/RESULT = {"MqttHost":"xxx.xxx.xxx.129"}
tasmota-XXXXXX-XXXX ESP-SRC: Backlog
tasmota-XXXXXX-XXXX ESP-CMD: Grp 0, Cmd 'MQTTPORT', Idx 1, Len 4, Pld 1883, Data '1883'
tasmota-XXXXXX-XXXX ESP-MQT: stat/tasmota_A39EA4/RESULT = {"MqttPort":1883}
tasmota-XXXXXX-XXXX ESP-SRC: Backlog
tasmota-XXXXXX-XXXX ESP-CMD: Grp 0, Cmd 'MQTTCLIENT', Idx 1, Len 9, Pld -99, Data 'DVES_%06X'
tasmota-XXXXXX-XXXX ESP-CFG: CR 346/699, Busy 0
tasmota-XXXXXX-XXXX ESP-MQT: stat/tasmota_A39EA4/RESULT = {"MqttClient":"DVES_%06X"}
tasmota-XXXXXX-XXXX ESP-SRC: Backlog
tasmota-XXXXXX-XXXX ESP-CMD: Grp 0, Cmd 'MQTTUSER', Idx 1, Len 1, Pld 1, Data '1'
tasmota-XXXXXX-XXXX ESP-CFG: CR 346/699, Busy 0
tasmota-XXXXXX-XXXX ESP-MQT: stat/tasmota_A39EA4/RESULT = {"MqttUser":"DVES_USER"}
tasmota-XXXXXX-XXXX ESP-SRC: Backlog
[... 34 lines cut ...]
tasmota-XXXXXX-XXXX ESP-MQT: tele/tasmota_A39EA4/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-XXXXXX-XXXX","IPAddress":"xxx.xxx.xxx.114"}}
tasmota-XXXXXX-XXXX ESP-MQT: tele/tasmota_A39EA4/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":13}}
tasmota-XXXXXX-XXXX ESP-MQT: stat/tasmota_A39EA4/RESULT = {"POWER":"OFF"}
tasmota-XXXXXX-XXXX ESP-MQT: stat/tasmota_A39EA4/POWER = OFF
tasmota-XXXXXX-XXXX ESP-QPC: Reset
tasmota-XXXXXX-XXXX ESP-MQT: tele/tasmota_A39EA4/STATE = {"Time":"2022-07-16T09:40:07","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"hod","BSSId":"08:55:31:E5:63:97","Channel":8,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
tasmota-XXXXXX-XXXX ESP-HTP: Main Menu
tasmota-XXXXXX-XXXX ESP-APP: Boot Count 13

```
Obtained via syslog, because the restart made it very hard to get the weblog.

### TO REPRODUCE
Clear "User" and "Password" fields in web UI and press save

### EXPECTED BEHAVIOUR
Anonymous authentication to mqtt broker.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2022-07-16:
Hi,

Sorry, blank user and/or blank password are not supported by design.

You should configure an user and password in your mqtt broker in order to use Tasmota.

closed 2022-07-16 by ascillato2

sfromis, 2022-07-16:
You do not need to have any user/password in your broker, my Tasmota + Mosquitto setup runs just fine allowing anonymous connections.

ascillato, 2022-07-17:
> You do not need to have any user/password in your broker, my Tasmota + Mosquitto setup runs just fine allowing anonymous connections.

Yes, that is because by default Tasmota has user/pass as blank.

@jpicht In order to put the default blank user/pass, please, go to the console and type:
```JS
backlog MqttPassword 0;MqttUser 0
```

Clearing the user/pass by the webUI is not supported.

sfromis, 2022-07-17:
While my standard Tasmota config sequence clears MQTT user/password, the default is not blank, but `DVES_USER/DVES_PASS`. This still works with my Mosquitto broker allowing anonymous connections, and not having any `DVES_USER` configured.
`New client connected from 192.168.xxx.xxx:58282 as xxxxxxxx (p2, c1, k30, u'DVES_USER')`

joba-1, 2022-07-17:
Ha! For me this is the longest standing tasmota mystery. What is DVES?

Have configured my mqtt broker to allow anonymous access, which just works. So I never looked it up

arendst, 2022-07-17:
DVES is short for device esp