arendst/Tasmota #15997: Hostname not working with static IP

Sineos, 2022-07-16:
### PROBLEM DESCRIPTION
When assigning a static IP, the hostname is not registered in the routers DNS server. Using DHCP it works.
Other devices, e.g. Raspberry Pi and other SBCs work in the combination of static IP and hostname

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): All affected (~15 different devices)
- [x] Tasmota binary firmware version number used: Currently running v10.x up to v12.x
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
12:33:14.316 CMD: Backlog Template; Module; GPIO 255
12:33:14.378 MQT: stat/TM_JonaBlind/RESULT = {"NAME":"SC500W","GPIO":[0,0,0,576,160,161,0,0,224,32,225,0,0,0],"FLAG":0,"BASE":18}
12:33:14.586 MQT: stat/TM_JonaBlind/RESULT = {"Module":{"0":"SC500W"}}
12:33:14.841 MQT: stat/TM_JonaBlind/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"576":"LedLink_i"},"GPIO4":{"160":"Switch1"},"GPIO5":{"161":"Switch2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"32":"Button1"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
12:34:23.427 CMD: Status 0
12:34:23.435 MQT: stat/TM_JonaBlind/STATUS = {"Status":{"Module":0,"DeviceName":"TM-JonaBlind","FriendlyName":["TM-JonaBlind",""],"Topic":"TM_JonaBlind","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:34:23.441 MQT: stat/TM_JonaBlind/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T14:04:10","StartupUTC":"2022-07-15T20:30:13","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2022-07-06T19:37:20","SaveCount":80,"SaveAddress":"F4000"}}
12:34:23.447 MQT: stat/TM_JonaBlind/STATUS2 = {"StatusFWR":{"Version":"12.0.2.3(tasmota)","BuildDateTime":"2022-07-06T15:03:32","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"370/699"}}
12:34:23.452 MQT: stat/TM_JonaBlind/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SweetHomeSpot24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A190000000000","400000C0","00006000","00004000","00000000"]}}
12:34:23.465 MQT: stat/TM_JonaBlind/STATUS4 = {"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
12:34:23.473 MQT: stat/TM_JonaBlind/STATUS5 = {"StatusNET":{"Hostname":"TM-JonaBlind-2989","IPAddress":"192.168.178.39","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"192.168.178.1","Mac":"7C:87:CE:8E:EB:AD","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:34:23.478 MQT: stat/TM_JonaBlind/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.8","MqttPort":1883,"MqttClientMask":"TM_JonaBlind","MqttClient":"TM_JonaBlind","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:34:23.484 MQT: stat/TM_JonaBlind/STATUS7 = {"StatusTIM":{"UTC":"2022-07-16T10:34:23","Local":"2022-07-16T12:34:23","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"05:35","Sunset":"21:14"}}
12:34:23.491 MQT: stat/TM_JonaBlind/STATUS10 = {"StatusSNS":{"Time":"2022-07-16T12:34:23","Switch1":"OFF","Switch2":"OFF","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
12:34:23.497 MQT: stat/TM_JonaBlind/STATUS11 = {"StatusSTS":{"Time":"2022-07-16T12:34:23","Uptime":"0T14:04:10","UptimeSec":50650,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SweetHomeSpot24","BSSId":"BA:FB:E4:F4:13:A4","Channel":11,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:07"}}}
12:34:23.501 MQT: stat/TM_JonaBlind/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":190,"Close":183,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1"}}}
12:35:26.465 MQT: tele/TM_JonaBlind/STATE = {"Time":"2022-07-16T12:35:26","Uptime":"0T14:05:13","UptimeSec":50713,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Hostname working with static IP

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

`SetOption55` makes no difference.

Query DNS for Tasmota device
```lua
root@orangepi3-lts ~ # dig -x 192.168.178.39

; <<>> DiG 9.11.5-P4-5.1+deb10u7-Debian <<>> -x 192.168.178.39
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NXDOMAIN, id: 8586
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 1, ADDITIONAL: 0

;; QUESTION SECTION:
;39.178.168.192.in-addr.arpa.   IN      PTR

;; AUTHORITY SECTION:
178.168.192.in-addr.arpa. 9     IN      SOA     fritz.box. admin.fritz.box. 1657967996 21600 1800 43200 10

;; Query time: 5 msec
;; SERVER: 192.168.178.1#53(192.168.178.1)
;; WHEN: Sat Jul 16 12:39:56 CEST 2022
;; MSG SIZE  rcvd: 105
```

Query DNS for a RPi with static IP 
```lua
root@orangepi3-lts ~ # dig -x 192.168.178.20

; <<>> DiG 9.11.5-P4-5.1+deb10u7-Debian <<>> -x 192.168.178.20
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 42722
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 1, ADDITIONAL: 3

;; QUESTION SECTION:
;20.178.168.192.in-addr.arpa.   IN      PTR

;; ANSWER SECTION:
20.178.168.192.in-addr.arpa. 9  IN      PTR     piheizung.fritz.box.

;; AUTHORITY SECTION:
20.178.168.192.in-addr.arpa. 9  IN      NS      fritz.box.

;; ADDITIONAL SECTION:
fritz.box.              9       IN      A       192.168.178.1
fritz.box.              9       IN      AAAA    fd00::3e37:12ff:feae:b3e9
fritz.box.              9       IN      AAAA    2a02:810d:4c3f:ce14:3e37:12ff:feae:b3e9

;; Query time: 9 msec
;; SERVER: 192.168.178.1#53(192.168.178.1)
;; WHEN: Sat Jul 16 12:44:58 CEST 2022
;; MSG SIZE  rcvd: 164
```

Any pointer is appreciated. Thanks in advance

sfromis, 2022-07-16:
My expectation would be that Tasmota is doing nothing extra for host name DNS resolution, in my network this is handled by my router running dnsmasq for DHCP+DNS, which indeed does not kick in if I use a static IP instead of best practice.

ascillato, 2022-07-16:
Hi, 

The behavior you see is the expected one. When using static IPs, the DHCP information is not being exchanged. In fact, the router don't know what devices are using static IP. It only knows the DHCP ones. When it receives a request to an IP that is not in its table, it will reroute it and that is why it works. For that it uses the MAC to IP table instead. That table don't have hostnames. So, when setting any device to static IP, the hostname is not populated anymore.

The case of your raspberry is a different one. When you set an static IP in the Raspberry OS without disabling the DHCP service, you will have both (static and dynamic IPs). Your raspberry will answer to the static IP and to the DHCP assigned IP. Depending on how you have configured it, the IP can be the same for both. So, seems that your raspberry is populating it's hostname due to DHCP.

So, in order to use Hostnames AND fixed IPs, please set in Tasmota to use DHCP and in your router set a reserved IP for your device.

closed 2022-07-16 by ascillato2

Sineos, 2022-07-17:
@ascillato 
Mayn thanks for your reply and of course you are right.
In Linux and according to my experience this is solved via Multicast DNS and not via DHCP, e.g. /etc/systemd/resolved.conf:

```
[Resolve]
MulticastDNS=yes
```

I was expecting that `SetOption55` does the same for Tasmota but apparently it is not working.

sfromis, 2022-07-17:
mDNS is not enabled in default Tasmota builds, as it was found to be too much trouble on ESP8266. Thus, flipping the `SetOption55` bit does nothing unless you use a custom build to enable the deprecated feature.