arendst/Tasmota #16001: Two SONOFF MS01 sensors don't work in Tasmota32

bob-tm, 2022-07-16:
### PROBLEM DESCRIPTION
Two SONOFF MS01 sensors don't work in Tasmota32 12.0.2.3
1. I connect two sensors to ESP32 board to GPIO 2 and GPIO 4. MS01_4 = valid result. MS01_2  = null. 
2. I reconnect sensor from GPIO2 to GPIO12. In this configuration MS01_4 = null; MS01_12 = valid result. 
3. I reconnect sensor from GPIO4 to GPIO13. Now MS01_12=null; MS01_13 = valid result.
So MS01 sensor with lower GPIO always display null. 

I swap sensors each other, check all wires. Always sensor with lower GPIO = null.
Stable Tasmota32 12.0.2 do not detect sensors at all. 

Everything works fine with ESP8266 running 12.0.2.2, I can not find Tasmota32 12.0.2.2 to check it. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): RoboDyn ESP32
- [x] Tasmota binary firmware version number used: 12.0.2.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:19:02.849 MQT: stat/tasmota_142214/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
15:19:03.049 MQT: stat/tasmota_142214/RESULT = {"Module":{"1":"ESP32-DevKit"}}
15:19:03.313 MQT: stat/tasmota_142214/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"8768":"MS01"},"GPIO13":{"8768":"MS01"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"224":"Relay1"},"GPIO26":{"225":"Relay2"},"GPIO27":{"0":"None"},"GPIO32":{"227":"Relay4"},"GPIO33":{"226":"Relay3"},"GPIO34":{"32":"Button1"},"GPIO35":{"33":"Button2"},"GPIO36":{"34":"Button3"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"35":"Button4"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:20:12.800 CMD: Status 0
15:20:12.812 MQT: stat/tasmota_142214/STATUS = {"Status":{"Module":1,"DeviceName":"fitowall_water_controller","FriendlyName":["fitowall_irrigation_pump","fitowall_valve_add_water","fitowall_pump_out","fitowall_valve_remove_water"],"Topic":"tasmota_142214","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:20:12.837 MQT: stat/tasmota_142214/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T01:16:09","StartupUTC":"2022-07-16T13:04:03","Sleep":50,"CfgHolder":4617,"BootCount":54,"BCResetTime":"2022-04-12T14:27:52","SaveCount":162}}
15:20:12.867 MQT: stat/tasmota_142214/STATUS2 = {"StatusFWR":{"Version":"12.0.2.3(tasmota)","BuildDateTime":"2022-07-16T10:51:18","Core":"2_0_4","SDK":"v4.4.2","CpuFrequency":80,"Hardware":"ESP32-D0WD rev.1","CR":"528/699"}}
15:20:12.895 MQT: stat/tasmota_142214/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["/dev/null",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000","00000000"]}}
15:20:12.939 MQT: stat/tasmota_142214/STATUS4 = {"StatusMEM":{"ProgramSize":1404,"Free":1856,"Heap":150,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9F9AC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
15:20:12.979 MQT: stat/tasmota_142214/STATUS5 = {"StatusNET":{"Hostname":"smarthome-fitowall-water-contro","IPAddress":"192.168.101.33","Gateway":"192.168.101.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.101.1","DNSServer2":"0.0.0.0","Mac":"10:97:BD:14:22:14","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:20:13.016 MQT: stat/tasmota_142214/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_142214","MqttUser":"bob","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:20:13.047 MQT: stat/tasmota_142214/STATUS7 = {"StatusTIM":{"UTC":"2022-07-16T14:20:13","Local":"2022-07-16T15:20:13","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:04","Sunset":"20:48"}}
15:20:13.083 MQT: stat/tasmota_142214/STATUS10 = {"StatusSNS":{"Time":"2022-07-16T15:20:13","MS01-12":{"Humidity":null,"Raw":0},"MS01-13":{"Humidity":18.7,"Raw":17258},"ESP32":{"Temperature":62.8},"TempUnit":"C"}}
15:20:13.115 MQT: stat/tasmota_142214/STATUS11 = {"StatusSTS":{"Time":"2022-07-16T15:20:13","Uptime":"0T01:16:10","UptimeSec":4570,"Heap":148,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

15:21:11.680 DHT: Timeout waiting for start signal high pulse
15:21:11.689 DHT: Read 430A0FA07C
15:21:11.691 DHT: Checksum failure 43 0A 0F A0 7C =? FC
15:21:12.003 HTP: Consoles
15:21:13.888 HTP: Console
15:21:15.662 DHT: Timeout waiting for start signal high pulse
15:21:15.671 DHT: Read 430A0F207C
15:21:19.705 DHT: Timeout waiting for start signal high pulse
15:21:19.716 DHT: Read 432A0F205C
15:21:19.718 DHT: Checksum failure 43 2A 0F 20 5C =? 9C
15:21:23.706 DHT: Timeout waiting for start signal high pulse
15:21:23.715 DHT: Read 432A07A01C
15:21:23.717 DHT: Checksum failure 43 2A 07 A0 1C =? 14
15:21:27.694 DHT: Timeout waiting for start signal high pulse
15:21:27.703 DHT: Read 436A0FA05C
15:21:30.684 WIF: Checking connection...
15:21:31.691 DHT: Timeout waiting for start signal high pulse
15:21:31.700 DHT: Read 430A0FA07C
15:21:31.702 DHT: Checksum failure 43 0A 0F A0 7C =? FC

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect two sensors to ESP32 with latest firmware

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-07-16:
> Stable Tasmota32 12.0.2 do not detect sensors at all.

Sonoff sensors (SI7021, THS01 and MS01) on ESP32 are supported starting with version v12.0.2.3. See https://github.com/arendst/Tasmota/blob/development/CHANGELOG.md

For debugging purposes pls connect only one MS01 and provide ``weblog 4`` for over a 1 minute to check if any DHT related messages pop up.

bob-tm, 2022-07-16:
I don't have physical access to device right now, so I change GPIO in settings and virtually disconnected it.

Log1. 
GPIO12 - connected sensor. 
GPIO13 - connected sensor, but set to NONE in settings.
GPIO14 - Second MS01, physically not connected. 

```
16:57:45.605 TFS: File 'autoexec.bat' not found
16:57:46.690 SRC: MQTT
16:57:46.692 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1'
16:57:46.701 MQT: stat/tasmota_142214/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:06","StartupUTC":"2022-07-16T15:57:40","Sleep":50,"CfgHolder":4617,"BootCount":54,"BCResetTime":"2022-04-12T14:27:52","SaveCount":169}}
16:57:46.744 SRC: MQTT
16:57:46.746 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11'
16:57:46.752 BRY: GC from 6009 to 3594 bytes, objects freed 29/40 (in 1 ms) - slots from 84/122 to 40/122
16:57:46.765 MQT: stat/tasmota_142214/STATUS11 = {"StatusSTS":{"Time":"2022-07-16T16:57:46","Uptime":"0T00:00:06","UptimeSec":6,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":53,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}}
16:57:46.802 SRC: MQTT
16:57:46.805 CMD: Grp 0, Cmd 'STATE', Idx 1, Len 0, Pld -99, Data ''
16:57:46.809 BRY: GC from 4054 to 3594 bytes, objects freed 1/40 (in 1 ms) - slots from 41/122 to 40/122
16:57:46.817 MQT: stat/tasmota_142214/RESULT = {"Time":"2022-07-16T16:57:46","Uptime":"0T00:00:06","UptimeSec":6,"Heap":152,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":57,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:57:46.859 SRC: MQTT
16:57:46.862 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 10, Data '10'
16:57:46.877 MQT: stat/tasmota_142214/STATUS10 = {"StatusSNS":{"Time":"2022-07-16T16:57:46","MS01-12":{"Humidity":null,"Raw":0},"MS01-14":{"Humidity":null,"Raw":0},"ESP32":{"Temperature":61.7},"TempUnit":"C"}}
16:57:47.109 HTP: Consoles
16:57:47.847 QPC: Reset
16:57:47.904 DHT: Timeout waiting for start signal high pulse
16:57:47.957 DHT: Timeout waiting for start signal high pulse
16:57:48.420 HTP: Console
16:57:49.881 APP: Boot Count 55
16:57:49.886 BRY: GC from 4322 to 3594 bytes, objects freed 3/40 (in 1 ms) - slots from 43/122 to 40/122
16:57:49.899 MQT: tele/tasmota_142214/STATE = {"Time":"2022-07-16T16:57:49","Uptime":"0T00:00:09","UptimeSec":9,"Heap":155,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":50,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:57:49.917 MQT: tele/tasmota_142214/SENSOR = {"Time":"2022-07-16T16:57:49","MS01-12":{"Humidity":null,"Raw":0},"MS01-14":{"Humidity":null,"Raw":0},"ESP32":{"Temperature":61.1},"TempUnit":"C"}
16:57:50.245 CFG: Saved, Count 170, Bytes 4096
16:57:51.897 DHT: Timeout waiting for start signal high pulse
16:57:51.952 DHT: Timeout waiting for start signal high pulse
16:57:54.904 MQT: tasmota/discovery/1097BD142214/config = {"ip":"192.168.101.33","dn":"fitowall_water_controller","fn":["fitowall_irrigation_pump","fitowall_valve_add_water","fitowall_pump_out","fitowall_valve_remove_water",null,null,null,null],"hn":"smarthome-fitowall-water-contro","mac":"1097BD142214","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.0.2.3","t":"tasmota_142214","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":1,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
16:57:54.937 MQT: tasmota/discovery/1097BD142214/sensors = {"sn":{"Time":"2022-07-16T16:57:54","MS01-12":{"Humidity":null,"Raw":0},"MS01-14":{"Humidity":null,"Raw":0},"ESP32":{"Temperature":61.1},"TempUnit":"C"},"ver":1} (retained)
16:57:55.914 DHT: Timeout waiting for start signal high pulse
16:57:55.969 DHT: Timeout waiting for start signal high pulse
16:57:56.287 SRC: MQTT
16:57:56.292 CMD: Grp 0, Cmd 'STATE', Idx 1, Len 0, Pld -99, Data ''
16:57:56.298 BRY: GC from 4410 to 3594 bytes, objects freed 4/40 (in 0 ms) - slots from 41/122 to 40/122
16:57:56.311 MQT: stat/tasmota_142214/RESULT = {"Time":"2022-07-16T16:57:56","Uptime":"0T00:00:16","UptimeSec":16,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:57:56.360 SRC: MQTT
[... cut ...]
17:00:19.888 DHT: Timeout waiting for start signal high pulse
17:00:19.942 DHT: Timeout waiting for start signal high pulse
17:00:23.912 DHT: Timeout waiting for start signal high pulse
17:00:23.966 DHT: Timeout waiting for start signal high pulse
17:00:24.933 WIF: Checking connection...
17:00:27.896 DHT: Timeout waiting for start signal high pulse
17:00:27.950 DHT: Timeout waiting for start signal high pulse
17:00:31.919 DHT: Timeout waiting for start signal high pulse
17:00:31.973 DHT: Timeout waiting for start signal high pulse
```

bob-tm, 2022-07-16:
Log2.
GPIO2 - first MS01, physically not connected.
GPIO12 - connected sensor, but set to NONE in settings.
GPIO13 - Second MS01, connected.

```
17:06:19.826 MQT: tele/tasmota_142214/STATE = {"Time":"2022-07-16T17:06:19","Uptime":"0T00:00:10","UptimeSec":10,"Heap":156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":45,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:06:19.848 MQT: tele/tasmota_142214/SENSOR = {"Time":"2022-07-16T17:06:19","MS01-02":{"Humidity":null,"Raw":0},"MS01-13":{"Humidity":19.3,"Raw":16387},"ESP32":{"Temperature":62.2},"TempUnit":"C"}
17:06:20.514 CFG: Saved, Count 172, Bytes 4096
17:06:21.881 DHT: Timeout waiting for start signal high pulse
17:06:21.891 DHT: Read 40230FA012
17:06:24.860 MQT: tasmota/discovery/1097BD142214/config = {"ip":"192.168.101.33","dn":"fitowall_water_controller","fn":["fitowall_irrigation_pump","fitowall_valve_add_water","fitowall_pump_out","fitowall_valve_remove_water",null,null,null,null],"hn":"smarthome-fitowall-water-contro","mac":"1097BD142214","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.0.2.3","t":"tasmota_142214","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":1,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
17:06:24.899 MQT: tasmota/discovery/1097BD142214/sensors = {"sn":{"Time":"2022-07-16T17:06:24","MS01-02":{"Humidity":null,"Raw":0},"MS01-13":{"Humidity":19.3,"Raw":16419},"ESP32":{"Temperature":62.2},"TempUnit":"C"},"ver":1} (retained)
17:06:25.883 DHT: Timeout waiting for start signal high pulse
17:06:25.893 DHT: Read 40230F2012
17:06:25.896 DHT: Checksum failure 40 23 0F 20 12 =? 92
17:06:26.015 SRC: MQTT
17:06:26.018 CMD: Grp 0, Cmd 'STATE', Idx 1, Len 0, Pld -99, Data ''
17:06:26.026 BRY: GC from 4423 to 3594 bytes, objects freed 4/40 (in 1 ms) - slots from 41/122 to 40/122
17:06:26.040 MQT: stat/tasmota_142214/RESULT = {"Time":"2022-07-16T17:06:26","Uptime":"0T00:00:17","UptimeSec":17,"Heap":155,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:06:26.096 SRC: MQTT
17:06:26.099 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 10, Data '10'
17:06:26.115 MQT: stat/tasmota_142214/STATUS10 = {"StatusSNS":{"Time":"2022-07-16T17:06:26","MS01-02":{"Humidity":null,"Raw":0},"MS01-13":{"Humidity":19.3,"Raw":16419},"ESP32":{"Temperature":62.2},"TempUnit":"C"}}
17:06:26.152 SRC: MQTT
17:06:26.156 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11'
17:06:26.164 BRY: GC from 4357 to 3594 bytes, objects freed 4/40 (in 1 ms) - slots from 45/122 to 40/122
17:06:26.180 MQT: stat/tasmota_142214/STATUS11 = {"StatusSTS":{"Time":"2022-07-16T17:06:26","Uptime":"0T00:00:17","UptimeSec":17,"Heap":153,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":52,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}}
17:06:26.229 SRC: MQTT
17:06:26.233 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1'
17:06:26.246 MQT: stat/tasmota_142214/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:17","StartupUTC":"2022-07-16T16:06:09","Sleep":50,"CfgHolder":4617,"BootCount":56,"BCResetTime":"2022-04-12T14:27:52","SaveCount":172}}
17:06:29.857 DHT: Timeout waiting for start signal high pulse
17:06:29.867 DHT: Read 40430FA032
17:06:33.875 DHT: Timeout waiting for start signal high pulse
17:06:33.885 DHT: Read 40430F2032
17:06:33.887 DHT: Checksum failure 40 43 0F 20 32 =? B2
17:06:35.054 WIF: Checking connection...
17:06:37.882 DHT: Timeout waiting for start signal high pulse
17:06:37.891 DHT: Read 40430FA032
17:06:41.882 DHT: Timeout waiting for start signal high pulse
17:06:41.936 DHT: Timeout waiting for start signal high pulse
17:06:44.509 HTP: Configuration
17:06:45.887 DHT: Timeout waiting for start signal high pulse
17:06:45.898 DHT: Read 40030F2072
17:06:45.961 HTP: Configure Module
17:06:49.832 DHT: Timeout waiting for start signal high pulse
17:06:49.841 DHT: Read 40430F2032
17:06:49.844 DHT: Checksum failure 40 43 0F 20 32 =? B2
17:06:53.875 DHT: Timeout waiting for start signal high pulse
17:06:53.884 DHT: Read 40230F2012
17:06:53.886 DHT: Checksum failure 40 23 0F 20 12 =? 92
17:06:55.356 WIF: Checking connection...
17:06:57.857 DHT: Timeout waiting for start signal high pulse
17:06:57.866 DHT: Read 40230FA012
[... cut ...]
17:09:25.849 DHT: Checksum failure 40 23 0F 20 12 =? 92
17:09:29.874 DHT: Timeout waiting for start signal high pulse
17:09:29.883 DHT: Read 40430FA032
17:09:33.867 DHT: Timeout waiting for start signal high pulse
17:09:33.877 DHT: Read 40430F2032
17:09:33.880 DHT: Checksum failure 40 43 0F 20 32 =? B2
17:09:35.599 WIF: Checking connection...
17:09:37.857 DHT: Timeout waiting for start signal high pulse
17:09:37.867 DHT: Read 40630FA052
17:09:41.855 DHT: Timeout waiting for start signal high pulse
17:09:41.864 DHT: Read 40630FA052
```

arendst, 2022-07-16:
Thx.

I'm about to release an updated version with changed timings specifically for the ESP32. I wasn't able to test the MS01 on an ESP32 until today and noticed it doesn't work with the ESP8266 timings. I've changed it and now I see valid readings with one MS01 connected.

Stay tuned...

bob-tm, 2022-07-16:
Log3
GPIO12 - connected sensor, but set to NONE in settings.
GPIO13 - Second MS01, connected.
So Tasmota see only one MS01_13 sensor, and it stop working. 
GPIO13 = null, but when GPIO12 set as MS01 sensor, GPIO13 reads correct value.

```
17:20:29.108 SRC: MQTT
17:20:29.111 CMD: Grp 0, Cmd 'STATE', Idx 1, Len 0, Pld -99, Data ''
17:20:29.118 BRY: GC from 4243 to 3594 bytes, objects freed 2/40 (in 1 ms) - slots from 41/122 to 40/122
17:20:29.134 MQT: stat/tasmota_142214/RESULT = {"Time":"2022-07-16T17:20:29","Uptime":"0T00:00:18","UptimeSec":18,"Heap":155,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:20:29.190 SRC: MQTT
17:20:29.195 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 10, Data '10'
17:20:29.214 MQT: stat/tasmota_142214/STATUS10 = {"StatusSNS":{"Time":"2022-07-16T17:20:29","MS01":{"Humidity":null,"Raw":0},"ESP32":{"Temperature":53.3},"TempUnit":"C"}}
17:20:29.254 SRC: MQTT
17:20:29.257 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11'
17:20:29.265 BRY: GC from 4314 to 3594 bytes, objects freed 4/40 (in 1 ms) - slots from 45/122 to 40/122
17:20:29.278 MQT: stat/tasmota_142214/STATUS11 = {"StatusSTS":{"Time":"2022-07-16T17:20:29","Uptime":"0T00:00:18","UptimeSec":18,"Heap":155,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":50,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}}
17:20:29.329 SRC: MQTT
17:20:29.334 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1'
17:20:29.348 MQT: stat/tasmota_142214/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:18","StartupUTC":"2022-07-16T16:20:11","Sleep":50,"CfgHolder":4617,"BootCount":67,"BCResetTime":"2022-04-12T14:27:52","SaveCount":192}}
17:20:31.860 DHT: Timeout waiting for start signal high pulse
17:20:35.887 DHT: Timeout waiting for start signal high pulse
17:20:37.820 WIF: Checking connection...
17:20:39.878 DHT: Timeout waiting for start signal high pulse
17:20:43.877 DHT: Timeout waiting for start signal high pulse
17:20:47.871 DHT: Timeout waiting for start signal high pulse
17:20:51.856 DHT: Timeout waiting for start signal high pulse
17:20:55.861 DHT: Timeout waiting for start signal high pulse
17:20:57.828 WIF: Checking connection...
17:20:59.846 DHT: Timeout waiting for start signal high pulse
17:21:03.888 DHT: Timeout waiting for start signal high pulse
17:21:07.846 DHT: Timeout waiting for start signal high pulse
17:21:08.961 HTP: Main Menu
17:21:10.706 HTP: Consoles
17:21:11.861 DHT: Timeout waiting for start signal high pulse
17:21:12.278 HTP: Console
17:21:15.843 DHT: Timeout waiting for start signal high pulse
17:21:17.817 WIF: Checking connection...
17:21:19.883 DHT: Timeout waiting for start signal high pulse
17:21:23.871 DHT: Timeout waiting for start signal high pulse
17:21:27.850 DHT: Timeout waiting for start signal high pulse
17:21:31.845 DHT: Timeout waiting for start signal high pulse
17:21:35.867 DHT: Timeout waiting for start signal high pulse
17:21:37.842 WIF: Checking connection...
17:21:39.845 DHT: Timeout waiting for start signal high pulse
17:21:43.877 DHT: Timeout waiting for start signal high pulse
17:21:47.877 DHT: Timeout waiting for start signal high pulse
17:21:51.856 DHT: Timeout waiting for start signal high pulse
17:21:55.844 DHT: Timeout waiting for start signal high pulse
17:21:57.811 WIF: Checking connection...
17:21:59.868 DHT: Timeout waiting for start signal high pulse
17:22:03.857 DHT: Timeout waiting for start signal high pulse
17:22:07.849 DHT: Timeout waiting for start signal high pulse
17:22:11.886 DHT: Timeout waiting for start signal high pulse
17:22:15.860 DHT: Timeout waiting for start signal high pulse
17:22:17.827 WIF: Checking connection...
17:22:19.886 DHT: Timeout waiting for start signal high pulse
17:22:23.868 DHT: Timeout waiting for start signal high pulse
17:22:27.852 DHT: Timeout waiting for start signal high pulse
17:22:31.848 DHT: Timeout waiting for start signal high pulse
17:22:35.875 DHT: Timeout waiting for start signal high pulse
17:22:37.849 WIF: Checking connection...
17:22:39.854 DHT: Timeout waiting for start signal high pulse
17:22:43.889 DHT: Timeout waiting for start signal high pulse
17:22:47.876 DHT: Timeout waiting for start signal high pulse
17:22:51.852 DHT: Timeout waiting for start signal high pulse
17:22:55.883 DHT: Timeout waiting for start signal high pulse
17:22:57.853 WIF: Checking connection...
17:22:59.860 DHT: Timeout waiting for start signal high pulse
17:23:03.882 DHT: Timeout waiting for start signal high pulse
17:23:07.875 DHT: Timeout waiting for start signal high pulse
```

arendst, 2022-07-16:
Try latest changes either as self-compile or wait an hour for the compiled development version become available for OTA upload at http://ota.tasmota.com/tasmota32/tasmota32.bin

Let me know what results you get with one and two sensors configured.

sfromis, 2022-07-16:
`"GPIO12":{"8768":"MS01"},"GPIO13":{"8768":"MS01"},`

This config cannot work, as same component number can only be used for one pin.

arendst, 2022-07-16:
@sfromis Yes that's true EXCEPT... The DHT driver was the first to support more than one sensor and it's implementation is a bit different from all others.

The above config is fine and should work (at least it works for my two AM2301 configured the same as above). The DHT sensor enumerates the names by gpio name and detects up to four sensors starting from the lowest gpio number.

SO in short, it's OK.

bob-tm, 2022-07-16:
Works!!!
But with some checksum errors

```
17:48:58.194 SRC: MQTT
17:48:58.197 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 10, Data '10'
17:48:58.217 MQT: stat/tasmota_142214/STATUS10 = {"StatusSNS":{"Time":"2022-07-16T17:48:58","MS01-12":{"Humidity":19.0,"Raw":16806},"MS01-13":{"Humidity":19.1,"Raw":16709},"ESP32":{"Temperature":53.3},"TempUnit":"C"}}
17:48:58.257 SRC: MQTT
17:48:58.261 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11'
17:48:58.268 BRY: GC from 4361 to 3594 bytes, objects freed 4/40 (in 1 ms) - slots from 45/122 to 40/122
17:48:58.283 MQT: stat/tasmota_142214/STATUS11 = {"StatusSTS":{"Time":"2022-07-16T17:48:58","Uptime":"0T00:00:17","UptimeSec":17,"Heap":152,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":49,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:2B:F3:37","Channel":12,"Mode":"11n","RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:04"}}}
17:48:58.333 SRC: MQTT
17:48:58.337 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1'
17:48:58.349 MQT: stat/tasmota_142214/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:17","StartupUTC":"2022-07-16T16:48:41","Sleep":50,"CfgHolder":4617,"BootCount":69,"BCResetTime":"2022-04-12T14:27:52","SaveCount":197}}
17:49:01.918 DHT: Pin12 read 41A60FA096
17:49:01.928 DHT: Pin13 read 41440FA035
17:49:01.930 DHT: Pin13 checksum failure 41440FA035 =? 34
17:49:02.512 HTP: Main Menu
17:49:05.862 DHT: Pin12 read 41A60FA096
17:49:05.870 DHT: Pin13 read 41650F2055
17:49:05.872 DHT: Pin13 checksum failure 41650F2055 =? D5
17:49:06.399 HTP: Consoles
17:49:06.851 WIF: Checking connection...
17:49:07.996 HTP: Console
17:49:09.905 DHT: Pin12 timeout waiting for high pulse
17:49:09.914 DHT: Pin13 read 41450FA035
17:49:13.901 DHT: Pin12 read 41260FA096
17:49:13.904 DHT: Pin12 checksum failure 41260FA096 =? 16
17:49:13.914 DHT: Pin13 read 41410FA035
17:49:13.916 DHT: Pin13 checksum failure 41410FA035 =? 31
17:49:17.872 DHT: Pin12 read 41A60F2096
17:49:17.875 DHT: Pin12 checksum failure 41A60F2096 =? 16
17:49:17.884 DHT: Pin13 read 41610F2055
17:49:17.886 DHT: Pin13 checksum failure 41610F2055 =? D1
17:49:21.882 DHT: Pin12 read 41A60FA096
17:49:21.890 DHT: Pin13 read 41450F2035
17:49:21.892 DHT: Pin13 checksum failure 41450F2035 =? B5
17:49:25.885 DHT: Pin12 read 41A60FA096
17:49:25.893 DHT: Pin13 read 41450FA031
17:49:25.895 DHT: Pin13 checksum failure 41450FA031 =? 35
17:49:26.876 WIF: Checking connection...
17:49:29.890 DHT: Pin12 read 41A60FA096
17:49:29.897 DHT: Pin13 read 41650FA051
17:49:29.901 DHT: Pin13 checksum failure 41650FA051 =? 55
17:49:33.935 DHT: Pin12 timeout waiting for high pulse
17:49:33.943 DHT: Pin13 read 41440FA035
17:49:33.945 DHT: Pin13 checksum failure 41440FA035 =? 34
17:49:37.886 DHT: Pin12 read 41A60FA016
17:49:37.889 DHT: Pin12 checksum failure 41A60FA016 =? 96
17:49:37.899 DHT: Pin13 read 41450FA035
17:49:41.893 DHT: Pin12 read 41C60FA0B6
17:49:41.904 DHT: Pin13 read 41650FA055
17:49:45.956 DHT: Pin12 timeout waiting for high pulse
17:49:45.965 DHT: Pin13 read 41450FA034
17:49:45.968 DHT: Pin13 checksum failure 41450FA034 =? 35
17:49:46.838 WIF: Checking connection...
17:49:49.895 DHT: Pin12 read 41C60FA0B6
17:49:49.903 DHT: Pin13 read 41450FA035
17:49:53.910 DHT: Pin12 read 41A60FA096
17:49:53.918 DHT: Pin13 read 41650F2055
17:49:53.920 DHT: Pin13 checksum failure 41650F2055 =? D5
17:49:57.869 DHT: Pin12 read 41260FA096
17:49:57.872 DHT: Pin12 checksum failure 41260FA096 =? 16
17:49:57.881 DHT: Pin13 read 41450F2035
17:49:57.883 DHT: Pin13 checksum failure 41450F2035 =? B5
17:50:01.872 DHT: Pin12 read 41C60FA0B6
17:50:01.881 DHT: Pin13 read 01060F2076
17:50:01.885 DHT: Pin13 checksum failure 01060F2076 =? 36
17:50:05.928 DHT: Pin12 timeout waiting for high pulse
17:50:05.938 DHT: Pin13 read 41450FA035
[... 42 lines cut ...]
17:50:45.935 DHT: Pin13 checksum failure 41650FA054 =? 55
17:50:47.355 WIF: Checking connection...
17:50:49.878 DHT: Pin12 read 41A60FA096
17:50:49.888 DHT: Pin13 read 416507A055
17:50:49.891 DHT: Pin13 checksum failure 416507A055 =? 4D
17:50:53.881 DHT: Pin12 read 41C60FA0B6
17:50:53.890 DHT: Pin13 read 41450FA035
17:50:57.883 DHT: Pin12 read 41A60FA096
17:50:57.892 DHT: Pin13 read 41650FA055
17:50:58.571 BRY: GC from 7221 to 3688 bytes, objects freed 29/40 (in 1 ms) - slots from 43/122 to 40/122
17:51:01.885 DHT: Pin12 read 41A60FA096
17:51:01.895 DHT: Pin13 read 41410FA035
17:51:01.897 DHT: Pin13 checksum failure 41410FA035 =? 31
17:51:05.888 DHT: Pin12 read 41A60FA016
17:51:05.890 DHT: Pin12 checksum failure 41A60FA016 =? 96
17:51:05.898 DHT: Pin13 read 41450F2035
17:51:05.900 DHT: Pin13 checksum failure 41450F2035 =? B5
17:51:07.379 WIF: Checking connection...
17:51:09.873 DHT: Pin12 read 41A60FA086
17:51:09.876 DHT: Pin12 checksum failure 41A60FA086 =? 96
17:51:09.884 DHT: Pin13 read 41650FA055
17:51:13.928 DHT: Pin12 timeout waiting for high pulse
17:51:13.938 DHT: Pin13 read 41610F2055
17:51:13.942 DHT: Pin13 checksum failure 41610F2055 =? D1
```

bob-tm, 2022-07-16:
Both sensors connected with extended wires (4 meter).

arendst, 2022-07-16:
Thx. Long wires may indeed be the cause of the checksum errors. Maybe someone has an idea how to solve this. BTW sonoff does support longer cables but I thought to know not for the MS01.

bob-tm, 2022-07-16:
it was sonoff extenders for MS01 (up to 5 meters). But I cut connectors, because they were very glitches

bob-tm, 2022-07-17:
ESP32+MS01 is much more unstable than ESP8266. A lot of checksum errors, so peak values from sensors are lost. 
I can try different timing settings, to find stable setup. 
1. what values can I try and what dont?
2. maybe it's possible to change them from web console using temp param?

arendst, 2022-07-17:
Checksum errors are detected by wrong pulse lengths during gathering of 80 pulses providing the sensor data. You'll have to analyse the 80 values to find a common value for a zero and a one.

Also notice that the original sonoff th hardware is different from just connecting the sensor to esps. you need to look at the schematics of the th10/16 to find out if they use external pull up resistors. Alas there is no logics for the esp32 based thr3xxD devices but from the pictures I'm quite sure they use external pullup to get a better signal over long wires.

bob-tm, 2022-07-17:
pin13: a lot of checksum errors
pin12: a lot of timeout errors, sometimes correct values.  

07:55:21.943 DHT: Pin12 timeout waiting for high pulse
07:55:25.979 DHT: Pin12 timeout waiting for high pulse
07:55:29.963 DHT: Pin12 timeout waiting for high pulse
07:55:33.962 DHT: Pin12 timeout waiting for high pulse

I will try to add some pull up resistors and check

bob-tm, 2022-07-22:
1. add pull up 6.7k
2. Moved from gpio12 to pin4 (won't boot with pool up on 12)

It's definitely helps!
No more checksum errors!
Much less time outs.

Sometimes (up to 5% of readings) time out on lower pin. Swapping sensors do not help. Seems like it depends on CPU Load. Disconnecting sensor from Higher gpio helps a little. 

I will monitor sensors in this configuration

bob-tm, 2022-07-23:
Peak readings from ESP32 are lower than from ESP8266. 
What it can be? 

[image]
[image]

sfromis, 2022-07-23:
When it comes to "soil moisture" and such overly simple sensors, asking "what is correct" is not a reasonable question. Neither is "correct" as you cannot measure an "absolute" soil humidity using a little bit of electronics. Values depends a lot on factors like exact electrolytic composition of both soil and the water. Two sensors nearby may not get the same numbers, where you already have so many digits ("precision") that it misleads you to think in "accuracy". Getting a trend up/down and a weak indication of try/moist/wet is just about what you can expect.

Jason2866, 2022-07-24:
Closing since the issue is fixed.

closed 2022-07-24 by Jason2866