arendst/Tasmota #16007: RGB+CW bulb not working as expected

dietmarczekay, 2022-07-17:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
beginning with version 11 my bulbs did not behave like expected. I have a Athom TB15 RGB+CW bulb. Switching on, the bulb did not recall the last setting and switching CT to 500, the bulb goes off.
00:00:00.002 HDW: ESP8266EX
00:00:00.047 CFG: Loaded from flash at FB, Count 21
00:00:00.052 QPC: Count 1
00:00:00.066 Project tasmota - Lampe15W1 Version 12.0.2(tasmota)-2_7_4_9(2022-06-20T12:41:03)

A 7W Bulb from the same supplier behaves like expected
00:00:00.002 HDW: ESP8266EX
00:00:00.051 CFG: Loaded from flash at F7, Count 7141
00:00:00.056 QPC: Count 1
00:00:00.067 Project tasmota - lampe7W1 Version 11.0.0(tasmota)-2_7_4_9(2022-02-12T14:13:50)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Athorm
- [ ] Tasmota binary firmware version number used: _____
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:19:13.933 MQT: stat/lampe7W1/RESULT = {"NAME":"LB01-7W-E27-TAS-1","GPIO":[0,0,0,0,416,419,0,0,417,420,418,0,0,0],"FLAG":0,"BASE":18}
17:19:14.142 MQT: stat/lampe7W1/RESULT = {"Module":{"0":"LB01-7W-E27-TAS-1"}}
17:19:14.347 MQT: stat/lampe7W1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"419":"PWM4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"420":"PWM5"},"GPIO14":{"418":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:19:50.844 MQT: stat/lampe7W1/STATUS11 = {"StatusSTS":{"Time":"2022-07-17T17:19:50","Uptime":"0T00:17:34","UptimeSec":1054,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"00000000FF","HSBColor":"117,100,0","White":100,"CT":500,"Channel":[0,0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxxxxxx","BSSId":"98:9B:CB:17:D1:D0","Channel":6,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
 switch on and send CT150 and CT 500

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-07-17:
Can you please provide the logs when you start, and issue CT150 and CT500?

s-hadinger, 2022-07-17:
I pulled back an old 5 ways Sonoff bulb (but not PWM) and I couldn't reproduce. Can you try `reset 4` to reinitialize all settings except Wifi? You will need to reapply the template

dietmarczekay, 2022-07-17:
sending "SetOption92 1" solved the problem with CT, but storing the last settings didn't work

s-hadinger, 2022-07-17:
What about `savedata`?

dietmarczekay, 2022-07-17:
savedata on saved the day, thanks.

closed 2022-07-17 by dietmarczekay