arendst/Tasmota #16022: Wrong lux calculation for BH1750 sensor

KestasG, 2022-07-20:
### PROBLEM DESCRIPTION
lux calculation line for BH1750 looks:
illuminance /= (1.2 * (69 / (float)Bh1750_sensors[sensor_index].mtreg));
and it does not correspond to the datasheet stating:
[image]

You can test it by checking if the results correspond datasheet provided table:
[image]

The current formula works fine for MTReg=69, but gives the wrong result for any other values. As an example let's say MTreg=254. We read 10 count from the sensor. Applying calculations gives as 10/(1.2*(69/254))=30.7 lux which is wrong as per data table 1 count corresponds to 0.23lux, so 10 counts = 2.3lux.

Correct code should be (parenthises for clarity)
illuminance *= (1/1.2) * (69 / (float)Bh1750_sensors[sensor_index].mtreg));
or simplified
illuminance *= 57.5 / (float)Bh1750_sensors[sensor_index].mtreg));
Calculating previous case - 10*57.5/254=2.3lux

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
use BH1750sensor with non default value for Bh1750MTime

### EXPECTED BEHAVIOUR
Lux values correspond datasheet

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-07-20:
Thx. I'll do some tests and if positive will make the change.

ascillato2, 2022-07-20:
Closing as it has been fixed by Theo. Thanks.

closed 2022-07-20 by ascillato2