arendst/Tasmota #16041: Prometheus + INA226 I2C sensor does not handle negative current values

davidhorrock0949, 2022-07-21:
### PROBLEM DESCRIPTION
Although the web client correctly displays negative values for current using an INA226 I2C module, the Prometheus metric is filtered out and does not display at <ip>/metrics if the value is less than zero.  This is a problem for measuring bidirectional current (for example, in a battery charge/discharge system).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [X] Tasmota binary firmware version number used: 12.0.2.4
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

18:56:06.450 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:56:06.670 RSL: RESULT = {"Module":{"0":"Generic"}}
18:56:06.924 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

Rules not used

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:57:44.738 CMD: Status 0
18:57:44.743 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_19588A","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:57:44.748 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:34:39","StartupUTC":"2022-07-21T17:23:05","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2022-07-21T12:58:47","SaveCount":10,"SaveAddress":"3FB000"}}
18:57:44.752 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2.4(TasmoCompiler-esp82664M)","BuildDateTime":"2022-07-21T12:49:47","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"354/699"}}
18:57:44.757 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mango",""],"TelePeriod":300,"Resolution":"558D80C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:57:44.769 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":786,"Free":1260,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FA0458F","04261FA1","740207CF","590C13C0","D832FDD1","083D60AC","003E9000","54000060","00000000"],"Drivers":"1,2,3,4,6,8,9,10,12,13,18,19,22,24,26,27,29,30,35,37,38,45,50,62","Sensors":"1,2,3,4,5,6,7,8,9,14,26,27,38,39,46,47,54,55,58,59,63,65,72,73,74,75,88"}}
18:57:44.775 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-19588A-6282","IPAddress":"192.168.4.95","Gateway":"192.168.4.1","Subnetmask":"255.255.252.0","DNSServer1":"45.90.28.193","DNSServer2":"45.90.30.193","Mac":"A0:20:A6:19:58:8A","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:57:44.778 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_19588A","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:57:44.784 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-07-21T17:57:44","Local":"2022-07-21T18:57:44","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:09","Sunset":"20:43"}}
18:57:44.790 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-07-21T18:57:44","INA226-1":{"Id":0,"Voltage":13.33,"Current":-0.653,"Power":8.7}}}
18:57:44.795 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-07-21T18:57:44","Uptime":"0T00:34:39","UptimeSec":2079,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"mango","BSSId":"9C:A5:70:7D:48:E5","Channel":6,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [X ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Not applicable

```

### TO REPRODUCE
Enable Prometheus and an INA226 module, any measured currents will not be logged in metrics.

### EXPECTED BEHAVIOUR
Negative currents will show up in metrics.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
Negative Current is correctly displayed in the web interface:

INA226-1 Voltage | 13.32 V
INA226-1 Current | -0.647 A
INA226-1 Power | 8.7 W

However, current is not available as a prometheus metrics when current is negative:

# TYPE tasmota_info gauge
tasmota_info{version="12.0.2.4",image="(TasmoCompiler-esp82664M)",build_timestamp="2022-07-21T12:49:47",devicename="Tasmota",friendlyname="Tasmota"} 1
# TYPE tasmota_uptime_seconds gauge
tasmota_uptime_seconds{} 767
# TYPE tasmota_boot_count counter
tasmota_boot_count{} 4
# TYPE tasmota_flash_writes_total counter
tasmota_flash_writes_total{} 10
# TYPE tasmota_wifi_station_info gauge
tasmota_wifi_station_info{bssid="9C:A5:70:7D:48:E5",ssid="mango"} 1
# TYPE tasmota_wifi_station_signal_dbm gauge
tasmota_wifi_station_signal_dbm{mac_address="9C:A5:70:7D:48:E5"} -74
# TYPE tasmota_memory_free_bytes gauge
tasmota_memory_free_bytes{memory="heap"} 19152
# TYPE tasmota_energy_voltage_volts gauge
tasmota_energy_voltage_volts{} 0.00
# TYPE tasmota_energy_current_amperes gauge
tasmota_energy_current_amperes{} 0.000
# TYPE tasmota_energy_power_active_watts gauge
tasmota_energy_power_active_watts{} 0.0
# TYPE tasmota_energy_power_kilowatts_daily counter
tasmota_energy_power_kilowatts_daily{} 0.000
# TYPE tasmota_energy_power_kilowatts_total counter
tasmota_energy_power_kilowatts_total{} 0.000
# TYPE tasmota_sensors_id_untyped gauge
tasmota_sensors_id_untyped{sensor="ina226-1",id="0"} 1
# TYPE tasmota_sensors_voltage_volts gauge
tasmota_sensors_voltage_volts{sensor="ina226-1"} 13.30
# TYPE tasmota_sensors_power_ gauge
tasmota_sensors_power_{sensor="ina226-1"} 8.5

barbudor, 2022-07-21:
Could you please copy/paste the json returned by Tasmota on the '/metrics' url?

barbudor, 2022-07-21:
OK, I found the issue in this test
https://github.com/arendst/Tasmota/blob/40a5d661fefd9fc8c5fb2d84f819d231767d6f90/tasmota/tasmota_xsns_sensor/xsns_75_prometheus.ino#L301
I can make a PR but I won't be able to test
Would you be able to compile from my fork or should I provide you with a binary?
If binary, do you need anything else than INA226 + prometheus to validate?

barbudor, 2022-07-21:
I prepared a fix, can you please test with the attached tasmota-sensors binaries ?
They are the standard tasmota-sensors + USE_INA226 + USE_PROMETHEUS
Alternatively if you can compile yourself from my branch: https://github.com/barbudor/Tasmota/tree/prometheus_negative_values 

If successfull, please report here : https://github.com/arendst/Tasmota/pull/16044

davidhorrock0949, 2022-07-21:
Thanks for the quick fix!  I'll compile and test it here this evening and provide an update.

davidhorrock0949, 2022-07-21:
Awesome, that fixed it!  Works perfectly now.

PR #16044 (2022-07-22): fix removal of negative values
## Description:

**Related issue (if applicable):** fixes #16041

Fix removal of values starting with a non digit, such as starting with '-' for negative values or '.' for floating with leading 0

Testing performed by @davidhorrock0949 as original issue reporter

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [ ] [... cut]
closed 2022-07-22 by arendst