arendst/Tasmota #16055: PowerSet Limit seems to be 2000W

BelgarathS, 2022-07-23:
### PROBLEM DESCRIPTION
PowerSet with value above 2000 does not change the powerset value
If that is physical calibration that is being done where the power meter is present you can set the value with a smaller load and then measure the higher values however if you use that as a method of setting a dummy power meter (OptionA = 2) there is no way to set that value above 1999 which in case of some relays makes that unusable (i.e using a dummy meter for pump/heater with high power usage where regular plugs won't work)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"Relay_Switch_X4_30A","GPIO":[0,0,0,0,32,576,0,0,226,227,225,5729,224,0],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
 {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":39,"Free":472,"Rules":"ON Power1#State DO Power3 %value% ENDON"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:20:01.153 MQT: stat/water_heater/STATUS = {"Status":{"Module":0,"DeviceName":"water_heater","FriendlyName":["water_heater_1","water_heater_2","water_heater_3","water_heater_4"],"Topic":"water_heater","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:20:01.164 MQT: stat/water_heater/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T04:30:14","StartupUTC":"2022-07-23T13:49:47","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2022-07-23T12:29:34","SaveCount":43,"SaveAddress":"F9000"}}
19:20:01.172 MQT: stat/water_heater/STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"413/699"}}
19:20:01.181 MQT: stat/water_heater/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["banda_ha",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00208009","0405C80001000600003C5A0A000000000000","08000080","00006000","00004000","00000000"]}}
19:20:01.196 MQT: stat/water_heater/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
19:20:01.205 MQT: stat/water_heater/STATUS5 = {"StatusNET":{"Hostname":"water-heater-8018","IPAddress":"10.0.40.241","Gateway":"10.0.40.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.40.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:B3:BF:52","Webserver":2,"HTTP_API":1,"WifiConfig":0,"WifiPower":17.0}}
19:20:01.214 MQT: stat/water_heater/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.40.80","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B3BF52","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:20:01.224 MQT: stat/water_heater/STATUS7 = {"StatusTIM":{"UTC":"2022-07-23T18:20:01","Local":"2022-07-23T19:20:01","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:12","Sunset":"20:40"}}
19:20:01.230 MQT: stat/water_heater/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:20:01.245 MQT: stat/water_heater/STATUS10 = {"StatusSNS":{"Time":"2022-07-23T19:20:01","ENERGY":{"TotalStartTime":"2022-07-23T12:29:34","Total":6.096,"Yesterday":0.000,"Today":6.096,"Power":[ 0, 0, 0],"ApparentPower":[ 0, 0, 0],"ReactivePower":[ 0, 0, 0],"Factor":[0.00,0.00,0.00],"Frequency":50,"Voltage":237,"Current":[0.000,0.000,0.000]}}}
19:20:01.255 MQT: stat/water_heater/STATUS11 = {"StatusSTS":{"Time":"2022-07-23T19:20:01","Uptime":"0T04:30:14","UptimeSec":16214,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"banda_ha","BSSId":"80:2A:A8:47:29:00","Channel":6,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
19:20:49.801 CMD: Grp 0, Cmd 'POWERSET', Idx 1, Len 4, Pld 2800, Data '2800'
19:20:49.807 MQT: stat/water_heater/RESULT = {"PowerSetCal":186700}

The powersetcal value is the value set previously, i.e the PowerSet command do not change the value
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Issue PowerSet with Value above 1999

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Value would be set as requested

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-07-23:
Seems to be as designed, the driver deliberately caps the value as "Between 1.00 and 2000.00 W".
https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xnrg_energy/xnrg_30_dummy.ino#L86

BelgarathS, 2022-07-23:
This makes sense for normal calibration with a physical sensor as you can select the load for calibration. With a dummy sensor, there is no way to set up the proper load value as there is no way to scale the "default value". With a dummy sensor, we have to set the actual load with SetPower (or SetCurrent)

sfromis, 2022-07-23:
The code I found is specifically not for normal calibration, where no such limiting is done, but only for dummy energy monitoring.

BelgarathS, 2022-07-23:
I'm sorry, I missed that it was specifically for the dummy monitor, do you, or anyone else here know why this limit was imposed?
I think it should be increased if possible as I have some relays that are capable of 100A switching and it would be useful if that can be added to the standard build unless there is a specific reason not to.
I understand this is not a standard requirement but if it does not impede normal setup I see no reason to limit appliances to 2000W as even some of the standard home appliances exceed this rating.