arendst/Tasmota #16061: Sonoff devices reboot every ~20 minutes  with "Exception":29

darkdragondraco, 2022-07-24:
### PROBLEM DESCRIPTION
Sonoff devices after upgrade to Tasmota 12.0.2 reboot every ~20 minutes with StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["40252ca8",...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff TH16, Sonoff Mini, Sonoff RF, Sonoff S55
- [ ] Tasmota binary firmware version number used: 12.0.2.4
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Ota from TasmoAdmin, before that used Tasmotizer for initial flash
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:42:14.935 CMD: Backlog Template; Module; GPIO 255
13:42:14.938 SRC: WebConsole from 192.168.2.143
13:42:14.941 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
13:42:14.958 SRC: Backlog
13:42:14.961 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
13:42:14.968 MQT: stat/sf-s55-01/RESULT = {"NAME":"Sonoff S55","GPIO":[32,1,0,1,1,0,0,0,224,320,0,0,0,0],"FLAG":0,"BASE":1}
13:42:15.181 SRC: Backlog
13:42:15.184 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
13:42:15.189 MQT: stat/sf-s55-01/RESULT = {"Module":{"0":"Sonoff S55"}}
13:42:15.431 SRC: Backlog
13:42:15.434 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
13:42:15.442 MQT: stat/sf-s55-01/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:43:03.435 CMD: Backlog Rule1; Rule2; Rule3
13:43:03.437 SRC: WebConsole from 192.168.2.143
13:43:03.439 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
13:43:03.461 SRC: Backlog
13:43:03.464 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
13:43:03.472 MQT: stat/sf-s55-01/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:43:03.680 SRC: Backlog
13:43:03.683 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
13:43:03.689 MQT: stat/sf-s55-01/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:43:03.930 SRC: Backlog
13:43:03.933 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
13:43:03.940 MQT: stat/sf-s55-01/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:43:35.567 CMD: Status 0
13:43:35.569 SRC: WebConsole from 192.168.2.143
13:43:35.571 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
13:43:35.579 MQT: stat/sf-s55-01/STATUS = {"Status":{"Module":0,"DeviceName":"sf-s55-01","FriendlyName":["sf-s55-01"],"Topic":"sf-s55-01","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:43:35.588 MQT: stat/sf-s55-01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.2.6:5555/data/firmwares/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:16:36","StartupUTC":"2022-07-24T10:26:59","Sleep":50,"CfgHolder":4617,"BootCount":86,"BCResetTime":"2021-06-09T11:15:56","SaveCount":119,"SaveAddress":"FB000"}}
13:43:35.595 MQT: stat/sf-s55-01/STATUS2 = {"StatusFWR":{"Version":"12.0.2.4(tasmota)","BuildDateTime":"2022-07-22T13:15:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"343/699"}}
13:43:35.602 MQT: stat/sf-s55-01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shadow",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
13:43:35.617 MQT: stat/sf-s55-01/STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
13:43:35.625 MQT: stat/sf-s55-01/STATUS5 = {"StatusNET":{"Hostname":"sf-s55-01-1694","IPAddress":"192.168.2.136","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"192.168.1.234","Mac":"24:A1:60:10:E6:9E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:43:35.630 MQT: stat/sf-s55-01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.6","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_10E69E","MqttUser":"draco","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:43:35.640 MQT: stat/sf-s55-01/STATUS7 = {"StatusTIM":{"UTC":"2022-07-24T10:43:35","Local":"2022-07-24T13:43:35","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+03:00","Sunrise":"07:13","Sunset":"22:39"}}
13:43:35.647 MQT: stat/sf-s55-01/STATUS10 = {"StatusSNS":{"Time":"2022-07-24T13:43:35"}}
13:43:35.655 MQT: stat/sf-s55-01/STATUS11 = {"StatusSTS":{"Time":"2022-07-24T13:43:35","Uptime":"0T00:16:36","UptimeSec":996,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shadow","BSSId":"48:8F:5A:21:1E:27","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:43:35.671 MQT: stat/sf-s55-01/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["401014c8","40268f93","40268f28","40268ecf","40268028","40268051","40265ac0","4025db44","4010392a","40261e2d","40256ab1","402667b7","40000f58","4026622a","4026f987","4026f247","4024d1dc","40000f49","40000f49","40000e19","40105aa9","40105aaf","4010000d","4026e124","4026e0d5","40100a7c","40102b21","401026f0","4010598b","40104b66","40104fd7"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
[... 44 lines cut ...]
14:12:59.781 WIF: Checking connection...
14:13:19.760 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Reboots every ~20 minutes 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Does not reboot without clear reason

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Started few days ago and only in one location.  in other location sonoff devices with same firmware work ok.  Problem devices use different power lines some of them are UPSed. Tried to downgrade  to 11.1 without clearing config but it did not help. 
The only this was changed at environment were DNS and DHCP server reconfiguration.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-07-24:
It looks like you're running out of heap space. 

A value of 14 is low and the exception seems to occur during heap allocation for a flash memcpy_P action. 

> The only this was changed at environment were DNS and DHCP server reconfiguration.

As it only takes 20 minutes I suggest you enable heap logging (command ``SO130 1``) and ``weblog 4`` and try to catch as much logging as possible using the console or ``mqttlog 4`` using a mqqt terminal tool like tdmgr.

darkdragondraco, 2022-07-24:
> What do you mean by this?

I've migrated from old DNS to new one,  and changed DHCP settings accordingly. But that shaped not affect anything

This is console log from tdmgr during reboot

[18:34:04] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:04"}}
[18:34:05] stat/sf-s55-01/LOGGING 18:34:04.678-017 SRC: MQTT
[18:34:05] stat/sf-s55-01/LOGGING 18:34:04.681-017 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:05] stat/sf-s55-01/LOGGING 18:34:04.687-014 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:04"}}
[18:34:05] stat/sf-s55-01/LOGGING 18:34:04.829-017 WIF: Checking connection...
[18:34:09] cmnd/sf-s55-01/STATUS 8
[18:34:09] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:09"}}
[18:34:10] stat/sf-s55-01/LOGGING 18:34:09.686-017 SRC: MQTT
[18:34:10] stat/sf-s55-01/LOGGING 18:34:09.689-017 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:10] stat/sf-s55-01/LOGGING 18:34:09.694-014 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:09"}}
[18:34:14] cmnd/sf-s55-01/STATUS 8
[18:34:14] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:14"}}
[18:34:15] stat/sf-s55-01/LOGGING 18:34:14.741-017 SRC: MQTT
[18:34:15] stat/sf-s55-01/LOGGING 18:34:14.744-017 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:15] stat/sf-s55-01/LOGGING 18:34:14.750-014 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:14"}}
[18:34:19] cmnd/sf-s55-01/STATUS 8
[18:34:19] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:19"}}
[18:34:20] stat/sf-s55-01/LOGGING 18:34:19.652-017 SRC: MQTT
[18:34:20] stat/sf-s55-01/LOGGING 18:34:19.655-017 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:20] stat/sf-s55-01/LOGGING 18:34:19.662-014 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:19"}}
[18:34:24] cmnd/sf-s55-01/STATUS 8
[18:34:24] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:24"}}
[18:34:25] stat/sf-s55-01/LOGGING 18:34:24.662-017 SRC: MQTT
[18:34:25] stat/sf-s55-01/LOGGING 18:34:24.665-017 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:25] stat/sf-s55-01/LOGGING 18:34:24.674-015 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:24"}}
[18:34:25] stat/sf-s55-01/LOGGING 18:34:24.863-017 WIF: Checking connection...
[18:34:29] cmnd/sf-s55-01/STATUS 8
[18:34:29] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:29"}}
[18:34:30] stat/sf-s55-01/LOGGING 18:34:29.674-016 SRC: MQTT
[18:34:30] stat/sf-s55-01/LOGGING 18:34:29.677-016 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:30] stat/sf-s55-01/LOGGING 18:34:29.684-013 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:29"}}
[18:34:34] cmnd/sf-s55-01/STATUS 8
[18:34:34] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:34"}}
[18:34:35] stat/sf-s55-01/LOGGING 18:34:34.678-016 SRC: MQTT
[18:34:35] stat/sf-s55-01/LOGGING 18:34:34.680-016 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:35] stat/sf-s55-01/LOGGING 18:34:34.686-013 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:34"}}
[18:34:39] cmnd/sf-s55-01/STATUS 8
[18:34:39] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:34:39"}}
[18:34:40] stat/sf-s55-01/LOGGING 18:34:39.679-016 SRC: MQTT
[18:34:40] stat/sf-s55-01/LOGGING 18:34:39.682-016 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:34:40] stat/sf-s55-01/LOGGING 18:34:39.688-013 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:34:39"}}
[18:34:41] stat/sf-s55-01/LOGGING 18:34:40.388-015 WIF: Sending Gratuitous ARP
[18:34:42] stat/sf-s55-01/STATUS {"Status":{"Module":0,"DeviceName":"sf-s55-01","FriendlyName":["sf-s55-01"],"Topic":"sf-s55-01","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
[18:34:42] stat/sf-s55-01/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.2.6:5555/data/firmwares/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:14:03","StartupUTC":"2022-07-24T15:20:38","Sleep":50,"CfgHolder":4617,"BootCount":102,"BCResetTime":"2021-06-09T11:15:56","SaveCount":139,"SaveAddress":"F7000"}}
[18:34:42] stat/sf-s55-01/STATUS2 {"StatusFWR":{"Version":"12.0.2.4(tasmota)","BuildDateTime":"2022-07-22T13:15:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"343/699"}}
[18:34:42] stat/sf-s55-01/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shadow",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00014000","00000000"]}}
[18:34:42] stat/sf-s55-01/STATUS4 {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
[... cut ...]
[18:41:40] stat/sf-s55-01/LOGGING 18:41:39.728-020 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:41:39"}}
[18:41:44] cmnd/sf-s55-01/STATUS 8
[18:41:44] stat/sf-s55-01/STATUS8 {"StatusSNS":{"Time":"2022-07-24T18:41:44"}}
[18:41:45] stat/sf-s55-01/LOGGING 18:41:44.676-023 SRC: MQTT
[18:41:45] stat/sf-s55-01/LOGGING 18:41:44.679-023 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 8, Data '8'
[18:41:45] stat/sf-s55-01/LOGGING 18:41:44.685-020 MQT: stat/sf-s55-01/STATUS8 = {"StatusSNS":{"Time":"2022-07-24T18:41:44"}}

Jason2866, 2022-07-24:
Wow `[18:40:35] stat/sf-s55-01/STATUS4 {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":2,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":`
Heap decreased to 2k. Theo was right. Thats the problem ending in a crash.

arendst, 2022-07-24:
Find the cause of executing command status 8 every 5 seconds. The comand is executed from MQTT.

Remove this and see if stability returns.

darkdragondraco, 2022-07-24:
Removed Status call but this did not help 

[22:08:17] tele/sf-s55-01/STATE {"Time":"2022-07-24T22:08:17","Uptime":"0T00:00:11","UptimeSec":11,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shadow","BSSId":"48:8F:5A:21:1E:27","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}
[22:08:18] stat/sf-s55-01/LOGGING 22:08:17.410-022 MQT: tele/sf-s55-01/STATE = {"Time":"2022-07-24T22:08:17","Uptime":"0T00:00:11","UptimeSec":11,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shadow","BSSId":"48:8F:5A:21:1E:27","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}
[22:08:23] stat/sf-s55-01/LOGGING 22:08:22.434-020 MQT: tasmota/discovery/24A16010E69E/config = {"ip":"192.168.2.136","dn":"sf-s55-01","fn":["sf-s55-01",null,null,null,null,null,null,null],"hn":"sf-s55-01-1694","mac":"24A16010E69E","md":"Sonoff S55","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.0.2.4","t":"sf-s55-01","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
[22:08:23] stat/sf-s55-01/LOGGING 22:08:22.440-020 MQT: tasmota/discovery/24A16010E69E/sensors = {"sn":{"Time":"2022-07-24T22:08:22"},"ver":1} (retained)
[22:08:32] stat/sf-s55-01/LOGGING 22:08:32.261-023 WIF: Checking connection...
[22:08:52] stat/sf-s55-01/LOGGING 22:08:52.277-023 WIF: Checking connection...
[22:09:09] stat/sf-s55-01/LOGGING 22:09:08.400-023 WIF: Sending Gratuitous ARP
[22:09:12] stat/sf-s55-01/LOGGING 22:09:12.274-023 WIF: Checking connection...
[22:09:32] stat/sf-s55-01/LOGGING 22:09:32.257-022 WIF: Checking connection...
[22:09:52] stat/sf-s55-01/LOGGING 22:09:52.283-022 WIF: Checking connection...
[22:10:09] stat/sf-s55-01/LOGGING 22:10:08.414-022 WIF: Sending Gratuitous ARP
[22:10:12] stat/sf-s55-01/LOGGING 22:10:12.269-022 WIF: Checking connection...
[22:10:32] stat/sf-s55-01/LOGGING 22:10:32.253-022 WIF: Checking connection...
[22:10:52] stat/sf-s55-01/LOGGING 22:10:52.279-022 WIF: Checking connection...
[22:11:09] stat/sf-s55-01/LOGGING 22:11:08.405-022 WIF: Sending Gratuitous ARP
[22:11:12] stat/sf-s55-01/LOGGING 22:11:12.263-022 WIF: Checking connection...
[22:11:32] stat/sf-s55-01/LOGGING 22:11:32.293-022 WIF: Checking connection...
[22:11:52] stat/sf-s55-01/LOGGING 22:11:52.270-022 WIF: Checking connection...
[22:12:10] stat/sf-s55-01/LOGGING 22:12:08.395-022 WIF: Sending Gratuitous ARP
[22:12:12] stat/sf-s55-01/LOGGING 22:12:12.279-022 WIF: Checking connection...
[22:12:32] stat/sf-s55-01/LOGGING 22:12:32.255-022 WIF: Checking connection...
[22:12:52] stat/sf-s55-01/LOGGING 22:12:52.285-022 WIF: Checking connection...
[22:13:09] stat/sf-s55-01/LOGGING 22:13:08.386-022 WIF: Sending Gratuitous ARP
[22:13:12] stat/sf-s55-01/LOGGING 22:13:12.297-022 WIF: Checking connection...
[22:13:18] tele/sf-s55-01/STATE {"Time":"2022-07-24T22:13:17","Uptime":"0T00:05:11","UptimeSec":311,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shadow","BSSId":"48:8F:5A:21:1E:27","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
[22:13:18] stat/sf-s55-01/LOGGING 22:13:17.414-019 MQT: tele/sf-s55-01/STATE = {"Time":"2022-07-24T22:13:17","Uptime":"0T00:05:11","UptimeSec":311,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shadow","BSSId":"48:8F:5A:21:1E:27","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
[22:13:32] stat/sf-s55-01/LOGGING 22:13:32.273-022 WIF: Checking connection...
[22:13:52] stat/sf-s55-01/LOGGING 22:13:52.257-022 WIF: Checking connection...
[22:14:09] stat/sf-s55-01/LOGGING 22:14:08.381-022 WIF: Sending Gratuitous ARP
[22:14:12] stat/sf-s55-01/LOGGING 22:14:12.285-022 WIF: Checking connection...
[22:14:32] stat/sf-s55-01/LOGGING 22:14:32.268-022 WIF: Checking connection...
[22:14:53] stat/sf-s55-01/LOGGING 22:14:52.298-022 WIF: Checking connection...
[22:15:09] stat/sf-s55-01/LOGGING 22:15:08.420-022 WIF: Sending Gratuitous ARP
[22:15:12] stat/sf-s55-01/LOGGING 22:15:12.279-021 WIF: Checking connection...
[22:15:21] cmnd/sf-s55-01/state 
[22:15:21] stat/sf-s55-01/RESULT {"Time":"2022-07-24T22:15:21","Uptime":"0T00:07:15","UptimeSec":435,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shadow","BSSId":"48:8F:5A:21:1E:27","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
[22:15:22] stat/sf-s55-01/LOGGING 22:15:21.745-021 SRC: MQTT
[22:15:22] stat/sf-s55-01/LOGGING 22:15:21.748-021 CMD: Grp 0, Cmd 'STATE', Idx 1, Len 0, Pld -99, Data ''
[22:15:22] stat/sf-s55-01/LOGGING 22:15:21.755-018 MQT: stat/sf-s55-01/RESULT = {"Time":"2022-07-24T22:15:21","Uptime":"0T00:07:15","UptimeSec":435,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shadow","BSSId":"48:8F:5A:21:1E:27","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
[22:15:23] cmnd/sf-s55-01/State 
[... cut ...] 0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
[22:31:13] stat/sf-s55-01/LOGGING 22:31:13.400-021 MQT: tasmota/discovery/24A16010E69E/sensors = {"sn":{"Time":"2022-07-24T22:31:13"},"ver":1} (retained)
[22:31:23] stat/sf-s55-01/LOGGING 22:31:23.557-023 WIF: Checking connection...
[22:31:43] stat/sf-s55-01/LOGGING 22:31:43.540-023 WIF: Checking connection...

arendst, 2022-07-24:
Thx.

So we still see heap degrading probably due to a memory leak....

For now I have no idea why considering you use the standard development release with a simple template. I'll dive into the stack trace again tomorrow to see if it contains further hints.

darkdragondraco, 2022-07-24:
i'm using binary from http://ota.tasmota.com/tasmota/release/tasmota.bin.gz
template from https://templates.blakadder.com/sonoff_S55.html

arendst, 2022-07-25:
Alas, I cannot find any Tasmota related cause for this. The stacktrace is long and only contains SDK network calls:

```
"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":[
"4000df64", memcpy
"00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":[
"401014c8", pvPortZalloc
"40268f93", tkip_encap
"401037fb", wDev_ProcessFiq
"40268f28", ieee80211_setup_ratetable
"40268ecf", ieee80211_phy_init
"40268028", cnx_update_bss_more
"40268051", cnx_update_bss_more
"40265ac0", scan_parse_beacon
"40261e2d", ieee80211_parse_beacon
"402667b7", sta_recv_mgmt
"4025e1d0", mem_malloc
"4026622a", sta_input
"4026f987", pp_tx_idle_timeout
"4026f247", ppPeocessRxPktHdr
"4024d1dc", _ZL9loop_taskP11ETSEventTag
"40000f49", ets_post_rom
"40000f49", ets_post_rom
"40000e19", ets_run
"40105aa9", call_user_start_local
"40105aaf", call_user_start_local
"4010000d", call_user_start
"4026e124", pm_for_bcn_only_mode
"4026e0d5", cont_yield
"40100a7c", ets_post
"401026f0", pp_post
"40100a7c", ets_post
"40102b21", rcUpdateTxDone
"401026f0", pp_post
"4010598b", lmacTxDone
"40103468", wDev_ProcessRxSucData
"40104b66"  lmacRecycleMPDU
```
It's a WiFi network issue eating the heap but I would expect a RSSI of 78 should still be enough for a stable connection.

Perhaps the wifi hardware is broken leading to bad connection attempts.

darkdragondraco, 2022-07-25:
Actually, all the devices with issues use 3  Mikrotik AP. On other site i have AP of the same model  but newer firmware and have no issues. Will try to upgrade Wifi firmware to newer version  and report back

darkdragondraco, 2022-07-25:
Did the update Mikrotik AP to 6.49.6,  connected additional AP from ASUS,  Installed fresh Sonoff mini tasmotized to 12.0.2 but sonoffs continue to reboot about every 20  minutes because eating  the heap.  Other network devices like cameras work normally. What additional info i need to provide to continue investigation of network part?

Jason2866, 2022-07-25:
First remove the ASUS AP. It is worst thing you can use in combination with espressif devices. Endless story....

Jason2866, 2022-07-25:
Sorry accidently closed, internet connection problems..

darkdragondraco, 2022-07-26:
Looks like i found the issue. During reconfiguration i've added ntp server that send multicast and manycast packets (https://wiki.mikrotik.com/wiki/Manual:System/Time). After i disabled it, heap stopped to grow. Looks like there is some issue on network stack for working with multicast/manycast.  I will investigate further during weekend and report back.

arendst, 2022-07-26:
This is great news!

I looked for this issue in the old issue list and came across mikrotik and ntp server misconfigurations. As I rewrote the ntp client part since then I couldn't link it to your issue.

Pls find time to investigate and let me know what to change to finally solve this.

arendst, 2022-07-26:
Did you also experience heap degrading on ESP32 based devices?

I searched for UDP multicast packet handling on ESP8266 and find this https://github.com/esp8266/Arduino/issues/7907#issuecomment-790536709. Obviously the dropping of overflow packets doesn't seem to work within the core code.

arendst, 2022-07-26:
I noticed for ESP8266 we use our own UDP listener driver. It might well be this driver doesn't test for overflow packets. I will dive in.

NielsPiersma, 2022-08-03:
I would like to report that I have got the same issues. Our policy doesn't allow using public NTP servers for time resolving, so our DHCP scope assigns internal NTP servers. These are Server 2019 boxes and may not be working as expected.

For testing, I'll open port 123 for the Tasmota devices and see if the problem disappears when running stock firmware.

Niels

NielsPiersma, 2022-08-03:
Well, removing the ntp settings from the DHCP pool did not work as expected.

Running 
backlog ntpserver1 0; ntpserver2 0; ntpserver3 0
did not result in a "default" list of ntp servers, but in an empty list.

I fetch the list of NTP servers from an other site of Tasmos and injectected the NTP-config

backlog NtpServer1 pool.ntp.org; NtpServer2 nl.pool.ntp.org; NtpServer3 0.nl.pool.ntp.org

However, after 21 minutes the devices rebooted. It is happening across multiple devices.
1 out of 4 d1 mini's showing this issue
1 sonoff RF433 bridge showing this issue
1 sonoff 4ch pro not showing this issue

Running several different 11.x and 12.x tasmo releases, but still not able to predict why this is happening.

Niels

barbudor, 2022-08-03:
`backlog ntpserver1 0; ntpserver2 0; ntpserver3 0backlog ntpserver1 0; ntpserver2 0; ntpserver3 0`
will not work. Tasmota do not accept NTP settings from DHCP
While this feature had been included in the past, it has been removed as being much more confusing for home users that do not know how to, or even can't configure the DHCP in their home router to provide this information.
Tasmota only support static NTP server configuration based either on a domain name (can be a local domain name not necesserally an Internet name) or a static IP value

 `backlog NtpServer1 pool.ntp.org; NtpServer2 nl.pool.ntp.org; NtpServer3 0.nl.pool.ntp.org`
This is valid, as far as your Tasmota has access to the Internet, at least port 123 as you listed
So you should see your Tasmota get the time.

Now what is reported in this Issue is a problem with some NTP servers sending multicast/manycast UDP packets.
Can you check if this is enabled on your NTP server and could be disabled ?

darkdragondraco, 2022-08-03:
> Now what is reported in this Issue is a problem with some NTP servers sending multicast/manycast UDP packets. Can you check if this is enabled on your NTP server and could be disabled ?

Hi @barbudor  I enabled multicast/manycast on my ntp server and Tasmota devices started to reboot. When i disabled multicast/manycast  on ntp reboots stopped.  the issue is that some device in network (in my case it was ntp) sending lots of multicast/manycast packets causing excessive heap usage in Tasmota. When heap goes to ~0 Tasmota reboots. 
You can monitor progress of this issue by checking heap value of the Tasmota device on its information web  page or in on of status/state requests in mqtt. if you have this issue heap will quickly shrink.
I plan to do some additional testing but first thing i suggest you to check - does your ntp really need multicast/manycast and disable it if not needed?

arendst, 2022-08-03:
I'm OoO. When I'm back I'll dive into the UDP multicast problem. Tasmota uses different UDP receive drivers so first issue is finding out which one is the culprit. I should be able to debug with either a mikrotik virtual router or a cheap hardware one.

NielsPiersma, 2022-08-03:
> `backlog ntpserver1 0; ntpserver2 0; ntpserver3 0backlog ntpserver1 0; ntpserver2 0; ntpserver3 0` will not work. Tasmota do not accept NTP settings from DHCP While this feature had been included in [...]

@barbudor, Unfortunately moving to " pool.ntp.org" servers and disabling the NTP server option from the DHCP server did not resolve the issue. Still spot-on 20 minutes a restart of the devices. Strangely it doesn't seem to be affecting all boxes, for now it is pinned to that network. I've got two other networks running tasmota devices (also windows dhcp servers) and don't show the issue. 

For testing purposes I am going to do the following.
-- assign fixed IP
-- assign static NTP servers

See if that does something.

NIels

NielsPiersma, 2022-08-03:
For testing purposes, I am going to do the following.
-- assign fixed IP
-- assign static NTP servers

See if that does something.
--> Yes, on the failing Tasmos I see the heap size dropping from 26 to 14 and rebooting. Roughly 1 per every two minutes.
On the Tasmo with the fixed IP address, the heap size stays at 26.

Not ideal, but for me, this works as a workaround for now.
Niels

barbudor, 2022-08-03:
@darkdragondraco I know, 
my message was addressed to @NielsPiersma who reported similar problem as yours
So the question "Does your NTP need multicast/manycast ?" is exactly what I asked @NielsPiersma 

@NielsPiersma 
So that's interresting, you found a configuration where the heap doesn't drop anymore and one where it does.
Would it means the problem is not actually related or not fully related to multicast/manycast ....

NielsPiersma, 2022-08-03:
Sorry
Clarification
Problem is gone (for all tasmos) when using fixed up addresses
Problems surfaces on 'some ' tasmos when set to DHCP .

Have not figured out the root cause.

Will update all tasmos to development version. So they all have the same firmware
Then I will enable DHCP again and see what is happening.

There must be some config causing this to happen.

I've over 100 tasmos running on various hardware but only at this site am I having this issue. It must be related to DHCP.

Also got some new d1 that work fine, so likely will move over the sensors and reprovision them.

Niels

NielsPiersma, 2022-08-04:
I've updated the Tamos (NOUS A1 devices) to 12.0.2.4 (sensors) and assigned DHCP addresses by issuing 
backlog ipaddress1 0.0.0.0; restart 1
I will let you know if there is any chance of the behavior.

TD-er, 2022-08-05:
Just an idea triggered by the word "multicast".
Do you have a switch which is (semi)manageable? (e.g. TP-link 108E, where the "E" is making all the difference, or one that's fully managed)
Those can be set to enable IGMP snooping.
If this can be place somewhere in-between the device sending the multicast packets and the access point which serves the ESP devices, then the switch can filter out unsolicited multicast packets.  The switch preferrable "close" to the access point, so no other devices connected to the same ethernet port can ask for multicast packets.

It is a known issue that ESPs don't handle multicast traffic too well.

Jason2866, 2022-08-05:
@TD-er  I have some experience regarding multicast traffic (my IP-TV provider does provide the streams via multicast). The only problem i have encountered when IGMP snooping does not work as it should is flooding the wlan with traffic. This ends in a horrible slow wlan. Every wlan device is terrible slow. BUT none of the Tasmota devices did crash. After removing the flooding traffic, all devices (including esp Tasmota) are working normal. I had no crashed devices. The uptimes in the management tool where still there. I see many wifi any mqtt reconnects in this time slot. Same problem as my mobiles and tablets do have when multicast flooding happens.
In short multicast can be a DoS, but Tasmota does not crash in this phase.
(To get IPTV well going i had to do some Hardware and Software upgrades...)

TD-er, 2022-08-05:
> In short multicast can be a DoS, but Tasmota does not crash in this phase.
> So it is not a general multicast/anycast issue, it has to be a very specific case.

I guess it highly depends on the amount of multicast packets (not traffic) as the IP stack in the ESP seems to keep track of packets until they are served or timeout.
Thus there are already 2 factors here:
- nr. of multicast packets per timeframe
- Amount of free memory to begin with

Your APs apparently slow down enormously due to multicast packets (multicast over WiFi is a terrible idea how it is currently implemented in WiFi 4 and 5 standards used in most popular APs)
So I guess this may also lead to more excessive memory usage in the IP stack as pending replies also take up memory.

NielsPiersma, 2022-08-05:
"I've updated the Tamos (NOUS A1 devices) to 12.0.2.4 (sensors) and assigned DHCP addresses by issuing
backlog ipaddress1 0.0.0.0; restart 1
I will let you know if there is any change of the behavior."

Nothing changed...
DHCP ones keep rebooting every 20 minutes.
When moving to fixed IP, reboots are gone.
However, one of my NOUS A1, running the same firmware as the other three, doesn't reboot.
I am quite puzzled, as I would have expected all of them. But this one didn't had the issue when it started to occur.

Quite strange, really looking for why DHCP ones are affected and fixed ones are not.

Niels

TD-er, 2022-08-05:
> Quite strange, really looking for why DHCP ones are affected and fixed ones are not.

Another idea might be that the NTP info in the DHCP differs from what the ESP may expect.
I have no idea how Tasmota does retrieve this info from the DHCP as I have not been able to get it in my project ESPEasy.

barbudor, 2022-08-05:
Tasmota doesn't retreive NTP server info from DHCP
I think this was already explained above

Only what you enter with NtpServer commands is taken into account where it is Ip@ or domain name

NielsPiersma, 2022-08-05:
> > Quite strange, really looking for why DHCP ones are affected and fixed ones are not.
> 
> Another idea might be that the NTP info in the DHCP differs from what the ESP may expect. I have no idea how Tasmota does retrieve this info from the DHCP as I have not been able to get it in my project ESPEasy.

TD-er,
I've removed offering NTP servers from the DHCP scope (as one of the first things to test). I did a Wireshark trace, and as far as I can tell, no NTP servers are included in the DHCP offer. And I agree with @barbudor, Tasmota (at this moment) doesn't seem to request / honor ntp server options handed out by the DHCP server (if any).

I still am trying to get my head around why it is only happening when the Tamo is in DHCP.
Niels

NielsPiersma, 2022-08-05:
I've done a Wireshark and noticed that DHCP Option 42 (NTP Server) is being sent out by the Tasmota. From @barbudor   "Tasmota do not accept NTP settings from DHCP" does seem conflicting.  It could be it is requesting it, but not using it of course.

wireshark trace.
[dhcp.zip](https://github.com/arendst/Tasmota/files/9268674/dhcp.zip)

Little extra info; our core switch is relaying the DHCP request. 

Niels

Jason2866, 2022-08-05:
DHCP Option 42 sent out is a left over from the time Tasmota used NTP from DHCP.
@arendst can you remove?

arendst, 2022-08-05:
> @arendst can you remove?

It's in the core code. Not in Tasmota. Tasmota has no DHCP code. It relies on the core DHCP code for IP address only.

arendst, 2022-08-06:
@darkdragondraco I recevied my MikroTik AP and try to simulate the NTP multicast/manycast situation.

Using broadcast address 192.168.2.255 (on my segment 192.168.2.0) I see no issues.

What was/is your NTP broadcast address?

darkdragondraco, 2022-08-06:
> What was/is your NTP broadcast address?

Actually broadcast address was not configured in my setup, i was using it in unicast mode. I've enabled broadcast mode with 192.168.2.255 (also have 192.168.2.0 network here) and looks like it is working OK. 
I currently have about 60 clients on this network and about 20 of them are tasmota devices
Also i've managed to get my hands on ESP32 with tasmota and it has no issues.  Tasmota version is 12.0.2(bluetooth)

darkdragondraco, 2022-08-07:
After few hours problem reappeared again, moved all  tasmotas to another VLAN, dedicated to tasmotas. Continue to monitor situation

NielsPiersma, 2022-08-07:
> After few hours problem reappeared again, moved all tasmotas to another VLAN, dedicated to tasmotas. Continue to monitor situation

My Tamotas are in a dedicated VLAN by design and have been for over 3 years. The VLAN only houses ESP8266 based devices running Tasmota, 1 raspberry Pi, and the router/gateway (this being a Cisco 3750 switch).

Still, I have the same issues when the Tasmotas are on DHCP. The reboot is predictable, every 20 minutes. No NTP server is offered by DHCP. I am thinking about capturing all data from the VLAN for 25 minutes; that may shed some light on what is going on.

Niels

darkdragondraco, 2022-08-07:
@NielsPiersma
What are the services hosted on pi? As i also have several pi's in network. Moving to other lan did not completely help, some devices still reboot. Disabled NTP server. Monitoring for now.

Jason2866, 2022-08-07:
A Pi in the network does no harm in general. I have 3 Pis running together in the same network with all my Tasmota devices.

NielsPiersma, 2022-08-07:
The pi is running flightaware so just port ,80 open and posting to flightradar.

I am not expecting this being any issue. Just want to be so complete as possible

rayzilt, 2022-08-16:
I like to jump into this thread as that I experience the same problems. Got a couple of Sonoff devices but also Shelly's.
They all reboot every 20 minutes. Devices where running version 8.x - 10.x. I upgraded them all to version 12.0.2 yesterday but that didn't solve the problem for me.

The devices connect to Unifi AP's

There is a major change in my vlan, and I'm guessing that this triggered the situation. I moved the router IP from a Juniper switch to VyOs router. 

Still troubleshooting.

NielsPiersma, 2022-08-16:
Unifi aps here as well .

However one site is experiencing this issue. The other unifi AP site is not.i will compare settings.

Niels

NielsPiersma, 2022-08-16:
Update,
Besides the SSID and the WPA password, the site settings are identical.
(IGMP, etc)

Still didn't find time to do a full Wireshark capture. Hopefully will get
it sorted out later this week.

NIels

rayzilt, 2022-08-16:
I'm positive that I found the issue on my network what was causing all devices to reboot every ~20 minutes.
I noticed that the reboots happen due to "Free Memory", slowly the free memory was eating away.
I verified the dhcp server settings, the option ntp-server was set for this range, but was not reachable due to firewall restrictions between vlans.

I removed the ntp-server option and all devices are running longer than 2 hours now. They are allowed to query the time on the internet.

This thread helped me to find the issue. Thanks

arendst, 2022-08-17:
@rayzilt could you do a test by changing the below code in file ``support_wifi.ino`` from
```
void WifiConnect(void)
{
  if (!Settings->flag4.network_wifi) { return; }

  WifiSetState(0);
  WifiSetOutputPower();
  WiFi.persistent(false);     // Solve possible wifi init errors
  Wifi.status = 0;

```
to
```
#ifdef ESP8266
#include <sntp.h>                       // sntp_servermode_dhcp()
#endif  // ESP8266

void WifiConnect(void)
{
  if (!Settings->flag4.network_wifi) { return; }

  WifiSetState(0);
  WifiSetOutputPower();

#ifdef ESP8266
  // https://github.com/arendst/Tasmota/issues/16061#issuecomment-1216970170
  sntp_servermode_dhcp(0);
#endif  // ESP8266

  WiFi.persistent(false);     // Solve possible wifi init errors
  Wifi.status = 0;

```
to see if it stops heap degrading.

rayzilt, 2022-08-17:
I'll test later this day when I get home from work

rayzilt, 2022-08-18:
@arendst An update on your patch request.

Before trying your patch I wanted to have the same situation. Unfortunately I'm unable to regain the same situation by restoring the old settings that I had when all devices where running out of memory. I've had it running for a day now.

The setting that I restored are:
- DHCP ntp-server set to a internal IP that is not reachable;
- DNS server set to a different IP that is not reachable.

I'm quite puzzled. I'll let this faulty configuration running for a time being.

arendst, 2022-09-13:
Keep active

NielsPiersma, 2022-10-11:
That is what I am seeing as well. Not all devices in my network are affected. Some work as expected.
Niels

closed 2022-11-10 by github-actions[bot]

kristoficko, 2024-01-10:
Hello,

I have moved all my ESP devices from Tuya to Tasmota. At this moment I have 5 devices running and all of them are restarting every 20min due to the free memory being exhausted throwing Exception 29.

I was following this thread, couple weeks ago realize I had a container with NTP running, disabled and removed the container and the devices stopped restarting.

Unfortunately after 10 days or so the issue re-appeared and I can't seem to figure out what the root cause is. 

My wifi is provided by multiple Mikrotik APs. 

Analyzing network traffic I can see the Tasmota devices are connecting to the NTP server specified in DHCP option 42...

Kindly requesting help.

Mike

NielsPiersma, 2024-01-10:
Mike,

Just for eliminating the obvious, can you pls configure a static IP address and configure the ntp server address manually and retest.

If the issue persists it is not related to the DHCp issue that was resolved last year .

Kind regards
Niels

kristoficko, 2024-01-10:
Hi Niels,

many thanks for the suggestion, I did setup static address on all 5 devices. Although it's a bit early to tell, I can already see the following improvement:

1. Heap is not decreasing constantly but rather oscillating between 21-23
2. all devices have already passed an uptime longer than 20min

It doesn't seem to make a difference whether I setup my own internal NTP server or leave the 3 default ones.

One more observation, which may be relevant to this issue -  there seems to be a massive difference in the frequency of NTP requests:

- with DHCP enabled on Tasmota devices I can see NTP communication passing through the FW multiple times per minute (2-3x)
- with static IP NTP was contacted once after reboot and I didn't see any request since

Kind regards

Mike

kristoficko, 2024-01-29:
Hi Jason, I did a few tests, exporting the DHCP traffik from my Mikrotik router into wireshark and I beg to differ. This is what I captured:

1. DHCP Discover from tasmota device contains in DHCP Option 55 (Parameter Request List) also DHCP Option 42 (Network Time Procotocol Servers)
2. Mikrotik sends DHCP 42 Option in the DHCP Offer
3. DHCP Request from tasmota again contains Option 42 inside option 55
4. DHCP ACK from the router confirms previously offered NTP servers in Option 42
5. Next Tasmota sends multiple requests to NTP servers - one to the server offered by my router in option 42, next to one of the public NTP servers that come preconfigured in the firmware.

This confirms that tasmota not only actively requests option 42 from the router, it also actively uses the NTP server offered from DHCP,
[image]

xdanik, 2025-04-19:
Hint for anybody debugging this issue with Mikrotik router - make sure the NTP Server on Mikrotik is really working correctly. Enabling NTP Server is not enought!

You also have to allow and configure NTP Client because Mikrotik's NTP Server will refuse to provide clock without it (as the "Use Local Clock" option is disabled by default). It's quite confusing as the router may have correct time set and it will respond to the NTP requests but it will not provide time... It took me several evenings of debugging to realise this.

Also note that it seems that the ESP8266 will try to use the default gateway as NTP Server regardless of DHCP config or servers configured in Tasmota. I was seeing many NTP requests heading from the Sonoff to the router even when DHCP was not configured to provide NTP server. Not sure why. ESP32 are not affected.

sfromis, 2025-04-19:
Tasmota is not using the router DHCP configuration to get NTP servers, but only the internal configuration of NTP servers, including default values.

xdanik, 2025-04-19:
> Tasmota is not using the router DHCP configuration to get NTP servers, but only the internal configuration of NTP servers, including default values.

I do not dispute that. But something in the ESP stack seems to use the DHCP NTP as I clearly saw many NTP requests to the router every few seconds in the network dump.

Tasmota was configured with the default NTP servers:
```
{"NtpServer1":"2.pool.ntp.org","NtpServer2":"2.europe.pool.ntp.org","NtpServer3":"2.nl.pool.ntp.org"}
```

heikok88, 2025-04-29:
I can confirm what @xdanik  said:
I discovered the exception 29 problem today. 10 out of 10 units have this problem. I was running 14.4.0 firmware, so I upgraded to 14.6.0 stable, but the problem was still there.
After rebooting all devices at the same time, the first devices rebooted after 32 minutes due to the exception and the last devices rebooted after 58 minutes.
After disabling the NTP option in my DHCP server (pi-hole), all devices are working fine, at the moment for 2 hours.

[3 more replies]