arendst/Tasmota #16070: Events Http#Initialized, wifi#connected, mqtt#connected, Power1#Boot, Time#Initialized, Time#Minute don't trigger

achim42, 2022-07-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
None of the events Http#Initialized, wifi#connected, mqtt#connected, Power1#Boot, Time#Initialized, Time#Minute triggers

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff CH4 Pro
- [X] Tasmota binary firmware version number used: 9.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: esptool.py
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: stat/tasmota_EAF5C4/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:stat/tasmota_EAF5C4/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":231,"Free":280,"Rules":"ON Http#Initialized  DO Var4 4 ENDON   ON wifi#connected    DO Var4 4 ENDON   ON mqtt#connected    DO Var4 4 ENDON   ON Power1#Boot       DO Var4 4 ENDON   ON Time#Initialized  DO Var4 5 ENDON   ON Time#Minute       DO Var5 1 ENDON"}}
22:32:02.503 SRC: Backlog
22:32:02.506 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
22:32:02.512 MQT: stat/tasmota_EAF5C4/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
22:32:02.753 SRC: Backlog
22:32:02.756 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
22:32:02.763 MQT: stat/tasmota_EAF5C4/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:MQT: stat/tasmota_EAF5C4/STATUS0 = {"Status":{"Module":23,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota_EAF5C4","ButtonTopic":"0","Power":7,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:05:17","StartupUTC":"2022-07-24T20:27:46","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2022-07-24T20:29:37","SaveCount":29,"SaveAddress":"F7000"},"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021.06.17 08:28:30","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"366/699"},"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":3,"SysLog":3,"LogHost":"","LogPort":514,"SSId":["ImpTest",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]},"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"tasmota_EAF5C4-5572","IPAddress":"192.168.4.180","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"C4:4F:33:EA:F5:C4","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.4.142","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EAF5C4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-07-24T21:33:03","Local":"2022-07-24T22:33:03","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:13","Sunset":"20:39"},"StatusSNS":{"Time":"2022-07-24T22:33:03"},"StatusSTS":{"Time":"2022-07-24T22:33:03","Uptime":"0T01:05:17","UptimeSec":3917,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"OFF","Wifi":{"AP":1,"SSId":"ImpTest","BSSId":"44:4E:6D:1A:26:30","Channel":1,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8285
00:00:00.050 CFG: aus Flash geladen am F8, Anzahl 28
00:00:00.056 QPC: Count 1
00:00:00.058 CFG: CR 366/699, Busy 0
00:00:00.062 SRC: Restart
00:00:00.064 Projekt tasmota Tasmota Version 9.5.0(tasmota)-2_7_4_9(2021.06.17 08:28:30)
00:00:00.216 WIF: Prüfe Verbindung...
00:00:00.216 WIF: Verbindungsversuch...
00:00:00.552 WIF: verbinden mit AP1 ImpTest Channel 1 BSSId 44:4E:6D:2B:2B:2B in Modus 11n als tasmota_EAF5C4-5572...
00:00:01.752 WIF: Prüfe Verbindung...
00:00:01.753 WIF: verbunden
00:00:02.004 HTP: Web-Server aktiv bei tasmota_EAF5C4-5572 mit IP-Adresse 192.168.xxx.180
00:00:02.109 HTP: Hauptmenü
00:00:02.490 RTC: UTC 2022-07-24T20:27:50, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
21:27:51.027 MQT: Verbindungsversuch...
21:27:51.122 MQT: verbunden
21:27:51.125 MQT: tele/tasmota_EAF5C4/LWT = Online (beibehalten)
21:27:51.127 MQT: cmnd/tasmota_EAF5C4/POWER =
21:27:51.129 MQT: abonniere cmnd/tasmota_EAF5C4/#
21:27:51.132 MQT: abonniere cmnd/tasmotas/#
21:27:51.133 MQT: abonniere cmnd/DVES_EAF5C4_fb/#
21:27:51.135 MQT: löse abo. von homeassistant/status
21:27:51.141 MQT: tele/tasmota_EAF5C4/INFO1 = {"Info1":{"Module":"Sonoff 4CH Pro","Version":"9.5.0(tasmota)","FallbackTopic":"cmnd/DVES_EAF5C4_fb/","GroupTopic":"cmnd/tasmotas/"}}
21:27:51.156 MQT: tele/tasmota_EAF5C4/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_EAF5C4-5572","IPAddress":"192.168.xxx.180"}}
21:27:51.167 MQT: tele/tasmota_EAF5C4/INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
21:27:51.174 MQT: stat/tasmota_EAF5C4/RESULT = {"POWER1":"ON"}
21:27:51.176 MQT: stat/tasmota_EAF5C4/POWER1 = ON
21:27:51.179 MQT: stat/tasmota_EAF5C4/RESULT = {"POWER2":"ON"}
21:27:51.185 MQT: stat/tasmota_EAF5C4/POWER2 = ON
[... 28 lines cut ...]
_Steps to reproduce the behavior:_
Rule1
ON Http#Initialized DO Var4 4 ENDON
ON wifi#connected DO Var4 4 ENDON
ON mqtt#connected DO Var4 4 ENDON
ON Power1#Boot DO Var4 4 ENDON
ON Time#Initialized DO Var4 5 ENDON
ON Time#Minute DO Var5 1 ENDON
Rule1 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Except for events of the type "ON Power1#state=1" (it works), no event is triggered.
In the log is no line with a "RUL" execution.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Apart from these simple commands, variants with delays or timers do not work either.
Variants with defined ruletimerX work - as soon as they are started in the console or via a Power1#state=1 trigger. But never via the events listed at the beginning.
By the way: event#myPing is also not triggered.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-07-24:
Too many spaces in your rules, especially after trigger before DO
This is not supported by v9.5.0

sfromis, 2022-07-24:
I tested the unchanged rule set with latest Tasmota. They all worked, even with the spaces which may cause problems with older versions.

achim42, 2022-07-25:
Hi barbudor
Thank you very much for the tip about the blanks. That was exactly the problem. Problem solved !

closed 2022-07-25 by achim42

barbudor, 2022-07-25:
I remember adding space triming on triggers some months back but clearly after 9.5.0