arendst/Tasmota #16077: SetOption 19 1 Tasmota 12

f0rc3r, 2022-07-25:
### PROBLEM DESCRIPTION
SetOption19 1seems not Working after Tasmota 12

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): yes and Shelly 1
- [ X] Tasmota binary firmware version number used: 11 and 12
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: Tasmotizer and OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-07-25:
Working as intended. 

USE_HOME_ASSISTANT has been removed from standard binaries in favor of TASMOTA_DISCOVERY
You are strongly advised to move to the HATASMOTA plug-in as the old way is in way of absolescence

If you absolutely want to use the old and deprecated way you'll have to compile yourself.

jeroenst, 2022-07-25:
More Info:

https://tasmota.github.io/docs/Home-Assistant/
https://www.home-assistant.io/integrations/tasmota/

closed 2022-07-25 by Jason2866