arendst/Tasmota #16085: Can't set host for syslog logging

gajes9, 2022-07-26:
### PROBLEM DESCRIPTION
I can't set host for syslog, both from WebGui and console.
When I try to set host using command: LogHost 192.168.5.179 
I receives the response:
11:21:31.148 CMD: LogHost 192.168.5.179
11:21:31.150 SRC: WebConsole from 192.168.5.50
11:21:31.153 CMD: Grp 0, Cmd 'LOGHOST', Idx 1, Len 13, Pld 192, Data '192.168.5.179'
11:21:31.156 CFG: Text overflow by 6 char(s)
11:21:31.160 MQT: stat/sterownik_nawadniania1/RESULT = {"LogHost":""}

Propably the problem is caused by too long ip adress, because when i issue command : LogHost 3.2.3.4 
I recives the response:
11:23:32.719 CMD: LogHost 3.2.3.4
11:23:32.721 SRC: WebConsole from 192.168.5.50
11:23:32.724 CMD: Grp 0, Cmd 'LOGHOST', Idx 1, Len 7, Pld 3, Data '3.2.3.4'
11:23:32.727 CFG: CR 699/699, Busy 0
11:23:32.731 MQT: stat/sterownik_nawadniania1/RESULT = {"LogHost":"3.2.3.4"}
11:23:33.655 CFG: Saved to flash at F5, Count 447, Bytes 4096

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pro R3 and Sonoff Pro R2
- [ ] Tasmota binary firmware version number used: 12.02
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Issue:
LogHost 192.168.5.179

### EXPECTED BEHAVIOUR
Saved host adress

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-07-26:
Tasmota stores certain values in a 699 bytes text buffer. Apparently you have used up this buffer with buttonnames, friendlynames, ntpservers etc.

So change/remove one of those to make room for your loghost.

BTW "normal" buffer fill is about 450/699 (See command ``status 2``, item CR): 
```
11:48:43.443-021 MQT: stat/wemos4/STATUS2 = {"StatusFWR":{"Version":"12.0.2.4(tasmota)","BuildDateTime":"2022-07-24T15:43:34","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"417/699"}}
``

gajes9, 2022-07-26:
You are right. The reason was no more space in buffer. When I reduce names of my dummys WebButtons, I can set host of syslog to proper value. Thanks a lot for the clarification and for your help.

closed 2022-07-26 by gajes9