arendst/Tasmota #16095: Stuck in Reset Loop after flashing Sonoff POW Elite - POWR316D

DefCon-CC, 2022-07-27:
### PROBLEM DESCRIPTION
After flashing development release of tasmota for ESP32 I'm stuck in endless reset. Switch to FBL and reflashing is possible via pressing Button. Output from UART:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x00000410
invalid header: 0x00000410
invalid header: 0x00000410
invalid header: 0x00000410
invalid header: 0x00000410
invalid header: 0x00000410
invalid header: 0x00000410
ets Jul 29 2019 12:21:46

I've already tried a more stable power supply but it did not help. Searching in the internet also pointed me to no direct conclusion.
I'm not sure if this reset happens in FBL or Application (Tasmota)

Also tried images mentioned in https://github.com/arendst/Tasmota/issues/15856

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW Elite - POWR316D
- [x] Tasmota binary firmware version number used: latest development of tasmota for ESP32
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESPEasyFlasher-Tool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I've just flashed latest development release of tasmota for ESP32 on Sonoff POW Elite - POWR316D

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Sonoff POW Elite starts up using tasmota

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-07-27:
ESP32 not flashed correctly
Please strictly follow the instructions with a supported method
https://tasmota.github.io/docs/ESP32/#flashing

DefCon-CC, 2022-07-27:
@barbudor Yes you were right. Flashing using [ESP_Flasher](https://github.com/Jason2866/ESP_Flasher/releases) solved the problem. 

Can you say something why it is so important to use the correct flash tool?

closed 2022-07-27 by DefCon-CC

barbudor, 2022-07-27:
There are some flashing parameters that need to be set correctly
ESP32 binaries are also made from multiple parts that needs to be flashed individually (bootloader, partition, app,..) and the easiest is to flash the full flash image '.factory.bin' which includes them all.
Tasmota documentation addresses those.

DefCon-CC, 2022-07-27:
Ahh ok. Thanks for the explanation.

gampam2000, 2024-01-25:
I Had a similar issue, you also need to power it with the 230V mains after flashing. Disconnect the serial and close the case before you do this. It also takes very long after connecting to the AP until the captive portal is shown.