arendst/Tasmota #16118: Sonoff POWR320D 20A EnergyToday value not restored after graceful restart

ejohb, 2022-07-30:
### PROBLEM DESCRIPTION
EnergyToday resets to 0 after power is lost, even after a graceful `restart 1` to save the value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POWR320D 20A
- [x] Tasmota binary firmware version number used: 12.0.2.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web Installer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
13:23:17.026 MQT: stat/immersion-heater-lower/RESULT = {"NAME":"Sonoff POWR320D","GPIO":[32,0,9313,0,9312,576,0,0,0,0,9280,0,3104,0,320,0,0,0,0,0,0,9184,9248,9216,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
13:23:17.275 MQT: stat/immersion-heater-lower/RESULT = {"Module":{"0":"Sonoff POWR320D"}}
13:23:17.536 MQT: stat/immersion-heater-lower/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"9313":"Relay_b2"},"GPIO3":{"0":"None"},"GPIO4":{"9312":"Relay_b1"},"GPIO5":{"576":"LedLink_i"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"9280":"TM1621 DAT"},"GPIO15":{"0":"None"},"GPIO16":{"3104":"CSE7766 Rx"},"GPIO17":{"0":"None"},"GPIO18":{"320":"Led_i1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"9184":"TM1621 CS"},"GPIO26":{"9248":"TM1621 RD"},"GPIO27":{"9216":"TM1621 WR"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
--N/A
```
- [x] Provide the output of this command: `Status 0`:
```lua
13:24:26.643 MQT: stat/immersion-heater-lower/STATUS = {"Status":{"Module":0,"DeviceName":"Immersion Heater Lower","FriendlyName":["Immersion Heater Lower"],"Topic":"immersion-heater-lower","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
13:24:26.686 MQT: stat/immersion-heater-lower/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:09:44","StartupUTC":"2022-07-30T12:14:42","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2022-07-23T13:03:31","SaveCount":104}}
13:24:26.730 MQT: stat/immersion-heater-lower/STATUS2 = {"StatusFWR":{"Version":"12.0.2.4(tasmota)","BuildDateTime":"2022-07-22T13:15:48","Core":"2_0_4","SDK":"v4.4.2","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"494/699"}}
13:24:26.752 MQT: stat/immersion-heater-lower/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["oac-not-2.4",""],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008029","2805C80001000680003C5A0A192800000000","000000A8","00006000","00004000","00000000"]}}
13:24:26.800 MQT: stat/immersion-heater-lower/STATUS4 = {"StatusMEM":{"ProgramSize":1407,"Free":1472,"Heap":148,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
13:24:26.865 MQT: stat/immersion-heater-lower/STATUS5 = {"StatusNET":{"Hostname":"immersion-heater-lower","IPAddress":"REDACTED","Gateway":"REDACTED","Subnetmask":"255.0.0.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"REDACTED","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:24:26.911 MQT: stat/immersion-heater-lower/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.2","MqttPort":1883,"MqttClientMask":"immersion-heater-lower","MqttClient":"immersion-heater-lower","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:24:26.944 MQT: stat/immersion-heater-lower/STATUS7 = {"StatusTIM":{"UTC":"2022-07-30T12:24:26","Local":"2022-07-30T13:24:26","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:21","Sunset":"20:31"}}
13:24:26.976 MQT: stat/immersion-heater-lower/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:24:27.010 MQT: stat/immersion-heater-lower/STATUS10 = {"StatusSNS":{"Time":"2022-07-30T13:24:26","ENERGY":{"TotalStartTime":"2022-07-23T13:03:31","Total":5.098,"Yesterday":0.178,"Today":0.490,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000},"ESP32":{"Temperature":47.8},"TempUnit":"C"}}
13:24:27.047 MQT: stat/immersion-heater-lower/STATUS11 = {"StatusSTS":{"Time":"2022-07-30T13:24:27","Uptime":"0T00:09:45","UptimeSec":585,"Heap":150,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":45},"POWER":"OFF","Wifi":{"AP":1,"SSId":"REDACTED","BSSId":"REDACTED","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
--N/A
```

### TO REPRODUCE
Run the following commands:

`energytoday`: `13:25:26.054 MQT: stat/immersion-heater-lower/RESULT = {"EnergyToday":{"Total":5.098,"Yesterday":0.178,"Today":0.490}}`
`restart 1`

Check value was retained:
`energytoday`: `13:25:26.054 MQT: stat/immersion-heater-lower/RESULT = {"EnergyToday":{"Total":5.098,"Yesterday":0.178,"Today":0.490}}`

Now kill power and check again:
`energytoday`: `13:27:43.622 MQT: stat/immersion-heater-lower/RESULT = {"EnergyToday":{"Total":5.098,"Yesterday":0.178,"Today":0.000}}`

Note: This process (i.e. restarting to save the value before killing power) seems to work for other energy-monitoring devices (e.g. Shelly EM) as the EnergyToday is restored after the power loss.

### EXPECTED BEHAVIOUR
EnergyToday should be retained.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-04:
Verified to fail. Investigating...

arendst, 2022-08-05:
For a better graceful power cycle execute command ``restart 2``. This will save changes and halt the device waiting for you to power cycle.

ascillato2, 2022-08-06:
Closing as the issue has been fixed by Theo. Thanks !

closed 2022-08-06 by ascillato2