arendst/Tasmota #16135: ESP32-CAM in 12.0.2:  AEC target level can't be controlled & WcAECLevel command is unknown

mrneutron42, 2022-08-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

For sunlit scenes, the automatic exposure (AEC) level that the camera seeks is too long and thus the image is too bright.
After reading the command names from this file:
[https://github.com/arendst/Tasmota/blob/master/tasmota/tasmota_xdrv_driver/xdrv_81_esp32_webcam.ino]
I tried to adjust the target AEC value using the "WcAECValue" command - no effect on image brightness.
I tried to adjust the brightness with "WcAECLevel" command - the command is unknown.
```
10:20:06.517 CMD: wcaeclevel
10:20:06.522 RSL: RESULT = {"Command":"Unknown"}
```
I tried "WcAELevel" command, and that is a valid command, but has no effect on image brightness.
```
10:20:13.918 CMD: wcaelevel
10:20:13.923 RSL: RESULT = {"WCAELevel":-2}

```
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-CAM from AI THINKER
- [x] Tasmota binary firmware version number used: 12.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used:  flash download tool 3.8.8
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

10:36:14.842 CMD: Backlog Template; Module; GPIO 255
10:36:14.861 RSL: RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,1,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}
10:36:15.091 RSL: RESULT = {"Module":{"0":"AITHINKER CAM"}}
```

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:37:28.571 CMD: Status 0
10:37:28.575 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ESP32-CAM1","FriendlyName":["ESP32-CAM1"],"Topic":"ESP32-CAM","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:37:28.584 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"RTC Watch dog reset digital core and rtc module","Uptime":"29T19:20:10","StartupUTC":"2022-07-03T20:17:18","Sleep":50,"CfgHolder":4617,"BootCount":123,"BCResetTime":"2021-07-13T16:53:45","SaveCount":479}}
10:37:28.589 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2(webcam)","BuildDateTime":"2022-06-20T12:42:51","Core":"2_0_3","SDK":"v4.4.1","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"392/699"}}
10:37:28.594 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["*****","*****"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000","00000000"]}}
10:37:28.605 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1242,"Free":1856,"Heap":117,"StackLowMark":2,"PsrMax":4093,"PsrFree":3312,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F9005CA","00040001","00000082","00000000","40000981","80000080","00200000","14000008","00000000"],"Drivers":"1,2,4,7,8,9,10,12,24,27,29,38,50,52,62,81","Sensors":"2,6,127"}}
10:37:28.610 RSL: STATUS5 = {"StatusNET":{"Hostname":"ESP32-CAM1","IPAddress":"192.168.8.71","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.8.1","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:C1:EF:40","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:37:28.614 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C1EF40","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:37:28.619 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-08-02T15:37:28","Local":"2022-08-02T10:37:28","StartDST":"2022-03-13T02:00:00","EndDST":"2022-11-06T02:00:00","Timezone":99,"Sunrise":"06:03","Sunset":"20:44"}}
10:37:28.628 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-08-02T10:37:28","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
10:37:28.637 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-08-02T10:37:28","Uptime":"29T19:20:10","UptimeSec":2575210,"Heap":117,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":2,"Objects":30},"POWER":"OFF","Dimmer":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"LOONLINK","BSSId":"10:DA:43:8C:40:A6","Channel":8,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":3,"Downtime":"0T00:02:29"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

In Command Window:
make sure AEC is turned on with the command "WcAEC 1"
enter WcAECValue command with various values from 1-1024 : example, "WcAECValue 10"
enter WcAECLevel command with various values from -2 to +2 : example, "WcAELevel -2"  (see that command is unknown)
enter WcAELevel command with various values from -2 to +2 : example, "WcAELevel -2"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect to see a change in the brightness of the camera image when sending different values for WcAECValue and WcAECLevel (WcAELevel).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

philrich, 2022-08-09:
> @philrich Can you have a look? You added this features in a PR. THX!

I am currently on vacation but will have a look in 2 days.

mrneutron42, 2022-08-30:
I was looking at the OV2640 datasheet, top of page 26.
[Download OV2640 Datasheet](https://datasheetspdf.com/pdf-file/548838/OmniVisionTechnologies/OV2640/1)
In the Table 13 registers, register 0x24 is called AEW.  It's value sets the frame brightness level that triggers the AGC to reduce the exposure time.  The data sheet says before you write to register 0x24, you must first write 0x01 to register 0xFF.
I believe there needs to be a register write to register 0x24 in order to set this target brightness level for the AGC.
I have searched through the Tasmota source code files and have not found any references to AEW or writes to register 0x24.
I believe a new variable need to be created to store this AEW value and write it to the OV2640's AEW register

mrneutron42, 2022-10-11:
From what I saw in the source code file (https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_81_esp32_webcam.ino), the command name AECValue does not represent what it actually controls.
The function of the AECValue command sets the MANUAL exposure value of the image sensor.
This value only takes effect than AECMode is 0 (disabled) and the camera module is in Manual Exposure Mode.
I suggest renaming the command to MECValue, to reflect the fact that it is for manual exposure mode control.

Further down in the same file, I see a code comment confirming that the variable .aec_value is actually a manual exposure control.
`Settings->webcam_config2.aec_value = 204; // manual exposure value`
Since this variable controls a manual exposure value, it would be more intuitive to rename it to mec_value.

Jason2866, 2022-10-13:
@mrneutron42 Does it work with the settings you found out? If, yes could you provide a PR?

mrneutron42, 2022-10-17:
@Jason2866, I don't think I'm the right one to change the code.  I don't have a code compiling environment set up.  I'm primarily a hardware engineer (circuit design, PCB layout) and have been having a hard time following the control flow of the code.  I'm not even sure of all the files to look in to get this changed.  It would be better if someone more familiar with this code make the necessary changes.

closed 2022-11-16 by github-actions[bot]

sfromis, 2024-03-25:
What is valid is the spelling `WcAELevel` already mentioned in the first post.

mrneutron42, 2024-03-25:
1. "WcAECLevel" is an unknown command.
2.  "WcAELevel" is a valid command.
3.  Sending new values with the "WcAELevel" command has no effect on the image brightness.

sfromis, 2024-03-25:
I saw a clear difference with `WcAELevel` values 0/1/2. Test case was the camera pointed at a lamp, with significant difference in the captured glare around the lamp.

mrneutron42, 2024-03-25:
With Tasmota 13.4.0 (webcam), I do not see any image brightness difference between WcAELevel -2, 0, or 2.
What is the output when you enter the wc command in the console?

sfromis, 2024-03-25:
```
{"WC":{"Stream":1,"Resolution":8,"Mirror":0,"Flip":0,"Saturation":0,"Brightness":0,"Contrast":0,"SpecialEffect":0,"AWB":1,"WBMode":0,"AWBGain":1,"AEC":1,"AECValue":204,"AELevel":2,"AECDSP":1,"AGC":1,"AGCGain":5,"GainCeiling":0,"GammaCorrect":1,"LensCorrect":1,"WPC":1,"DCW":1,"BPC":0,"Colorbar":0,"Feature":0,"Auth":0,"Rtsp":0}}
```

mrneutron42, 2024-03-25:
My expectation is that the AELevel variable is setting the target auto exposure level of the camrea (the target brightness).
Thus, we should expect the brightness of the whole frame to change when the AELevel numbers are changed.

My brightness and exposure settings are mostly the same as yours, except that you have an AGCgain of 5, and mine is 0.
Your resolution is 8 and mine is 12, but that should be irrelivant to frame brightness.
{"WC":{"Stream":1,"Resolution":12,"Mirror":0,"Flip":0,"Saturation":0,"Brightness":-2,"Contrast":0,"SpecialEffect":0,"AWB":1,"WBMode":0,"AWBGain":1,"AEC":1,"AECValue":204,"AELevel":-2,"AECDSP":1,"AGC":1,"AGCGain":0,"GainCeiling":0,"GammaCorrect":1,"LensCorrect":1,"WPC":1,"DCW":1,"BPC":0,"Colorbar":0,"Feature":0,"Auth":0,"Rtsp":0}}

sfromis, 2024-03-25:
Again, did you try my *specific* test case?

And I disagree that you should always see the whole frame change.

mrneutron42, 2024-03-25:
My brightness and exposure settings are mostly the same as yours, except that you have an AGCgain of 5, and mine is 0.
Your resolution is 8 and mine is 12, but that should be irrelivant to frame brightness.

The target automatic exposure value sets the exposure (time) of the frame, which controls how bright the image is.
That is how cameras work.

sfromis, 2024-03-25:
Well, I saw very clear differences, meaning that there can be an effect. 0/1/2:
[image]

mrneutron42, 2024-03-26:
Thank you for posting the pictures @sfromis.  
Now, I can see that the adjustment is VERY little and it's only makes a difference with unusually bright objects.

My use of the ESP32-CAMS is for my Weewx weather station.  
I have several cameras in several windows grabbing images of the current weather.  
The default ESP32-CAM Automatic Exposure levels are brighter than I want.
I want to be able to set the target average brightness of the scene. 
The current amount of Automatic Exposure adjustment is NOT useful for normal images of the weather!  
I need MORE adjustment range!

mrneutron42, 2024-06-04:
I made a discovery!  
The AECDSP control is key to allowing the AECLevel variable to be effective.

When AECDSP is active (WcAECDSP set to 1), the AECLevel variable is mostly ignored or counteracted!
With sunlit outdoor scenes, the image is overexposed and has washed out colors levels when AECDSP is on.
Turning off AECDSP (WcAECDSP set to 0) allows AECLevel to actually control the automatic exposure target level!
The same scene is not overexposed and has good color saturation when AECDSP is off.

joysfera, 2025-03-28:
@mrneutron42 just wanted to thank you for the `WcAECDSP 0` discovery. It really saved my camera on sunny days.

mrneutron42, 2025-04-01:
@joysfera I'm glad this info helped you!
I just shared it in a more prominent place, so that more people might see it.
https://github.com/arendst/Tasmota/discussions/23222