arendst/Tasmota #16136: Device unreachable after mqtt reconnect fails

reimerp, 2022-08-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

A very strange problem occurs on an ESP32-U4WDH rev.3 based device -- possible even not tasmota is the root cause, but some esp32 libraries / TCP stack / or even the mqtt server (mosquitto)
After a while, I noticed some missing MQTT events from it, and after digging into this (using syslog 2) I noticed that the device lost mqtt connection. "Connect failed to broker, rc -5"
This seems to be a DNS related problem, so to rule that out I used IP addresses instead of names in configuration for
MqttHost / NtpServer1 / LogHost.
The problem occures again, but the error changes to -2, indicating a MQTT connection problem.
Having syslog in place, I find this is still functioning ang logging unsuccessfull mqtt attempts, as well as other results from that device (bluetooth beacons).
The mqtt server (mosquitto version 2.0.11, configured to nearly default) is running fine.
The IP Stack on this device sems not reacting on icmp (ping) messages (or with ~90% packet loss)  while this happens.
WLAN is running fine, and the device is showing best signal when its operationable.
I fired up an tcpdump on the server to capture the mqtt messages, I can see that the device sends regulary MQTT ping messages every 30 seconds which are responded from the server.
From time to time the server is re-sending packages and this seems to completely confuses the device. I don't know why this happens, the resent should occur when no ACK is received, but there are only ~500ms between the resent. The device has only ACKS previous packages, not the duplicate ones and then there is this mqtt lost connection & trying, and maybe the device recoveres, but I have also seen cases where it was completely lost for hours (mqtt in this case) -- syslog seems operational.
The device itself is unreachable, tried ping or reach tcp port 80, so I cannot even issue a reset command.
I have dozens of esp8266 based devices running fine, so this behavior seems esp32 related (or only on the solo1) -- maybe buried inside the TCP/IP stack.
I also dumped a whole day of mqtt traffic for other devices and found not a single duplicate transaction, also no increasing in MqttCount.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Shelly Plus 1 ()
- [x] Tasmota binary firmware version number used: 12.0.2 (release but also checked development branch)
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just wait....

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
the device shoud ignore duplicate packages, and ack accordingly

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

reimerp, 2022-08-02:
I have read https://github.com/arendst/Tasmota/discussions/10262
with some hints but nothing really helped (yet)

Normally the mqtt Request is responded by my mosqutto server in 0.3-0-5 ms
This packet is than acked by the shelly after 13ms quickest, bur normally after 160ms up to 300ms
If there is no ACK the server  does the TCP Retransmission (mqtt Response) again after ~400ms

I saw this retransmission occasionally on other devices, but there seems no packet loss but only a timeout, so this leads to a duplicate ACK and everything goes back to nomal.

I will do some traces again regarding the ARP traffic, see link above

barbudor, 2022-08-02:
As you haven't provided the information required by the issue template I must ask : what binary are you using ?
Shelly Plus 1 requires [tasmota32solo1.bin](http://ota.tasmota.com/tasmota32/tasmota32solo1.bin)

You definitively seems to have some WIfi problems. First Tasmota cannot resolve DNS, then when you switch to static you can't connect to MQTT. But suprising that you can get logs through syslog
Have you tried to increase the log level to 3 or 4 ? It may give you more hints on what's Tasmoa is up to which make its so busy

reimerp, 2022-08-02:
Yes, I used tasmota32solo1.bin first release version, than selft-build from development branch.
The shelly reports 
"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,
"Wifi": "RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:05"
"MqttCount":27
It has one of the best WIFI connection from all my devices but the highest MqttCount, others count +1 in 2-3 days....
Currently I am running a full TCPDump and to not disturb that I set syslog to 0

barbudor, 2022-08-02:
MqttCount clearly shows that it has problem with MQTT connections
So the problem is not the Wifi
Are you sure to use unique MqttClient for each of your devices ?
Using the same MqttClient on 2 different devices would lead to continuous disconnection/reconnections as Mosquitto want only 1 connection per MqttClient
(I believe you know the difference between the MttqClient and MqttUser/MqttPassword)