arendst/Tasmota #16144: Tasmota LVGL on M5Stack Core2 showing MPU6886 values twice

msp1974, 2022-08-03:
### PROBLEM DESCRIPTION
Tasmota 12.0.2.4 on an M5 Stack Core2 displays the MPU6886 values twice on both the web console and in the MQTT data,  I have serial flashed as per the LVGL in less than 10 minutes with Tasmota document.

Everything works fine but just displays twice.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): M5 Stack Core2
- [x] Tasmota binary firmware version number used: Tasmota32 LVGL 12.0.2.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmota Web Installer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
14:41:28.964 MQT: stat/tasmota_C46F4C/RESULT = {"NAME":"M5Stack Core 2","GPIO":[6210,1,1,1,6720,768,0,0,0,1,1,800,0,0,736,1,0,641,609,704,0,0,0,1,0,0,0,0,640,608,1,1,1,0,672,7968],"FLAG":0,"BASE":1}
14:41:29.171 MQT: stat/tasmota_C46F4C/RESULT = {"Module":{"0":"M5Stack Core 2"}}
14:41:29.389 MQT: stat/tasmota_C46F4C/RESULT = {"GPIO0":{"6210":"Option A3"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6720":"SDCard CS"},"GPIO5":{"768":"SPI CS1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"800":"SPI DC1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"736":"SPI CLK1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"641":"I2C SDA2"},"GPIO22":{"609":"I2C SCL2"},"GPIO23":{"704":"SPI MOSI1"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"640":"I2C SDA1"},"GPIO33":{"608":"I2C SCL1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"672":"SPI MISO1"},"GPIO39":{"7968":"Interrupt1"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  No rules used

```
- [x] Provide the output of this command: `Status 0`:
```lua
14:43:00.172 MQT: stat/tasmota_C46F4C/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_C46F4C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:43:00.211 MQT: stat/tasmota_C46F4C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:08:39","StartupUTC":"2022-08-03T13:34:21","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2022-07-30T11:57:40","SaveCount":62}}
14:43:00.247 MQT: stat/tasmota_C46F4C/STATUS2 = {"StatusFWR":{"Version":"12.0.2.4(lvgl-haspmota)","BuildDateTime":"2022-07-29T18:01:58","Core":"2_0_4_1","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6-V3 rev.3","CR":"419/699"}}
14:43:00.268 MQT: stat/tasmota_C46F4C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NAAccessPoint",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
14:43:00.304 MQT: stat/tasmota_C46F4C/STATUS4 = {"StatusMEM":{"ProgramSize":1800,"Free":1079,"Heap":136,"StackLowMark":19,"PsrMax":4090,"PsrFree":4031,"ProgramFlashSize":4096,"FlashSize":16384,"FlashChipId":"186020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAAC7CE","00048001","000007C3","00000000","40000981","00000080","00200000","54000019","00000000"],"Drivers":"1,2,4,5,8,9,10,12,13,16,20,24,26,27,29,38,50,52,54,55,56,62","Sensors":"1,2,5,6,7,8,9,85,127"}}
14:43:00.347 MQT: stat/tasmota_C46F4C/STATUS5 = {"StatusNET":{"Hostname":"tasmota-C46F4C-3916","IPAddress":"192.168.x.x","Gateway":"192.168.x.x","Subnetmask":"255.255.255.0","DNSServer1":"192.168.x.x","DNSServer2":"192.168.x.x","Mac":"B8:F0:09:C4:6F:4C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:43:00.376 MQT: stat/tasmota_C46F4C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.200.240","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C46F4C","MqttUser":"themqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:43:00.407 MQT: stat/tasmota_C46F4C/STATUS7 = {"StatusTIM":{"UTC":"2022-08-03T13:43:00","Local":"2022-08-03T14:43:00","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:26","Sunset":"20:25"}}
14:43:00.452 MQT: stat/tasmota_C46F4C/STATUS10 = {"StatusSNS":{"Time":"2022-08-03T14:43:00","MPU6886":{"AX":-33,"AY":-21,"AZ":1057,"GX":1,"GY":-2,"GZ":-3},"ESP32":{"Temperature":37.2},"AXP192":{"VBusVoltage":5.115,"VBusCurrent":5.1,"BattVoltage":4.147,"BattCurrent":0.0,"Temperature":30.4},"MPU6886":{"AX":-38,"AY":-22,"AZ":1060,"GX":2,"GY":-2,"GZ":-2},"TempUnit":"C"}}
14:43:00.505 MQT: stat/tasmota_C46F4C/STATUS11 = {"StatusSTS":{"Time":"2022-08-03T14:43:00","Uptime":"0T00:08:39","UptimeSec":519,"Heap":137,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":11,"Objects":162},"POWER":"ON","Wifi":{"AP":1,"SSId":"WifiHouseAP","BSSId":"04:D4:C4:3A:DA:60","Channel":11,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Flash M5 Stack Core2 via Tasmota Web Installer, set Autoconfig to M5Stack Core2.

### EXPECTED BEHAVIOUR
Only expecting 1 iteration of the MPU6886 values

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

msp1974, 2022-08-03:
UPDATE:  Sorry hadn't really noticed but they also have different values.

s-hadinger, 2022-08-03:
Please re-download the autoconf file for M5Stack Core 2 (select it again in the drop list, don't reapply). It should be fixed now.

Jason2866, 2022-08-05:
@msp1974 Please test.

Jason2866, 2022-08-05:
Tested and verified issue is fixed.

closed 2022-08-05 by Jason2866

msp1974, 2022-08-05:
Perfect. Thanks for fixing so quickly