arendst/Tasmota #16173: IRrecv GPIO and IRsend GPIO set on ESP32 crash

thanasiskalos, 2022-08-07:
### PROBLEM DESCRIPTION
On ESP32 with tasmota 12.0.2 when  IRsend and Irresv is configured on any GPIO ( even without attaching an IR LED) the esp32 crash and reboot itself when the following occurs:

- Send any Irsend command though console or MQTT to the tasmota . 
for example : send though console the following command:
   IRhvac {"Vendor":"Mitsubishi_Heavy_152", "Power":"On","Mode":"Hot","FanSpeed":3,"Temp":22.5}

After some repetition of the same command esp32 crashes.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ESP32 ESP32-D0WD rev.1 and ESP32 and ESP32-D0WDQ6 rev.1
- [ ] Tasmota binary firmware version number used: _12.0.2 IR 
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: Tasmota Installer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:31:57.866 SRC: Backlog
08:31:57.869 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
08:31:57.883 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"1056":"IRsend"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"1088":"IRrecv"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
08:31:57.968 BRY: GC from 6654 to 3858 bytes, objects freed 7/30 (in 1 ms) - slots from 35/76 to 31/76

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
08:32:43.311 SRC: Backlog
08:32:43.315 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
08:32:43.321 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:33:04.351 CMD: Grp 0, Cmd 'STATUS', Idx 0, Len 0, Pld -99, Data ''
08:33:04.381 BRY: GC from 4438 to 3052 bytes, objects freed 4/30 (in 1 ms) - slots from 34/76 to 31/76
08:33:04.392 RSL: STATUS0 = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9F2040","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:09:44","StartupUTC":"2022-08-07T07:23:20","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2022-08-07T08:18:31","SaveCount":13},"StatusFWR":{"Version":"12.0.2(ir)","BuildDateTime":"2022-06-20T12:41:13","Core":"2_0_3","SDK":"v4.4.1","CpuFrequency":80,"Hardware":"ESP32-D0WD rev.1","CR":"406/699"},"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Apartments by the port",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00190000000000","00000080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":1199,"Free":1680,"Heap":163,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1A05C2","00000001","00000003","00000000","00000000","00000080","00200000","04000008","00000000"],"Drivers":"1,2,4,5,7,9,10,12,38,50,52,62","Sensors":"1,2,127"},"StatusNET":{"Hostname":"tasmota-9F2040-0064","IPAddress":"10.0.16.24","Gateway":"10.0.16.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.200.147","DNSServer2":"1.1.1.1","Mac":"A4:CF:12:9F:20:40","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F2040","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-08-07T07:33:04","Local":"2022-08-07T08:33:04","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:32","Sunset":"20:19"},"StatusSNS":{"Time":"2022-08-07T08:33:04","ESP32":{"Temperature":53.9},"TempUnit":"C"},"StatusSTS":{"Time":"2022-08-07T08:33:04","Uptime":"0T00:09:44","UptimeSec":584,"Heap":162,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":2,"Objects":30},"Wifi":{"AP":1,"SSId":"Apartments by the port","BSSId":"70:97:41:94:A4:79","Channel":11,"Mode":"11n","RSSI":100,"Signal":-21,"LinkCount":1,"Downtime":"0T00:00:04"}},"StatusSTK":{"Exception":28,"Reason":"LoadProhibited","EPC":"40081612","EXCVADDR":"000000a0","CallChain":["4008160f"]}}
08:33:05.438 WIF: Checking connection...

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12596
load:0x40080400,len:2916
entry 0x400805c4

00:00:00.002 HDW: ESP32-D0WD
00:00:00.088 UFS: FlashFS mounted with 304 kB free
00:00:00.156 CFG: Loaded from File, Count 12
00:00:00.172 QPC: Count 1
00:00:00.368 BRY: Berry initialized, RAM used=3085 bytes
00:00:00.394 Project tasmota - Tasmota Version 12.0.2(ir)-2_0_3(2022-06-20T12:41:13)
00:00:01.042 WIF: Connecting to AP1 Apartments by the port Channel 11 BSSId 70:97:41:94:A4:79 in mode 11n as tasmota-9F2040-0064...
[... 86 lines cut ...]
Core  1 register dump:
PC      : 0x40081612  PS      : 0x00060033  A0      : 0x400858a4  A1      : 0x3ffbf1cc
A2      : 0x3ffb3aac  A3      : 0x3ffbdd10  A4      : 0x00000000  A5      : 0x3ffbf1bc
A6      : 0x00000001  A7      : 0x80000002  A8      : 0x00000000  A9      : 0x3ffbf18c
A10     : 0x00000001  A11     : 0x3ffc7710  A12     : 0x3ffc7714  A13     : 0x0000abab
A14     : 0x00060023  A15     : 0x00060021  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x000000a0  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4008160f:0x3ffbf1cc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12596
load:0x40080400,len:2916
entry 0x400805c4

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-08-07:
I confirm I can reproduce.

s-hadinger, 2022-08-07:
Narrowing down, when sending IR, Tasmota disables receiving using `irrecv->disableIRIn();` to avoid receiving our own messages, and re-enables right after with `irrecv->enableIRIn();`.

Calling those two methods repeatedly creates the crash; probably a problem while initializing an interrupt handler.

thanasiskalos, 2022-08-07:
Hello
Sometimes when controlling an HVAC it is good to have the option to listen to your own IR send commands in order to update the climate status on the Home Assistant.
Can we enable / Disable such functionality from the console ?

s-hadinger, 2022-08-07:
True but I've always been uncomfortable to have the interrupt handler triggered by received IR competing with the sender code with tight timing (40 kHz). 

Although it seemed to work. It would be different if it used RMT hardware support. But it's not in the roadmap foe the short term. 

For the bug above, I will implement a much lighter "interrupt pause" mechanism that should avoir d the crash.

s-hadinger, 2022-08-08:
> Hello Sometimes when controlling an HVAC it is good to have the option to listen to your own IR send commands in order to update the climate status on the Home Assistant. Can we enable / Disable such functionality from the console ?

You can enable this with compile time option `#define IR_RCV_WHILE_SENDING  1`

PR #16179 (2022-08-08): Fixed IR crash on ESP32 (#16173)
## Description:

Quick fix for ESP32, will be submitted upstream if proved effective.

**Related issue (if applicable):** fixes #16173

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.4.1
  - [x] I accept the [CLA](https://github.com/arendst/T [... cut]
closed 2022-08-08 by s-hadinger