arendst/Tasmota #16175: Zbbridge: After Battery Auto-Probe, Battery Voltage & Percentage Reporting "0"

rebel7580, 2022-08-07:
### PROBLEM DESCRIPTION
After upgrading to 12.0.2.4, Battery Auto-Probes at first report correct battery values, but then almost immediately revert to "0"'s. Reporting "0"'s seems to continue for about an hour, then reports contain valid battery values start to appear.
Observed with Aqara motion sensors, water leak sensors, and buttons.
Not seen with Aqara temp/hum sensor, as this appears to send battery values all the time (I never saw an auto-probe for it).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [x] Tasmota binary firmware version number used: 12.0.2.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
13:18:22.342 CMD: Backlog Template; Module; GPIO 255;
13:18:22.408 MQT: stat/tasmota_zbBridge/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:18:22.616 MQT: stat/tasmota_zbBridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
13:18:22.870 MQT: stat/tasmota_zbBridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
  Configuration output here:

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
  None

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:19:31.285 CMD: status 0
13:19:31.289 MQT: stat/tasmota_zbBridge/STATUS = {"Status":{"Module":75,"DeviceName":"zbbridge","FriendlyName":["zbBridge"],"Topic":"tasmota_zbBridge","ButtonTopic":"0","Power":-1,"PowerOnState":4,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:19:31.295 MQT: stat/tasmota_zbBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"Boston","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:03:14","StartupUTC":"2022-08-07T16:16:17","Sleep":50,"CfgHolder":4617,"BootCount":176,"BCResetTime":"2021-02-15T07:07:42","SaveCount":278,"SaveAddress":"FC000"}}
13:19:31.300 MQT: stat/tasmota_zbBridge/STATUS2 = {"StatusFWR":{"Version":"12.0.2.4(zbbridge)","BuildDateTime":"2022-08-07T14:29:01","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"408/699"}}
13:19:31.305 MQT: stat/tasmota_zbBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["beantown",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00000000000000","00000080","40006080","00004030","00000000"]}}
13:19:31.314 MQT: stat/tasmota_zbBridge/STATUS4 = {"StatusMEM":{"ProgramSize":685,"Free":1104,"Heap":21,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","04000000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50,62","Sensors":"1"}}
13:19:31.319 MQT: stat/tasmota_zbBridge/STATUS5 = {"StatusNET":{"Hostname":"tasmota-zbBridge-5317","IPAddress":"192.168.123.31","Gateway":"192.168.123.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.123.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:9D:14:C5","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:19:31.324 MQT: stat/tasmota_zbBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.123.176","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D14C5","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:19:31.330 MQT: stat/tasmota_zbBridge/STATUS7 = {"StatusTIM":{"UTC":"2022-08-07T17:19:31","Local":"2022-08-07T13:19:31","StartDST":"2022-03-13T02:00:00","EndDST":"2022-11-06T02:00:00","Timezone":99,"Sunrise":"05:59","Sunset":"20:03"}}
13:19:31.335 MQT: stat/tasmota_zbBridge/STATUS10 = {"StatusSNS":{"Time":"2022-08-07T13:19:31"}}
13:19:31.342 MQT: stat/tasmota_zbBridge/STATUS11 = {"StatusSTS":{"Time":"2022-08-07T13:19:31","Uptime":"0T01:03:14","UptimeSec":3794,"Vcc":3.387,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"beantown","BSSId":"3C:37:86:52:48:14","Channel":11,"Mode":"11n","RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
I captured this before weblog was set to 4, not seen after some time when weblog 4 was used.

One of my Aqara motion sensors (Similar behavior for all my battery operated sensor except the temp/hum one):

12:33:05.078 ZIG: Battery auto-probe `ZbSend {"Device":"0x3C06","Read":{"BatteryPercentage":true,"BatteryVoltage":true}}`
12:33:05.085 MQT: tele/tasmota_zbBridge/MotionGD/ZbReceived = {"Device":"0x3C06","Name":"MotionGD","BatteryVoltage":3.05,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":66}
12:33:05.646 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0xB344,0A97DFEF62696400000000"}
12:33:05.662 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x6331,0AE3DFEF62476400000000"}
12:33:05.677 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x6D52,0A2FE6EF6264FF000000000506010F0000"}
12:33:05.693 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x2354,0A7EE7EF623DFF000000000506010F0000"}
12:33:05.709 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x8429,0A2BE5EF62476400000000"}
12:33:05.724 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0xD5F1,0A57DFEF62696400000000"}
12:33:05.740 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0xB051,0AEFDFEF62476400000000"}
12:33:05.755 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x97B2,0A5FDFEF62666400000000"}
12:33:05.771 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x3744,0A47DFEF62446400000000"}
12:33:05.797 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x3466,0A47B8F8613FFF000000000506010F00000E01010F00FFFEFFFFFFFFFFFFFFFF"}
12:33:05.814 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0xE3D3,0AEAE7EF62665C000000000E05010F00400CFD031A19FF000080"}
12:33:05.831 MQT: stat/tasmota_zbBridge/RESULT = {"ZbData":"ZbData 0x2E00,0ABBE4EF62426400000000"}
[... 8 lines cut ...]
About a half hour later, this was observed. The first two are periodic status reports (from two motion sensors), The third is when the MotionGD sensor was triggered:

13:12:56.511 MQT: tele/tasmota_zbBridge/MotionGD/ZbReceived {"Device":"0x3C06","Name":"MotionGD","Occupancy":0,"BatteryPercentage":0}
13:13:03.102 ZIG: {"ZbEZSPReceived":"7D0054237F0CCEFEFF72025C01"}
13:13:07.034 MQT: tele/tasmota_zbBridge/MotionGS/ZbReceived = {"Device":"0x5196","Name":"MotionGS","Occupancy":0,"BatteryPercentage":0}
13:13:08.516 ZIG: {"ZbEZSPReceived":"7D0069FE367BFCFEFF6F0D0001"}
13:13:36.867 MQT: tele/tasmota_zbBridge/MotionGD/ZbReceived = {"Device":"0x3C06","Name":"MotionGD","Illuminance":67,"Occupancy":1,"BatteryPercentage":0,"Endpoint":1,"LinkQuality":61}

```

### TO REPRODUCE
- Update to latest (12.0.2.4 in my case).
- Watch for auto-probes - will see initial ZbRecieved message has correct battery values, then shortly after a memory save, another ZbRecieved message is sent with battery values set to 0.
- Observed that, about an hour later, battery values started to be reported correctly.

### EXPECTED BEHAVIOUR
Battery values report correct values immediately after auto-probe.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-08-07:
Thanks. Do you have the same logs with `Weblog 3`?

Aqara typically report battery via a proprietary field and I don't know what happens when you query the standard field. 

I could also just ignore a value of zero

rebel7580, 2022-08-07:
Changed to weblog 3. The following is me triggering the Motion5 motion sensor, which in turn (via my automation system) turns on a light (Upstairs_Hall_Master). Probably more extranious info here than you want/need!
You can see that bat % is reported for the motion sensor now: 17:01:36.852, 17:03:06.493

It seems, based on the initial logs, that right after the auto-probe, it reports correctly, but then "forgets" the % value until ~1hr later (next periodic report?) After that, seems to work OK.

I tried restarting the bridge (restart 1), but I didn't see any more auto-probe messages and the bat values were reporting OK. So I can't get back to that initial condition where every battery powered device was auto-probed.

```
17:00:48.826 CMD: weblog 3
17:00:48.827 SRC: WebConsole from 192.168.123.186
17:00:48.829 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 3, Data '3'
17:00:48.833 MQT: stat/tasmota_zbBridge/RESULT = {"WebLog":3}
17:00:49.096 CFG: Saved to flash at FD, Count 281, Bytes 4096
17:00:51.211 ZIG: {"ZbEZSPReceived":"7D0069FE367BFCFEFF6F0D0001"}
17:01:18.501 RTC: UTC 2022-08-07T21:01:18, DST 2022-03-13T02:00:00, STD 2022-11-06T02:00:00
17:01:18.000 RTC: Synced by NTP
17:01:32.358 ZIG: {"ZbEZSPReceived":"7D00526D481FCEFEFF72025C01"}
17:01:36.238 ZIG: {"ZbEZSPReceived":"7D0054237F0CCEFEFF72025C01"}
17:01:36.438 ZIG: {"ZbEZSPReceived":"4500000401000401010001000006B8CA02EDFFFF08181B0A0000211000"}
17:01:36.440 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0400","srcaddr":"0xED02","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":87,"securityuse":0,"seqnumber":6,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":27,"cmdid":"0x0A","payload":"0000211000"}}
17:01:36.443 ZIG: ZbZCLRawReceived: {"0xED02":{"0400/0000":16,"Endpoint":1,"LinkQuality":87}}
17:01:36.448 ZIG: {"ZbEZSPReceived":"4500000401060401010001000007B8CA02EDFFFF07181C0A00001801"}
17:01:36.449 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0406","srcaddr":"0xED02","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":87,"securityuse":0,"seqnumber":7,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":28,"cmdid":"0x0A","payload":"00001801"}}
17:01:36.451 ZIG: ZbZCLRawReceived: {"0xED02":{"0406/0000":1,"Endpoint":1,"LinkQuality":87}}
17:01:36.852 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","Illuminance":16,"Occupancy":1,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":87}
17:01:36.950 SRC: MQTT
17:01:36.952 CMD: Grp 0, Cmd 'ZBSEND', Idx 1, Len 55, Pld -99, Data '{"device":"Upstairs_Hall_Master","send":{"Dimmer":254}}'
17:01:36.954 ZIG: guessing endpoint 1
17:01:36.956 ZigbeeZCLSend device: 0xFE69, endpoint:1, cluster:0x0008, cmd:0x04, send:"FE0A00"
17:01:36.957 ZIG: ZbEZSPSend 34000069FE04010800010140010000150106111504FE0A00
17:01:36.961 MQT: stat/tasmota_zbBridge/RESULT = {"ZbSend":"Done"}
17:01:36.001 SRC: MQTT
17:01:36.002 CMD: Grp 0, Cmd 'ZBSEND', Idx 1, Len 55, Pld -99, Data '{"device":"Upstairs_Hall_Master","send":{"POWER":"on"}}'
17:01:36.005 ZIG: guessing endpoint 1
17:01:36.006 ZigbeeZCLSend device: 0xFE69, endpoint:1, cluster:0x0006, cmd:0x01, send:""
17:01:36.007 ZIG: ZbEZSPSend 34000069FE04010600010140010000160103111601
17:01:37.001 MQT: stat/tasmota_zbBridge/RESULT = {"ZbSend":"Done"}
17:01:37.007 ZIG: {"ZbEZSPReceived":"3400009C"}
17:01:37.018 ZIG: {"ZbEZSPReceived":"3400009D"}
17:01:37.065 ZIG: {"ZbEZSPReceived":"3F000069FE040108000101400100009C010000"}
17:01:37.073 ZIG: {"ZbEZSPReceived":"4500000401060001014000000014C8CE69FEFFFF0708010A00001001"}
17:01:37.075 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0006","srcaddr":"0xFE69","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":97,"securityuse":0,"seqnumber":20,"fc":"0x08","frametype":0,"direction":1,"disableresp":0,"manuf":"0x0000","transact":1,"cmdid":"0x0A","payload":"00001001"}}
17:01:37.076 ZigbeeZCLSend device: 0xFE69, endpoint:1, cluster:0x0006, cmd:0x0B, send:"0A00"
17:01:37.077 ZIG: ZbEZSPSend 34000069FE0401060001010000000001010510010B0A00
17:01:37.079 ZIG: ZbZCLRawReceived: {"0xFE69":{"0006/0000":1,"Endpoint":1,"LinkQuality":97}}
17:01:37.090 ZIG: {"ZbEZSPReceived":"3400009E"}
17:01:37.092 ZIG: {"ZbEZSPReceived":"3F000069FE040106000101400100009D010000"}
17:01:37.145 ZIG: {"ZbEZSPReceived":"3F000069FE040106000101000000009E010000"}
17:01:37.449 MQT: tele/tasmota_zbBridge/Upstairs_Hall_Master/ZbReceived = {"Device":"0xFE69","Name":"Upstairs_Hall_Master","Power":1,"Endpoint":1,"LinkQuality":97}
17:01:38.070 ZigbeeZCLSend device: 0xFE69, endpoint:1, cluster:0x0008, cmd:0x00, send:"0000"
17:01:38.071 ZIG: ZbEZSPSend 34000069FE040108000101400100001701050017000000
17:01:38.085 ZIG: {"ZbEZSPReceived":"3400009F"}
17:01:38.150 ZIG: {"ZbEZSPReceived":"4500000401080001014000000015CCCF69FEFFFF0808170100000020FE"}
17:01:38.152 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0008","srcaddr":"0xFE69","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":100,"securityuse":0,"seqnumber":21,"fc":"0x08","frametype":0,"direction":1,"disableresp":0,"manuf":"0x0000","transact":23,"cmdid":"0x01","payload":"00000020FE"}}
17:01:38.154 ZIG: ZbZCLRawReceived: {"0xFE69":{"0008/0000":254,"Endpoint":1,"LinkQuality":100}}
17:01:38.158 ZIG: {"ZbEZSPReceived":"3F000069FE040108000101400100009F010000"}
17:01:38.257 ZigbeeZCLSend device: 0xFE69, endpoint:1, cluster:0x0006, cmd:0x00, send:"0000"
[... cut ...] leresp":0,"manuf":"0x0000","transact":26,"cmdid":"0x01","payload":"0000001000"}}
17:03:37.374 ZIG: ZbZCLRawReceived: {"0xFE69":{"0006/0000":0,"Endpoint":1,"LinkQuality":100}}
17:03:37.377 ZIG: {"ZbEZSPReceived":"3F000069FE04010600010140010000A4010000"}
17:03:37.731 MQT: tele/tasmota_zbBridge/Upstairs_Hall_Master/ZbReceived = {"Device":"0xFE69","Name":"Upstairs_Hall_Master","Power":0,"Dimmer":254,"Endpoint":1,"LinkQuality":100}
17:03:40.408 ZIG: {"ZbEZSPReceived":"7D0054237F0CCEFEFF72025C01"}
```

s-hadinger, 2022-08-08:
Unfortunately the new logs do not show when the Battery goes down to 0. I wanted to see what was the faulty packet sent by the Aqara device.

rebel7580, 2022-08-08:
I tried to do a restart, but that didn't cause the auto-probes to re-occur as they did when I first upgraded the sw.  I'll try reflashing  to see if that triggers them again, if you think that might work.

rebel7580, 2022-08-08:
Re-flash didn't prompt a new set of auto-probes. Any suggestions, hopefully short of a full reset of the Zigbee info stored in memory? I'd hate to start over for something that seems to be an "initial" issue that resolves itself after a hour. :)

rebel7580, 2022-08-08:
Caught an auto-probe, but this one is weird, as it isn't a battery powered device, it's a Jasco 43076 powered wall switch (short addr 0x6D52)! Note at 11:55:22.113 it actually sends an MQTT message with Battery Percentage included.

Likely a different problen altogether, but thought I should post this.
```

11:54:53.101 HTP: Console
11:54:54.155 ZIG: {"ZbEZSPReceived2":"372690017D00526D481FCEFEFF72025C01"}
11:54:54.156 ZIG: ZbEZSPSentRaw 84
11:54:54.157 ZIG: {"ZbEZSPReceived":"7D00526D481FCEFEFF72025C01"}
11:55:04.043 ZIG: {"ZbEZSPReceived2":"472690017D0054237F0CCEFEFF72025C01"}
11:55:04.044 ZIG: ZbEZSPSentRaw 85
11:55:04.045 ZIG: {"ZbEZSPReceived":"7D0054237F0CCEFEFF72025C01"}
11:55:17.920 ZIG: {"ZbEZSPReceived2":"572690017D0069FE367BFCFEFF6F0D0001"}
11:55:17.921 ZIG: ZbEZSPSentRaw 86
11:55:17.922 ZIG: {"ZbEZSPReceived":"7D0069FE367BFCFEFF6F0D0001"}
11:55:21.643 ZIG: {"ZbEZSPReceived2":"672690014500000401060001014000000001C4CD526DFFFF0708350A00001000"}
11:55:21.644 ZIG: ZbEZSPSentRaw 87
11:55:21.646 ZIG: {"ZbEZSPReceived":"4500000401060001014000000001C4CD526DFFFF0708350A00001000"}
11:55:21.648 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0006","srcaddr":"0x6D52","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":94,"securityuse":0,"seqnumber":1,"fc":"0x08","frametype":0,"direction":1,"disableresp":0,"manuf":"0x0000","transact":53,"cmdid":"0x0A","payload":"00001000"}}
11:55:21.650 ZigbeeZCLSend device: 0x6D52, endpoint:1, cluster:0x0006, cmd:0x0B, send:"0A00"
11:55:21.651 ZIG: ZbEZSPSend 340000526D0401060001010000000035010510350B0A00
11:55:21.652 ZIG: adding packet to_send, to_ack:7, to_send:7, to_end:7
11:55:21.653 ZIG: Battery auto-probe `ZbSend {"Device":"0x6D52","Read":{"BatteryPercentage":true,"BatteryVoltage":true}}`
11:55:21.655 ZIG: ZbZCLRawReceived: {"0x6D52":{"0006/0000":0,"Endpoint":1,"LinkQuality":94}}
11:55:21.657 ZIG: Something to_send, to_ack:7, to_send:7, to_end:0
11:55:21.658 ZIG: ZbEZSPSentRaw 77270001340000526D0401060001010000000035010510350B0A00
11:55:21.660 ZigbeeZCLSend device: 0x6D52, endpoint:1, cluster:0x0001, cmd:0x00, send:"20002100"
11:55:21.661 ZIG: ZbEZSPSend 340000526D0401010001014001000001010700010020002100
11:55:21.662 ZIG: adding packet to_send, to_ack:7, to_send:0, to_end:0
11:55:21.665 ZIG: {"ZbEZSPReceived2":"702780013400007D"}
11:55:21.666 ZIG: new ack/data received, was 7 now 0
11:55:21.667 ZIG: freeing packet 7 from memory
11:55:21.668 ZIG: ZbEZSPSentRaw 80
11:55:21.668 ZIG: {"ZbEZSPReceived":"3400007D"}
11:55:21.669 ZIG: Something to_send, to_ack:0, to_send:0, to_end:1
11:55:21.670 ZIG: ZbEZSPSentRaw 00280001340000526D0401010001014001000001010700010020002100
11:55:21.680 ZIG: {"ZbEZSPReceived2":"012880013400007E"}
11:55:21.681 ZIG: new ack/data received, was 0 now 1
11:55:21.682 ZIG: freeing packet 0 from memory
11:55:21.683 ZIG: ZbEZSPSentRaw 81
11:55:21.684 ZIG: {"ZbEZSPReceived":"3400007E"}
11:55:21.724 ZIG: {"ZbEZSPReceived2":"112890013F0000526D040106000101000000007D010000"}
11:55:21.726 ZIG: ZbEZSPSentRaw 82
11:55:21.727 ZIG: {"ZbEZSPReceived":"3F0000526D040106000101000000007D010000"}
11:55:21.747 ZIG: {"ZbEZSPReceived2":"212890014500000401010001014000000002C4CD526DFFFF09080101200086210086"}
11:55:21.749 ZIG: ZbEZSPSentRaw 83
11:55:21.750 ZIG: {"ZbEZSPReceived":"4500000401010001014000000002C4CD526DFFFF09080101200086210086"}
11:55:21.752 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0001","srcaddr":"0x6D52","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":94,"securityuse":0,"seqnumber":2,"fc":"0x08","frametype":0,"direction":1,"disableresp":0,"manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"200086210086"}}
11:55:21.754 ZIG: ZbZCLRawReceived: {"0x6D52":{"Endpoint":1,"LinkQuality":94}}
11:55:21.756 ZIG: {"ZbEZSPReceived2":"312890013F0000526D040101000101400100007E010000"}
11:55:21.758 ZIG: ZbEZSPSentRaw 84
11:55:21.759 ZIG: {"ZbEZSPReceived":"3F0000526D040101000101400100007E010000"}
11:55:22.113 MQT: tele/tasmota_zbBridge/UnderCabinet/ZbReceived = {"Device":"0x6D52","Name":"UnderCabinet","Power":0,"BatteryPercentage":48,"Endpoint":1,"LinkQuality":94}
11:55:30.419 WIF: Sending Gratuitous ARP
11:55:56.415 ZIG: {"ZbEZSPReceived2":"412890017D00526D481FCEFEFF72025C01"}
11:55:56.416 ZIG: ZbEZSPSentRaw 85
11:55:56.418 ZIG: {"ZbEZSPReceived":"7D00526D481FCEFEFF72025C01"}
11:56:06.159 ZIG: {"ZbEZSPReceived2":"512890017D0054237F0CCEFEFF72025C01"}
11:56:06.161 ZIG: ZbEZSPSentRaw 86
11:56:06.162 ZIG: {"ZbEZSPReceived":"7D0054237F0CCEFEFF72025C01"}
11:56:20.171 ZIG: {"ZbEZSPReceived2":"612890017D0069FE367BFCFEFF6F0D0001"}
11:56:20.172 ZIG: ZbEZSPSentRaw 87
11:56:20.173 ZIG: {"ZbEZSPReceived":"7D0069FE367BFCFEFF6F0D0001"}
11:56:30.416 WIF: Sending Gratuitous ARP
```

s-hadinger, 2022-08-09:
The device responded with an error after auto-probing. This is expected. The detection of non-battery powered device is done at pairing time, and wasn't recorded before the auto-probe feature. So any device paired before, will continue to receive battery requests - no big harm

s-hadinger, 2022-08-09:
Looking closer at the log, the first reported (valid) value was from Aqara proprietary format, so the second was from cluster 0x0001. I will add some code to ignore if both voltage and battery are zero.

closed 2022-08-09 by s-hadinger

rebel7580, 2022-08-09:
Thanks for the quick fix! I'll try it out.

BTW, is there a way to tell expicitly if a merged commit has actually included in a build? Other than "wait a few hours"?
I see that the latest built is timestamped 20220809 19:17, which I think is CEDT, so that would be just a few minutes ago, w.r.t the timeI post this. And I would assume it includes the commit, based on the merge timestamp.

s-hadinger, 2022-08-09:
There used to be a commit id for the builds, but it's gone. Indeed the only way is to check the build time.

rebel7580, 2022-08-11:
FWIW, after updating to the build of 20220809 19:17, which should have included your fix, I've seen some auto-probes, but they are exclusively for the three powered switches/dimmer (2 Jasco switches, 1 Jasco dimmer), Don't see any probes for all the actual battery powered devices. Not sure if there is something important that could use some "clean-up" in the auto probe code w.r.t. a powered device, if you can even tell the type. Otherwise, probably not a big deal.

I went back to weblog 2, so I don't have any detailed logs. I can change it back to 3 or 4 if you want logs.

s-hadinger, 2022-08-11:
It should be fine for now.