arendst/Tasmota #16199: Curtain Switch reboots when pressing the open or close button on the switch (physical)

Dark-Noir, 2022-08-10:
### PROBLEM DESCRIPTION
Hey, i use this switch: https://templates.blakadder.com/jinvoo_SM-SW101-C_b.html but with this config: https://templates.blakadder.com/anccy_LKTDUOZC.html for shutter mode (just switched buttons/relais).
Everything works fine, but i can't use the up or down buttons on the switch itself (manually). 
Every time i do it the shutter moves for a second and than the ESP reboots. 
No issue with the stop button. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Jinvoo SM-SW101-C Curtain Switch
- [x] Tasmota binary firmware version number used: 12.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: [Tasmota Web Installer](https://tasmota.github.io/install/)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
09:28:47.975 CMD: Backlog Template; Module; GPIO 255
09:28:48.055 MQT: stat/TCS01/RESULT = {"NAME":"Jinvoo Curtain","GPIO":[288,0,0,34,226,33,0,0,32,224,0,225,0,0],"FLAG":0,"BASE":18}
09:28:48.268 MQT: stat/TCS01/RESULT = {"Module":{"0":"Jinvoo Curtain"}}
09:28:48.522 MQT: stat/TCS01/RESULT = {"GPIO0":{"288":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"34":"Button3"},"GPIO4":{"226":"Relay3"},"GPIO5":{"33":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"224":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"225":"Relay2"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
09:29:27.356 CMD: Backlog Rule1; Rule2; Rule3
09:29:27.423 MQT: stat/TCS01/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":39,"Free":472,"Rules":"ON Power3#State=1 DO ShutterStop1 ENDON"}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
09:29:45.239 CMD: Status 0
09:29:45.247 MQT: stat/TCS01/STATUS = {"Status":{"Module":0,"DeviceName":"TCS01","FriendlyName":["TCS01-Open","TCS01-Close","TCS01-Stop"],"Topic":"TCS01","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:29:45.257 MQT: stat/TCS01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T09:27:36","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":40,"BCResetTime":"1970-01-01T00:00:00","SaveCount":84,"SaveAddress":"F8000"}}
09:29:45.267 MQT: stat/TCS01/STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"411/699"}}
09:29:45.276 MQT: stat/TCS01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Dark-6660",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C00B","2805C80001000600003C5A0A190000000000","40000080","00006000","00004000","00000000"]}}
09:29:45.293 MQT: stat/TCS01/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
09:29:45.305 MQT: stat/TCS01/STATUS5 = {"StatusNET":{"Hostname":"TCS01-7392","IPAddress":"192.168.178.170","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"80:7D:3A:74:1C:E0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:29:45.315 MQT: stat/TCS01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.160","MqttPort":1952,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_741CE0","MqttUser":"Dark-MQTT","MqttCount":13,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:29:45.326 MQT: stat/TCS01/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T09:29:45","Local":"1970-01-01T09:29:45","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
09:29:45.340 MQT: stat/TCS01/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T09:29:45","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}}
09:29:45.352 MQT: stat/TCS01/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T09:29:45","Uptime":"0T09:27:36","UptimeSec":34056,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":13,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Dark-6660","BSSId":"2C:3A:FD:FE:8A:1C","Channel":11,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":70,"Downtime":"0T00:46:24"}}}
09:29:45.369 MQT: stat/TCS01/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4025305e","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40231028","4023ddf7","4000050c","4023eace","40252b14","40101917","40101b9f","4028eb1b","40248d52","402490fd","402490fd","4024bc7c","40101b9f","4023ef40","40101206","4021e8da","4023a224","40228615","40101206","40101206","4023a292","4023d434","4023d4ec","40100a7c","4024cc10","40101e95"]}}
09:29:45.380 MQT: stat/TCS01/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":200,"Close":200,"50perc":68,"Delay":0,"Opt":"0000","Calib":[1440,256,3136,900,1000],"Mode":"1"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 84
00:00:00.055 FRC: Some settings have been reset (2)
00:00:00.076 Project tasmota - TCS01 Version 12.0.2(tasmota)-2_7_4_9(2022-06-20T12:41:03)
00:00:00.519 WIF: Connecting to AP1 Dark-6660 Channel 13 BSSId 2C:3A:FD:FE:8A:1C in mode 11n as TCS01-7392...
00:00:05.242 WIF: Connected
00:00:05.495 HTP: Web server active on TCS01-7392 with IP address 192.168.178.170
00:00:09.038 MQT: Attempting connection...
00:00:09.249 MQT: Connect failed to 192.168.178.160:1952, rc -2. Retry in 10 sec
00:00:36.456 MQT: Attempting connection...
00:00:36.498 MQT: Connected
00:00:36.502 MQT: tele/TCS01/LWT = Online (retained)
00:00:36.503 MQT: cmnd/TCS01/POWER = 
00:00:36.511 MQT: tele/TCS01/INFO1 = {"Info1":{"Module":"Jinvoo Curtain","Version":"12.0.2(tasmota)","FallbackTopic":"cmnd/DVES_741CE0_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:36.525 MQT: tele/TCS01/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"TCS01-7392","IPAddress":"192.168.178.170"}}
00:00:36.542 MQT: tele/TCS01/INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4025305e","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40231028","4023ddf7","4024caf8","4023eace","40100a7c","40102b21","401026f0","4010598b","40104b66","40104fd7","40220a90","4022d8e1","40104a9f","40103964","4022124b","40233bec","4000050c","4010364c","4023ef40","40101206","4021e8da","4023a224","402286cc","40101206","40101206","4023a292","4023d434","4023d4ec","40100a7c","4024cc10","40101e95"]},"BootCount":41}}
00:00:36.589 MQT: stat/TCS01/RESULT = {"POWER1":"OFF"}
00:00:36.591 MQT: stat/TCS01/POWER1 = OFF
00:00:36.593 MQT: stat/TCS01/RESULT = {"POWER2":"OFF"}
00:00:36.595 MQT: stat/TCS01/POWER2 = OFF
00:00:36.599 MQT: stat/TCS01/RESULT = {"POWER3":"OFF"}
00:00:36.604 MQT: stat/TCS01/POWER3 = OFF
00:00:40.902 MQT: tele/TCS01/STATE = {"Time":"1970-01-01T00:00:40","Uptime":"0T00:00:41","UptimeSec":41,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Dark-6660","BSSId":"2C:3A:FD:FE:8A:1C","Channel":11,"Mode":"11n","RSSI":66 [... cut ...]
00:00:40.937 MQT: tele/TCS01/SENSOR = {"Time":"1970-01-01T00:00:40","Shutter1":{"Position":50,"Direction":0,"Target":50,"Tilt":0}}
```

### TO REPRODUCE
Press one of the physical shutter buttons (open or close).

### EXPECTED BEHAVIOUR
The shutters open or close without a reboot.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-08-10:
Can you please try latest development version. There are a few changes in the shutter code since the release version.

Dark-Noir, 2022-08-10:
> Can you please try latest development version. There are a few changes in the shutter code since the release version.

Yeah thanks, that did the trick. Now it works as expected 👍

closed 2022-08-10 by Dark-Noir